FreeCAD/homebrew-freecad #10: pivy failed to build on 10.9.5

thomasgi, 2015-02-02:
When building pivy clang fails to create the shared library _soqt.so, because it is unable to find QtGui and QtCore:

clang++ -bundle -undefined dynamic_lookup -L/usr/local/lib -L/usr/local/opt/sqlite/lib -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.9.sdk build/temp.macosx-10.9-x86_64-2.7/pivy/gui/soqt_wrap.o -o build/lib/pivy/gui/_soqt.so -L/usr/local/opt/coin/lib -Wl,-framework,OpenGL -Wl,-multiply_defined,suppress -lSoQt -F/usr/local/Cellar/qt/4.8.6/lib -framework QtOpenGL QtGui QtCore -lCoin -lpthread
clang: error: no such file or directory: 'QtGui'
clang: error: no such file or directory: 'QtCore'
error: command 'clang++' failed with exit status 1

https://gist.github.com/69514e7589573bf83656

sanelson, 2015-06-17:
I believe that you'll get this exact problem if Coin is not built with the options specified below:

brew install --without-framework --without-soqt sanelson/freecad/coin

I've updated the README to reflect this for the build instructions.  Let me know if you are still having an issue with this.

evansharp, 2016-09-20:
Hi,

I'm still having this issue, though I think for a different reason. I built coin using the flags prescribed before install FreeCAD, but I get:

`Failed to download resource "pivy"
Failure while executing:  clone https://bitbucket.org/Coin3D/pivy /Users/*******/Library/Caches/Homebrew/pivy--hg
`

I am able to install pivy standalone with `brew install pivy`, is there a flag for the FreeCAD make that will utilize this external version as a dependancy?

evansharp, 2016-09-21:
If you're here with this issue, the problem is that OS X does not ship with Murcurial, which is the source engibe used by Bitbucket where the pivy source is being cloned.

If you run `brew install hg` first and then `brew install freecad` you'll get pivy installed, but then run into issue #18.

See pull #17 where this is being addressed.

bblacey, 2016-10-03:
Pull request #17 has been merged.

closed 2016-10-03 by bblacey