FreeCAD/homebrew-freecad #11: command 'clang' failed on 10.10.2

TLQuest, 2015-03-14:
When setting Homebrew Dependencies:     brew install --HEAD pivy
I get: 
==> Installing pivy from sanelson/homebrew-freecad
==> Cloning https://bitbucket.org/Coin3D/pivy
Updating /Library/Caches/Homebrew/pivy--hg
==> python setup.py install --prefix=/usr/local/Cellar/pivy/HEAD
/usr/local/Cellar/coin/3.1.3/Frameworks/Inventor.framework/Headers/Qt/devices/SoQtDevice.h:39:10: fatal error: 'qevent.h' file not found
#include <qevent.h>
         ^
1 error generated.
error: command 'clang' failed with exit status 1

sanelson, 2015-06-17:
It is possible that Coin was built with SoQt. I think this could cause this to happen.  As far as I know, SoQt is no longer needed by FreeCAD.  Try building Coin with the following options:

brew install --without-framework --without-soqt sanelson/freecad/coin

bblacey, 2016-10-07:
Closing as OBE/no response.

closed 2016-10-07 by bblacey