FreeCAD/homebrew-freecad #15: FreeCAD brew failed at "Build target FreeCADGui"

philchacko, 2016-02-04:
Attempted both:
brew install sanelson/freecad/freecad
brew install --HEAD sanelson/freecad/freecad

Each failed at:

...ct.cpp.o CMakeFiles/FreeCADGui.dir/Selection.cpp.o CMakeFiles/FreeCADGui.dir/SelectionFilter.cpp.o CMakeFiles/FreeCADGui.dir/PreCompiled.cpp.o -F/usr/local/Cellar/qt/4.8.7_2/lib ../../lib/libFreeCADApp.dylib /usr/local/lib/libCoin.dylib -framework QtOpenGL -framework QtSvg /usr/local/Cellar/qt/4.8.7_2/lib/libQtUiTools.a -framework QtWebKit -framework QtGui -framework QtXml -framework QtNetwork -framework QtCore /usr/local/lib/libboost_filesystem-mt.dylib /usr/local/lib/libboost_program_options-mt.dylib /usr/local/lib/libboost_regex-mt.dylib /usr/local/lib/libboost_signals-mt.dylib /usr/local/lib/libboost_system-mt.dylib /usr/local/lib/libboost_thread-mt.dylib -framework OpenGL /usr/local/Cellar/shiboken/1.2.2/lib/libshiboken-python2.7.dylib /usr/local/Cellar/pyside/1.2.2_1/lib/libpyside-python2.7.dylib ../../lib/libFreeCADBase.dylib /usr/local/Cellar/python/2.7.11/Frameworks/Python.framework/Versions/2.7/Python /usr/local/lib/libxerces-c.dylib /usr/lib/libz.dylib -lutil -ldl -framework QtCore /usr/local/lib/libboost_filesystem-mt.dylib /usr/local/lib/libboost_program_options-mt.dylib /usr/local/lib/libboost_regex-mt.dylib /usr/local/lib/libboost_signals-mt.dylib /usr/local/lib/libboost_system-mt.dylib /usr/local/lib/libboost_thread-mt.dylib 
[ 64%] Built target FreeCADGui
make: *** [all] Error 2

dmingram66, 2016-03-21:
I have the same problem, it fails when compiling the GUI.  Looks from the log file as though the linker is unable to resolve the necessary dependencies.  Please see the attached log file

[02.make.txt](https://github.com/sanelson/homebrew-freecad/files/183183/02.make.txt)

bblacey, 2016-10-07:
Resolved as of 3cf52e234ba58053ea5098de3f0cd07e2c7fc65e

closed 2016-10-07 by bblacey