FreeCAD/homebrew-freecad #18: Empty installation error

macfree, 2016-08-08:
Hi, I try to install on osx10.11.6, but it fails. 

**Normal install**
The normal install option created this error:
🍺  /usr/local/Cellar/orocos-kdl/1.3.0: 95 files, 900.3K, built in 28 seconds
Error: /usr/local/Cellar/pivy/0.5.0 is not a directory

I uninstalled pivy and reinstalled it (rev 0.5.0) and the following error comes up (both normal and HEAD installation) :

**--HEAD Installation**
test:usr markusenberg$ brew install --HEAD sanelson/freecad/freecad
==> Installing freecad from sanelson/freecad
==> Installing dependencies for sanelson/freecad/freecad: pivy
==> Installing sanelson/freecad/freecad dependency: pivy
==> Cloning https://bitbucket.org/Coin3D/pivy
Updating /Users/markusenberg/Library/Caches/Homebrew/pivy--hg
==> python setup.py install --prefix=/usr/local/Cellar/pivy/HEAD-8eab90908f2a
Error: Empty installation

macfree, 2016-08-08:
using the without external pivy option worked it seems, but are there some paths that needs to be set?

macfree, 2016-08-08:
It worked, but failed calling up issue #15

evansharp, 2016-09-21:
I also have this exact issue.

The pivy dep install script is pulling the HEAD of pivy from bitbucket and installing it to `/usr/local/Cellar/pivy/HEAD-8eab90908f2a` rather than `../pivy/0.5.0` where the freecad.rb looks for it. 

`brew link pivy` creates 22 syslinks, but none apparently send a memo to freecad's build script. My ruby is non-existent or I'd go looking myself.

bblacey, 2016-10-07:
I believe if you have mercurial installed, as referenced by #10, this will work.  I also encourage you guys to try the new, but not quite finished freecad.rb script to perform installs.  If you are installing on Yosemite, it will use FreeCAD-specific bottles that we have created and hosted in the release section of this repo.  Finally, there is an outstanding issue with nglib on El Capitan so unless you are modifying the FreeCAD source code, you can just grab the latest pre-built build from the [FreeCAD GitHub Releases](https://github.com/FreeCAD/FreeCAD/releases/tag/0.17_pre)

closed 2016-10-07 by bblacey