FreeCAD/homebrew-freecad #20: Transfer repo ownership to FreeCAD Github account?

bblacey, 2016-09-30:
Hey Sam @sanelson, 

For about a year, we have enjoyed continuous integration builds for FreeCAD that rely upon the pivy and coin formula that you originally created.  However, some of these formula are becoming outdated as the world around them changes.  In addition, to improve the speed of our CI builds, we also have a developed a FreeCAD-ports-cache that employs a custom homebrew caching scheme and custom bottles.  I would like make some changes to your formulae to improve them a bit but also to reference a keg for pre-built bottles to improve build times.  I plan to fork your repo but it seems that these FreeCAD-specific formula should be hosted on the FreeCAD web-site, especially pre-built bottles.  Are you open to the idea of transferring your repo to the FreeCAD website?  I can add you as a collaborator once the transfer is complete.  

This seems like the best approach for the FreeCAD community, your prior art, and the best way to ensure we retain an accurate record of your original work (also, any forks will follow the transfer).

Thanks,
Bruce

sanelson, 2016-10-01:
Hi Bruce!

Thanks for reaching out regarding the homebrew-freecad repo.  I think this is a great idea, I have unfortunately been unable to contribute much to the repo lately and would welcome it becoming a more official part of the FreeCAD ecosystem and getting the attention it really deserves.  I'd also love to remain a collaborator on the repo if possible so I can help out with maintenance when I get the time.

I haven't performed a repo transfer before so if you have any tips on that, please let me know.

Best regards,

Sam

bblacey, 2016-10-01:
Hi Sam, excellent.  The transfer is very straight-forward - instructions are [here](https://help.github.com/articles/transferring-a-repository-owned-by-your-personal-account/).  Once the transfer is complete, I will add you as a collaborator to the repo and look forward your jumping back in to help as time permits.  Thanks in advance for transferring your prior work/art.

Cheers,
Bruce

evansharp, 2016-10-01:
Thanks for this collaboration guys! Some of us can only stand on the shoulders of giants like y'all.

sanelson, 2016-10-02:
Hi Bruce,

I attempted to transfer the repository to the "FreeCAD" organization on GitHub but since I don't have Admin rights in that organization, it didn't work.  I'm assuming you have admin rights in the "FreeCAD" organization, would it be better to just transfer the repo to you and then you can then transfer it to the "FreeCAD" organization?

Thanks!

Sam

bblacey, 2016-10-02:
Hi Sam,

Sounds like a plan.  I just renamed my fork of your repo to avoid any conflict during the transfer so give it a shot and let's see what happens.

Thanks,
Bruce

sanelson, 2016-10-02:
Hey Bruce,

Looks github detected your fork of my repo and wouldn't transfer it.  Which seems kind of silly to me :).  Here's the error I get:

`bblacey already has a repository in the sanelson/homebrew-freecad network`

I'll poke around a bit and see if I can figure out a way around this.  If you have any ideas, please let me know.  Another option would be for me to transfer the repo to someone else on the FreeCAD dev team who hasn't forked homebrew-freecad and see if they can then transfer it to the main FreeCAD organization.

Thanks,

Sam

bblacey, 2016-10-02:
Sam, I just deleted my fork.  Give it another shot.

sanelson, 2016-10-03:
Hey Bruce,

I was able to start the transfer after you deleted the fork.  Let me know if you need anything else.

Thanks!

Sam

bblacey, 2016-10-03:
Sam, looks good so far  I am updating this issue while the repository is on my personal github account.  My next step is to transfer the repo to FreeCAD.  Thanks!

bblacey, 2016-10-03:
Sam, done!  Your original work has been transferred in entirety to the FreeCAD organization.  I have confirmed that you are a collaborator on the repo.  Thanks for your contribution and support!

closed 2016-10-03 by bblacey

sanelson, 2016-10-04:
Thanks Bruce!  I really appreciate you taking this on and keeping the momentum going for this project.

Best regards,

Sam