FreeCAD/homebrew-freecad #24: Refactor freecad.rb for FreeCAD 0.16+, Travis-CI and FreeCAD-ports-cache

bblacey, 2016-10-03:
Some of the changes necessary:

- Now hosted on github (not sourceforge), so url resource needs to be updated accordingly
- 0.16 has since been released
- Opencascade 7 is preferred over OCE (OCE is languishing and is somewhat obviated by OCCT 7)
- FreeCAD is supported on Mavericks and later
- Formula logic can be simplified because CMake configs are more sophisticated (i.e. better dependency location)
- New dependencies (vtk, nglib, med-file)
- Tailored bottles 
- New cmake build defines

bblacey, 2016-10-07:
This formula is still _in work_ but the following configurations have been tested and this formula is now used by Travis CI to install all FreeCAD dependencies for each continuous integration build.

Install FreeCAD dependencies
`brew install --only-dependencies FreeCAD/freecad/freecad`

Install FreeCAD 0.17 head
`brew install FreeCAD/freecad/freecad -HEAD`

Install FreeCAD 0.17_pre
`brew install FreeCAD/freecad/freecad`

Still need to add ability to package as an Application but I see this mainly for installing FreeCAD dependencies and most non-developers should simply use the CI builds available in the Releases section of the FreeCAD/FreeCAD Github repo.

ianrrees, 2018-01-02:
Safe to say this is now done?

bblacey, 2018-01-02:
Yes.  Thanks :)

commit 856ede0d (2018-01-02): Tap layout complies with homebrew guideline

Fixes #24
closed 2018-01-02 by bblacey