FreeCAD/homebrew-freecad #26: nglib fails to brew on El Capitan when --with-opencascade option is used

bblacey, 2016-10-04:
The [nglib](https://github.com/FreeCAD/homebrew-freecad/blob/master/nglib.rb) formula fails to build on El Capitan using Xcode 7.3.1 when the --with-opencascade option is used.

- The same formula, builds fine on Mavericks and Yosemite using --with-opencascade against OCCT7
- The same formula builds on El Capitan is the user does not specify --with-opencascade

More details in this [forum post](http://forum.freecadweb.org/viewtopic.php?f=10&t=17557&p=139772#p139772)

bblacey, 2016-10-04:
@ianrrees, two other observations for this formula that you might want to consider.

1. [Patching configure.ac](https://github.com/FreeCAD/homebrew-freecad/blob/master/nglib.rb#L96-L107) causes make to regenerate the configure script by invoking automake/aclocal.  This results in errors during make unless you have automake installed however `brew install automake` does not seem to resolve the errors.  All of that said, perhaps we don't need to patch configure.ac because we are [patching the configure script](https://github.com/FreeCAD/homebrew-freecad/blob/master/nglib.rb#L84-L95) in the distro with the same changes...  That way make would not try to rebuild the patched configure script. Unless I am missing something, I see two options and we should only need one.
 - Retain configure.ac patch, add appropriate :build options to ensure that make can regenerate the configure script.  This makes the configure script patch redundant and should be eliminated.
 - Remove configure.ac patch to avoid regeneration.  It doesn't work as is and is redundant with the configure script patches.

2. The opencascade [cad_kernal opt_prefix is appended with /include](https://github.com/FreeCAD/homebrew-freecad/blob/master/nglib.rb#L37).  This results in invalid paths such as `OCCFLAGS = -DOCCGEOMETRY -I/usr/local/opt/opencascade/include/include/opencascade -I/usr/local/include/opencascade` and `OCCLIBS = -L/usr/local/opt/opencascade/include/lib` that are invalid/non-existant.  A quick test using the following change, corrected the paths.  Why this doesn't affect builds on Mavericks and Yosemite is beyond me.

```
diff --git a/nglib.rb b/nglib.rb
index 80d8667..be07308 100644
--- a/nglib.rb
+++ b/nglib.rb
@@ -34,7 +34,7 @@ class Nglib < Formula
       cad_kernel = Formula[build.with?("opencascade") ? "opencascade" : "oce"]
 
       if build.with? "opencascade"
-        args << "--with-occ=#{cad_kernel.opt_prefix}/include"
+        args << "--with-occ=#{cad_kernel.opt_prefix}"
 
       else
         args << "--with-occ=#{cad_kernel.opt_prefix}/include/oce"
```

bblacey, 2016-11-07:
Looks like this may be an Apple toolchain problem - https://openradar.appspot.com/27822079

bblacey, 2016-11-07:
@ianrrees is aware and on this issue - see https://github.com/Homebrew/homebrew-science/issues/3624 and http://www.freecadweb.org/tracker/view.php?id=2500

ianrrees, 2016-11-12:
Hey Bruce, sorry I've been super busy for the last couple months and haven't been able to contribute to FreeCAD.  I'm going to try to get my machine updated to have a look at that issue 3624 later today.

bblacey, 2016-11-12:
@ianrrees, good to hear from you.  I completely understand being too busy ;)  You might review #33 when you investigate because it seems like the same underlying issue but with different output when built on Sierra...  I am hopeful that you can figure it out because I haven't been able to...  That said, I did send an collaborator invite for this repo - I will resend it in case it was lost in the shuffle.

ianrrees, 2016-11-12:
Yes, I accepted the collaborator invite a few minutes ago, thanks!

bblacey, 2016-11-12:
Ah, well that explains it.  I went to re-send the invite and saw you were a collaborator so I edited my remark above to remove that part ;)  Thanks in advance for the help.

bblacey, 2016-11-12:
@ianrrees, I was reviewing your [nglib PR](https://github.com/Homebrew/homebrew-science/pull/4587) and realize that point 1) in my [comment above](https://github.com/FreeCAD/homebrew-freecad/issues/26#issuecomment-251293870) about patching configure.ac is incorrect.  That must have been a patch introduced by @vejmarie, so you can disregard however I believe my point 2) is still valid.  Thoughts?

ianrrees, 2016-11-13:
Yeah, configure.ac doesn't matter in the current formula since we're not running autoconf during the homebrew build process.

Haven't had a chance to try --with-opencascade against the new version of opencascade, perhaps in the next few days.  I'm pretty sure I had it working for the initial commit, so perhaps we'll need to make a condition on the opencascade version.

bblacey, 2016-11-13:
Ian, are you planning to also upgrade the nglib formula in this freecad-homebrew repo?  The one here supports OCCT7 and is used to build a custom bottle with the --with-opencascade option.

ianrrees, 2016-11-14:
Not immediately, feel free to adapt those changes here if you have a minute.  I think it would be best if we can get to a point where the homebrew-science formula for nglib does what we need for FreeCAD, so was going to work towards that goal.

bblacey, 2016-11-14:
Ok, OCCT7 support falls into "nglib does what we need for FreeCAD" hence my question...  Maybe that is already part of your goal because hombrew-science/opencascade switched to OCCT7 a couple months ago.

bblacey, 2017-01-09:
@ianrrees - I am working on patching the occt7 nglib formula here to incorporate your recent changes in homebrew/science however the [Makefile.in patch](https://github.com/Homebrew/homebrew-science/blob/bde9c1b2ef944516cb6a055df032a8cbba9a3cc0/nglib.rb#L118-L132) does not exist in the Makefile.in if I simply download the netgen 5.3.1 source using```curl -L -o - "https://downloads.sourceforge.net/project/netgen-mesher/netgen-mesher/5.3/netgen-5.3.1.tar.gz" | tar xvz```.  It doesn't seem to be a showstopper because it is resolved with a fuzzy merge but I'm wondering if your intended change is now absent.

```
$ brew install --interactive freecad/freecad/nglib
==> Installing nglib from freecad/freecad
==> Downloading https://downloads.sourceforge.net/project/netgen-mesher/netgen-mesher/5.3/netgen-5.3.1.tar.gz
Already downloaded: /Users/blacey/Library/Caches/Homebrew/nglib-5.3.1.tar.gz
==> Patching
patching file configure
patching file libsrc/occ/Partition_Loop2d.cxx
patching file ng/Makefile.in
Hunk #1 succeeded at 327 with fuzz 1.
patching file libsrc/meshing/improve2.hpp
patching file libsrc/meshing/meshclass.hpp
patching file libsrc/meshing/meshtype.hpp
patching file libsrc/occ/Partition_Inter2d.hxx
patching file libsrc/occ/Partition_Inter3d.hxx
patching file libsrc/occ/Partition_Loop.hxx
patching file libsrc/occ/Partition_Loop2d.hxx
patching file libsrc/occ/Partition_Loop3d.hxx
patching file libsrc/occ/Partition_Spliter.hxx
patching file libsrc/occ/occgeom.cpp
patching file libsrc/occ/occgeom.hpp
patching file libsrc/occ/occmeshsurf.hpp
patching file libsrc/occ/vsocc.cpp
patching file ng/ngpkg.cpp
==> Entering interactive mode
Type `exit' to return and finalize the installation
Install to this prefix: /usr/local/Cellar/nglib/5.3.1

bash-3.2$ diff Makefile.in /tmp/netgen-5.3.1/Makefile.in 
```

bblacey, 2017-01-09:
cc: @jimhigson because this is the same root cause as #33 that I closed as a duplicate of this issue.

ianrrees, 2017-01-10:
Hi @bblacey, sorry for the slow response (and lack of work on my part...)

The recent "fix" for building nglib is, I think, more to do with the new XCode rather than the new MacOS.  It was just to cut out the -lTK* from the configure script - see https://github.com/Homebrew/homebrew-science/pull/4587/files .  Very hacky; my guess is that there's a bug in ld where it goes in to an infinite loop in that situation (several non-existent libraries specified), but I haven't yet gotten around to making a simpler test case.

I think there are some OCCT7-related changes that aren't my work, can't remember.  Hope this helps?

bblacey, 2017-01-10:
@ianrrees - thanks.  I will include the Makefile.in patch even though it is benign in its current form.

bblacey, 2017-01-10:
@ianrrees - the homebrew patch regresses opencascade.  Re-opening to resolve.  You will need to include the same change to the Homebrew-science nglib patch.

closed 2017-01-10 by bblacey

peterl94, 2017-01-15:
@bblacey, thank you for taking care of this! Works fine with macOS 10.12/xcode 8.2.1. Haven't had time for FreeCAD these past couple of weeks.

commit e21da7d8 (2017-01-10): Merge Hombrew nglib patch to resolve ld issue on 10.11+

  * Updated formula to include Homebrew/homebrew-science#4587 fix
  * Corrected opencascade include path
  * Removed unneccesarry Makefile.ac patch
  * Bottled for Yosemite, El Capitan and Sierra

Fixes #26
commit 616c43f7 (2017-01-10): Set opencascade include path

  * Fix OCCFLAGS patch regression from Homebrew/homebrew-science#4587
  * Re-bottled for Yosemite, El Capitan and Sierra

Fixes #26