FreeCAD/homebrew-freecad #33: Build fails on Sierra

jimhigson, 2016-10-13:
With a fully updated brew on Sierra I get this:

```
patching file libsrc/occ/vsocc.cpp
patching file ng/ngpkg.cpp
==> ./configure --prefix=/usr/local/Cellar/nglib/5.3.1 --disable-gui --enable-nglib --enable-occ --with-occ=/usr/local/o
==> make install
Last 15 lines from /Users/jimhigson/Library/Logs/Homebrew/nglib/02.make:
0  0x10f1905cc  __assert_rtn + 144
1  0x10f1dd7c8  ld::tool::InputFiles::findDylib(char const*, ld::dylib::File const*, bool) + 882
2  0x10f17232b  generic::dylib::File<x86_64>::processIndirectLibraries(ld::dylib::File::DylibHandler*, bool) + 83
3  0x10f1deab9  ld::tool::InputFiles::createIndirectDylibs() + 271
4  0x10f1dfcff  ld::tool::InputFiles::forEachInitialAtom(ld::File::AtomHandler&, ld::Internal&) + 513
5  0x10f1eb318  ld::tool::Resolver::resolve() + 44
6  0x10f19152b  main + 1050
7  0x7fffe8b7b255  start + 1
A linker snapshot was created at:
	/tmp/libocc.0.dylib-2016-09-13-173230.ld-snapshot
ld: Assertion failed: (counter() < 0xffff), function nextCounter, file /Library/Caches/com.apple.xbs/Sources/ld64/ld64-274.1/src/ld/ld.hpp, line 108.
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [libocc.la] Error 1
make[1]: *** [install-recursive] Error 1
make: *** [install-recursive] Error 1

READ THIS: https://git.io/brew-troubleshooting
If reporting this issue please do so at (not Homebrew/brew):
  https://github.com/freecad/homebrew-freecad/issues

These open issues may also help:
Promote nglib formula for OCCT7 to homebrew/science as pull request https://github.com/FreeCAD/homebrew-freecad/issues/30
nglib fails to brew on El Capitan when --with-opencascade option is used https://github.com/FreeCAD/homebrew-freecad/issues/26
```

bblacey, 2016-10-13:
Hi Jim, thanks for reporting this issue.  I haven't tried to build on Sierra yet but I will to see if I can reproduce the issue.  There is an open nglib build issue #26, where I was not able to build on El Capitan and it is possible that it is the same issue perhaps manifested slightly differently which may give insight into the root cause.  In the mean time, you can grab builds from our [FreeCAD releases on GitHub](https://github.com/FreeCAD/FreeCAD/releases)

bblacey, 2016-10-13:
@jimhigson, here's another quick follow-up; My test failed out of the gate on macOS Sierra because QT is not supported on macOS > 10.11.  How did you get around that issue?

```
brew install --only-dependencies --verbose FreeCAD/freecad/freecad --with-packaging-utils
==> Installing freecad from freecad/freecad
qt: This formula either does not compile or function as expected on macOS
versions newer than El Capitan due to an upstream incompatibility.
Error: An unsatisfied requirement failed this build.
```

```
$ brew info qt
qt: stable 4.8.7, HEAD
Cross-platform application and UI framework
https://www.qt.io/
Not installed
From: https://github.com/Homebrew/homebrew-core/blob/master/Formula/qt.rb
==> Dependencies
Required: openssl ✘
Optional: dbus ✘, mysql ✘, postgresql ✘
==> Requirements
Build: git ✔
Required: macOS <= 10.11 ✘
==> Options
--with-dbus
	Build with dbus support
--with-docs
	Build documentation
--with-mysql
	Build with mysql support
--with-postgresql
```

jimhigson, 2016-10-13:
To my knowledge I didn't do anything to get around that. Maybe it found an old install from before I upgraded to Sierra? Let me know if I can provide any information that would help here.

Thanks for the links to the snapshot builds - I didn't know those existed.

bblacey, 2016-11-07:
Further to [gift's forum post](http://forum.freecadweb.org/viewtopic.php?f=10&t=17832&p=143870#p143509), you can install qt4 using cartr's qt4 port.  I have reproduced this issue on Sierra and it is the same nglib issue #26 that occurs on El Capitan.

jimhigson, 2016-12-22:
Is there any workaround for building on Sierra?

bblacey, 2016-12-23:
Hi Jim, I believe @ianrrees has already fixed the nglib issue that prevented builds on Sierra but neither of us has had a chance to back-port the changes to the OCCT7 nglib formula.  Unfortunately I am traveling in Indonesia through January 7th with very limited access...  If you are up for it, you can clone this repo and back-port ian's changes to the nglib formula - I believe it will work.   Details are here https://github.com/Homebrew/homebrew-science/issues/3624 and here https://github.com/FreeCAD/homebrew-freecad/issues/26

ianrrees, 2016-12-24:
Hi @jimhigson, sorry I've been super busy with work stuff lately as Bruce mentioned.  I believe the nglib in the Homebrew main line will build for you, but I don't actually have a Sierra machine to work on.  If you could test it and report back, that would be great!

As has been discussed on another thread, I think the ideal solution is to get the changes we need in to the main Homebrew recipe (or we could get rid of that one - I added it initially to support FreeCAD on Mac) so that we're not trying to maintain two copies of essentially the same thing.

jimhigson, 2016-12-24:
This is what I get now:

I've previously done `brew update` and `brew upgrade`

```
jimhigson@budapest ~/3d> brew install --HEAD freecad/freecad/freecad
==> Installing freecad from freecad/freecad
==> Installing dependencies for freecad/freecad/freecad: FreeCAD/freecad/nglib
==> Installing freecad/freecad/freecad dependency: FreeCAD/freecad/nglib
==> Downloading https://downloads.sourceforge.net/project/netgen-mesher/netgen-mesher/5.3/netgen-5.3.1.tar.gz
Already downloaded: /Users/jimhigson/Library/Caches/Homebrew/nglib-5.3.1.tar.gz
==> Patching
patching file configure
patching file configure.ac
patching file libsrc/meshing/improve2.hpp
patching file libsrc/meshing/meshclass.hpp
patching file libsrc/meshing/meshtype.hpp
patching file libsrc/occ/Partition_Inter2d.hxx
patching file libsrc/occ/Partition_Inter3d.hxx
patching file libsrc/occ/Partition_Loop.cxx
patching file libsrc/occ/Partition_Loop.hxx
patching file libsrc/occ/Partition_Loop2d.cxx
patching file libsrc/occ/Partition_Loop2d.hxx
patching file libsrc/occ/Partition_Loop3d.hxx
patching file libsrc/occ/Partition_Spliter.hxx
patching file libsrc/occ/occgeom.cpp
patching file libsrc/occ/occgeom.hpp
patching file libsrc/occ/occmeshsurf.hpp
patching file libsrc/occ/vsocc.cpp
patching file ng/ngpkg.cpp
==> ./configure --prefix=/usr/local/Cellar/nglib/5.3.1 --disable-gui --enable-nglib --enable-occ --with-occ=/usr/local/opt/opencascade/include
==> make install
Last 15 lines from /Users/jimhigson/Library/Logs/Homebrew/nglib/02.make:
libtool: link: clang++ -dynamiclib -Wl,-flat_namespace -Wl,-undefined -Wl,suppress -o .libs/libocc.0.dylib  .libs/Partition_Inter2d.o .libs/Partition_Inter3d.o .libs/Partition_Loop.o .libs/Partition_Loop2d.o .libs/Partition_Loop3d.o .libs/Partition_Spliter.o .libs/occconstruction.o .libs/occgenmesh.o .libs/occgeom.o .libs/occmeshsurf.o   -L/usr/local/opt/opencascade/include/lib -lTKernel -lTKGeomBase -lTKMath -lTKG2d -lTKG3d -lTKXSBase -lTKOffset -lTKFillet -lTKShHealing -lTKMesh -lTKMeshVS -lTKTopAlgo -lTKGeomAlgo -lTKBool -lTKPrim -lTKBO -lTKIGES -lTKBRep -lTKSTEPBase -lTKSTEP -lTKSTL -lTKSTEPAttr -lTKSTEP209 -lTKXDESTEP -lTKXDEIGES -lTKXCAF -lTKLCAF -lFWOSPlugin  -O2   -install_name  /usr/local/Cellar/nglib/5.3.1/lib/libocc.0.dylib -compatibility_version 1 -current_version 1.0 -Wl,-single_module
0  0x10c477608  __assert_rtn + 144
1  0x10c4c46b8  ld::tool::InputFiles::findDylib(char const*, ld::dylib::File const*, bool) + 882
2  0x10c45936b  generic::dylib::File<x86_64>::processIndirectLibraries(ld::dylib::File::DylibHandler*, bool) + 83
3  0x10c4c59a9  ld::tool::InputFiles::createIndirectDylibs() + 271
4  0x10c4c6bef  ld::tool::InputFiles::forEachInitialAtom(ld::File::AtomHandler&, ld::Internal&) + 513
5  0x10c4d2208  ld::tool::Resolver::resolve() + 44
6  0x10c478567  main + 1050
A linker snapshot was created at:
	/tmp/libocc.0.dylib-2016-11-24-133043.ld-snapshot
ld: Assertion failed: (counter() < 0xffff), function nextCounter, file /Library/Caches/com.apple.xbs/Sources/ld64/ld64-274.2/src/ld/ld.hpp, line 108.
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [libocc.la] Error 1
make[1]: *** [install-recursive] Error 1
make: *** [install-recursive] Error 1

If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
https://github.com/freecad/homebrew-freecad/issues

These open issues may also help:
nglib fails to brew on El Capitan when --with-opencascade option is used https://github.com/FreeCAD/homebrew-freecad/issues/26
Promote nglib formula for OCCT7 to homebrew/science as pull request https://github.com/FreeCAD/homebrew-freecad/issues/30
```

ianrrees, 2016-12-26:
Hi Jim,

     Could you try instead to "brew install homebrew/science/nglib" and see what happens?  I'm not sure whether the current FreeCAD Homebrew recipe will be happy with that nglib recipe, but am interested in whether that nglib can be built on Sierra.

Sorry, I misspoke earlier - the nglib isn't in the homebrew main line as such, but is in homebrew-science.  You'll probably have already "tapped" homebrew-science, but if not you'll get an error above, and can then "brew tap homebrew/science".

jimhigson, 2016-12-26:
@ianrrees - unfortunately very similar:

```
jimhigson@budapest ~> brew install homebrew/science/nglib
Updating Homebrew...
==> Auto-updated Homebrew!
Updated 6 taps (homebrew/boneyard, homebrew/core, homebrew/games, homebrew/python, homebrew/versions, homebrew/x11).
==> New Formulae
handbrake                  homebrew/games/bubbros     homebrew/games/zsdx
homebrew/boneyard/pixie    homebrew/games/sdlpop      homebrew/games/zsxd
==> Updated Formulae
agda                                     p11-kit
emscripten                               parallel
exploitdb                                pazpar2
gofabric8                                pngquant
graphviz ✔                               proxychains-ng
homebrew/games/minetest                  pwntools
homebrew/games/roll                      ranger
homebrew/games/stormlib                  rmlint
homebrew/python/pygame                   ruby
homebrew/versions/jenkins-lts            ruby-build
jenkins                                  ruby-install
libstfl                                  skinny
libwebsockets                            suomi-malaga-voikko
lighttpd                                 ttyd
mikutter                                 vim
mongo-orchestration                      vnstat
mosquitto                                voltdb
mpv                                      zinc
notmuch                                  znc
==> Deleted Formulae
homebrew/x11/pixie

==> Installing nglib from homebrew/science
==> Downloading https://homebrew.bintray.com/bottles-science/nglib-5.3.1.sierra.
######################################################################## 100.0%
==> Pouring nglib-5.3.1.sierra.bottle.1.tar.gz
🍺  /usr/local/Cellar/nglib/5.3.1: 208 files, 5.3M
jimhigson@budapest ~> brew install --HEAD freecad/freecad/freecad
==> Installing freecad from freecad/freecad
==> Installing dependencies for freecad/freecad/freecad: FreeCAD/freecad/nglib
==> Installing freecad/freecad/freecad dependency: FreeCAD/freecad/nglib
==> Downloading https://downloads.sourceforge.net/project/netgen-mesher/netgen-mesher/5.3/netgen-5.3.1.tar.gz
Already downloaded: /Users/jimhigson/Library/Caches/Homebrew/nglib-5.3.1.tar.gz
==> Patching
patching file configure
patching file configure.ac
patching file libsrc/meshing/improve2.hpp
patching file libsrc/meshing/meshclass.hpp
patching file libsrc/meshing/meshtype.hpp
patching file libsrc/occ/Partition_Inter2d.hxx
patching file libsrc/occ/Partition_Inter3d.hxx
patching file libsrc/occ/Partition_Loop.cxx
patching file libsrc/occ/Partition_Loop.hxx
patching file libsrc/occ/Partition_Loop2d.cxx
patching file libsrc/occ/Partition_Loop2d.hxx
patching file libsrc/occ/Partition_Loop3d.hxx
patching file libsrc/occ/Partition_Spliter.hxx
patching file libsrc/occ/occgeom.cpp
patching file libsrc/occ/occgeom.hpp
patching file libsrc/occ/occmeshsurf.hpp
patching file libsrc/occ/vsocc.cpp
patching file ng/ngpkg.cpp
==> ./configure --prefix=/usr/local/Cellar/nglib/5.3.1 --disable-gui --enable-nglib --enable-occ --with-occ=/usr/local/opt/opencascade/include
==> make install
Last 15 lines from /Users/jimhigson/Library/Logs/Homebrew/nglib/02.make:
0  0x106263608  __assert_rtn + 144
1  0x1062b06b8  ld::tool::InputFiles::findDylib(char const*, ld::dylib::File const*, bool) + 882
2  0x10624536b  generic::dylib::File<x86_64>::processIndirectLibraries(ld::dylib::File::DylibHandler*, bool) + 83
3  0x1062b19a9  ld::tool::InputFiles::createIndirectDylibs() + 271
4  0x1062b2bef  ld::tool::InputFiles::forEachInitialAtom(ld::File::AtomHandler&, ld::Internal&) + 513
5  0x1062be208  ld::tool::Resolver::resolve() + 44
6  0x106264567  main + 1050
7  0x7fff969e5255  start + 1
A linker snapshot was created at:
	/tmp/libocc.0.dylib-2016-11-26-101439.ld-snapshot
ld: Assertion failed: (counter() < 0xffff), function nextCounter, file /Library/Caches/com.apple.xbs/Sources/ld64/ld64-274.2/src/ld/ld.hpp, line 108.
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [libocc.la] Error 1
make[1]: *** [install-recursive] Error 1
make: *** [install-recursive] Error 1

If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
https://github.com/freecad/homebrew-freecad/issues

These open issues may also help:
nglib fails to brew on El Capitan when --with-opencascade option is used https://github.com/FreeCAD/homebrew-freecad/issues/26
Promote nglib formula for OCCT7 to homebrew/science as pull request https://github.com/FreeCAD/homebrew-freecad/issues/30
```

ianrrees, 2016-12-26:
Hmm, that's a bummer.  I'm heading to the hills for a few days, but will try to get back on this in the next week or two.  Thanks for testing!

peterl94, 2016-12-29:
@ianrrees, @bblacey, I recently upgraded to Sierra. Your fix in homebrew/science does resolve the above error . The reason @jimhigson is still getting it is because the freecad recipe still specifies the nglib recipe from homebrew-freecad. However, am I missing something? The opencascade include path is not correct for homebrew occt 7 (it was looking for headers in opencascade/inc) and after I corrected that it fails because Handle_BRepAlgo_AsDes.hxx does not exist.

peterl94, 2016-12-29:
Ah, after reading a litter more of these issues am I correct in thinking that the recipe in science doesn't support occt7, only the one in this repo? I applied @ianrrees patch to hombrew-freecad/nglib and it built fine.

bblacey, 2017-01-07:
Hi @peterl94, I was in Indonesia for the past 3 weeks during much of which I didn't have Internet access and am just catching up.  I noticed the occ paths were incorrect in https://github.com/FreeCAD/homebrew-freecad/issues/26#issuecomment-251293870.  Neither Ian nor I have had a chance to update the nglib formula here with Ian's latest patch.  You are also a contributor on this repo so feel free to provide an update if you have time - hint, hint ;)

bblacey, 2017-01-09:
@ianrrees - I am working on patching the occt7 nglib formula here to incorporate your recent changes in homebrew/science however the [Makefile.in patch](https://github.com/Homebrew/homebrew-science/blob/bde9c1b2ef944516cb6a055df032a8cbba9a3cc0/nglib.rb#L118-L132) does not exist in the Makefile.in if I simply download the netgen 5.3.1 source using```curl -L -o - "https://downloads.sourceforge.net/project/netgen-mesher/netgen-mesher/5.3/netgen-5.3.1.tar.gz" | tar xvz```.  It doesn't seem to be a showstopper because it is resolved with a fuzzy merge but I'm wondering if your intended change is now absent.

bblacey, 2017-01-09:
Closing as duplicate of #26 which I plan to resolve as soon as @ianrrees answers my outstanding question.

closed 2017-01-09 by bblacey

jimhigson, 2017-01-15:
I now get a different failure. Not sure if this is the same issue as #26 

```
jimhigson@budapest ~> brew install --HEAD freecad/freecad/freecad
==> Installing freecad from freecad/freecad
==> Cloning https://github.com/FreeCAD/FreeCAD.git
Updating /Users/jimhigson/Library/Caches/Homebrew/freecad--git
==> Checking out branch master
==> Caught inreplace exception
==> cmake .. -DCMAKE_C_FLAGS_RELEASE=-DNDEBUG -DCMAKE_CXX_FLAGS_RELEASE=-DNDEBUG -DCMAKE_INSTALL_PREFIX=/usr/local/Cellar/freecad/HEAD-d8afbe5 -DCMAKE_BUILD_TYPE=Release -DCM
==> make install
Last 15 lines from /Users/jimhigson/Library/Logs/Homebrew/freecad/02.make:
[ 10%] Building CXX object src/Base/CMakeFiles/FreeCADBase.dir/__/zipios++/zipheadio.cpp.o
cd /tmp/freecad-20170115-27569-17umk3/Build/src/Base && /usr/local/bin/ccache /usr/local/Homebrew/Library/Homebrew/shims/super/clang++   -DFreeCADBase_EXPORTS -DHAVE_CONFIG_H -DHAVE_SWIG=1 -DQT_CORE_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_NO_DEBUG -DQT_OPENGL_LIB -DQT_SVG_LIB -DQT_UITOOLS_LIB -DQT_WEBKIT_LIB -DQT_XML_LIB -D_OCC64 -I/tmp/freecad-20170115-27569-17umk3/Build -isystem /usr/local/Cellar/qt/4.8.7_3/include -iframework /usr/local/Cellar/qt/4.8.7_3/lib -isystem /usr/local/Cellar/qt/4.8.7_3/include/QtOpenGL -isystem /usr/local/Cellar/qt/4.8.7_3/include/QtSvg -isystem /usr/local/Cellar/qt/4.8.7_3/include/QtUiTools -isystem /usr/local/Cellar/qt/4.8.7_3/include/QtWebKit -isystem /usr/local/Cellar/qt/4.8.7_3/include/QtGui -isystem /usr/local/Cellar/qt/4.8.7_3/include/QtXml -isystem /usr/local/Cellar/qt/4.8.7_3/include/QtNetwork -isystem /usr/local/Cellar/qt/4.8.7_3/lib/QtCore.framework/Headers -I/tmp/freecad-20170115-27569-17umk3/Build/src -I/tmp/freecad-20170115-27569-17umk3/src -I/tmp/freecad-20170115-27569-17umk3/Build/src/Base -I/tmp/freecad-20170115-27569-17umk3/src/Base -I/usr/local/include -I/usr/local/Cellar/python/2.7.12/Frameworks/Python.framework/Versions/2.7/include/python2.7  -Wall -Wextra -Wno-write-strings  -std=c++11 -D_OCC64 -DNDEBUG -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk -mmacosx-version-min=10.12 -fPIC   -o CMakeFiles/FreeCADBase.dir/__/zipios++/zipheadio.cpp.o -c /tmp/freecad-20170115-27569-17umk3/src/zipios++/zipheadio.cpp
[ 10%] Building CXX object src/Base/CMakeFiles/FreeCADBase.dir/__/zipios++/zipinputstream.cpp.o
cd /tmp/freecad-20170115-27569-17umk3/Build/src/Base && /usr/local/bin/ccache /usr/local/Homebrew/Library/Homebrew/shims/super/clang++   -DFreeCADBase_EXPORTS -DHAVE_CONFIG_H -DHAVE_SWIG=1 -DQT_CORE_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_NO_DEBUG -DQT_OPENGL_LIB -DQT_SVG_LIB -DQT_UITOOLS_LIB -DQT_WEBKIT_LIB -DQT_XML_LIB -D_OCC64 -I/tmp/freecad-20170115-27569-17umk3/Build -isystem /usr/local/Cellar/qt/4.8.7_3/include -iframework /usr/local/Cellar/qt/4.8.7_3/lib -isystem /usr/local/Cellar/qt/4.8.7_3/include/QtOpenGL -isystem /usr/local/Cellar/qt/4.8.7_3/include/QtSvg -isystem /usr/local/Cellar/qt/4.8.7_3/include/QtUiTools -isystem /usr/local/Cellar/qt/4.8.7_3/include/QtWebKit -isystem /usr/local/Cellar/qt/4.8.7_3/include/QtGui -isystem /usr/local/Cellar/qt/4.8.7_3/include/QtXml -isystem /usr/local/Cellar/qt/4.8.7_3/include/QtNetwork -isystem /usr/local/Cellar/qt/4.8.7_3/lib/QtCore.framework/Headers -I/tmp/freecad-20170115-27569-17umk3/Build/src -I/tmp/freecad-20170115-27569-17umk3/src -I/tmp/freecad-20170115-27569-17umk3/Build/src/Base -I/tmp/freecad-20170115-27569-17umk3/src/Base -I/usr/local/include -I/usr/local/Cellar/python/2.7.12/Frameworks/Python.framework/Versions/2.7/include/python2.7  -Wall -Wextra -Wno-write-strings  -std=c++11 -D_OCC64 -DNDEBUG -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk -mmacosx-version-min=10.12 -fPIC   -o CMakeFiles/FreeCADBase.dir/__/zipios++/zipinputstream.cpp.o -c /tmp/freecad-20170115-27569-17umk3/src/zipios++/zipinputstream.cpp
[ 10%] Building CXX object src/Base/CMakeFiles/FreeCADBase.dir/__/zipios++/zipinputstreambuf.cpp.o
cd /tmp/freecad-20170115-27569-17umk3/Build/src/Base && /usr/local/bin/ccache /usr/local/Homebrew/Library/Homebrew/shims/super/clang++   -DFreeCADBase_EXPORTS -DHAVE_CONFIG_H -DHAVE_SWIG=1 -DQT_CORE_LIB -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_NO_DEBUG -DQT_OPENGL_LIB -DQT_SVG_LIB -DQT_UITOOLS_LIB -DQT_WEBKIT_LIB -DQT_XML_LIB -D_OCC64 -I/tmp/freecad-20170115-27569-17umk3/Build -isystem /usr/local/Cellar/qt/4.8.7_3/include -iframework /usr/local/Cellar/qt/4.8.7_3/lib -isystem /usr/local/Cellar/qt/4.8.7_3/include/QtOpenGL -isystem /usr/local/Cellar/qt/4.8.7_3/include/QtSvg -isystem /usr/local/Cellar/qt/4.8.7_3/include/QtUiTools -isystem /usr/local/Cellar/qt/4.8.7_3/include/QtWebKit -isystem /usr/local/Cellar/qt/4.8.7_3/include/QtGui -isystem /usr/local/Cellar/qt/4.8.7_3/include/QtXml -isystem /usr/local/Cellar/qt/4.8.7_3/include/QtNetwork -isystem /usr/local/Cellar/qt/4.8.7_3/lib/QtCore.framework/Headers -I/tmp/freecad-20170115-27569-17umk3/Build/src -I/tmp/freecad-20170115-27569-17umk3/src -I/tmp/freecad-20170115-27569-17umk3/Build/src/Base -I/tmp/freecad-20170115-27569-17umk3/src/Base -I/usr/local/include -I/usr/local/Cellar/python/2.7.12/Frameworks/Python.framework/Versions/2.7/include/python2.7  -Wall -Wextra -Wno-write-strings  -std=c++11 -D_OCC64 -DNDEBUG -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk -mmacosx-version-min=10.12 -fPIC   -o CMakeFiles/FreeCADBase.dir/__/zipios++/zipinputstreambuf.cpp.o -c  [... cut ...] lib/libboost_chrono-mt.dylib /usr/local/lib/libboost_date_time-mt.dylib /usr/local/lib/libboost_atomic-mt.dylib /usr/lib/libz.dylib -framework QtCore
[ 10%] Built target FreeCADBase
make: *** [all] Error 2

If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
https://github.com/freecad/homebrew-freecad/issues

These open issues may also help:
Refactor freecad.rb for FreeCAD 0.16+, Travis-CI and FreeCAD-ports-cache https://github.com/FreeCAD/homebrew-freecad/issues/24
```

bblacey, 2017-01-15:
Jim, I haven't really run out the build from head yet so could you try the following:

```
#  Uninstall existing FreeCAD ports
brew deps FreeCAD/freecad/freecad | xargs brew uninstall --ignore-dependencies

#  Install FreeCAD dependencies
brew tap cartr/qt4
brew tap FreeCAD/freecad
brew install --only-dependencies FreeCAD/freecad/freecad
brew install --only-dependencies FreeCAD/freecad/freecad ##Run twice to ensure all dependencies are satisfied.

#  Clone FreeCAD repo and build once all port dependencies are satisfied
git clone https://github.com/FreeCAD/FreeCAD.git
cd FreeCAD
mkdir build && cd build
cmake -DBUILD_FEM_NETGEN=ON -DFREECAD_USE_EXTERNAL_KDL=ON -DFREECAD_CREATE_MAC_APP=ON -DCMAKE_BUILD_TYPE=DEBUG
```

The advantage of first installing the dependencies and then building within your own repo is that ccache will be installed so future re-compiles will be faster.  On the other hand, the formula will always download a new set of source from GitHub into a clean directory and recompile everything. Once you have the dependencies installed and your own FreeCAD repo clone, you can use ```git pull``` to update your repo from master and then run cmake again from the build directory and it should only re-compile the files necessary to accommodate the git change set since the last compile.

Looking forward to hearing your results.

jimhigson, 2017-01-21:
@bblacey - thanks for the script. Am building now - let's see where this gets.

jimhigson, 2017-01-21:
Guessing you didn't mean `cd build`? That doesn't work. Running cmake in freecad having run all of the above, I get: 

```

CMake Error at CMakeLists.txt:533 (MESSAGE):
  ========================================

  boost not found, install the components:

   * filesystem
   * program_options
   * regex
   * signals
   * system
   * thread

  ========================================
```

I guess that means that the dependencies didn't install properly. At this point I'm not sure I want to continue - my decision to use brew to install software was mostly so that I can just install things - I'm not very experienced with building C programs and at this point I think it makes more sense to put up with an old version of Freecad than to keep trying to build it.

bblacey, 2017-01-21:
Hi Jim, you do want to cd to build and then append .. to the cmake command.  Sorry for the error in the cmake command.  Let me how how it works out.  If it doesn't work, try another git clone of FreeCAD because I am not sure what running cmake in the root might have done to your instance.

jimhigson, 2017-01-21:
Ok, I did `git reset --hard` and then cd into make and `cmake` with `..` on the end. Same error. Looks like brew didn't install all the build dependencies.

bblacey, 2017-01-21:
Hmmm, did you run ```brew install --only-dependencies FreeCAD/freecad/freecad``` twice?  What happens if you run it again?  Does it say, 'all dependencies satisfied'?

Also, please provide the output of ```brew deps FreeCAD/freecad/freecad``` and ```brew list -1```

bblacey, 2017-01-21:
Jim, also, if you want to run the latest and greatest builds, they are available at https://github.com/FreeCAD/FreeCAD/releases/tag/0.17_pre.  The macOS build is automatically built, tested and uploaded each time a PR is merged into master.  You can refer to the SHA on the file name or in the about box to understand the changes since the last build you downloaded.