FreeCAD/homebrew-freecad #35: Add option to build with QT5

bblacey, 2017-01-09:
The broader ecosystem is forging forward with QT5 and initial QT5 support has been added to FreeCAD.  The scope of this issue is to add an option to build FreeCAD with QT5 which also requires moving to pyside2.

bblacey, 2017-01-09:
To resolve CI build issues quickly, I exhumed the qt4 formulae from homebrew/boneyard to provide efficient qt4 support on Travis.  A better approach would be to use cartr/homebrew-qt4 formula repo but currently cartr only provides bottles for Sierra.  I have [provided instructions and offered to support](https://github.com/cartr/homebrew-qt4/issues/8#issuecomment-271370307) to migrate existing homebrew bottles to cartr's bintray account obviating the need for maintaining qt4 formula on our FreeCAD homebrew repo.  This would provide more robust and longer-term qt4 support for the broader community and clear the way for QT5 dependency formula development so building on qt4 would tap-pin cartr/qt4 and building on qt5 would use hombrew/core and FreeCAD/freecad.

bblacey, 2017-01-10:
I have worked with Cartr to provide bottles and updated formulae for legacy QT4 formulae in cartr's QT4 repo under issue cartr/homebrew-qt4#8 and PR cartr/homebrew-qt#9.  Cartr's repo is recognized by the Homebrew Team and community as the official source for QT4 formula.  Once cartr accepts and merges the pull request, we can remove the qt4 formulae (qt, shiboken, pyside and pyside-tools) from the FreeCAD/homebrew-freecad repo. 

There is also an outstanding issue cartr/homebrew-qt4#5 to rename the qt formula to qt@4 following the current Homebrew versioning scheme to explicitly deconflict qt4 from qt5 which is becoming the default (i.e. qt ==> qt@4 and qt5 ==> qt).  Merging the associated PR and closing the rename issue is blocked by a Homebrew bug that is in work.  I believe we should change our FreeCAD/freecad formula to use Cartr's repo AFTER the rename PR is complete.

Once we finalize the QT4 legacy support by moving to Cartr's repo after the rename, I will refresh the ports-cache and begin integrating QT5 support into the FreeCAD/freecad formula.  Ideally, the Homebrew QT5 rename will not be too disruptive.

bblacey, 2017-01-10:
The phased approach is detailed in [this forum post](http://forum.freecadweb.org/viewtopic.php?f=10&t=19724&p=152718#p152718)

ianrrees, 2017-01-15:
As discussed on the forum, it looks we're going to need to stick with Qt <5.7 for a while, because PySide2 isn't yet compatible with Qt >=5.7 ( [PYSIDE-323](https://bugreports.qt.io/browse/PYSIDE-323) and [PYSIDE-322](https://bugreports.qt.io/browse/PYSIDE-322)).  Homebrew has moved on to Qt 5.7 already, and there's been a small change to Homebrew which means the original 5.6.2 recipe doesn't build anymore without a small modification...

Python 3 is in a similar situation, as Homebrew is at 3.6 but Pyside2 only supports up to 3.5.  I suppose that won't be an issue with FreeCAD if we're still using Python 2.7.

Anyway, I've built and installed PySide2 on OSX 10.11 today.  My plan is to make a document describing downgrading to (and sticking with) Qt 5.6 in Homebrew, then make a formula for PySide2, once I've got FreeCAD going with Qt5.

Any thoughts on how well this will work with the CI system?  Building Qt from source takes some time, and it doesn't seem like the Qt 5.6 bottles are still around.

bblacey, 2017-01-15:
@ianrrees - sounds great!  Homebrew still officially supports QT 5.5, including bottles, so that is what I used for my alpha Qt5 testing.  If you think we can stick with QT 5.5, then we won't need to worry about bottling 5.6.  If on the other hand, we need QT 5.6, then once you tweak the formula and commit here, I can make a one-time investment to bottle QT 5.6 for Yosemite, El Capitan and Sierra on some VMs and post them to GitHub.

That said, I have started a qt5 branch locally that includes the following:

- Modified FreeCAD/freecad/freecad.rb with a new option --with-qt4 (i.e. next freecad.rb revision will default to QT5 in what ever form we get it to work)
- Modified Shiboken formula to build Shiboken dev from qt.io
- Began modifying PySide formula to build PySide dev from qt.io but ran into problems that perhaps you have already resolved.  (Here is my post to the qt.io forums - https://forum.qt.io/topic/75150/latest-pyside2-with-qt-5-5-build-failure-on-mac-os-10-10-with-xcode-6-4) that perhaps you have already overcome or it is specific to 10.10/XC6.4/QT5.5...

We are both contributors here so I can push my in-work qt5 branch to the repo so you can merge/add your commits.  We don't need to worry about striving for a clean commit history on our working branch at this point, we can just push what we have working for now.  Once the branch has everything we need, we can make a pass to squash commits and clean up the commit history.  Does that sound good?  If so, we probably want to keep the main discussion on the forum so others working the qt5 can help (e.g. f3nix/Matuesz) and benefit or enjoy this effort vicariously and keep this GitHub thread to items very specific to the qt5 change set we are developing.

It would be awesome if we can chip away at this together.  Thoughts?

ianrrees, 2017-01-15:
All sounds good!

How did you get Homebrew to install Qt 5.5?

I haven't run in to the same problem you mentioned with PySide (yet?), though I did find a bug in their Tetrix test/example...

A Qt5 branch here seems like the right way forward.

bblacey, 2017-01-15:
Using the new Homebrew core version support - ```brew install qt@5.5```

I just found the culprit commit, at least for 5.5 - https://codereview.qt-project.org/gitweb?p=pyside/pyside.git;a=commit;h=eb28469e920696a1d0833c55b5c20c5c5a6f9d5e

I am testing a pyside formula that builds up through the previous commit.  If it passes, I will commit and push the qt5 branch.  That will give you freecad that defaults to qt5, shiboken and pyside for qt5.  I can start work on pyside-tools and perhaps you can see if you can resolve the issue with the aforementioned pyside commit?  Or if you have other good stuff, I push and wait for you to add ;)

bblacey, 2017-01-15:
@ianrrees, I just pushed the in-work/incomplete [qt5 branch](https://github.com/FreeCAD/homebrew-freecad/tree/qt5) for collaboration.  Much more fun to collaborate ;)

**State of affairs on qt5 branch:**

Depends upon qt@5.5
Shiboken ported to qt.io
PySide ported to qt.io but pinned at commit 9f3aba9 due to compilation failure with commit eb28469
FreeCAD formula extended to add optional ```--with-qt4``` but defaults to installing qt@5.5 and associated dependencies.

**Known issues/gaps:**
- Port pyside-tools to qt.io
- Resolve build issue with pyside commit eb28469
- Test with FreeCAD
- ```make install``` erroneously installs /Applications into qtplugins

bblacey, 2017-01-15:
Let's move the general conversation to the [forum for transparency](http://forum.freecadweb.org/viewtopic.php?f=10&t=19724&start=20#p153591)

bblacey, 2017-01-15:
@ianrrees - in case you haven't yet subscribed to the forum topic, I have pushed the full set of pyside formula ports to the qt5 branch...  You should be able to pull, use the freecad formula to install the dependencies and build the app.  Details in the forum posts.  We have a blocking issue with ```make install``` that hopefully either you or someone else will recognize the root cause for immediately.

bblacey, 2017-01-15:
@ianrrees, I don't know if you ran into this yet or not but there is a [Homebrew bug](https://github.com/Homebrew/brew/issues/737) that makes it difficult to work with a branch within a tap (I found out because I reset by Homebrew ports and every time I tried to install, homebrew would checkout master in the freecad tap).  The trick is to [set homebrew.developer in the homebrew git config file](https://github.com/Homebrew/brew/pull/881/commits/42c2ed5e7f1aa8adc719a095a753c248c0ecbee0).  

**So here is how you get up and running with the qt5 branch:**
```
git config --file $(brew --prefix)/HomeBrew/.git/config --add homebrew.developer 1
brew tap --full FreeCAD/freecad
cd $(brew --prefix)/HomeBrew/Library/Taps/freecad/homebrew-freecad
git checkout -b qt5 origin/qt5
```
Once you have done that, you are ready to install the freecad dependencies using the qt5 branch.
```
brew install --only-dependencies freecad/freecad
brew install --only-dependencies freecad/freecad  #for some unresolved reason, you  have to run it twice to install all dependencies
```
I hope this helps.

bblacey, 2017-01-16:
FYI, I also pushed a [cmake change](https://github.com/bblacey/FreeCAD-MacOS-CI/commit/af3906ef75f276afa53af601aa2a66d00042ed9d) to https://github.com/bblacey/FreeCAD-MacOS-CI/commits/qt5 that is required for QT5 on macOS.

bblacey, 2017-03-12:
Qt 5.6.2 has been integrated into the CI builds.

closed 2017-03-12 by bblacey