FreeCAD/homebrew-freecad #37: Upgrade to Coin 4.0.0a

bblacey, 2017-02-16:
Recently, @wmayer asked why the version of Coin in the macOS builds was "so old".  It was relatively hard to answer because no one really new where to download 4.0.0a that is used in the distributions for other platforms.

I found, what I believe to be, the original 4.0.0a Coin source and companion source packages on Emergent's ftp server - https://grey.colorado.edu/emergent_ftp/?C=M;O=A.  This seems to be the last version before Emergent turned Coin, and the companion packages, over to the community for continued development on https://bitbucket.org/Coin3D/coin/wiki/Home

For the macOS builds, we will use the latest tip from the Coin Mercurial repository but pinned to commit 035e53e, made on 15-JUL-16, to avoid build surprises if/when someone updates the Coin tip.  The coin formula will also support a —HEAD option for the adventurous who would like to build and install the absolute latest version of coin.

The Coin developers have moved, or should I say, are in the process of moving, to cmake and the macOS formula will take advantage of the new build configuration.  However, pivy relies upon coin-config and an associated configuration file that are not installed by default by the CMake rules.  The Coin formula will take extra steps to create and install the coin-default.cfg file.

bblacey, 2017-02-16:
[Patch set submitted to Coin3D](https://bitbucket.org/Coin3D/coin/issues/130/cmake-have_inttypes_h-implementation-issue) as an issue.  Re-opening to track.  Will also modify formula to use patch set from BitBucket as a resource as that might be more palatable to the Homebrew mods.

bblacey, 2017-02-16:
Submitted [upstream issue ](https://bitbucket.org/Coin3D/coin/issues/131/cmake-doesnt-create-coin-defaultcfg-or) for cmake build scripts to install coin-config and associated coin-default.cfg so we can avoid this [formula work-around](https://github.com/FreeCAD/homebrew-freecad/blob/master/coin.rb#L47-L53).

bblacey, 2018-01-02:
The INTYPES patch I submitted to Coin has been incorporated into the source tree so we can probably remove [this patch](https://github.com/FreeCAD/homebrew-freecad/blob/master/coin.rb#L36-L40) at some point

vejmarie, 2021-01-05:
@bblacey any issue at closing this issue ?

ipatch, 2021-01-05:
> The INTYPES patch I submitted to Coin has been incorporated into the source tree so we can probably remove [this patch](https://github.com/FreeCAD/homebrew-freecad/blob/master/coin.rb?rgh-link-date=2018-01-02T18%3A13%3A48Z#L36-L40) at some point

link to patch is 404

commit 968e2875 (2017-02-16): Upgrade coin to 4.0.0a

  * Upgraded to latest coin source available on BitBucket
  * Rebottled coin and pivy with coin 4.0.0a

Fixes #37
commit 4e135a0d (2017-02-16): Upgrade coin to 4.0.0a

  * Upgraded to latest coin source available on BitBucket
  * Rebottled coin and pivy with coin 4.0.0a

Fixes #37
closed 2021-04-12 by vejmarie