FreeCAD/homebrew-freecad #42: Add Graphviz and Calculix as dependencies with options to not install

bblacey, 2017-03-07:
FreeCAD uses both Graphviz and Calculix.  We should add them to the FreeCAD formula dependencies to ensure proper configuration of FreeCAD.

luzpaz, 2017-07-05:
Reference forum thread: https://forum.freecadweb.org/viewtopic.php?f=21&t=21094#p163309

luzpaz, 2017-09-25:
Will you make them optional ? 
`depends_on "graphviz" => :optional`

luzpaz, 2017-10-24:
CalculiX-ccx 2.13 PR: https://github.com/Homebrew/homebrew-science/pull/6417

luzpaz, 2018-08-23:
@peterlama just a follow up regarding this issue. Hopefully this can be done before 0.18 release.

peterlama, 2018-08-24:
@luzpaz Ah yes, I forgot about it for the 0.17 release... :/

luzpaz, 2018-12-16:
Elmer FEM should be also added.

- [ ] CalculiX
- [ ] Graphviz
- [ ] Gmsh
- [ ] ElmerFEM

luzpaz, 2019-03-27:
Updated the above post with relevant version numbers a links to pre-existing formulas in Homebrew

vejmarie, 2021-01-05:
Did somebody ever worked on this ? I can give it a try if needed.

vejmarie, 2021-04-12:
I have added Elmer and even created the associated formula which is laid down into our Formula's now. Got some users in Singapor who like it. Will close the current issue for the moment, and add calculix as a potneitl dependency.

closed 2021-04-12 by vejmarie

luzpaz, 2021-04-12:
@vejmarie what about graphviz and gmsh?