FreeCAD/homebrew-freecad #47: Problem finding qt5 cmake files

fcacarminati, 2017-05-21:
[/Users/fca] brew install --only-dependencies freecad
==> Installing freecad from freecad/freecad
All dependencies for freecad/freecad/freecad are satisfied.

[...r/local/FreeCAD/build] cd .. ; rm -rf build ; mkdir build; cd build ; cmake -DBUILD_QT5=1 -DFREECAD_USE_EXTERNAL_KDL=1 -DBUILD_FEM_NETGEN=1 -DFREECAD_CREATE_MAC_APP=1 -DCMAKE_BUILD_TYPE="Release" ../FreeCAD-git/
[...]
-- Found Eigen3: /sw/include/eigen3 (Required is at least version "2.91.0")
CMake Warning at CMakeLists.txt:772 (find_package):
By not providing "FindQt5Network.cmake" in CMAKE_MODULE_PATH this project
has asked CMake to find a package configuration file provided by
"Qt5Network", but CMake did not find one.

Could not find a package configuration file provided by "Qt5Network" with
any of the following names:

Qt5NetworkConfig.cmake
qt5network-config.cmake

Add the installation prefix of "Qt5Network" to CMAKE_PREFIX_PATH or set
"Qt5Network_DIR" to a directory containing one of the above files. If
"Qt5Network" provides a separate development package or SDK, be sure it has
been installed.

[... the same for all the cmake files of QT5 ...]

CMake Error at CMakeLists.txt:789 (MESSAGE):
========================

Qt5 libraries not found.

========================

-- Configuring incomplete, errors occurred!
See also "/usr/local/FreeCAD/build/CMakeFiles/CMakeOutput.log".
See also "/usr/local/FreeCAD/build/CMakeFiles/CMakeError.log".

I can make it work by the following configuration for CMAKE

cmake -DBUILD_QT5=1 -DFREECAD_USE_EXTERNAL_KDL=1 -DBUILD_FEM_NETGEN=1 -DFREECAD_CREATE_MAC_APP=1 -DCMAKE_BUILD_TYPE="Release" -DCMAKE_PREFIX_PATH=/usr/local/Cellar/qt\@5.6/5.6.2/ ../FreeCAD-git/

But I believe it should be automatic.

ianrrees, 2017-05-21:
Hi @fcacarminati - thanks for the issue!  I'm not in a good position to work on this at the moment, and I know @bblacey has also been busy lately.  I think that you'll find running `brew link -f qt` (might be qt5 not qt, I can't remember) before the cmake will let cmake work properly.

What version of MacOS are you using?

fcacarminati, 2017-05-21:
Hi,
  thanks for your answer. I have tried the link and still does not work. I am running Sierra 10.12.5 (16F73). Best regards,

Federico Carminati
CERN-PH
1211 Geneva 23
Switzerland
Tel: +41 22 76 74959
Fax: +41 22 76 68505
Mobile: +41 75 411 4843

luzpaz, 2017-05-26:
Linking to:
https://freecadweb.org/tracker/view.php?id=3042
https://freecadweb.org/tracker/view.php?id=3043

fcacarminati, 2017-05-29:
For completeness I am quoting here the CMAKE command that allows compilation

cmake -DBUILD_QT5=1 -DFREECAD_USE_EXTERNAL_KDL=1 -DBUILD_FEM_NETGEN=1 -DFREECAD_CREATE_MAC_APP=1 -DCMAKE_BUILD_TYPE="Release" -DCMAKE_PREFIX_PATH=/usr/local/Cellar/qt\@5.6/5.6.2/ -DCMAKE_INSTALL_PREFIX=/Applications -DBUILD_JTREADER=ON -DBUILD_SANDBOX=ON -DBUILD_TEMPLATE=ON -DFREECAD_USE_FREETYPE=OFF ../FreeCAD-git/

bblacey, 2018-01-02:
@fcacarminati - the .travis.yml config is always a good reference for local build issues - note the eigen include on this [line of the config](https://github.com/FreeCAD/FreeCAD/blob/master/.travis.yml#L194)

Does that resolve the issue for you?

peterlama, 2018-09-01:
A lot has changed since this issue was opened.

closed 2018-09-01 by peterlama