FreeCAD/homebrew-freecad #48: not working for me

gedw99, 2017-05-29:
````

x-MacBook-Pro:_demo apple$ brew install freecad
Updating Homebrew...
==> Auto-updated Homebrew!
Updated 2 taps (caskroom/cask, homebrew/core).
==> Updated Formulae
cimg                          dpkg                          folly                         libxml2                       mosquitto                     qbs                           sfcgal                        tnef                          wget

Error: No available formula with the name "freecad" 
==> Searching for a previously deleted formula...
Error: No previously deleted formula found.
==> Searching for similarly named formulae...
Error: No similarly named formulae found.
==> Searching taps...
Error: No formulae found in taps.
x-MacBook-Pro:_demo apple$ 

````

ianrrees, 2017-05-29:
Looks like you just need to `brew tap FreeCAD/freecad`.  If that doesn't help, please reopen with more information about your computer (version of MacOS, etc).

closed 2017-05-29 by ianrrees

gedw99, 2017-05-29:
ok that worked.

I have to bale on installing it though. The deps list is big and its going to mess up some develpment code i use QT for.

My god its a beast !!!!
==> Installing dependencies for freecad/freecad/freecad: ccache, boost, boost-python, xerces-c, qt@5.6, sphinx-doc, FreeCAD/freecad/shiboken, FreeCAD/freecad/pyside, FreeCAD/freecad/pyside-tools, freeimage, tbb, gl2ps, homebrew/science/opencascade, eigen@3.2, homebrew/science/orocos-kdl, numpy, homebrew/science/matplotlib, szip, homebrew/science/hdf5, homebrew/science/med-file, hdf5, netcdf, homebrew/science/vtk, opencascade, homebrew/science/nglib, FreeCAD/freecad/coin, swig, FreeCAD/freecad/pivy

The OSX binary version has a bug that causes it to fail to stat too :)
https://github.com/FreeCAD/FreeCAD/releases

I really wanted to try it out and i did 3 years ago on windows. but the GUI was crashy for me at least.

ianrrees, 2017-05-29:
FreeCAD is much improved from 3 years ago.

Could you please elaborate on "The OSX binary version has a bug that causes it to fail to stat"?  It's much more efficient for us to find/fix issues when we have good bug reports to work with.