FreeCAD/homebrew-freecad #53: OSX High Sierra build fails

maltitco, 2017-11-29:
==> ./configure -verbose -prefix /usr/local/Cellar/qt@5.6/5.6.2 -release -opensource -confirm-license -system-zlib -qt-libpng -qt-libj
==> make
Last 15 lines from /Users/malt/Library/Logs/Homebrew/qt@5.6/02.make:
    require_action(inImage != NULL, InvalidImage, err = paramErr);
                                    ^
3 errors generated.
make[5]: *** [.obj/qcocoahelpers.o] Error 1
make[5]: *** Waiting for unfinished jobs....
make[4]: *** [sub-cocoa-make_first] Error 2
make[3]: *** [sub-platforms-make_first] Error 2
make[3]: *** Waiting for unfinished jobs....
2 warnings generated.
rm -f libqsqlite.dylib
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/clang++ -headerpad_max_install_names -stdlib=libc++ -Wl,-syslibroot,/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.13.sdk -mmacosx-version-min=10.7 -single_module -dynamiclib -o libqsqlite.dylib .obj/smain.o .obj/qsql_sqlite.o .obj/sqlite3.o .obj/moc_qsql_sqlite_p.o  -F/private/tmp/qt@5.6-20171129-31214-tvf3b5/qt-everywhere-opensource-src-5.6.2/qtbase/lib -framework QtSql -framework QtCore -framework DiskArbitration -framework IOKit  
mv -f libqsqlite.dylib ../../../../plugins/sqldrivers/ 
make[2]: *** [sub-plugins-make_first] Error 2
make[1]: *** [sub-src-make_first] Error 2
make: *** [module-qtbase-make_first] Error 2

chernals, 2017-11-29:
I am facing the exact same issue. Could someone provide any advice or another build of qt5.6 for high sierra ?

bblacey, 2018-01-02:
@peterlama is working on upgrading our Qt 5.x stack to something newer which should resolve this issue.