FreeCAD/homebrew-freecad #54: building opencamlib failed

PLChris, 2017-12-09:
Tried to install opencamlib according to this forum post: https://forum.freecadweb.org/viewtopic.php?f=15&t=21267.
I get the following errors. The mentioned directories and files in the /tmp directory do not exist.

maccb:tmp cb$ brew install opencamlib
==> Installing opencamlib from freecad/freecad
==> Cloning https://github.com/aewallin/opencamlib.git
Updating /Users/cb/Library/Caches/Homebrew/opencamlib--git
==> Checking out revision 398f7c7fac60169fda067ea0f5ebc15083361f5c
==> cmake -DCMAKE_C_FLAGS_RELEASE=-DNDEBUG -DCMAKE_CXX_FLAGS_RELEASE=-DNDEBUG -DCMAKE_INSTALL_PREFIX=/usr/l
Last 15 lines from /Users/cb/Library/Logs/Homebrew/opencamlib/01.cmake:
--  called with version  = 11.10.83
-- architecture is: 
-- architecture is: 
--  CPACK_DEBIAN_DISTRIBUTION_RELEASES : lucid;maverick;natty;oneiric;precise;quantal
--  CMAKE_SOURCE_DIR is = /tmp/opencamlib-20171209-33252-n24h8m/src
--  descr file = /tmp/opencamlib-20171209-33252-n24h8m/src/deb/debian_package_description.txt
-- type:
--  'make' for a normal build
--  'make -j8' to build faster (if you have many cpus)
--  'make install' to install
--  'make package' to build a binary deb-packate
--  'make spackage' to build debian source-packages
--  'make test' to run the tests
-- Configuring incomplete, errors occurred!
See also "/tmp/opencamlib-20171209-33252-n24h8m/build/CMakeFiles/CMakeOutput.log".

If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
https://github.com/freecad/homebrew-freecad/issues

These open issues may also help:
Reminder: update opencamlib https://github.com/FreeCAD/homebrew-freecad/issues/43
maccb:tmp cb$

ianrrees, 2017-12-09:
I'll try to have a look at this soon.

ianrrees, 2017-12-16:
Looks like there are two aspects to this problem:

  1 I guess the Homebrew Python 2 distribution has changed slightly, so there's no longer a symlink called just "python" where there used to be.  Our Homebrew formula tells the OCL cmake file the wrong path to Python as a result, so the OCL cmake fails - I'll fix that shortly.
  2 There's an issue with newer cmakes, which is fixed by a PR upstream - https://github.com/aewallin/opencamlib/pull/25

I'm inclined to just fix 1 locally, and not setup a patch for 2 which would be obviated by that PR.  If you need to build OCL sooner though, do something like "brew edit opencamlib", change the "head" to use https://github.com/neomilium/opencamlib.git , and then install like "brew install --HEAD opencamlib"

closed 2017-12-16 by ianrrees

ianrrees, 2018-01-19:
@PLChris - just a note that @aewallin updated opencamlib, and I've updated our brew formula to match.  So, if you run in to more problems please let us know!