FreeCAD/homebrew-freecad #55: Homebrew/science will be deprecated at the end of the year

bblacey, 2017-12-15:
@peterlama and @ianrrees, It looks like the Homebrew team is deprecating homebrew/science and moving select (i.e. popular) formula from science to core.  I discovered this while submitting a pull request to upgrade the opencascade formula from 7.1 to 7.2.

Freecad has the following dependencies that we need to ensure are migrated from science to core.
```
depends_on "homebrew/science/orocos-kdl"
depends_on "homebrew/science/matplotlib"
depends_on "homebrew/science/med-file"
depends_on "homebrew/science/nglib"
```

Looks like matplotlib is in work leaving orocos-kdl, med-file and nglib, most likely to our team.  I have started the opencascade port to core but am currently traveling so I don't know if I will be able to tackle any of the others in the near term.

Pertinent Homebrew issues:

Homebrew/homebrew-science/issues/6331
Homebrew/homebrew-core/issues/15619

ianrrees, 2017-12-15:
I'm pretty swamped with work stuff at the moment (and more immediately, putting a roof on a friends new shop :) ).  Should have more time after CES, but was hoping to have a look at Issue #54 next.

Any idea where we're meant to move projects that aren't popular enough for core?  nglib in particular probably doesn't have much interest...

bblacey, 2017-12-31:
I just finished the port of opencascade 7.2 to homebrew-core but homebrew-science/nglib is not compatible with OCCT 7.2 (just an include issue).  We have a number of options - add a patch for OCCT 7.2 and see if Homebrew-core will accept it into core.  Move to NGSolve/netgen and try to get Homebrew-core to accept it.  Or we can host one or the other here but then we incur bottle maintenance when one of the homebrew-core dependencies is updated.  I spent a bit of time moving formula out of our tap into homebrew-science because of the maintenance so it is tough for me to think about going backwards.  If we have to, then we should create a .travis.yml job here to rebuild dependencies as needed and upload the bottles to bintray (can't use GitHub with `@version` syntax).

Please weigh in.

peterlama, 2018-01-01:
netgen still needs a bit of work: https://github.com/NGSolve/netgen/pull/4

Do you think we will use python 2 for 0.17 release? If so, it probably would be easier to patch netgen 5.3 and put in our repo for now. Then we can work on getting the latest netgen to work and getting it into core. Do you know if core would accept "bleeding edge" software? There is no actual 6.2 release as far as I can tell. Then there is the issue of whether having a separate formula for netgen would work if someone later creates a formula for NGSolve. From the NGSolve docs, I get the impression that it was designed to have everything built at once.

bblacey, 2018-01-01:
This is the closest I can find - https://forum.freecadweb.org/viewtopic.php?f=8&t=25443&start=20 (wish I found this sooner).

0.17 looks like: 
- Python2 (thank goodness)
- Qt4 on Linux/Win & Qt5 on macOS
- OCCT 7.1?  Arghhh - I wonder if I can get the Homebrew Core team to accept an opencascade@7.1 but I doubt it because the usage isn't high enough.

I had to effectively rewrite the opencascade formula in order for the homebrew-core team to accept it so I experiment to see if the formula rewrite will work with opencascade 7.1.  If it does, then I will give it a shot and see if the homebrew-core guys are open to accepting it.

peterlama, 2018-01-01:
Actually, I don't think it is worth having 7.1 just to be consistent with other platforms. Newer the better for OCC, IMO. I mean it will also be using qt5, so mac is already different anyway.

peterlama, 2018-01-01:
I've just updated the nglib 5.3 patch to work with 7.2. I'll make a PR once I confirm it works with freecad.

xzcvczx, 2018-01-01:
qt 5.9 i believe is an LTS so might be worth trying to migrate to that as the standard qt, although there are issues with pyqt

peterlama, 2018-01-01:
Hmm, Homebrew already change to 5.10, that's annoying. Think it would be a problem to use 5.10 instead of 5.9?

xzcvczx, 2018-01-01:
hmmm weird they should have @'d it but it doesn't seem they have

qt5.10 isn't a big problem but its not an LTS which i guess is a bit annoying

bblacey, 2018-01-01:
Let the fun begin...
```
Updating Homebrew...
==> Auto-updated Homebrew!
Updated Homebrew from 93a2e9fc to 4e08da95.
Updated 5 taps (caskroom/cask, homebrew/livecheck, homebrew/python, homebrew/science, homebrew/versions).
==> Deleted Formulae
homebrew/science/matplotlib ✔                     homebrew/science/gmap-gsnap                       homebrew/science/phylip
homebrew/science/med-file ✔                       homebrew/science/gmcloser                         homebrew/science/phyml
homebrew/science/nglib ✔                          homebrew/science/gmtk                             homebrew/science/phyutility
homebrew/science/orocos-kdl ✔                     homebrew/science/gnuastro                         homebrew/science/phyx
homebrew/science/a5                               homebrew/science/gnudatalanguage                  homebrew/science/piler
homebrew/science/abacas                           homebrew/science/grabix                           homebrew/science/pilercr
homebrew/science/abinit                           homebrew/science/graph-tool                       homebrew/science/pilon
...snip...
```

@xzcvczx - do you want to pitch in and do some of the heavy lifting on macOS? Maybe updating the qt5 stack? You can either fork this repo or I can add you as a contributor - your choice.

peterlama, 2018-01-01:
I have confirmed the 5.9 branch of pyside2 builds fine with qt 5.10. Now the question is will it work with freecad. I seem to remember seeing issues reported with shiboken in the install/compile forum.

bblacey, 2018-01-01:
@peterlama - did you try QT@5.7 (in Homebrew core) with the 5.6 pyside2 branch?  I gave it a quick test before I had to run an errand but it failed to compile (most like a config issue).

peterlama, 2018-01-01:
No. I only tested the "official" way to build pyside (setup.py) to make sure things worked before looking into brew formulas.

bblacey, 2018-01-01:
@peterlama - One of the bigger changes in pyside2, after Qt took it over, was to use the clang parser for shiboken so it could handle c++11 constructs so it can handle Qt 5.7+  Have you had a chance to test it yet?

peterlama, 2018-01-01:
Yeah, you can't compile the 5.9 branch with out llvm-config available. I didn't see it in the xcode app, so I installed it with `brew install llvm` and added its location to the PATH var.

bblacey, 2018-01-02:
Yeah, I did pretty much the same thing but only shiboken built, pyside2 failed.  I tried
`python setup.py build --qmake=/usr/local/Cellar/qt@5.7/5.7.1/bin/qmake  --openssl=/usr/local/opt/openssl --ignore-git --jobs=4 --verbose`

But it fails with `/Library/Frameworks/QtCore.framework/Headers/qglobal.h:306:12: fatal error: 'AvailabilityMacros.h' file not found`

I tried setting `--osx-sysroot=/Applications/Xcode.app//Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk` but no go.

I'm on High Sierra.

peterlama, 2018-01-02:
Why 5.7? IMO, we should use the newest version.

I'm on High Sierra as well.

bblacey, 2018-01-02:
No specific reason other than it was something later than 5.7 and earlier than 5.10 which isn’t released yet.  Have you tested it with an FC build yet?

peterlama, 2018-01-02:
No, I haven't had the chance to build freecad yet.

bblacey, 2018-01-02:
Build fails with the same error using Qt 5.10 (as expected).  What are you doing differently to get it to build?

peterlama, 2018-01-02:
Not sure, this is what I did:

```
brew install python
brew install qt
brew install llvm
export PATH=/usr/local/opt/llvm/bin:$PATH
git clone https://codereview.qt-project.org/pyside/pyside-setup
cd pyside-setup
git checkout 5.9
python2 setup.py build --qmake=/usr/local/opt/qt/bin/qmake
```

peterlama, 2018-01-02:
By the way, I uninstalled homebrew after upgrading to High High Sierra yesterday sot that I would be starting from a clean slate.

bblacey, 2018-01-02:
Ok, well pyside2 won't build on my system, even doing exactly what you did, so I am going to leave Qt re-baselining to you and look into the deprecated ports like orocos-kdl, etc.  Ok?

bblacey, 2018-01-02:
Also, if we are stuck hosting the formula ourselves - this might help improve bottle maintenance https://gist.github.com/maelvalais/068af21911c7debc4655cdaa41bbf092

peterlama, 2018-01-02:
Strange. Anyway, yeah I'll plan on taking care of pyside then.

+1 for test-bot. That looks great.

bblacey, 2018-01-02:
@peterlama & @ianrrees - I have created a [science-deprecation project](https://github.com/FreeCAD/homebrew-freecad/projects/1?) and will open the issues I think we need to address for this project.  I will try to file and organize the issues today so if you review the project board after today and see something missing, then please file the issues.  Also, feel free to break any issues assigned to you into smaller pieces if it makes it easier.

ianrrees, 2018-01-02:
@bblacey - That project feature is neat - haven't noticed it before on github.  Thanks for setting it up!  I'll try to chip in here-and-there as time permits.

bblacey, 2018-01-02:
@ianrrees - yeah, a simple drag-n-drop scrum-type board that should help us maintain perspective on progress.  It seems appropriate to file and manage these issues here because they do not affect the FreeCAD source code in anyway (i.e. they are macOS port dependency issues only).

bblacey, 2018-01-02:
Guys, I'de like to complete the tap refactoring under #24 before we work these issues.  Are you guys good with my doing that right now?

ianrrees, 2018-01-02:
Sure!

bblacey, 2018-01-02:
Done.  Brew update should pull the refactored tap.

bblacey, 2018-01-02:
Argh, I meant #62 - had to amend commit to reference `Fixes #62`.  Anyway, this prerequisite is done so hack away on the other items at your convenience ;)

bblacey, 2018-01-03:
I'm going to start work on orocos-kdl #59 next.  If the core team accepts it, then all will be good.  If not, we will have to host it here.

bblacey, 2018-01-03:
Submitted orocos-kdl PR to homebrew-core.

bblacey, 2018-01-03:
`orocos-kdl` lives again in Homebrew/core.  I will start on `med-file` next but it will be a background task to my other commitments.  Still hoping @ianrrees will grab the nglib rains from @peterlama or jump on @matplotlib ;)

peterlama, 2018-01-03:
FYI, it doesn't look like I'll be able to work on this until the weekend.

bblacey, 2018-01-03:
No worries Peter - I am just trying to chip away at the deprecated ports in the background and probably won't be able to do much before the weekend either.

bblacey, 2018-01-04:
Guys, I have created a [science-update](https://github.com/FreeCAD/homebrew-freecad/tree/science-update) branch that we should all be able to collaborate on.  We probably shouldn't merge any **new** or **updated** ports into master until all of them are ready because it might impact the CI builds.  The goal would be to complete all of our work, verify it, merge science-update to master and then rebuild the ports-cache.

ianrrees, 2018-01-04:
I'll try to chip in over the weekend, currently a bit too busy with work.

Separate branch sounds like a good idea - wouldn't be too surprised if changing the Python library install method means we need to adjust the application bundling script, or something along those lines.

bblacey, 2018-01-04:
FYI, I submitted med-file to the homebrew-core as a PR a couple of days ago but it requires patches and their policy is to not accept new formula into core if they require patches.  There was some activity but it has stopped so I think it caused pause.  Bottom-line, we may be on our own here guys which raises the need for an auto-bottling solution.  I will open an issue that cites some prior art that could help us.  The core problem is that we aren't able to detect when our bottles need to be updated due to an updated dependency.  It's too bad that Homebrew doesn't have federated dependency detection/events so all taps within the ecosystem would know if they need to update their bottles.  If I had more time, I'de try to get something started with them in that arena...