FreeCAD/homebrew-freecad #58: Pyside 5.9 / Qt 5.10

bblacey, 2018-01-02:
Currently we host the Qt@5.6 formula and bottles along with the dependent pyside formulae pinned to an early version of pyside2.  The scope of this issue is to upgrade the pyside formulae to a more recent version with any necessary refactoring (i.e. eliminating shiboken2 and pyside-tools because they are wrapped into the new and simply setup.py?)

peterlama, 2018-01-02:
Yeah, it would be really convenient to have shiboken and pyside-tools in the same formula.

peterlama, 2018-01-02:
Although, I just realized getting a snapshot with the git submodules could be an issue.

bblacey, 2018-01-02:
I "think" if you reference the right commit, it should pull it all.  Also, if you don't pass `--ignore-git` to `setup.py`, I think it pulls the submodules.  It would be better to use their higher-level setup script but I guess we could always drop down to cmake for each of the parts because that is all that setup.py is doing (searching for dependencies and setting cmake args).  It's easy to get the dependency location using #{Formula['<formula>'].opt_prefix} so setup.py may not offer us much at this point.

ianrrees, 2018-01-05:
Am I understanding correctly, that we expect the current FreeCAD, and PySide2 +friends to be compatible with Qt5.10?

bblacey, 2018-01-05:
That is @peterlama's recommendation and the goal.  If it works, he should also be able to coalesce the existing shiboken, pyside-tools and pyside formulae into a single pyside formula.  I'm wondering if we should rename it to pyside2 at this point because that seems to be the generally accepted name.

bblacey, 2018-01-25:
For reference - pyside2 issues/dependencies:

https://bugreports.qt.io/browse/PYSIDE-590
https://bugreports.qt.io/browse/PYSIDE-594

PR #65 (2018-01-26): Adds pyside2 formulas
Adds support for pyside2, pyside2-tools, and shiboken2.
Bypasses the setup.py based build as it's problematic, and simply relies on the
(working) cmake based build.

fixes #58
closed 2018-01-26 by dansrogers