FreeCAD/homebrew-freecad #60: matplotlib deprecated in homebrew-science

bblacey, 2018-01-02:
[matplotlib](https://github.com/Homebrew/homebrew-science/blob/19f75951641d3a5e70ea105f76a6a77bc0553d07/med-file.rb) is no longer available via homebrew-science. Need to either push for it to be included in homebrew-core or host it ourselves. From my experience with opencascade, it will be more near-term work to get it into core than it will be to host it ourselves however, longer term it should be less overall work and will avoid breakages when dependent packages change (i.e. freetype, libpng, numpy).  

Given the number of homebrew-core dependencies & the [usage metrics](https://gist.github.com/sjackman/d1db0d73597b674bb23b51221a2914cd#file-metrics-science-macos-tsv-L2), this is a strong candidate to move to homebrew-core

ianrrees, 2018-01-02:
I think matplotlib is a bit outside the scope of FreeCAD, and has a whole lot of other users so we should probably let someone else handle this.

How feasible is it to use pip for installing matplotlib instead of brew?

bblacey, 2018-01-02:
Actually, that is a great question and I would think you python experts (you and @peterlama) would have a better understanding than I about the feasibility and maintenance.  If we were to tackle this as a Homebrew port, then I think it should be to move it into core versus hosting it ourselves.

sgrogan, 2018-01-03:
Not an expert, but matplotlib, depends on numpy. I think Ship. Plot, Path, and FEM, depend on both. Ship and Plot will build without numpy. I think Path and FEM will too, but will fail at runtime.

peterlama, 2018-01-03:
Yeah, we should definitely try to get it into core if we are going to use it. I don't think it will take that much time to prepare a pull request, but we have to have version 2.1.1 for it to work with pyside2. Maybe we can start with this: https://github.com/Homebrew/homebrew-science/pull/6545

Actually, maybe we could use pip instead? We would just need to change the travis build scripts.

bblacey, 2018-01-03:
While working with the homebrew-core team on my `med-file` PR (which we may find that they won't accept because it requires patches), they stated that it was unlikely that they would accept `matplotlib`.  I think we can simply add it as a manual pip install step within the freecad.rb formula.

ianrrees, 2018-01-03:
Interesting - was the justification that they'd prefer to use Python's package management stuff for installing matplotlib?

Just did a little grepping through my homebrew tree, it looks like we should be able to "just" do something like:

```
  resource "matplotlib" do
    url "https://files.pythonhosted.org/packages/some/magic/string/that/ends/in.tar.gz"
    sha256 "111222333444555666777888999000"
  end

  venv = virtualenv_create(libexec)
  venv.pip_install resource("matplotlib")
  venv.pip_install_and_link buildpath
```

bblacey, 2018-01-03:
I think we can also just host the existing `matplotlib` formula without bottling it because it installs using pip, pulls all the prerequisites and is reasonably fast without a bottle.  We only need to bottle "expensive" formula we host like Qt5 but with Qt5.10, even that hosted bottle will go away.

bblacey, 2018-01-03:
I'll add this to my list - I think it is easier, at least for now, to simply host `matplotlib` here.  We are probably going to have to invest in automated bottling at some point but I just don't have the cycles to tackle that right now either.

ianrrees, 2018-01-03:
Sounds like a good plan - easy to do in the mean time, then maybe we can roll the matplotlib dependency in to the freecad formula (using pip or similar) down the track.

bblacey, 2018-01-04:
Guys, I have created a [science-update)[https://github.com/FreeCAD/homebrew-freecad/tree/science-update/Formula] branch that contains [matplotlib 2.1.1](https://github.com/FreeCAD/homebrew-freecad/commits/science-update/Formula/matplotlib.rb).  I imported 2.0.2, updated to 2.1.1 from the PR that @peterlama cited above.  I then upgraded some of the dependent ports.

Could you guys test matplotlib on your systems when you have a chance?

Also, I'm thinking that we shouldn't merge any **new** ports into master until all of them are ready because it might impact the CI builds.  The goal would be to complete all of our work, verify it, merge to master and then rebuild the ports-cache.  We should all be able to contribute to the science-update branch.

bblacey, 2018-01-06:
More color around the 0.17 release timeline - https://forum.freecadweb.org/viewtopic.php?f=10&t=26174

From Werner on 4-Jan

>As a suggestion for a time schedule I would say that we set the development into feature freeze on around 20th/21st of January which is a bit more than two weeks from now and then only accept bug fixes.

dansrogers, 2018-01-26:
So the pip version of matplotlib does not work for us.  It is unfortunately linked against the system libpng which conflicts with the brew installed version of freetype, as it's linked against brew libpng.  I had to build it myself to get freecad working.  I couldn't find a way to get matplotlib installed with pip that somehow ignored the prebuild binary version either (though maybe this is pip ignorance on my part).

dansrogers, 2018-01-26:
Also matplot lib works for me.  I was able to compile and start freecad master.

peterlama, 2018-03-24:
This seems to be done.

closed 2018-03-24 by peterlama