FreeCAD/homebrew-freecad #63: Auto-bottling for homebrew-freecad tap formulae

bblacey, 2018-01-04:
**SCOPE:** 
Implement an auto-bottling mechanism for this tap, that will run on Travis and deploy bottles to bintray, akin to what Homebrew does, albeit with Jenkins and bintray.  

**IMPETUS:**
The FreeCAD-ports-cache reduces Travis build times by archiving and restoring a fully instantiated homebrew ports tree that includes all the Freecad dependencies.  If a formula is updated, the Travis job naturally falls back to installing the updated formula using available bottles and if no bottles, it builds from source.  Without the ports-cache, all dependencies would be installed from bottles causing excessive Travis build times and in the case of this tap, all Freecad hosted formulae would build from source exceeding Travis build time limitations, beit for instantiating a ports-cache, or a routine FreeCAD CI build.

Furthermore, the Homebrew team recently deprecated and archived homebrew-science abandoning a number of important FreeCAD dependencies on macOS.  Wherever possible, we have ported existing science formulae (e.g. opencascade and orocos-kdl) to Homebrew-core but the Homebrew-core team has stringent acceptance criteria that prevents some formula from being ported (e.g. matplotlib).  

The advantage of a formulae living within the Homebrew-core repo is that they are broadly available and benefit from a well-established infrastructure that will rebuild the bottles whenever a dependency changes ensuring bottled-dependency compatibility/integrity.  We had made great strides moving most of our dependent Formulae to homebrew-science to leverage their automated bottling infrastructure, greatly reducing CI build failures, but those formulae have effectively been abandoned by Homebrew because they will not accept all formulae into homebrew-core (e.g. matplotlib).

Given the number of formulae that will likely return to this tap once again (e.g. nglib, med-file) and other Homebrew-abandoned formulae (e.g. matplotlib), we need our own mechanism to automatically create and deploy bottles using Travis CI to ensure FreeCAD CI build reliability, performance and integrity.

We are not alone in this custom-tap auto-bottling quest as there is some recent [prior art](https://gist.github.com/maelvalais/068af21911c7debc4655cdaa41bbf092) that proves the viability and that we can leverage. The prior art uses Hombrew's `brew test-bot`, albeit with some changes that ultimately should find their way back to Homebrew.

The scope of this issue is to implement an auto-bottling mechanism for this tap, that will run on Travis and deploy bottles to bintray, akin to what Homebrew does, albeit with Jenkins and bintray.

bblacey, 2019-06-05:
Update - nearly complete implementation based upon aforementioned prior art but enhanced to use BinTray as an intermediary between bottle build and final upload (avoid AWS account fees).  That said, Conda seems to be the wise FreeCAD choice for cross-platform build and test and I personally haven't had cycles to take the home-brew port the rest of the way.  Another user has prototyped and documented a fairly [clean approach using Azure Pipelines](https://github.com/ladislas/homebrew-greetings) - referencing it here in case anyone wants to carry this issue forward.

ipatch, 2021-08-12:
i've spent some time working on getting auto bottling going for this tap using github actions. i've definitely run into some roadblocks however.

1. the default homebrew _test-bot_ can only build a bottle for catalina at the moment. ie. homebrew-core uses it's own _self-hosted_ runner to build bottles for the last three major OSes. [learn more][1]
2. i definitely have a better understanding of using github actions from working with them for the past couple of days, and i created a little dumping ground (if you will) repo for experimenting with github actions, and generating bottles for the formulae contained within the repo. [learn more][2]

definitely not the most glamorous of work, and definitely not as visually appealing gratifying as working with the curves workbench, but i'll keep attempting to fight the good fight in hopes of streamlining this tap to hopefully autogenerate bottles using github actions, self-hosted runner(s), and github releases. if your reading this, and have experience working with any of the three mentioned topics and want to lend a hand 👐, and just play _rubber duck_ 🐥, you know where to find me.

[1]: <https://github.com/Homebrew/discussions/discussions/1970#discussioncomment-1161863>
[2]: <https://github.com/ipatch/homebrew-freecad-pg13>

ipatch, 2021-08-18:
hey a little heads up to the people watching this repo, ie. the people who've clicked the little _watch_ button in the upper right of the github interface.

i'll be checking in two workflow files sooner rather than later, and start the process of auto building the bottles for the last three major revisions of macos for the x86_64 platform.

however, if you've clicked the _watch_ button you'll more than likely see a notification from github about the release. (this will more than likely get annoying as using github actions with homebrew _test-bot_ each formula file will be contained within its own release, thus a number of notifications will be presented via github.

github does offer a more granular approach for _watching_ certain aspects of a repo, just click the little down arrow next to the _watch_ button and uncheck _releases_ to not be notified of all the releases that will be published in the coming days.

---

as for auto bottling and publishing releases. i think i have a _decent_ understanding of building bottles and publishing them to github releases, and i have setup 3 separate virtual machines of macos to use as _self-hosted_ github actions runners for building the bottles. so hopefully if all goes well i'll have all the formula file contained within this repo built using github actions and published using github actions as well in conjuction with my _self-hosted_ runners.

unfortunately, i don't have access to an m1 mac, so i can't build bottles for that platform. 🤷‍♂️, and github presently does not provide an environment for using github actions against the m1 platform.

all that said, once i get the formula within this repo setup using github actions i'll close this issue, and start addressing the other open issues and pull requests within this repo.

and i'll open another issue, related to setting up an action that will hopefully build and publish a dmg image of freecad will be a evergreen 🌲 release of freecad. 🤞

---

so as a friendly reminder for those _watching_ this repo **you may want to adjust your _watch_ settings**

ipatch, 2021-08-21:
it looks github took macos 11 out of preview and has officially released it. so github now supports catalina and big sur, which is good.

https://github.com/actions/virtual-environments/issues/2486#issuecomment-899752373

ipatch, 2021-08-22:
this is the current road block i'm running into 🤷‍♂️

https://github.com/Homebrew/discussions/discussions/2028

ipatch, 2021-08-23:
wow this has been quite an adventure. just now got the first bottle published for this repo, ie. swig using github actions ci/cd and the releases.

one down, and 25 to go.

things will get more interesting once i start running to python and python related releases.

ipatch, 2021-08-25:
_leaving this for future me 👴🏼_

if we (tap maintainers) decide to stick with a specific version of python ie. `python@3.9` then pretty sure all sorts of conflicts are going to arise, and using upstream python will more than likely cause some excessive friction with warp speed 🛸 that homebrew updates it's packages.

so _figures crossed_ i'm going to try and stick with `python@3.9.6` which will hopefully coexist with the homebrew-core python _presently `3.9.6` as well, but they (homebrew-core) truncate things at `3.9` so possible (more finger crossing) we can use `3.9.6` and use this method for formulae

```
#major_minor_patch
```

doing a little digging the homebrew api / ruby docs.

https://rubydoc.brew.sh/Version.html#major_minor-instance_method

ipatch, 2021-12-10:
closing this now, as most if not all formula within this repo have auto generated bottles using the provided github actions CI/CD to build bottles of the of the formula contained within this repo.

if there is a specific issue regarding a specific formula please open a new issue, but if there is a overall general issue / comment feel free to leave one below as i'm not opposed to reopening this issue if required.

closed 2021-12-10 by ipatch