FreeCAD/homebrew-freecad #80: Xcode not set up properly

kyirong6, 2018-02-06:
brew install freecad/freecad/qt@5.6

Warning: Calling 'depends_on :mysql' is deprecated!
Use 'depends_on "mysql"' instead.
/usr/local/Homebrew/Library/Taps/freecad/homebrew-freecad/Formula/qt@5.6.rb:33:in `<class:QtAT56>'
Please report this to the freecad/freecad tap!

Warning: Calling 'depends_on :postgresql' is deprecated!
Use 'depends_on "postgresql"' instead.
/usr/local/Homebrew/Library/Taps/freecad/homebrew-freecad/Formula/qt@5.6.rb:34:in `<class:QtAT56>'
Please report this to the freecad/freecad tap!

==> Installing qt@5.6 from freecad/freecad
Warning: Calling 'depends_on :mysql' is deprecated!
Use 'depends_on "mysql"' instead.
/usr/local/Homebrew/Library/Taps/freecad/homebrew-freecad/Formula/qt@5.6.rb:33:in `<class:QtAT56>'
Please report this to the freecad/freecad tap!

Warning: Calling 'depends_on :postgresql' is deprecated!
Use 'depends_on "postgresql"' instead.
/usr/local/Homebrew/Library/Taps/freecad/homebrew-freecad/Formula/qt@5.6.rb:34:in `<class:QtAT56>'
Please report this to the freecad/freecad tap!

==> Downloading https://download.qt.io/official_releases/qt/5.6/5.6.2/single/qt-everywhere-opensource-src-5.6.2.tar.xz
Already downloaded: /Users/choendenkyirong/Library/Caches/Homebrew/qt@5.6-5.6.2.tar.xz
==> Downloading https://raw.githubusercontent.com/Homebrew/formula-patches/04c2de3/qt5/qtconnectivity-bluetooth-fix.diff
Already downloaded: /Users/choendenkyirong/Library/Caches/Homebrew/qt@5.6--patch-41fd73cba0018180015c2be191d63b3c33289f19132c136f482f5c7477620931.diff
==> Downloading https://raw.githubusercontent.com/Homebrew/formula-patches/04c2de3/qt5/qtwebengine-bluetooth-fix.diff
Already downloaded: /Users/choendenkyirong/Library/Caches/Homebrew/qt@5.6--patch-218b8682d7e8a3f74618d0bd87e4797e13fc2cbfe49c21a2845e64da3fe8868a.diff
==> Patching
==> Applying qtconnectivity-bluetooth-fix.diff
patching file qtconnectivity/src/bluetooth/osx/osxbtcentralmanager_p.h
patching file qtconnectivity/src/bluetooth/osx/osxbtledeviceinquiry.mm
patching file qtconnectivity/src/bluetooth/osx/osxbtutility.mm
==> Applying qtwebengine-bluetooth-fix.diff
patching file qtwebengine/src/3rdparty/chromium/base/mac/sdk_forward_declarations.h
==> Downloading https://download.qt.io/community_releases/5.6/5.6.2/qtwebkit-opensource-src-5.6.2.tar.xz
Already downloaded: /Users/choendenkyirong/Library/Caches/Homebrew/qt@5.6--qt-webkit-5.6.2.tar.xz
==> ./configure -verbose -prefix /usr/local/Cellar/qt@5.6/5.6.2 -release -opensource -confirm-license -system-zlib -qt-libpng -qt-libjpeg -qt-freetype -qt-pcre -nomake tests -no-rpa
Last 15 lines from /Users/choendenkyirong/Library/Logs/Homebrew/qt@5.6/01.configure:
-qt-freetype
-qt-pcre
-nomake
tests
-no-rpath
-nomake
examples
-no-dbus

+ cd qtbase
+ /private/tmp/qt@5.6-20180206-33496-ph7qx/qt-everywhere-opensource-src-5.6.2/qtbase/configure -top-level -verbose -prefix /usr/local/Cellar/qt@5.6/5.6.2 -release -opensource -confirm-license -system-zlib -qt-libpng -qt-libjpeg -qt-freetype -qt-pcre -nomake tests -no-rpath -nomake examples -no-dbus

   Xcode not set up properly. You may need to confirm the license
   agreement by running /usr/bin/xcodebuild without arguments.

If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
https://github.com/freecad/homebrew-freecad/issues

ianrrees, 2018-02-06:
As the error message says: 

> Xcode not set up properly. You may need to confirm the license
agreement by running /usr/bin/xcodebuild without arguments.

Have you tried that?

kyirong6, 2018-02-06:
Yeah, I end up getting this response when I run that:

xcode-select: error: tool 'xcodebuild' requires Xcode, but active developer directory '/Library/Developer/CommandLineTools' is a command line tools instance

ianrrees, 2018-02-06:
Hmm, sounds like you've got a problem with your Xcode install then.  You'll need to figure that out before many Homebrew programs will be able to build properly.  That's not related to FreeCAD, so I'm going to close this ticket.

Also, FYI - we're working on updating the FreeCAD build stuff for OSX at the moment.  Qt5.6 is getting a bit old now - hopefully we'll be able to update the process/instructions/etc soon to work with a newer Qt.

closed 2018-02-06 by ianrrees

kyirong6, 2018-02-06:
Solved the issue: 

sudo xcode-select -s /Applications/Xcode.app/Contents/Developer

Thanks!