FreeCAD/homebrew-freecad #81: Error during "make"

kyirong6, 2018-02-06:
brew install freecad/freecad/qt@5.6

Warning: Calling 'depends_on :mysql' is deprecated!
Use 'depends_on "mysql"' instead.
/usr/local/Homebrew/Library/Taps/freecad/homebrew-freecad/Formula/qt@5.6.rb:33:in `<class:QtAT56>'
Please report this to the freecad/freecad tap!

Warning: Calling 'depends_on :postgresql' is deprecated!
Use 'depends_on "postgresql"' instead.
/usr/local/Homebrew/Library/Taps/freecad/homebrew-freecad/Formula/qt@5.6.rb:34:in `<class:QtAT56>'
Please report this to the freecad/freecad tap!

==> Installing qt@5.6 from freecad/freecad
Warning: Calling 'depends_on :mysql' is deprecated!
Use 'depends_on "mysql"' instead.
/usr/local/Homebrew/Library/Taps/freecad/homebrew-freecad/Formula/qt@5.6.rb:33:in `<class:QtAT56>'
Please report this to the freecad/freecad tap!

Warning: Calling 'depends_on :postgresql' is deprecated!
Use 'depends_on "postgresql"' instead.
/usr/local/Homebrew/Library/Taps/freecad/homebrew-freecad/Formula/qt@5.6.rb:34:in `<class:QtAT56>'
Please report this to the freecad/freecad tap!

==> Downloading https://download.qt.io/official_releases/qt/5.6/5.6.2/single/qt-everywhere-opensource-src-5.6.2.tar.xz
Already downloaded: /Users/choendenkyirong/Library/Caches/Homebrew/qt@5.6-5.6.2.tar.xz
==> Downloading https://raw.githubusercontent.com/Homebrew/formula-patches/04c2de3/qt5/qtconnectivity-bluetooth-fix.diff
Already downloaded: /Users/choendenkyirong/Library/Caches/Homebrew/qt@5.6--patch-41fd73cba0018180015c2be191d63b3c33289f19132c136f482f5c7477620931.diff
==> Downloading https://raw.githubusercontent.com/Homebrew/formula-patches/04c2de3/qt5/qtwebengine-bluetooth-fix.diff
Already downloaded: /Users/choendenkyirong/Library/Caches/Homebrew/qt@5.6--patch-218b8682d7e8a3f74618d0bd87e4797e13fc2cbfe49c21a2845e64da3fe8868a.diff
==> Patching
==> Applying qtconnectivity-bluetooth-fix.diff
patching file qtconnectivity/src/bluetooth/osx/osxbtcentralmanager_p.h
patching file qtconnectivity/src/bluetooth/osx/osxbtledeviceinquiry.mm
patching file qtconnectivity/src/bluetooth/osx/osxbtutility.mm
==> Applying qtwebengine-bluetooth-fix.diff
patching file qtwebengine/src/3rdparty/chromium/base/mac/sdk_forward_declarations.h
==> Downloading https://download.qt.io/community_releases/5.6/5.6.2/qtwebkit-opensource-src-5.6.2.tar.xz
Already downloaded: /Users/choendenkyirong/Library/Caches/Homebrew/qt@5.6--qt-webkit-5.6.2.tar.xz
==> ./configure -verbose -prefix /usr/local/Cellar/qt@5.6/5.6.2 -release -opensource -confirm-license -system-zlib -qt-libpng -qt-libjpeg -qt-freetype -qt-pcre -nomake tests -no-rpa
**==> make
Last 15 lines from /Users/choendenkyirong/Library/Logs/Homebrew/qt@5.6/02.make:
                                      ^
qcocoahelpers.mm:550:38: error: use of undeclared identifier 'InvalidBounds'
    require_action(inBounds != NULL, InvalidBounds, err = paramErr);
                                     ^
qcocoahelpers.mm:551:37: error: use of undeclared identifier 'InvalidImage'
    require_action(inImage != NULL, InvalidImage, err = paramErr);
                                    ^
3 errors generated.
make[5]: *** [.obj/qcocoahelpers.o] Error 1
make[5]: *** Waiting for unfinished jobs....
make[4]: *** [sub-cocoa-make_first] Error 2
make[3]: *** [sub-platforms-make_first] Error 2
make[2]: *** [sub-plugins-make_first] Error 2
make[1]: *** [sub-src-make_first] Error 2
make: *** [module-qtbase-make_first] Error 2**

If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
https://github.com/freecad/homebrew-freecad/issues

kyirong6, 2018-02-06:
The "make" gets stuck there for a while until it spits out these errors.

peterlama, 2018-09-01:
This issue is obsolete.

closed 2018-09-01 by peterlama