FreeCAD/homebrew-freecad #85: cMake error

rap-gti, 2018-03-25:
When trying to install pyside2 I get the followingon High Sierra 10.13.3:

==> Installing freecad/freecad/freecad dependency: FreeCAD/freecad/pyside2
==> Downloading https://codereview.qt-project.org/gitweb?p=pyside/pyside-setup.git;a=snapshot;h=b09fd
Already downloaded: /Users/rap/Library/Caches/Homebrew/pyside2-5.9-b09fde6.git;a=snapshot;h=b09fde6260b255e8b93b0d20a337e701bc940a99;sf=tgz
==> cmake -DCMAKE_C_FLAGS_RELEASE=-DNDEBUG -DCMAKE_CXX_FLAGS_RELEASE=-DNDEBUG -DCMAKE_INSTALL_PREFIX=
Last 15 lines from /Users/rap/Library/Logs/Homebrew/pyside2/01.cmake:
-- optional module Qt5WebKitWidgets skipped
-- optional module Qt5WebSockets found ()
-- optional module Qt53DCore found ()
-- optional module Qt53DRender found ()
-- optional module Qt53DInput found ()
-- optional module Qt53DLogic found ()
-- optional module Qt53DAnimation found ()
-- optional module Qt53DExtras found ()
-- Detected OS: mac
CMake Error at CMakeLists.txt:418 (message):
  Could not detect Python module installation directory.

Python was already installed by homebrew

peterlama, 2018-03-25:
That is strange. It seems to work fine for me. Can you post the output of `brew install pyside2 -v`

dansrogers, 2018-03-28:
it may not be working with python3.  I think the python3 -> python; python -> python2 rename rolled out.    Do you have a /usr/local/bin/python2?

rap-gti, 2018-04-09:
Sorry for not replying earlier.  My computer decided to die, so I am now back trying this again on a newer platform 10.13.4.  I am now seeing this error message in the cmake log file:

-- Found unsuitable Qt version "5.10.1" from /usr/local/opt/qt/bin/qmake

I haven't been able to find version 5.6 in homebrew anymore. It insists on installing 5.10. Do you have any suggestions where I can get the correct version?
Thanks.

peterlama, 2018-04-09:
That is strange, the 0.17 release build was created with homebrew deps and uses qt 5.10. https://github.com/FreeCAD/FreeCAD/releases/tag/0.17. Is this a fresh install?

By the way, why is it even trying to compile pyside2? It should be using the bottle.

rap-gti, 2018-04-09:
The original pyside problem was on a computer that was unstable.  I am now starting with a new install on a new laptop, and here are the messages:
brew tap freecad/freecad
==> Tapping freecad/freecad
Cloning into '/usr/local/Homebrew/Library/Taps/freecad/homebrew-freecad'...
remote: Counting objects: 16, done.
remote: Compressing objects: 100% (16/16), done.
remote: Total 16 (delta 0), reused 3 (delta 0), pack-reused 0
Unpacking objects: 100% (16/16), done.
Tapped 12 formulae (51 files, 66.9KB)
brew install freecad
==> Installing freecad from freecad/freecad
==> Installing dependencies for freecad/freecad/freecad: qtwebkit, FreeCAD/freecad/shiboken2, FreeCAD/freecad/pyside2, FreeCAD/freecad/pyside2-tools, freecad/freecad/matplotlib, freecad/freecad/med-file, FreeCAD/freecad/nglib, FreeCAD/freecad/coin, FreeCAD/freecad/pivy
==> Installing freecad/freecad/freecad dependency: qtwebkit
==> Downloading https://dl.bintray.com/freecad/bottles-freecad/qtwebkit-5.212-72cfbd.high_sierra.bottle
######################################################################## 100.0%
==> Pouring qtwebkit-5.212-72cfbd.high_sierra.bottle.tar.gz
==> Caveats
This formula is keg-only, which means it was not symlinked into /usr/local,
because Qt itself is keg only which implies the same for Qt modules.

For compilers to find this software you may need to set:
    LDFLAGS:  -L/usr/local/opt/qtwebkit/lib
    CPPFLAGS: -I/usr/local/opt/qtwebkit/include

==> Summary
🍺  /usr/local/Cellar/qtwebkit/5.212-72cfbd: 164 files, 45.7MB
==> Installing freecad/freecad/freecad dependency: FreeCAD/freecad/shiboken2
==> Downloading https://dl.bintray.com/freecad/bottles-freecad/shiboken2-5.9-b09fde6_1.high_sierra.bott
######################################################################## 100.0%
==> Pouring shiboken2-5.9-b09fde6_1.high_sierra.bottle.tar.gz
🍺  /usr/local/Cellar/shiboken2/5.9-b09fde6_1: 40 files, 2.3MB
==> Installing freecad/freecad/freecad dependency: FreeCAD/freecad/pyside2
==> Downloading https://dl.bintray.com/freecad/bottles-freecad/pyside2-5.9-b09fde6.high_sierra.bottle.t
######################################################################## 100.0%
==> Pouring pyside2-5.9-b09fde6.high_sierra.bottle.tar.gz
🍺  /usr/local/Cellar/pyside2/5.9-b09fde6: 145 files, 33.1MB
==> Installing freecad/freecad/freecad dependency: FreeCAD/freecad/pyside2-tools
==> Downloading https://dl.bintray.com/freecad/bottles-freecad/pyside2-tools-5.9-413ecc7.high_sierra.bo
######################################################################## 100.0%
==> Pouring pyside2-tools-5.9-413ecc7.high_sierra.bottle.tar.gz
🍺  /usr/local/Cellar/pyside2-tools/5.9-413ecc7: 40 files, 371KB
==> Installing freecad/freecad/freecad dependency: freecad/freecad/matplotlib
==> Downloading https://dl.bintray.com/freecad/bottles-freecad/matplotlib-2.1.1.high_sierra.bottle.tar.
######################################################################## 100.0%
==> Pouring matplotlib-2.1.1.high_sierra.bottle.tar.gz
==> Caveats
If you want to use the `wxagg` backend, do `brew install wxpython`.
This can be done even after the matplotlib install.
==> Summary
🍺  /usr/local/Cellar/matplotlib/2.1.1: 1,186 files, 20.0MB
==> Installing freecad/freecad/freecad dependency: freecad/freecad/med-file
==> Downloading https://dl.bintray.com/freecad/bottles-freecad/med-file-3.2.0.high_sierra.bottle.tar.gz
######################################################################## 100.0%
==> Pouring med-file-3.2.0.high_sierra.bottle.tar.gz
Warning: freecad/freecad/med-file dependency gcc was built with a different C++ standard
library (libstdc++ from clang). This may cause problems at runtime.
🍺  /usr/local/Cellar/med-file/3.2.0: 90 files, 8.0MB
==> Installing freecad/freecad/freecad dependency: FreeCAD/freecad/nglib
==> Downloading https://dl.bintray.com/freecad/bottles-freecad/nglib-5.3.1_1.high_sierra.bottle.tar.gz
######################################################################## 100.0%
==> Pouring nglib-5.3.1_1.high_sierra.bottle.tar.gz
🍺  /usr/local/Cellar/nglib/5.3.1_1: 208 files, 5.5MB
==> Installing freecad/freecad/freecad dependency: FreeCAD/freecad/coin
==> Downloading https://dl.bintray.com/freecad/bottles-freecad/coin-4.0.0a-92cea70.high_sierra.bottle.1
######################################################################## 100.0%
==> Pouring coin-4.0.0a-92cea70.high_sierra.bottle.1.tar.gz
🍺  /usr/local/Cellar/coin/4.0.0a-92cea70: 891 files, 10.0MB
==> Installing freecad/freecad/freecad dependency: FreeCAD/freecad/pivy
==> Downloading https://dl.bintray.com/freecad/bottles-freecad/pivy-0.5.0-4b84e76.high_sierra.bottle.3.
######################################################################## 100.0%
[... 4 lines cut ...]
==> Downloading from https://codeload.github.com/FreeCAD/FreeCAD/tar.gz/0.17_pre
######################################################################## 100.0%
==> cmake -DCMAKE_C_FLAGS_RELEASE=-DNDEBUG -DCMAKE_CXX_FLAGS_RELEASE=-DNDEBUG -DCMAKE_INSTALL_PREFIX=/u
Last 15 lines from /Users/reparker/Library/Logs/Homebrew/freecad/01.cmake:
Qt QTXML library not found.
Qt QTNETWORK library not found.
Qt QTCORE library not found.
CMake Error at CMakeLists.txt:705 (MESSAGE):
  =======================

  Qt libraries not found.

  =======================

-- Configuring incomplete, errors occurred!
See also "/tmp/freecad-20180409-15519-1x4dste/FreeCAD-0.17_pre/Build/CMakeFiles/CMakeOutput.log".
See also "/tmp/freecad-20180409-15519-1x4dste/FreeCAD-0.17_pre/Build/CMakeFiles/CMakeError.log".

If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
https://github.com/FreeCAD/homebrew-freecad/issues

These open issues may also help:
Auto-bottling for homebrew-freecad tap formulae https://github.com/FreeCAD/homebrew-freecad/issues/63

[01.cmake.txt](https://github.com/FreeCAD/homebrew-freecad/files/1891254/01.cmake.txt)

closed 2018-04-09 by rap-gti