FreeCAD/homebrew-freecad #86: Installing shiboken2 from freecad/freecad error

mahaidong, 2018-03-27:
==> Installing shiboken2 from freecad/freecad
==> Downloading https://codereview.qt-project.org/gitweb?p=pyside/pyside-setup.git;a=snapshot;h=b09f
Already downloaded: /Users/apple/Library/Caches/Homebrew/shiboken2-5.9-b09fde6.git;a=snapshot;h=b09fde6260b255e8b93b0d20a337e701bc940a99;sf=tgz
==> cmake -DCMAKE_C_FLAGS_RELEASE=-DNDEBUG -DCMAKE_CXX_FLAGS_RELEASE=-DNDEBUG -DCMAKE_INSTALL_PREFIX
==> make -j4 install
==> cmake -DCMAKE_C_FLAGS_RELEASE=-DNDEBUG -DCMAKE_CXX_FLAGS_RELEASE=-DNDEBUG -DCMAKE_INSTALL_PREFIX
==> make -j4 install
Last 15 lines from /Users/apple/Library/Logs/Homebrew/shiboken2/04.make:
[ 21%] Linking CXX shared library libsample.dylib
cd /tmp/shiboken2-20180327-19908-1wec1yb/pyside-setup-b09fde6/macbuild2.7/tests/libsample && /usr/local/Cellar/cmake/3.10.3/bin/cmake -E cmake_link_script CMakeFiles/libsample.dir/link.txt --verbose=1
/usr/local/Homebrew/Library/Homebrew/shims/super/clang++  -Wall -fvisibility=hidden -Wno-strict-aliasing -D QT_NO_CAST_FROM_ASCII -D QT_NO_CAST_TO_ASCII -DNDEBUG -dynamiclib -Wl,-headerpad_max_install_names  -o libsample.dylib -install_name @rpath/libsample.dylib CMakeFiles/libsample.dir/abstract.cpp.o CMakeFiles/libsample.dir/blackbox.cpp.o CMakeFiles/libsample.dir/bytearray.cpp.o CMakeFiles/libsample.dir/bucket.cpp.o CMakeFiles/libsample.dir/collector.cpp.o CMakeFiles/libsample.dir/complex.cpp.o CMakeFiles/libsample.dir/onlycopy.cpp.o CMakeFiles/libsample.dir/derived.cpp.o CMakeFiles/libsample.dir/echo.cpp.o CMakeFiles/libsample.dir/functions.cpp.o CMakeFiles/libsample.dir/handle.cpp.o CMakeFiles/libsample.dir/implicitconv.cpp.o CMakeFiles/libsample.dir/injectcode.cpp.o CMakeFiles/libsample.dir/listuser.cpp.o CMakeFiles/libsample.dir/modifications.cpp.o CMakeFiles/libsample.dir/mapuser.cpp.o CMakeFiles/libsample.dir/modified_constructor.cpp.o CMakeFiles/libsample.dir/multiple_derived.cpp.o CMakeFiles/libsample.dir/objectmodel.cpp.o CMakeFiles/libsample.dir/objecttype.cpp.o CMakeFiles/libsample.dir/objecttypeholder.cpp.o CMakeFiles/libsample.dir/objecttypelayout.cpp.o CMakeFiles/libsample.dir/objecttypeoperators.cpp.o CMakeFiles/libsample.dir/objectview.cpp.o CMakeFiles/libsample.dir/overload.cpp.o CMakeFiles/libsample.dir/overloadsort.cpp.o CMakeFiles/libsample.dir/pairuser.cpp.o CMakeFiles/libsample.dir/pen.cpp.o CMakeFiles/libsample.dir/photon.cpp.o CMakeFiles/libsample.dir/point.cpp.o CMakeFiles/libsample.dir/pointf.cpp.o CMakeFiles/libsample.dir/polygon.cpp.o CMakeFiles/libsample.dir/protected.cpp.o CMakeFiles/libsample.dir/reference.cpp.o CMakeFiles/libsample.dir/sample.cpp.o CMakeFiles/libsample.dir/samplenamespace.cpp.o CMakeFiles/libsample.dir/sbkdate.cpp.o CMakeFiles/libsample.dir/simplefile.cpp.o CMakeFiles/libsample.dir/size.cpp.o CMakeFiles/libsample.dir/sometime.cpp.o CMakeFiles/libsample.dir/str.cpp.o CMakeFiles/libsample.dir/strlist.cpp.o CMakeFiles/libsample.dir/templateptr.cpp.o CMakeFiles/libsample.dir/transform.cpp.o CMakeFiles/libsample.dir/virtualmethods.cpp.o CMakeFiles/libsample.dir/expression.cpp.o CMakeFiles/libsample.dir/filter.cpp.o 
[ 21%] Built target libsample
[ 21%] Building CXX object ApiExtractor/CMakeFiles/apiextractor.dir/qtdocparser.cpp.o
cd /tmp/shiboken2-20180327-19908-1wec1yb/pyside-setup-b09fde6/macbuild2.7/ApiExtractor && /usr/local/Homebrew/Library/Homebrew/shims/super/clang++  -DCLANG_BUILTIN_INCLUDES_DIR=\"\" -DHAVE_LIBXSLT -DQT_CORE_LIB -DQT_NETWORK_LIB -DQT_NO_DEBUG -DQT_PLUGIN -DQT_SHARED -DQT_XMLPATTERNS_LIB -DQT_XML_LIB -DRXX_ALLOCATOR_INIT_0 -I/tmp/shiboken2-20180327-19908-1wec1yb/pyside-setup-b09fde6/macbuild2.7/ApiExtractor/apiextractor_autogen/include -I/tmp/shiboken2-20180327-19908-1wec1yb/pyside-setup-b09fde6/sources/shiboken2/ApiExtractor -I/tmp/shiboken2-20180327-19908-1wec1yb/pyside-setup-b09fde6/macbuild2.7/ApiExtractor -I/tmp/shiboken2-20180327-19908-1wec1yb/pyside-setup-b09fde6/sources/shiboken2/ApiExtractor/parser -I/tmp/shiboken2-20180327-19908-1wec1yb/pyside-setup-b09fde6/sources/shiboken2/ApiExtractor/parser/rpp -I/usr/local/opt/llvm/include -I/usr/include/libxml2 -iframework /usr/local/opt/qt/lib -isystem /usr/local/opt/qt/lib/QtCore.framework/Headers -isystem /usr/local/opt/qt/./mkspecs/macx-clang -isystem /usr/local/opt/qt/lib/QtXml.framework/Headers -isystem /usr/local/opt/qt/lib/QtXmlPatterns.framework/Headers -isystem /usr/local/opt/qt/lib/QtNetwork.framework/Headers  -Wall -fvisibility=hidden -Wno-strict-aliasing -D QT_NO_CAST_FROM_ASCII -D QT_NO_CAST_TO_ASCII -DNDEBUG   -fPIC -fPIC -std=gnu++11 -o CMakeFiles/apiextractor.dir/qtdocparser.cpp.o -c /tmp/shiboken2-20180327-19908-1wec1yb/pyside-setup-b09fde6/sources/shiboken2/ApiExtractor/qtdocparser.cpp
[ 22%] Building CXX object ApiExtractor/CMakeFiles/apiextractor.dir/apiextractor_autogen/mocs_compilation.cpp.o
cd /tmp/shiboken2-20180327-19908-1wec1yb/pyside-setup-b09fde6/macbuild2.7/ApiExtractor && /usr/local/Homebrew/Library/Homebrew/shims/super/clang++  -DCLANG_BUILTIN_INCLUDES_DIR=\"\" -DHAVE_LIBXSLT -DQT_CORE_LIB -DQT_NETWORK_LIB -DQT_NO_DEBUG -DQT_PLUGIN -DQT_SHARED -DQT_XMLPATTERNS_LIB -DQT_XML_LIB -DRXX_ALLOCATOR_INIT_0 -I/tmp/shiboken2-20180327-19908-1wec1yb/pyside-setup-b09fde6/macbuild2.7/ApiExtractor/apiextractor_autogen/include -I/tmp/shiboken2-20180327-19908-1wec1yb/pyside-setup-b09fde6/sources/shiboken2/ApiExtractor -I/tmp/shiboken2-20180327-19908-1wec1yb/pyside-setup-b09fde6/macbuild2.7/ApiExtractor -I/tmp/shiboken2-20180327-19908-1wec1yb/pyside-setup-b09fde6/sources/shiboken2/ApiExtractor/parser -I/tmp/shiboken2-20180327-19908-1wec1yb/pyside-setup-b09fde6/sources/shiboken2/ApiExtractor/parser/rpp -I/usr/local/opt/llvm/include -I/usr/include/libxml2 -iframework /usr/local/opt/qt/lib -isystem /usr/local/opt/qt/lib/QtCore.framework/Headers -isystem /usr/local/opt/qt/./mkspecs/macx-clang -isystem /usr/local/opt/qt/lib/QtXml.framework/Headers -isystem /usr/local/opt/qt/lib/QtXmlPatterns.framework/Headers -isystem /usr/local/opt/qt/lib/QtNetwork.framework/Headers  -Wall -fvisibility=hidden -Wno-strict-aliasing -D QT_NO_CAST_FROM_ASCII -D QT_NO_CAST_TO_ASCII -DNDEBUG   -fPIC -fPIC -std=gnu++11 -o CMakeFiles/apiextractor.dir/apiextractor_autogen/mocs_compilation.cpp.o -c /tmp/shiboken2-20180327-19908-1wec1yb/pyside-setup-b09fde6/macbuild2.7/ApiExtractor/apiextractor_autogen/mocs_compilation.cpp
[ 22%] Linking CXX static library ../tests/libapiextractor.a
cd /tmp/shiboken2-20180327-19908-1wec1yb/pyside-setup-b09fde6/macbuild2.7/ApiExtractor && /usr/local/Cellar/cmake/3.10.3/bin/cmake -P CMakeFiles/apiextractor.dir/cmake_clean_target.cmake
cd /tmp/shiboken2-20180327-19908-1wec1yb/pyside-setup-b09fde6/macbuild2.7/ApiExtractor && /usr/local/Cellar/cmake/3.10.3/bin/cmake -E cmake_link_script CMakeFiles/apiextractor.dir/link.txt --verbose=1
/usr/bin/ar qc ../tests/libapiextractor.a  CMakeFiles/apiextractor.dir/apiextractor.cpp.o CMakeFiles/apiextractor.dir/abstractmetabuilder.cpp.o CMakeFiles/apiextractor.dir/abstractmetalang.cpp.o CMakeFiles/apiextractor.dir/fileout.cpp.o CMakeFiles/apiextractor.dir/graph.cpp.o CMakeFiles/apiextractor.dir/reporthandler.cpp.o CMakeFiles/apiextractor.dir/typeparser.cpp.o CMakeFiles/apiextractor.dir/typesystem.cpp.o CMakeFiles/apiextractor.dir/include.cpp.o CMakeFiles/apiextractor.dir/typedatabase.cpp.o CMakeFiles/apiextractor.dir/clangparser/compilersupport.cpp.o CMakeFiles/apiextractor.dir/clangparser/clangparser.cpp.o CMakeFiles/apiextractor.dir/clangparser/clangbuilder.cpp.o CMakeFiles/apiextractor.dir/clangparser/clangdebugutils.cpp.o CMakeFiles/apiextractor.dir/clangparser/clangutils.cpp.o CMakeFiles/apiextractor.dir/parser/codemodel.cpp.o CMakeFiles/apiextractor.dir/docparser.cpp.o CMakeFiles/apiextractor.dir/doxygenparser.cpp.o CMakeFiles/apiextractor.dir/qtdocparser.cpp.o CMakeFiles/apiextractor.dir/apiextractor_autogen/mocs_compilation.cpp.o
/usr/bin/ranlib ../tests/libapiextractor.a
[ 22%] Built target apiextractor
make: *** [all] Error 2

If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
https://github.com/freecad/homebrew-freecad/issues

peterlama, 2018-03-27:
I see that is the "Last 15 lines from /Users/apple/Library/Logs/Homebrew/shiboken2/04.make" but it doesn't show the actual error. Can you upload /Users/apple/Library/Logs/Homebrew/shiboken2/04.make somewhere and post a link here?

mahaidong, 2018-03-28:
here : https://github.com/caadxyz/homebrew-freecad/tree/master/make/shiboken2

dansrogers, 2018-03-28:
Looks like numpy is required.

peterlama, 2018-03-28:
What version of macOS do you have?

mahaidong, 2018-03-29:
version 10.13.3(17D102)

mahaidong, 2018-03-30:
finally, I rename `/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.13.sdk` to `/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk` , then I can compile the source codes

closed 2018-03-30 by mahaidong