FreeCAD/homebrew-freecad #90: after installing dependencies, brew install freecad

danielavdh, 2018-07-01:
==> Installing freecad from freecad/freecad
==> Downloading https://github.com/FreeCAD/FreeCAD/archive/0.17.tar.gz
==> Downloading from https://codeload.github.com/FreeCAD/FreeCAD/tar.gz/0.17
######################################################################## 100.0%
==> cmake -DCMAKE_C_FLAGS_RELEASE=-DNDEBUG -DCMAKE_CXX_FLAGS_RELEASE=-DNDEBUG -DCMAKE_INSTALL_PREFIX=/usr/local/Cellar/fre
Last 15 lines from /Users/daniela/Library/Logs/Homebrew/freecad/01.cmake:
-- Found Doxygen: /usr/local/bin/doxygen (found version "1.8.14") found components:  doxygen missing components:  dot
-- Note: Doxygen docs will look better with graphviz's dot installed.
-- /tmp/freecad-20180701-97167-sgw55l/FreeCAD-0.17/Build/src/Gui/DocumentPy.cpp
-- /tmp/freecad-20180701-97167-sgw55l/FreeCAD-0.17/Build/src/Gui/PythonWorkbenchPy.cpp
-- /tmp/freecad-20180701-97167-sgw55l/FreeCAD-0.17/Build/src/Gui/ViewProviderPy.cpp
-- /tmp/freecad-20180701-97167-sgw55l/FreeCAD-0.17/Build/src/Gui/ViewProviderDocumentObjectPy.cpp
-- /tmp/freecad-20180701-97167-sgw55l/FreeCAD-0.17/Build/src/Gui/WorkbenchPy.cpp
-- /tmp/freecad-20180701-97167-sgw55l/FreeCAD-0.17/Build/src/Gui/SelectionObjectPy.cpp
=======================================
Now run 'make' to build FreeCAD
=======================================

-- Configuring incomplete, errors occurred!
See also "/tmp/freecad-20180701-97167-sgw55l/FreeCAD-0.17/Build/CMakeFiles/CMakeOutput.log".
See also "/tmp/freecad-20180701-97167-sgw55l/FreeCAD-0.17/Build/CMakeFiles/CMakeError.log".

If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
https://github.com/FreeCAD/homebrew-freecad/issues

These open issues may also help:
Auto-bottling for homebrew-freecad tap formulae https://github.com/FreeCAD/homebrew-freecad/issues/63

YSHI17172, 2018-07-20:
same issue here

ianrrees, 2018-07-21:
Hi @danielavdh (and @YSHI17172 ) - I'm not in a position to help with this as I've switched back to Linux from MacOS, but wanted to suggest that you might share some more information about your problem, in order to help others understand the situation you're encountering.

For example, what version of MacOS are you using?  Have you been able to use the provided FreeCAD binary?  Could you attach the build log referred to by homebrew, rather than just pasting the truncated error report?  Thanks!

adcwilliams, 2018-08-31:
Same here, using High Sierra. Here's CMakeError.log:

```
Determining if the include file GL/gl.h exists failed with the following output:
Change Dir: /tmp/freecad-20180831-28394-kuejhp/FreeCAD-0.17/Build/CMakeFiles/CMakeTmp

Run Build Command:"/usr/local/Homebrew/Library/Homebrew/shims/mac/super/gmake" "cmTC_2d73e/fast"
/Library/Developer/CommandLineTools/usr/bin/make -f CMakeFiles/cmTC_2d73e.dir/build.make CMakeFiles/cmTC_2d73e.dir/build
Building C object CMakeFiles/cmTC_2d73e.dir/CheckIncludeFile.c.o
/usr/local/Homebrew/Library/Homebrew/shims/mac/super/clang    -o CMakeFiles/cmTC_2d73e.dir/CheckIncludeFile.c.o   -c /tmp/freecad-20180831-28394-kuejhp/FreeCAD-0.17/Build/CMakeFiles/CMakeTmp/CheckIncludeFile.c
/tmp/freecad-20180831-28394-kuejhp/FreeCAD-0.17/Build/CMakeFiles/CMakeTmp/CheckIncludeFile.c:1:10: fatal error: 'GL/gl.h' file not found
#include <GL/gl.h>
         ^~~~~~~~~
1 error generated.
make[1]: *** [CMakeFiles/cmTC_2d73e.dir/CheckIncludeFile.c.o] Error 1
make: *** [cmTC_2d73e/fast] Error 2
```

peterlama, 2018-09-01:
Compiling 0.17 with homebrew does not work regardless of the above issue, because the libraries that are available in homebrew no longer work with 0.17. Use the `--HEAD` option if you want to compile freecad with homebrew. However, I recommend you use the official builds if you just want to use FreeCAD, and if you want to develop FreeCAD, than homebrew would only be used for installing dependencies.

closed 2018-09-01 by peterlama

luzpaz, 2018-09-01:
@peterlama is there a way write the 0.17 homebrew formula to pin certain older libraries to build 0.17 ?

peterlama, 2018-09-01:
@luzpaz Not that I am aware of. It is just not how homebrew was designed to work. The version of a library is specified in the formula itself. By that I mean if you depend on Qt for example, your formula would specify "depends on qt" and homebrew would than lookup the qt formula on your system and download the version specified there. All the formulas are stored in a git repository, so a user could retrieve an older version of a formula with git, but it is not something that can be specified in the formula. For the official 0.17 builds, we use a snapshot of an entire homebrew install that was created when 0.17 was released. https://github.com/FreeCAD/FreeCAD-ports-cache/releases/tag/v0.17

bblacey, 2018-09-02:
@peterlama and @luzpaz, home-brew does allow pinning formulae but there are pros and cons in the context of CI.  We could "pin" the formulae in the ports-cache to avoid any ports updates during CI.  The pro is that the port dependencies would rarely, if ever, cause a build breakage.  If the goal of CI is to only test FreeCAD source code changes, then one could make an argument that this is the way to go, although it would be different than CI for other platforms.

If however, the goal of CI is to test the FreeCAD source code changes and to ensure it is buildable against the current Homebrew ports, then we should test it against the latest and greatest community ports which is what we do today.  The gap that is yet to be properly filled by us, is to detect ports dependency changes and rebuild the dependent ports automagically.  Since the Homebrew team has become more stringent on which ports they allow in their public repos and have deprecated and abandoned many, the broader community has been trying to improve the infrastructure to better support self-hosted taps (which our FreeCAD-ports-cache) is.  I haven't kept up with it but there were people looking into detecting dependency changes and performing a rebuild.  One approach contemplated is to register for the FreeCAD formula/build update web-service notifications that would kick-off a job to rebuild any of our self-hosted formula that depend upon that port.

Unfortunately, at the current time, I don't have the cycles required to tackle this improvement but it is doable and with a little spelunking, we might find that someone else has documented exactly what needs to be done or that the Homebrew team has made improvements to the out-of-the-box mechanisms to support it directly.