FreeCAD/homebrew-freecad #95: Installation of freecad/qtwebkit fails on Mojave.

fcacarminati, 2018-10-08:
==> Upgrading 4 outdated packages, with result:
freecad/freecad/pyside2 5.11.1 -> 5.11.2, freecad/freecad/pyside2-tools 5.11.1 -> 5.11.2, freecad/freecad/shiboken2 5.11.1 -> 5.11.2, freecad/freecad/qtwebkit 5.212-72cfbd_2 -> 5.212-72cfbd_3
==> Upgrading freecad/freecad/qtwebkit
==> Cloning https://code.qt.io/qt/qtwebkit.git
Cloning into '/Users/fca/Library/Caches/Homebrew/qtwebkit--git'...
Checking out files: 100% (13148/13148), done.
==> Checking out branch 5.212
Already on '5.212'
Your branch is up-to-date with 'origin/5.212'.
==> ./Tools/Scripts/build-webkit --qt --prefix=/usr/local/Cellar/qtwebkit/5.212-
Last 15 lines from /Users/fca/Library/Logs/Homebrew/qtwebkit/01.build-webkit:
xpc_array_apply(xpc_object_t xarray, XPC_NOESCAPE xpc_array_applier_t applier);
^
In file included from ../../Source/WebKit2/UIProcess/WebBackForwardList.cpp:27:
In file included from ../../Source/WebKit2/UIProcess/WebBackForwardList.h:31:
In file included from ../../Source/WebKit2/UIProcess/WebPageProxy.h:33:
In file included from ../../Source/WebKit2/Platform/IPC/Connection.h:48:
../../Source/WTF/wtf/spi/darwin/XPCSPI.h:94:15: error: conflicting types for 'xpc_dictionary_apply'
EXTERN_C bool xpc_dictionary_apply(xpc_object_t xdict, xpc_dictionary_applier_t applier);
              ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/xpc/xpc.h:1997:1: note: previous declaration is here
xpc_dictionary_apply(xpc_object_t xdict,
^
2 errors generated.
[3720/4527] Building CXX object Source/WebCore/CMakeFiles/WebCoreTestSupport.dir/testing/Internals.cpp.o
ninja: build stopped: subcommand failed.

kevinhendricks, 2018-10-29:
I have the exact same issue when building with OSX 10.14 SDK:

[ 81%] Building CXX object Source/WebKit2/CMakeFiles/WebKit2.dir/DatabaseProcess/DatabaseProcess.cpp.o
In file included from /Volumes/SSD-Drive/repo/qtwebkit/Source/WebKit2/DatabaseProcess/DatabaseProcess.cpp:27:
In file included from /Volumes/SSD-Drive/repo/qtwebkit/Source/WebKit2/DatabaseProcess/DatabaseProcess.h:31:
In file included from /Volumes/SSD-Drive/repo/qtwebkit/Source/WebKit2/Shared/ChildProcess.h:29:
In file included from /Volumes/SSD-Drive/repo/qtwebkit/Source/WebKit2/Platform/IPC/Connection.h:48:
/Volumes/SSD-Drive/repo/qtwebkit/Source/WTF/wtf/spi/darwin/XPCSPI.h:93:15: error: 
      conflicting types for 'xpc_array_apply'
EXTERN_C bool xpc_array_apply(xpc_object_t, xpc_array_applier_t);
              ^
/usr/include/xpc/xpc.h:1257:1: note: previous declaration is here
xpc_array_apply(xpc_object_t xarray, XPC_NOESCAPE xpc_array_applier_t applier);

In file included from /Volumes/SSD-Drive/repo/qtwebkit/Source/WebKit2/Platform/IPC/Connection.h:48:
/Volumes/SSD-Drive/repo/qtwebkit/Source/WTF/wtf/spi/darwin/XPCSPI.h:94:15: error: 
      conflicting types for 'xpc_dictionary_apply'
EXTERN_C bool xpc_dictionary_apply(xpc_object_t xdict, xpc_dictionary_ap...
              ^
/usr/include/xpc/xpc.h:1997:1: note: previous declaration is here
xpc_dictionary_apply(xpc_object_t xdict,

It appears that the XPC_NOESCAPE macro definition is the issues and this has changed in /usr/include/xpc/base.h

Not sure which is wrong.

kevinhendricks, 2018-10-29:
BTW, as a workaround to continue the build I edited the file:
      qtwebkit/Source/WTF/wtf/spi/darwin/XPCSPI.h
and changed lines 93 and 94 which are the forward declarations to insert the XPC_NOESCAPE macro in the right places in xpc_array_apply and xpc_dictionary_apply to match the definitions now used in the 10.14 SDK and the build continued just fine.
Since these are external references, it should not be an issue.

vejmarie, 2021-01-05:
I think we can close this issue as it has been fixed in latest release

closed 2021-01-05 by vejmarie