FreeCAD/homebrew-freecad #96: Created a Macports ticket for py3/qt5 build of 0.18dev

luzpaz, 2018-10-25:
https://trac.macports.org/ticket/57450#ticket
Trying to get all repos ready for 0.18 thanks to https://github.com/luzpaz/FreeCAD-dependencies

ianrrees, 2018-10-25:
@luzpaz - could you clarify what this ticket is about?  I'm confused, because it's referring to MacPorts, but is in our hombrew-specific tracker.

luzpaz, 2018-10-25:
@ianrrees I know, sorry. My logic was to direct the macport maintainers of the FC port to the FC homebrew team so that they could work out any workarounds that y'all have figured out. If this is not appropriate I can close it from https://github.com/luzpaz/FreeCAD-dependencies/issues/3

ianrrees, 2018-10-25:
I think it's fine to direct them here, thanks for coordinating!  That said, I don't see the value in having a ticket here about a MacPorts task, so am going to close this one.

closed 2018-10-25 by ianrrees

luzpaz, 2018-10-25:
makes sense. I realized after the fact that I could track it from the other repository. Apologies for the 'noise'

ianrrees, 2018-10-25:
NP