FreeCAD/homebrew-freecad #97: Get ifcopenshell in to homebrew

luzpaz, 2018-11-26:
https://forum.freecadweb.org/viewtopic.php?f=39&t=32158&p=270631#p270631

Currently it's only available via the Arch AUR
per Repology:
[image]

luzpaz, 2021-03-08:
@vejmarie are there any macOS folks open to making a homebrew recipe ?

ipatch, 2021-03-08:
@luzpaz 

are you asking for option to build/compile freecad with LfcOpenShell support?

a quick search, and i came across this, https://github.com/IfcOpenShell/IfcOpenShell/issues/1074

and i assume you're looking for something along the lines of?

```
brew install freecad --with-lfcopenshell
```

luzpaz, 2021-03-08:
Yes, I opened that ticket :)

ipatch, 2022-04-11:
following the instructions on their README

```
cmake \ 
-DOCC_LIBRARY_DIR=$bp/opt/opencascade/lib \
-DOCC_INCLUDE_DIR=$bp/opt/opencascade/include/opencascade \
-DCOLLADA_SUPPORT=0 \
-DCGAL_INCLUDE_DIR=$bp/opt/cgal/include \
-DGMP_LIBRARY_DIR=$bp/opt/gmp/lib \
-DMPFR_LIBRARY_DIR=$bp/opt/mpfr/lib \
-DCMAKE_BUILD_TYPE=RelWithDebInfo \
-DCMAKE_INSTALL_PREFIX=../../installs/v0.7.0 \
-L \
../../src/cmake
```

yields

```
[ 17%] Built target IfcParse
[ 17%] Built target ifcopenshell_wrapper_swig_compilation
[ 18%] Built target IfcParseExamples
[ 25%] Built target IfcGeom_ifc4x3_rc4
[ 32%] Built target IfcGeom_ifc4x3_rc3
[ 39%] Built target IfcGeom_ifc4x3_rc2
[ 46%] Built target IfcGeom_ifc4x3_rc1
[ 53%] Built target IfcGeom_ifc4x2
[ 60%] Built target IfcGeom_ifc4x1
[ 67%] Built target IfcGeom_ifc4
[ 74%] Built target IfcGeom_ifc2x3
[ 75%] Built target libsvgfill
[ 78%] Built target IfcGeom
[ 80%] Built target svgfill
[ 81%] Built target Serializers_ifc4x3_rc4
[ 82%] Built target Serializers_ifc4x3_rc3
[ 83%] Built target Serializers_ifc4x3_rc2
[ 84%] Built target Serializers_ifc4x3_rc1
[ 85%] Built target Serializers_ifc4x2
[ 87%] Built target Serializers_ifc4x1
[ 88%] Built target Serializers_ifc4
[ 89%] Built target Serializers_ifc2x3
[ 94%] Built target Serializers
make[2]: *** No rule to make target `/usr/lib/x86_64-linux-gnu/hdf5/serial/libhdf5_cpp.so', needed by `IfcConvert'.  Stop.
make[1]: *** [CMakeFiles/IfcConvert.dir/all] Error 2
make: *** [all] Error 2
```

luzpaz, 2022-04-11:
Looks like they need hdf5 https://formulae.brew.sh/formula/hdf5

kakke14, 2022-09-08:
Did you manage to get it to work?

I have the same problem, trying to compile IfcOpenShell on Mac apple silicon. 
I have tried installing hdf5 with the brew formula, but I am unsure how to update the path in Cmake.