FreeCAD/homebrew-freecad #98: opencamlib fails during cmake using homebrew

mcmillanrs, 2018-12-27:
Hello:  I am trying to install opencamlib on max osx high sierra using homebrew.  The installation fails during cmake. I am trying to use 3d surface command in freecad 0.17 as installed on the mac.  I have enclosed the output from the terminal so you might diagnose the problem.  I believe the issue is with boost.  I hope you can help.  Cheers Rod
[opencamlib_error.pdf](https://github.com/FreeCAD/homebrew-freecad/files/2711313/opencamlib_error.pdf)

bblacey, 2018-12-27:
@mcmillanrs, I have been prototyping auto-bottling the formula in this repo and in doing so upgraded several of the formula including the opencamlib formula.  You can pull it here https://github.com/FreeCAD/homebrew-freecad/tree/bottling.  Give that a shot and see if it gets you up and running.

ianrrees, 2018-12-27:
Thanks, and nice to see you back @bblacey!

mcmillanrs, 2018-12-27:
> Hi Rod, try this specifically:
>
> brew install --build-from-source --verbose https://raw.githubusercontent.com/FreeCAD/homebrew-freecad/bottling/Formula/opencamlib.rb

Hello Bruce:  Thank you very much for that.  The verbose flag meant that the failed brew indicated that flags need to be set pointing to the /usr/local/opt directory when of course I was puting them in the /usr/local directory so after editing the .bash_profile correctly the rebrew worked at lease I think it did.  The brew summary gave me
/usr/local/Cellar/opencamlib/0.0.1: 48 files, 451.9KB, built in 1 minute 37 seconds

Of course it is built in the usr/local/Cellar.  Opening up freecad and trying this it tells me that “This operation requires OpenCamLib to be installed so Freecad 0.17 cannot find the opencamlib.  I have set enableexperimentalfeatures thinking that was required but that is not it.  Do you know how to have freecad find the opencamlib installation?  Cheers Rod

bblacey, 2018-12-28:
@mcmillanrs, checkout this [forum thread](https://forum.freecadweb.org/viewtopic.php?t=20014) for additional activation ideas - I personally do not use the Path workbench so if that doesn't help then @ianrrees will need to point you in the right direction.  Sorry.

bblacey, 2018-12-28:
@mcmillanrs, ok, I see a couple issues:

1) There was a problem with the `opencamlib` formula so it didn't install the necessary python modules into the site-packages directory.  I have fixed the formula.  You will need to `brew uninstall opencamlib` followed by the `brew install ...` earlier in this ticket.
2) You either need to copy the necessary files into the FreeCAD app bundle or possibly build FreeCAD from source.  This should be easier but `opencamlib` is GPL3 so it can not be bundled with FreeCAD which is LGPL 2.1.  For example, the formula reports the following:
```
-- Installing: /usr/local/Cellar/opencamlib/0.0.1/lib/python2.7/site-packages/ocl.so
-- Up-to-date: /usr/local/Cellar/opencamlib/0.0.1/lib/python2.7/site-packages
-- Installing: /usr/local/Cellar/opencamlib/0.0.1/lib/python2.7/site-packages/camvtk.py
-- Installing: /usr/local/Cellar/opencamlib/0.0.1/lib/python2.7/site-packages/procmemory.py
-- Installing: /usr/local/Cellar/opencamlib/0.0.1/lib/python2.7/site-packages/pyocl.py
-- Installing: /usr/local/Cellar/opencamlib/0.0.1/lib/python2.7/site-packages/STLTools.py
-- Installing: /usr/local/Cellar/opencamlib/0.0.1/lib/opencamlib/libocl.11.10.90.dylib
-- Installing: /usr/local/Cellar/opencamlib/0.0.1/lib/opencamlib/libocl.dylib
```

So you will need to copy the `/usr/local/Cellar/opencamlib/0.0.1/lib/python2.7/site-packages/*` to the FreeCAD.app bundle so that the Path workbench can find them.  It may be as easy as copying everything to `/Applications/FreeCAD.app/Contents/lib/python2.7/site-packages`.  Hope this helps.  

@ianrrees, if you are lurking and remember how to activate `opencamlib` on macOS, please jump in and correct where my guesses are wrong ;)

mcmillanrs, 2018-12-28:
Hello Bruce: thank you for all your work on this. Have you been successful in brewing freecad from the freecad.rb tap at your github site you gave me in the original email? Cheers Rod

iPad

mcmillanrs, 2018-12-28:
> IIRC, OCL is a Python library. So, the easiest way to get it in to FreeCAD is to build FreeCAD from source after installing OCL.

Hello Ian: having successfully used homebrew to install opencamlib, if i can compile freecad successfuly how will the compilation of freecad find ocl if it is in the usr/local/Cellar directory. Will not the freecad compilation expect ocl to be in the usr/local/ directory? Cheers Rod

iPad

bblacey, 2018-12-28:
@mcmillanrs, I just submitted a couple important changes to the upstream opencamlib project under https://github.com/aewallin/opencamlib/pull/36 and pushed a new `opencamlib` formula with these patches to our repo until the pull request is accepted upstream so you should go through the brew uninstall / install process again.

When building on macOS, if you set the `-DFREECAD_CREATE_MAC_APP=ON"` when invoking cmake, it will copy all the dependencies from /usr/local to the FreeCAD.app bundle and set the loader paths so you can freely move FreeCAD around without load issues.  The [Travis-ci config file](https://github.com/FreeCAD/FreeCAD/blob/master/.travis.yml#L165-L219) is a good reference for building on macOS.

I believe, but I am not 100% certain, the you will be able to build on macOS high sierra using the latest formula from this repo from the **bottling branch**.  I have been working through formula and bottling issues and think I have everything working now.  It actually would be good if you give it a shot to see if it works for you.

You will need to do the following in order to instruct homebrew to use the freecad formula from the bottling branch.

```
export HOMEBREW_DEVELOPER=1
brew tap --full freecad/freecad
git -C $(brew --repo freecad/freecad) checkout bottling
```

mcmillanrs, 2018-12-28:
Hello Bruce: Thank you again for all your work.   I have brew uninstalled and brew installed opencamlib I believe succesffully I entered the folllowing text as you suggested but somehow it cannot get to the proper branch it gets tthe master branch.  This is the output in terminal I get:

Rods-Mac-mini-3:~ rodmcmillan$ export HOMEBREW_DEVELOPER=1
Rods-Mac-mini-3:~ rodmcmillan$ brew tap --full freecad/freecad
Rods-Mac-mini-3:~ rodmcmillan$ git -C $(brew --repo freecad/freecad) checkout bottling
error: pathspec 'bottling' did not match any file(s) known to git.

When you say"When building on macOS, if you set the -DFREECAD_CREATE_MAC_APP=ON" when invoking cmake” where do I set this as I thought I would be brewing the tap freecad.rb?  Cheers Rod

On Dec 28, 2018, at 11:07 AM, Bruce Lacey <notifications@github.com> wrote:
> 
> export HOMEBREW_DEVELOPER=1
> brew tap --full freecad/freecad
> git -C $(brew --repo freecad/freecad) checkout bottling

mcmillanrs, 2018-12-28:
Hello Bruce :  I think the "When building on macOS, if you set the -DFREECAD_CREATE_MAC_APP=ON” is an argument you set in the freecad.rb under ards like "
 -DBUILD_FEM_NETGEN:BOOL=ON
      -DFREECAD_USE_EXTERNAL_KDL=ON
      -DCMAKE_BUILD_TYPE=#{build.with?("debug") ? "Debug" : "Release”}”

is that correct?  Cheers Rod
> On Dec 28, 2018, at 11:07 AM, Bruce Lacey <notifications@github.com> wrote:
> 
> When building on macOS, if you set the -DFREECAD_CREATE_MAC_APP=ON"

mcmillanrs, 2018-12-28:
Hello Bruce:  Git branch says that only master is available.  Cheers Rod

> On Dec 28, 2018, at 4:08 PM, Rod McMillan <rodmcmillan@me.com> wrote:
> 
>  git -C $(brew --repo freecad/freecad) checkout bottling

bblacey, 2018-12-29:
@mcmillanrs, Rod, sorry was out most of the day.  Because my environment is configured for development, sometimes there are errors in the details of my instructions.  To access the bottling branch, you will need to pull the branches from git so the bottling branch is accessible.  Normally you wouldn't need to do this but we are in the midst of upgrading the formula before merging to master so try this to pull the git branches into your local repo:

`git -C $(brew --repo freecad/freecad) fetch --all` 
 
Yes, you should be able to build locally with freecad/freecad/freecad.  That said, I hardly ever do that because I need to iterate on code.  I use the following to install the freecad dependencies and run cmake.

Install dependencies and configure cmake arguments:
```
brew install --verbose --only-dependencies freecad
export CMAKE_BUILD_TYPE=${CMAKE_BUILD_TYPE:-Release}
export CMAKE_OPTS="-DBUILD_FEM_NETGEN=ON" 
export QT5_CMAKE_PREFIX=$(ls -d $(brew --cellar)/qt/*/lib/cmake)
export QT5_WEBKIT_CMAKE_PREFIX=$(ls -d $(brew --cellar)/qtwebkit/*/lib/cmake)
export CMAKE_OPTS="${CMAKE_OPTS} -DBUILD_QT5=ON -DCMAKE_PREFIX_PATH=${QT5_CMAKE_PREFIX};${QT5_WEBKIT_CMAKE_PREFIX}"
export CMAKE_ARGS="${CMAKE_OPTS} -DFREECAD_CREATE_MAC_APP=ON"
export INSTALLED_APP_PATH="/usr/local/FreeCAD.app/Contents/MacOS/FreeCAD"
mkdir build
```
Run cmake:
```
cd build
cmake ${CMAKE_ARGS} -DCMAKE_BUILD_TYPE=${CMAKE_BUILD_TYPE} ../
sudo make -j2 install
${INSTALLED_APP_PATH} --console --run-test 0
```

My use case differs from end-users however I'm sharing this in case you want to forge the developer route until we wrap up the formula revisions to make this easier on end users.

mcmillanrs, 2018-12-29:
Hello Bruce:  The brew of opencamlib works.
The brew of brew install freecad/freecad/freecad fails with many caveats.  I added this flag exports and path exportsd to .bash_profile and reran brew and it fails but just says there are configuration errors.

So I used cmake , make install as per last email building from git freecad clone into freecad-code.  The configured and made with 7 warnings and installed in /usr/local/FreeCAD.app/Contents/MacOS/Freecad.

${INSTALLED_APP_PATH} --console --run-test 0 fails so to run freecad I just entered 
/usr/local/FreeCAD.app/Contents/MacOS/Freecad in terminal and get
Rods-Mac-mini-3:~ rodmcmillan$ /usr/local/FreeCAD.app/Contents/MacOS/FreeCAD
Your PYTHONPATH points to a site-packages dir for Python 3.x but you are running Python 2.x!
     PYTHONPATH is currently: ""
     You should `unset PYTHONPATH` to fix this.

I have looked for the solution in the freecad forum but I am unsuccess.

in terminal /usr/local/FreeCAD.app/Contents/MacOS/FreeCADCmd  appears to start the command line Freecad.
also in terminal
Rods-Mac-mini-3:~ rodmcmillan$ python --version
Python 3.7.2

Thank you for all your help but I do not know how to start compiled Freecad successfully.  I hope you can help.  Cheers Rod

> .On Dec 28, 2018, at 8:55 PM, Bruce Lacey <notifications@github.com> wrote:
> 
> /usr/local/FreeCAD.app/Contents/MacOS/FreeCAD

bblacey, 2018-12-29:
Hi Rod, I haven't that issue but currently FreeCAD for macOS under Homebrew requires Python 2.7.  The freecad Homebrew formulae in the bottling branch strive to avoid inadvertent python3 installation by explicitly reference python@2.  During the app packaging, the python2 interpreter and associated site-packages are actually copied to the FreeCAD bundle to ensure that the app is running with the proper interpreter and dependencies.  

Do you require the brewed Python3 outside FreeCAD?  If not, then I recommend that you uninstall Python3 and build FreeCAD for Python2 explicitly.  The `brew install --verbose --only-dependencies freecad` should do so.  

If you try `python -c "import site; print(site.getsitepackages())"`, what does it report?

Also, what happens if you try to start the app by double-clicking it vs the command-line?  If it launches, then open the python console and check the site packages path to ensure it is running from within the bundle.

I'm sorry build for macOS is currently not a single-button-press process but we have very limited macOS developers on FreeCAD and the macOS builds aged a bit over the past 6-9 months because everyone has been busy with non-FreeCAD deliverables and we are in the midst of trying to bring them back on-line.

@looo has done some excellent cross-platform work with Conda and has builds for Linux, Windows and macOS under Python2 and 3 working.  That could be an alternative if you want to try that path to get up and running before we wring the kinks out with Homebrew.

mcmillanrs, 2018-12-29:
Hello Bruce:  appears to  be the correct  python
Rods-Mac-mini-3:~ rodmcmillan$  python -c "import site; print(site.getsitepackages())"
['/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages', '/System/Library/Frameworks/Python.framework/Versions/2.7/lib/site-python', '/Library/Python/2.7/site-packages']
I do not have python 3 installed.
To your script of last email I have added
export CMAKE_ARGS="${CMAKE_OPTS} -DPYTHON_EXECUTABLE=/usr/bin/python”
brew —config
and bre tap FreeCad/freecad
as per .travis.yml
then brew install —verbose —only-dependencies freecad….
i also git clone latest freecad source into freecad-code and cmake and make install from there.

We shall try that.  I will let you know.  Cheers Rpd

bblacey, 2018-12-29:
Ah, you need to use a brewed python 2.7, not the system python.  So, you need `-DPYTHON_EXECUTABLE=/usr/local/bin/python`

bblacey, 2018-12-29:
Also, is homebrew in your path?  For example, in my `~/.bash_profile`, I use `export PATH=/usr/local/bin:$PATH` to prepend Homebrew's bin directory to my `PATH` environment variable.

mcmillanrs, 2018-12-29:
Hello Bruce I have changed that and trying.  Cheers Rod

> On Dec 29, 2018, at 1:08 PM, Bruce Lacey <notifications@github.com> wrote:
> 
> export PATH=/usr/local/bin:$PATH

mcmillanrs, 2018-12-29:
Hello Bruce:  After making and double clicking freecad loads freecad but webgui is missing.  Do you know which packagecreates this?  Also this builds freecad 18 how would I create freecad 17?  Cheers Rod

peterlama, 2018-12-29:
@mcmillanrs, webgui is not built if cmake doesn't find qtwebkit. Search the cmake cache to see if it is using qtwebkit or not. Theoretically, you could build 0.17 by checking out the branch `releases/FreeCAD-0-17`, but don't be surprised if it doesn't work out of the box due to different versions of dependencies than what was available at the time of 0.17.

mcmillanrs, 2018-12-29:
Hello Peter:  Thank you for that.  I will stick with 0.18.  Cmake puts out this:
CMake Warning at /usr/local/Cellar/cmake/3.13.2/share/cmake/Modules/CMakeFindDependencyMacro.cmake:48 (find_package):
  Found package configuration file:

    /usr/local/Cellar/qtwebkit/5.212-72cfbd_3/lib/cmake/Qt5WebKit/Qt5WebKitConfig.cmake

  but it set Qt5WebKit_FOUND to FALSE so package "Qt5WebKit" is considered to
  be NOT FOUND.  Reason given by package:

  Qt5WebKit could not be found because dependency is required to have exact
  version 5.11.x.

Call Stack (most recent call first):
  /usr/local/Cellar/qtwebkit/5.212-72cfbd_3/lib/cmake/Qt5WebKitWidgets/Qt5WebKitWidgetsConfig.cmake:83 (find_dependency)
  CMakeLists.txt:876 (find_package)

CMake Warning at CMakeLists.txt:876 (find_package):
  Found package configuration file:

    /usr/local/Cellar/qtwebkit/5.212-72cfbd_3/lib/cmake/Qt5WebKitWidgets/Qt5WebKitWidgetsConfig.cmake

  but it set Qt5WebKitWidgets_FOUND to FALSE so package "Qt5WebKitWidgets" is
  considered to be NOT FOUND.  Reason given by package:

  Qt5WebKitWidgets could not be found because dependency Qt5WebKit could not
  be found.
and the CMakeCache,txt contains

//The directory containing a CMake configuration file for Qt5WebKit.
Qt5WebKit_DIR:PATH=/usr/local/Cellar/qtwebkit/5.212-72cfbd_3/lib/cmake/Qt5WebKit

Reading this I cannot figure out if it finds qtwebkit or not, I hope you can decypher it for me.  Cheers Rod

peterlama, 2018-12-29:
Ah yes, that means it found it but it is built with a different Qt version. @bblacey, is there a bottle for qtwebkit built with qt 5.12?

@mcmillanrs, if there is no qtwebkit bottle using qt 5.12, you will have to compile it yourself (note that it could take up to an hour or more to build).
```
brew uninstall qtwebkit
brew install qtwebkit --build-from-source
``` 
After that, rerunning cmake should find the correct qtwebkit.

mcmillanrs, 2018-12-29:
Hello Peter:  Thank you I will try that.  Cheers Rod

bblacey, 2018-12-29:
@peterlama - working on it ;)  Checkout https://travis-ci.org/FreeCAD/homebrew-freecad/builds/473425417.  You should have an email from me in your inbox about this that I sent to you yesterday.

mcmillanrs, 2018-12-29:
Hello Peter:  Tried your suggestion this is the terminal output. As it failed.  I am getting out of my knowledge base.  Cheers Rod
Rods-Mac-mini-3:~ rodmcmillan$ brew install qtwebkit --build-from-source
Updating Homebrew...
==> Auto-updated Homebrew!
Updated 1 tap (homebrew/core).
==> Updated Formulae
goenv                                          librsvg                                        signify-osx

==> Installing qtwebkit from freecad/freecad
==> Installing dependencies for freecad/freecad/qtwebkit: ninja
==> Installing freecad/freecad/qtwebkit dependency: ninja
==> Downloading https://homebrew.bintray.com/bottles/ninja-1.8.2.high_sierra.bottle.tar.gz
######################################################################## 100.0%
==> Pouring ninja-1.8.2.high_sierra.bottle.tar.gz
==> Caveats
Bash completion has been installed to:
  /usr/local/etc/bash_completion.d

zsh completions have been installed to:
  /usr/local/share/zsh/site-functions
==> Summary
🍺  /usr/local/Cellar/ninja/1.8.2: 7 files, 273.3KB
==> Installing freecad/freecad/qtwebkit
==> Cloning https://code.qt.io/qt/qtwebkit.git
Cloning into '/Users/rodmcmillan/Library/Caches/Homebrew/qtwebkit--git'...
Checking out files: 100% (13148/13148), done.
==> Checking out branch 5.212
Already on '5.212'
Your branch is up to date with 'origin/5.212'.
==> ./Tools/Scripts/build-webkit --qt --prefix=/usr/local/Cellar/qtwebkit/5.212-72cfbd_3 --install
Last 15 lines from /Users/rodmcmillan/Library/Logs/Homebrew/qtwebkit/01.build-webkit:
In file included from ../../Source/WebKit2/DatabaseProcess/DatabaseProcess.cpp:27:
In file included from ../../Source/WebKit2/DatabaseProcess/DatabaseProcess.h:31:
In file included from ../../Source/WebKit2/Shared/ChildProcess.h:29:
In file included from ../../Source/WebKit2/Platform/IPC/Connection.h:48:
../../Source/WTF/wtf/spi/darwin/XPCSPI.h:94:15: error: conflicting types for 'xpc_dictionary_apply'
EXTERN_C bool xpc_dictionary_apply(xpc_object_t xdict, xpc_dictionary_applier_t applier);
              ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/xpc/xpc.h:1997:1: note: previous declaration is here
xpc_dictionary_apply(xpc_object_t xdict,
^
../../Source/WebKit2/DatabaseProcess/DatabaseProcess.cpp:233:81: warning: lambda capture 'websiteDataTypes' is not used [-Wunused-lambda-capture]
        postDatabaseTask(std::make_unique<CrossThreadTask>([callbackAggregator, websiteDataTypes, this] {
                                                                                ^
1 warning and 2 errors generated.
ninja: build stopped: subcommand failed.

If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
https://github.com/freecad/homebrew-freecad/issues

These open issues may also help:
Installation of freecad/qtwebkit fails on Mojave. https://github.com/FreeCAD/homebrew-freecad/issues/95

> On Dec 29, 2018, at 4:12 PM, Peter Lama <notifications@github.com> wrote:
> 
> brew uninstall qtwebkit
> brew install qtwebkit --build-from-source

peterlama, 2018-12-29:
@mcmillanrs Sorry, I forgot about that. qtwebkit has to be patched before it will build on Mojave. At this point, I don't know what else to tell you other than just wait a few days, since we are in the middle of getting things working again.

mcmillanrs, 2018-12-29:
Hello Peter:Bruce:  Thank you.  Please let me know when I can try again.  Chers Rod

> On Dec 29, 2018, at 5:56 PM, Rod McMillan <rodmcmillan@me.com> wrote:
> 
> Hello Peter:  Tried your suggestion this is the terminal output. As it failed.  I am getting out of my knowledge base.  Cheers Rod
> Rods-Mac-mini-3:~ rodmcmillan$ brew install qtwebkit --build-from-source
> Updating Homebrew...
> ==> Auto-updated Homebrew!
> Updated 1 tap (homebrew/core).
> ==> Updated Formulae
> goenv                                          librsvg                                        signify-osx
> 
> ==> Installing qtwebkit from freecad/freecad
> ==> Installing dependencies for freecad/freecad/qtwebkit: ninja
> ==> Installing freecad/freecad/qtwebkit dependency: ninja
> ==> Downloading https://homebrew.bintray.com/bottles/ninja-1.8.2.high_sierra.bottle.tar.gz <https://homebrew.bintray.com/bottles/ninja-1.8.2.high_sierra.bottle.tar.gz>
> ######################################################################## 100.0%
> ==> Pouring ninja-1.8.2.high_sierra.bottle.tar.gz
> ==> Caveats
> Bash completion has been installed to:
>   /usr/local/etc/bash_completion.d
> 
> zsh completions have been installed to:
>   /usr/local/share/zsh/site-functions
> ==> Summary
> 🍺  /usr/local/Cellar/ninja/1.8.2: 7 files, 273.3KB
> ==> Installing freecad/freecad/qtwebkit
> ==> Cloning https://code.qt.io/qt/qtwebkit.git <https://code.qt.io/qt/qtwebkit.git>
> Cloning into '/Users/rodmcmillan/Library/Caches/Homebrew/qtwebkit--git'...
> Checking out files: 100% (13148/13148), done.
> ==> Checking out branch 5.212
> Already on '5.212'
> Your branch is up to date with 'origin/5.212'.
> ==> ./Tools/Scripts/build-webkit --qt --prefix=/usr/local/Cellar/qtwebkit/5.212-72cfbd_3 --install
> Last 15 lines from /Users/rodmcmillan/Library/Logs/Homebrew/qtwebkit/01.build-webkit:
> In file included from ../../Source/WebKit2/DatabaseProcess/DatabaseProcess.cpp:27:
> In file included from ../../Source/WebKit2/DatabaseProcess/DatabaseProcess.h:31:
> In file included from ../../Source/WebKit2/Shared/ChildProcess.h:29:
> In file included from ../../Source/WebKit2/Platform/IPC/Connection.h:48:
> ../../Source/WTF/wtf/spi/darwin/XPCSPI.h:94:15: error: conflicting types for 'xpc_dictionary_apply'
> EXTERN_C bool xpc_dictionary_apply(xpc_object_t xdict, xpc_dictionary_applier_t applier);
>               ^
> /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/xpc/xpc.h:1997:1: note: previous declaration is here
> xpc_dictionary_apply(xpc_object_t xdict,
> ^
> ../../Source/WebKit2/DatabaseProcess/DatabaseProcess.cpp:233:81: warning: lambda capture 'websiteDataTypes' is not used [-Wunused-lambda-capture]
>         postDatabaseTask(std::make_unique<CrossThreadTask>([callbackAggregator, websiteDataTypes, this] {
>                                                                                 ^
> 1 warning and 2 errors generated.
> ninja: build stopped: subcommand failed.
> 
> If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
> https://github.com/freecad/homebrew-freecad/issues <https://github.com/freecad/homebrew-freecad/issues>
> 
> These open issues may also help:
> Installation of freecad/qtwebkit fails on Mojave. https://github.com/FreeCAD/homebrew-freecad/issues/95 <https://github.com/FreeCAD/homebrew-freecad/issues/95>
> 
> 
>> On Dec 29, 2018, at 4:12 PM, Peter Lama <notifications@github.com <mailto:notifications@github.com>> wrote:
>> 
>> brew uninstall qtwebkit
>> brew install qtwebkit --build-from-source
>

mcmillanrs, 2018-12-30:
Hello Bruse:  I tried one more thing by running the below command but with the qtwebkit.rb  formula.

It has errors of course but I believe they are related to xcodebuild.  The output is 
==> ./Tools/Scripts/build-webkit --qt --prefix=/usr/local/Cellar/qtwebkit/5.212-72cfbd --install
xcode-select: error: tool 'xcodebuild' requires Xcode, but active developer directory '/Library/Developer/CommandLineTools' is a command line tools instance
*************************************************************
Xcode Version 4.6 or later is required to build WebKit.
You have an earlier version of Xcode, thus the build will
most likely fail. The latest Xcode is available from the App Store.

Xcode and command line tools are installed of course, xcode in applications and the command line tools in Library/Developer.  xcode-select says the path is application/xcode.app/Contnets.  I am reinstalling xcode and commandlinetools and we will see.  Cheers Rod

 

> On Dec 27, 2018, at 3:28 PM, Bruce Lacey <notifications@github.com> wrote:
> 
> brew install --build-from-source --verbose https://raw.githubusercontent.com/FreeCAD/homebrew-freecad/bottling/Formula/opencamlib.rb <https://raw.githubusercontent.com/FreeCAD/homebrew-freecad/bottling/Formula/opencamlib.rb>

mcmillanrs, 2018-12-30:
Hello Peter: isit patched for high sierra 10.13? That is what i am developing on. Cheers Rod

iPad

peterlama, 2018-12-30:
Yes, it should work with 10.13. You must be using the latest xcode though, because it is using the 10.14 SDK:
```
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/xpc/xpc.h:1997:1: note: previous declaration is here
```

mcmillanrs, 2018-12-30:
Hello Peter, Bruce, Ian:

Tried to build qtwebkit. today.  Attached is a text file that shows the Formula, the tools used and the errors.  I hope it helps you.

The logs mentioned do not appear to be created.
I perhaps have not told you why I am doing all this.  I just want to try Freecad 3dsurface operation on a model I have to see if it generates the proper g-code for my milling machine.  

Cheers Rod

bblacey, 2018-12-30:
@mcmillanrs, I think you forgot the attachment so please attach the logs when you have a chance.  That said, if configured properly, you should be able to build `qtwebkit` because we currently building it on both macOS 10.12 and macOS 10.13 on [Travis](https://travis-ci.org/FreeCAD/homebrew-freecad/builds/473602650).

mcmillanrs, 2018-12-30:
Hello Peter, Ian and Bruce:  I think  you are the people to let know about issues with the latest mac development dng I installed from Freecad today.  On startup a “global name encoding not defined error code is thrown.  When the path workbench is opened the path menus are not loaded ie job and all operations.  The report panel says path workbench is activated but it is not.  Reseting the preferences causes the tool table in my 0.17  to be lost.  Cheers Rod

mcmillanrs, 2018-12-31:
Hello Bruce.  the logs the output claims are never where they say and i can never find them , i think the tmp files get deleted when the make fails. I have excerts from the terminal output.  Yes attachments are a problem sometimes with apple mail. I will send them tomorrow but essentially i get the same errors as have been indicated in the github post  below about installation fails for qtwebkit on mojave that is xpc dictionary problems. This is also shown in my terminal output below.
Last 15 lines from /Users/rodmcmillan/Library/Logs/Homebrew/qtwebkit/01.build-webkit:
In file included from ../../Source/WebKit2/DatabaseProcess/DatabaseProcess.cpp:27:
In file included from ../../Source/WebKit2/DatabaseProcess/DatabaseProcess.h:31:
In file included from ../../Source/WebKit2/Shared/ChildProcess.h:29:
In file included from ../../Source/WebKit2/Platform/IPC/Connection.h:48:
../../Source/WTF/wtf/spi/darwin/XPCSPI.h:94:15: error: conflicting types for 'xpc_dictionary_apply'
EXTERN_C bool xpc_dictionary_apply(xpc_object_t xdict, xpc_dictionary_applier_t applier);
              ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/xpc/xpc.h:1997:1: note: previous declaration is here
xpc_dictionary_apply(xpc_object_t xdict,
^
../../Source/WebKit2/DatabaseProcess/DatabaseProcess.cpp:233:81: warning: lambda capture 'websiteDataTypes' is not used [-Wunused-lambda-capture]
        postDatabaseTask(std::make_unique<CrossThreadTask>([callbackAggregator, websiteDataTypes, this] {
                                                                                ^
1 warning and 2 errors generated.
ninja: build stopped: subcommand failed.

If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
https://github.com/freecad/homebrew-freecad/issues

These open issues may also help:
Installation of freecad/qtwebkit fails on Mojave. https://github.com/FreeCAD/homebrew-freecad/issues/95

Thank you cherrs Rod
iPad

bblacey, 2018-12-31:
Rod, ah yes, I don't believe GitHub accepts attachments to issue threads from email.  Please run the following commands and provide the output here:

```
sw_vers
brew doctor
xcode-select --print-path
xcodebuild -version -sdk
```

mcmillanrs, 2018-12-31:
Hello Bruce :  Here is output requested plus that of brew —config.
sw_vers
ProductName:	Mac OS X
ProductVersion:	10.13.6
BuildVersion:	17G4015
xcode-select -p
/Library/Developer/CommandLineTools
xcodebuild -version -sdk
xcode-select: error: tool 'xcodebuild' requires Xcode, but active developer directory '/Library/Developer/CommandLineTools' is a command line tools instance
brew --config
HOMEBREW_VERSION: 1.8.6-54-g0034b0f
ORIGIN: https://github.com/Homebrew/brew
HEAD: 0034b0f874822a99eb189cc31ff6eef1c8cf4bbc
Last commit: 2 days ago
Core tap ORIGIN: https://github.com/Homebrew/homebrew-core
Core tap HEAD: 9fe967ad7a9d194b08c28a5b6816110fc23f2663
Core tap last commit: 23 hours ago
HOMEBREW_PREFIX: /usr/local
HOMEBREW_LOGS: /Users/rodmcmillan/Library/Logs/Homebrew
CPU: quad-core 64-bit ivybridge
Homebrew Ruby: 2.3.7 => /System/Library/Frameworks/Ruby.framework/Versions/2.3/usr/bin/ruby
Clang: 10.0 build 1000
Git: 2.17.2 => /Library/Developer/CommandLineTools/usr/bin/git
Curl: 7.54.0 => /usr/bin/curl
macOS: 10.13.6-x86_64
CLT: 10.1.0.0.1.1539992718
Xcode: 10.1
XQuartz: 2.7.11 => /opt/X11

The xcodebuild is what other people appear to get but it appears to indicate a problem I do not know how to fix.  I reinstalled Xcode before this and had homebrew install commandlinetools.  
Cheers Rod

bblacey, 2018-12-31:
You can try using Xcode by doing a `sudo xcode-select -s /Applications/Xcode.app/Contents/Developer` assuming you installed Xcode in /Applications.  Other than that, everything looks good. 

Also, what does `brew doctor` show?

bblacey, 2018-12-31:
Reading this from my phone, it looks like you only have iOS SDK’s installed - that won’t work for macOS app builds.  Launch Xcode, go to preferences and select download macOS SDK.

mcmillanrs, 2018-12-31:
Hello Bruce:  The mac osx appears to be installed. The complete output of xcodebuild is below:
Rods-Mac-mini-3:~ rodmcmillan$ xcodebuild -version -sdk
iPhoneOS12.1.sdk - iOS 12.1 (iphoneos12.1)
SDKVersion: 12.1
Path: /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS12.1.sdk
PlatformVersion: 12.1
PlatformPath: /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform
BuildID: 6F0BDB7C-D34C-11E8-B4D5-64D77DC3894B
ProductBuildVersion: 16B91
ProductCopyright: 1983-2018 Apple Inc.
ProductName: iPhone OS
ProductVersion: 12.1

iPhoneSimulator12.1.sdk - Simulator - iOS 12.1 (iphonesimulator12.1)
SDKVersion: 12.1
Path: /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneSimulator.platform/Developer/SDKs/iPhoneSimulator12.1.sdk
PlatformVersion: 12.1
PlatformPath: /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneSimulator.platform
BuildID: 6F0BDB7C-D34C-11E8-B4D5-64D77DC3894B
ProductBuildVersion: 16B91
ProductCopyright: 1983-2018 Apple Inc.
ProductName: iPhone OS
ProductVersion: 12.1

MacOSX10.14.sdk - macOS 10.14 (macosx10.14)
SDKVersion: 10.14
Path: /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk
PlatformVersion: 1.1
PlatformPath: /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform
ProductBuildVersion: 18B71
ProductCopyright: 1983-2018 Apple Inc.
ProductName: Mac OS X
ProductUserVisibleVersion: 10.14.1
ProductVersion: 10.14.1

AppleTVOS12.1.sdk - tvOS 12.1 (appletvos12.1)
SDKVersion: 12.1
Path: /Applications/Xcode.app/Contents/Developer/Platforms/AppleTVOS.platform/Developer/SDKs/AppleTVOS12.1.sdk
PlatformVersion: 12.1
PlatformPath: /Applications/Xcode.app/Contents/Developer/Platforms/AppleTVOS.platform
BuildID: 82CCE69C-D0F4-11E8-93D5-A908D0A55C45
ProductBuildVersion: 16J602
ProductCopyright: 1983-2018 Apple Inc.
ProductName: Apple TVOS
ProductVersion: 12.1

AppleTVSimulator12.1.sdk - Simulator - tvOS 12.1 (appletvsimulator12.1)
SDKVersion: 12.1
Path: /Applications/Xcode.app/Contents/Developer/Platforms/AppleTVSimulator.platform/Developer/SDKs/AppleTVSimulator12.1.sdk
PlatformVersion: 12.1
PlatformPath: /Applications/Xcode.app/Contents/Developer/Platforms/AppleTVSimulator.platform
BuildID: 82CCE69C-D0F4-11E8-93D5-A908D0A55C45
ProductBuildVersion: 16J602
ProductCopyright: 1983-2018 Apple Inc.
ProductName: Apple TVOS
ProductVersion: 12.1

WatchOS5.1.sdk - watchOS 5.1 (watchos5.1)
SDKVersion: 5.1
Path: /Applications/Xcode.app/Contents/Developer/Platforms/WatchOS.platform/Developer/SDKs/WatchOS5.1.sdk
PlatformVersion: 5.1
PlatformPath: /Applications/Xcode.app/Contents/Developer/Platforms/WatchOS.platform
BuildID: 0EA5384C-D355-11E8-9BF8-A40624452C56
ProductBuildVersion: 16R591
ProductCopyright: 1983-2018 Apple Inc.
ProductName: Watch OS
ProductVersion: 5.1

WatchSimulator5.1.sdk - Simulator - watchOS 5.1 (watchsimulator5.1)
SDKVersion: 5.1
Path: /Applications/Xcode.app/Contents/Developer/Platforms/WatchSimulator.platform/Developer/SDKs/WatchSimulator5.1.sdk
PlatformVersion: 5.1
PlatformPath: /Applications/Xcode.app/Contents/Developer/Platforms/WatchSimulator.platform
BuildID: 0EA5384C-D355-11E8-9BF8-A40624452C56
ProductBuildVersion: 16R591
ProductCopyright: 1983-2018 Apple Inc.
ProductName: Watch OS
ProductVersion: 5.1

Xcode 10.1
Build version 10B61
Rods-Mac-mini-3:~ rodmcmillan$ 

Running qtwebkit bottle just know fails errors are as per documented by others on forum.  Last part of my output from terminal here:

In file included from ../../Source/WebKit2/UIProcess/API/qt/qquickwebview_p_p.h:24:
In file included from ../../Source/WebKit2/UIProcess/DefaultUndoController.h:26:
In file included from ../../Source/WebKit2/UIProcess/WebPageProxy.h:33:
In file included from ../../Source/WebKit2/Platform/IPC/Connection.h:48:
../../Source/WTF/wtf/spi/darwin/XPCSPI.h:93:15: error: conflicting types for 'xpc_array_apply'
EXTERN_C bool xpc_array_apply(xpc_object_t, xpc_array_applier_t);
              ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/xpc/xpc.h:1257:1: note: previous declaration is here
xpc_array_apply(xpc_object_t xarray, XPC_NOESCAPE xpc_array_applier_t applier);
^
In file included from ../../Source/WebKit2/UIProcess/qt/WebColorPickerQt.cpp:26:
In file included from ../../Source/WebKit2/UIProcess/API/qt/qquickwebview_p_p.h:24:
In file included from ../../Source/WebKit2/UIProcess/DefaultUndoController.h:26:
In file included from ../../Source/WebKit2/UIProcess/WebPageProxy.h:33:
In file included from ../../Source/WebKit2/Platform/IPC/Connection.h:48:
../../Source/WTF/wtf/spi/darwin/XPCSPI.h:94:15: error: conflicting types for 'xpc_dictionary_apply'
EXTERN_C bool xpc_dictionary_apply(xpc_object_t xdict, xpc_dictionary_applier_t applier);
              ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.14.sdk/usr/include/xpc/xpc.h:1997:1: note: previous declaration is here
xpc_dictionary_apply(xpc_object_t xdict,
^
2 errors generated.
ninja: build stopped: subcommand failed.

bblacey, 2018-12-31:
Hi Rod, are you certain that you are using the qrwebkit formula from the bottling branch?  That is the only one compatible with qt 5.212 that is the default in Homebrew.

peterlama, 2018-12-31:
@mcmillanrs, see if you can install the 10.13 sdk. Perhaps by installing an older xcode.

bblacey, 2018-12-31:
Rob, I just pushed a **_temporary_** `qtwebkit` bottle up to my bintray repo for you.  You can install it with:

`brew install https://dl.bintray.com/bblacey/travis-staging/qtwebkit-5.212-72cfbd.high_sierra.bottle.tar.gz`

However, in order to build FreeCAD, you will need the 10.13 SDK as @peterlama suggested. You can download Xcode 9.4 from developer.apple.com - you can sign up for a free developer account to download earlier versions of Xcode.  FYI, Travis is configured to build with Xcode 9.2 and 9.4 on macOS 10.12 and 10.13 respectively.

mcmillanrs, 2018-12-31:
Hello Bruce:  Happy New Year.  I loaded xcode 9.4.1 with macosx sdk 10.13.  xcode-select points to this   Your temporary qtwebkit bottle worked extremely fast and qtwebkit was built in the Cellar.  Then built freecad but Path does not work correctly.  Path menus will no lload after creating a box in Part.   this is the error from the report panel:

dlopen(/usr/local/FreeCAD.app/Contents/lib/MeshPart.so, 2): Symbol not found: __ZN9BRep_Tool14CurveOnSurfaceERK11TopoDS_EdgeRK11TopoDS_FaceRdS6_
  Referenced from: /usr/local/FreeCAD.app/Contents/lib/libocc.0.dylib
  Expected in: flat namespace
 in /usr/local/FreeCAD.app/Contents/lib/libocc.0.dylib
Traceback (most recent call last):
  File "<string>", line 75, in Initialize
  File "/usr/local/FreeCAD.app/Contents/Mod/Path/PathScripts/PathGuiInit.py", line 69, in Startup
    from PathScripts import PathSurfaceGui
  File "/usr/local/FreeCAD.app/Contents/Mod/Path/PathScripts/PathSurfaceGui.py", line 27, in <module>
    import PathScripts.PathSurface as PathSurface
  File "/usr/local/FreeCAD.app/Contents/Mod/Path/PathScripts/PathSurface.py", line 28, in <module>
    import MeshPart

Freecad built but mapplotlib and spnav were not found during the cmake process but mapplotlib is in the Cellar.  I hope you can help.  Cheers Rod

peterlama, 2018-12-31:
I'm pretty sure that is because netgen needs to be recompiled with the new opencascade version. For now you can try building without netgen support since the Path module does not use it. Try: `-DBUILD_FEM_NETGEN=OFF`

mcmillanrs, 2018-12-31:
Thank you Peter I will try that.  Cheers Rod

mcmillanrs, 2019-01-01:
Hello Peter, Bruce, Ian:  Thank you for all your help.  There is success.  NETGEN=OFF allows it all to work.  I now have Freecad 0.18 whose Path workbench has a 3dsurface operation that can create gcode for a sphere fussed into a box.  I will try on a more complex 3d surface now.  Cheers Rod

closed 2020-11-09 by vejmarie