FreeCAD/homebrew-freecad #101: Checksum for Cask 'freecad' does not match

jppiiroinen, 2019-12-09:
```
$ brew cask install freecad
==> Downloading https://github.com/FreeCAD/FreeCAD/releases/download/0.18.4/FreeCAD_0.18-16146-OSX-x86_64-c
Already downloaded: /Users/juhapekka/Library/Caches/Homebrew/downloads/ee94481e73396180ff93e07dc89c6335c4b53f1bc85fab9e71784a9e9047d15f--FreeCAD_0.18-16146-OSX-x86_64-conda-Qt5-Py3.dmg
==> Verifying SHA-256 checksum for Cask 'freecad'.
==> Note: Running `brew update` may fix SHA-256 checksum errors.
Error: Checksum for Cask 'freecad' does not match.
Expected: d0a7188f87bea7551d501c6a8459106f5d0b0f0e9fc34a342dcdac8c677349b2
  Actual: 5b27d3d0c1499768ddb8929251fb36965253e8f9bbc66ed0d04ec918e15c3e92
    File: /Users/juhapekka/Library/Caches/Homebrew/downloads/ee94481e73396180ff93e07dc89c6335c4b53f1bc85fab9e71784a9e9047d15f--FreeCAD_0.18-16146-OSX-x86_64-conda-Qt5-Py3.dmg
To retry an incomplete download, remove the file above.
If the issue persists, visit:
  https://github.com/Homebrew/homebrew-cask/blob/master/doc/reporting_bugs/checksum_does_not_match_error.md
```

Harald500, 2019-12-31:
Hello on the 31 of Dec 2019 i do
harald@iMac ~ % brew cask install freecad         
==> Downloading https://github.com/FreeCAD/FreeCAD/releases/download/0.18.4/FreeCAD_0.18-16146-OSX-x86_64-conda-Qt5-Py3.dmg
Already downloaded: /Users/harald/Library/Caches/Homebrew/downloads/ee94481e73396180ff93e07dc89c6335c4b53f1bc85fab9e71784a9e9047d15f--FreeCAD_0.18-16146-OSX-x86_64-conda-Qt5-Py3.dmg
==> Verifying SHA-256 checksum for Cask 'freecad'.
==> Note: Running `brew update` may fix SHA-256 checksum errors.
Error: Checksum for Cask 'freecad' does not match.
Expected: 9ce6dacef941d218742f6a90171af58360f82d879419018ee134813f6555d413
  Actual: 0a064d14faacaa9ac419b79a31b03edd0a323049613d4483f88bee03aa5d6a4c
I check the SHA-256 on the free cad webpage and it match with the Actual.
So the formulae need to updated.

Thanks in advance

peterlama, 2019-12-31:
Thanks for the report. However, homebrew cask is unrelated to this repository. You may be able to contact the maintainer of the cask here: https://github.com/Homebrew/homebrew-cask/pull/74825

closed 2019-12-31 by peterlama