FreeCAD/homebrew-freecad #102: Brew cask install checksum check failure

gdeshazer, 2020-01-14:
While trying to install through `brew cask install freecad` I receive a checksum failure.

**Output from homebrew**

```
==> Downloading https://github.com/FreeCAD/FreeCAD/releases/download/0.18.4/FreeCAD_0.18-16146-OSX-x
==> Downloading from https://github-production-release-asset-2e65be.s3.amazonaws.com/5736080/928df50
######################################################################## 100.0%
==> Verifying SHA-256 checksum for Cask 'freecad'.
==> Note: Running `brew update` may fix SHA-256 checksum errors.
Error: Checksum for Cask 'freecad' does not match.
Expected: 6de32db5acd921d9e118ebc7a0481046b8e7bd76ec55296df72f2f5512bc8a37
  Actual: fdb45ad5bd4338110a48b7d36e9a91d46004dc62976a9a8a2a4c9e407ccda1a9
```

Downloading the DMG file manually from the download link on the [freecad wiki](https://www.freecadweb.org/wiki/Install_on_Mac) and verifying the shasum works without issue.

judismith, 2020-01-15:
I am also having this issue with the home-brew installation.

Temporalis, 2020-01-22:
Error: Checksum for Cask 'freecad' does not match.
Expected: 8f9a13d3c9a3728cd8d4065995ddf002e010cbce157e506520506bc8773a2fb7
  Actual: 190dacd5716e6d50771135dec4824bac291503b9ec7628c260cc1cb62a05b49a
  File: /Users/silva/Library/Caches/Homebrew/downloads/ee94481e73396180ff93e07dc89c6335c4b53f1bc85fab9e71784a9e9047d15f--FreeCAD_0.18-16146-OSX-x86_64-conda-Qt5-Py3.dmg

The Download Version from the Web-Page has the same, i.e. Actual Checksum.

brainstorm, 2020-02-13:
```
==> Downloading https://github.com/FreeCAD/FreeCAD/releases/download/0.18.4/FreeCAD_0.18-16146-OSX-x86_64-conda-Qt5-Py3.dmg
==> Downloading from https://github-production-release-asset-2e65be.s3.amazonaws.com/5736080/04af6d80-4e74-11ea-8ab9-2888ca9f09
######################################################################## 100.0%
==> Verifying SHA-256 checksum for Cask 'freecad'.
==> Note: Running `brew update` may fix SHA-256 checksum errors.
Error: Checksum for Cask 'freecad' does not match.
Expected: a500b86b29b3e085835aee23f123c2d519d672730c93e7d53d8dbcb687eb0cd5
  Actual: ccea35b73912f138dd0190aab58bdf5a818076020540cc23295dbb2d38869a4e
    File: /Users/rvalls/Library/Caches/Homebrew/downloads/ee94481e73396180ff93e07dc89c6335c4b53f1bc85fab9e71784a9e9047d15f--FreeCAD_0.18-16146-OSX-x86_64-conda-Qt5-Py3.dmg
To retry an incomplete download, remove the file above.
If the issue persists, visit:
  https://github.com/Homebrew/homebrew-cask/blob/master/doc/reporting_bugs/checksum_does_not_match_error.md
Failed to install freecad. Continue? [y/N] [y/N]: 
```

ipatch, 2020-11-06:
```shell
brew cask info freecad-pre
echo "for latest cask info"

brew cask install freecad-pre
```

please close issue.

closed 2020-11-09 by vejmarie