FreeCAD/homebrew-freecad #103: Need assistance with building FreeCAD on Mojave 10.14.6

Gitmax20, 2020-01-21:
Following the instructions  from [https://www.freecadweb.org/wiki/Compile_on_MacOS](https://www.freecadweb.org/wiki/Compile_on_MacOS)

Getting the following error: 

My input: **git clone https://github.com/FreeCAD/FreeCAD FreeCAD-git**
Cloning into 'FreeCAD-git'...
remote: Enumerating objects: 24, done.
remote: Counting objects: 100% (24/24), done.
remote: Compressing objects: 100% (24/24), done.
error: RPC failed; curl 56 LibreSSL SSL_read: SSL_ERROR_SYSCALL, errno 54
fatal: the remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed

This instruction has failed, several times: What to do? 

I have another machine running High Sierra, but I can no longer download Xcode for High Sierra from the App Store, to compile FreeCAD. 

Is it possible to obtain a legacy version of Xcode for High Sierra, from another source?

ianrrees, 2020-01-21:
Hi @Gitmax20, thanks for the bug report.  That looks like a git problem rather than a FreeCAD build problem; for some reason git is failing to download the FreeCAD source code, something to do with [curl](https://curl.haxx.se/) and/or LibreSSL.  Searching "git", "clone", and that line starting with "error:" will yield some next steps.

As for your questions about High Sierra - my experience with MacOS is that development on/for old versions is hard, so I wouldn't suggest going in that direction.

closed 2020-01-21 by ianrrees

Gitmax20, 2020-01-21:
Thank you very much for the prompt reply!