FreeCAD/homebrew-freecad #104: Python@2 dependencies

DanielPa, 2020-03-28:
Since homebrew removed  python@2  from core (https://docs.brew.sh/Homebrew-and-Python), I get an error while installing the dependencies

`brew tap freecad/freecad`
`brew install --only-dependencies freecad --with-packaging-utils`
>>>
```
==> Installing freecad from freecad/freecad
Error: No available formula with the name "python@2" (dependency of freecad/freecad/freecad)
==> Searching for a previously deleted formula (in the last month)...
Warning: homebrew/core is shallow clone. To get complete history run:
  git -C "$(brew --repo homebrew/core)" fetch --unshallow

Error: No previously deleted formula found.
==> Searching for similarly named formulae...
Error: No similarly named formulae found.
==> Searching taps...
==> Searching taps on GitHub...
Error: No formulae found in taps.
```

FredericLeuba, 2020-04-01:
Hello Daniel,
did you find a way to fix the issue?
Best regards
Fred

DanielPa, 2020-04-02:
Hi Fred, unfortunately not. 
But I tried to `brew install` the dependencies from the freecad formula one by one and found 3 more formulae depending on python 2:

- freecad/freecad/matplotlib
- FreeCAD/freecad/pyside2-tools
- freecad/freecad/med-file

dblock247, 2020-08-15:
+1 Please fix.

vejmarie, 2020-09-21:
I just pushed a big PR which is building FreeCAD through python3 only with homebrew. Warning the installation time will be long as everything is built using source code currently. Their is no bottle available at the present time. I am checking how to do it.

peterlama, 2020-09-21:
Merged PR

closed 2020-09-21 by peterlama

sgrogan, 2020-09-21:
Hey @vejmarie if/when you get the bottles working can you ping me on the forum. Right now we have disabled the Travis builds for OSX because of this issue. Thanks for looking into this!

vejmarie, 2020-09-21:
hello @sgrogan I will let you know. It might work only on catalina for your reference. I do not really know how to specify the O/S version for MacOS into travis. I will try to build the bottle tonight. Do you have any permission rights on where they were initially stored ? If not that shouldn't be an issue I can find some place on the net.

sgrogan, 2020-09-21:
@vejmarie 
I think its here https://github.com/FreeCAD/FreeCAD-ports-cache
I have merge privileges as does @peterlama , I think.
There is a lot of traffic on the forum recently about this.
I don't know about Travis MacOS version, but I'll look.

peterlama, 2020-09-21:
correct, that is the repo with the bottle automation. @bblacey is the one who set everything up. I have access to the bintray account, but again not at a level where I can add new users. Maybe we can move the discussion to https://gitter.im/FreeCAD/macOS since it kind of off topic for this issue

sgrogan, 2020-09-21:
Maybe here is a good place https://forum.freecadweb.org/viewtopic.php?f=4&t=49769
@jmdzampieron  on the forum seems knowledgeable. 
Good to "see" you @peterlama

ianrrees, 2020-09-21:
Hi guys!  It's good to see this is still ticking along, thanks!

vejmarie, 2020-09-22:
Hi @ianrrees good to see you back. @sgrogan I just updated the various formula to use a prebuild for most of the bottles (except freeCAD). I opened up my own bintray account, but if somebody has credential on FreeCAD bintray account feel free to copy and past the bottles into it and push for an additionnal PR. Can you let me know if it could work for Travis build ? I believe it shall be working if you can specify Catalina as a build environment. I am curious to know if that works.