FreeCAD/homebrew-freecad #105: medfile installation fails

LuxiWang99, 2020-04-15:
I am trying to build FreeCAD on Mac OS high sierra. 
When I ran `brew install freecad/freecad/med-file`
I get:
`Installing med-file from freecad/freecad
==> Downloading http://files.salome-platform.org/Salome/other/med-4.0.0.tar.gz
Already downloaded: /Users/Library/Caches/Homebrew/downloads/ad1112ebc5e00ed868b887c890acbcd83d4bdc4ba930be839ff5d69063d0b062--med-4.0.0.tar.gz
==> cmake . -DMEDFILE_BUILD_PYTHON=ON -DMEDFILE_BUILD_TESTS=OFF -DMEDFILE_INSTALL_DOC=OFF -DPYTHON_INCLUDE_DIR=/usr/local/Cellar/python@2/2
Last 15 lines from /Users/luxiwang/Library/Logs/Homebrew/med-file/01.cmake:
-- Detecting Fortran/C Interface - Found GLOBAL and MODULE mangling
-- Check for HDF5 ...
-- HDF5: Using hdf5 compiler wrapper to determine C configuration
-- Found HDF5: /usr/local/lib/libhdf5.dylib;/usr/local/lib/libsz.dylib;/usr/lib/libz.dylib;/usr/lib/libdl.dylib;/usr/lib/libm.dylib (found version "1.12.0")  
-- Found HDF5 version 1.12.0 -- parallel flag is set to 
-- _versionComponents=1;12;0
-- HDF5_VERSION=1.12.0
CMake Error at config/cmake_files/medMacros.cmake:402 (MESSAGE):
  HDF5 version is 1.12.0.  Only versions >= 1.10.2 are supported.
Call Stack (most recent call first):
  CMakeLists.txt:101 (MED_FIND_HDF5)

-- Configuring incomplete, errors occurred!
See also "/tmp/med-file-20200415-22579-1n2scvh/med-4.0.0/CMakeFiles/CMakeOutput.log".
`

I cannot seem to build FreeCAD without having med-file installed.

luzpaz, 2020-04-21:
sorry about the lag in reponse. The freecad tap isn't as active as it used to be since we moved to [conda](https://github.com/FreeCAD/FreeCAD_Conda). You may want to discuss this on the FreeCAD forum. Please link to the thread that you create in this issue so we can track it.

vejmarie, 2020-09-21:
I just pushed a PR which shall be fixing this issue. Otherwise if you want to fix temporarly you just need to replace the 
depends_on "hdf5" by depends_on "hdf5@1.10" into med-file.rb formula file.

peterlama, 2020-09-21:
Merged PR

closed 2020-09-21 by peterlama