FreeCAD/homebrew-freecad #106: “FreeCAD.app” cannot be opened because the developer cannot be verified.

vishdev007, 2020-06-06:
How can I solve this ? In need asap. Been trying so many things but nothing working. Using MAC OS Catalina 10.15.5

vishdev007, 2020-06-06:
[image]

closed 2020-06-06 by vishdev007

nelsonic, 2023-01-05:
Just to expand on @vishdev007's instructions for anyone finding this in 2023 running `FreeCAD 0.20.1+` 

If you see: 

## "FreeCAD.app" cannot be opened because the developer cannot be verified
### macOS cannot verify that this app is free from malware.

[image]

Click **`Cancel`** to dismiss this modal and the follow @vishdev007's instructions.

1. Open the App from finder.
2. **_Right_-click** (<kbd>control</kbd> + `click`) on **`FreeCad.app`**
3. Single-Click on **`Show Package Contents`** from the menu

[image]

4. Expand the `Contents` directory
5. Expand `MacOS` directory
6. And finally, _double_-click on the `FreeCAD` (Unix Executable File) and let it run on terminal.

You should see something similar to the following:
[image]

The following _different_ dialog is displayed:

[image]

Clicked on **`Open`**  and this is what I saw:

[image]

_Initially_ appeared to not work ... 🤷‍♂️  (had to quit the terminal which appeared to be hanging ...)

But the _next_ time I tried to open `FreeCAD.app` from `Applications` I got the following _different_ Modal:

[image]

And when I clicked **`Open`**, it started to open:

[image]

And eventually: ⏳ 

[image]

🚀

bblacey, 2023-01-05:
On macOS, you can control-click an App that is not signed/notarized and select Open to intentionally open it.

[image]

nelsonic, 2023-01-05:
@bblacey for sure it would be awesome for FreeCAD to have an official build in the MacOS App Store too. it would really accelerate adoption especially in Education where installing random Apps is usually a No-go…. 🙅‍♀️

The core FreeCAD devs likely don’t even *use* Mac so whoever is publishing the App would need a Mac too. The cost of the annual Apple Dev membership/license is only (a relatively tiny) part of the requirements. 💭
The dev time is by far the scarcest resource…👨‍💻⏳

rutaslav, 2023-01-11:
Unfortunately for me running FreeCad in the MAcOS directory gives the same error message. Any ideas how to work around that? Running MacOS Ventura 13.0.1 
[image]

bblacey, 2023-01-11:
FYI, I took a quick look, and if there is interest, it seems that FreeCAD may be eligible for a free Developer license because it is a nonprofit organization that distributes free applications.  

[Fee Waiver](https://developer.apple.com/support/membership-fee-waiver/)
[Developer License](https://developer.apple.com/support/compare-memberships/)

I believe that only @yorikvanhavre would be able to apply for the fee waiver.

mcknightd, 2023-01-23:
> FYI, I took a quick look, and if there is interest, it seems that FreeCAD may be eligible for a free Developer license because it is a nonprofit organization that distributes free applications. [...]

I think this is the best solution. So many of us mac users out there need a good free/open cad solution.

adjwilley, 2023-03-08:
None of the above worked for me. No matter what I did the only options were "cancel" and "move to trash". Eventually after trying to open it several times I got a helpful popup from the "macOS User Guide" telling me to go to Apple Menu > System Settings > Privacy & Security (scroll down) "Allow applications downloaded from..." Right below that there was a message saying, "FreeCad was blocked from use because it is not from an identified developer." There was an "Open Anyway" button there. I clicked that and after a password challenge and a final "are you sure?" from Mac, FreeCAD opened.

jens1101, 2023-04-24:
On MacOS Ventura I followed the above advice, but pressing "Open Anyway" resulted in the same annoying "Cancel" and "Move to trash" modal 🙄. Turns out you can forcefully remove the quarantine flag from the app via the terminal:
```shell
xattr -d com.apple.quarantine /Applications/FreeCAD.app
```

Run the above command once and then you can open FreeCAD like any other app.

yorikvanhavre, 2023-04-26:
Update: We are in the process of obtaining an Apple developer ID for the FPA, with which we'll hopefully be able to sign the Mac packages. Hopefully this issue will get a happy ending soon!

LondonWebFactory, 2023-04-29:
> 

Works like a charm! Many thanks.

adaltonreis, 2023-06-25:
Hey everyone. I just need to draw some stuff for my own utilization ( making some changes at home).

Unfortunately, none of the above worked for me. 

I did try to go through Finder and open the Unix file directly, but it continued to give me the same annoying screen. And know I cannot open it anymore. When i click the downloaded file noting happens.

I appreciate any suggestions.

adaltonreis, 2023-06-25:
Update: What actually worked for me was to open the 'Disk Utility' and mounting it from there. I mean, the disk image for FreeCad was listed there, but was not 'mounted'. When I clicked 'mount' the folder containing the executable appeared in the screen, then i clicked in the application inside the mentioned folder and it opened the program. I hope this trick helps anyone...

rathinamt, 2023-06-26:
None of these seem to work with the latest Ventura 13.4.1 but got this in the "settings & privacy"... 
[image]

Clicked on the <open>  and got this dialogue...
[image]
Again clicked <open> and finally got it!
[image]

Hope this helps to someone.