FreeCAD/homebrew-freecad #114: Building FreeCAD on macOS Catalina fails during configuration.

MartinMajewski, 2020-10-01:
Hi there,

So I followed the (updated?!) instructions for building FreeCAD on my macOS machine and run into an issue.

Unfortunately, the mentioned output and error log files are not present. Therefore, I provide you the full terminal output.
I gladly provide more information if required, but I need some assistance with what you need.

System: Catalina 10.15.7 (19H2)

Terminal output:
---

> 
> ➜  FreeCAD brew tap FreeCAD/freecad
> Updating Homebrew...
> ==> Auto-updated Homebrew!
> Updated 1 tap (homebrew/core).
> ==> Updated Formulae
> Updated 13 formulae.
> ➜  FreeCAD brew install --only-dependencies freecad --with-packaging-utils 
> Updating Homebrew...
> ==> Auto-updated Homebrew!
> Updated 1 tap (homebrew/cask).
> ==> Updated Casks
> rider                                                                           rocket-chat
> 
> ==> Installing freecad from freecad/freecad
> ==> Downloading https://homebrew.bintray.com/bottles/boost-python3-1.73.0.catalina.bottle.tar.gz
> ==> Downloading from https://d29vzk4ow07wi7.cloudfront.net/36a3b2ffacb47649b51e0b3031b8f67bf7dcd87e61dfd6d594610cb3e21a1acc?response-content-disposition=attach
> ######################################################################## 100.0%
> ==> Downloading https://homebrew.bintray.com/bottles/node-14.13.0.catalina.bottle.tar.gz
> ==> Downloading from https://d29vzk4ow07wi7.cloudfront.net/e625a89ad797dff3d58056df96e528965794583946754b8cb68074c5fba7276f?response-content-disposition=attach
> ######################################################################## 100.0%
> ==> Installing dependencies for freecad/freecad/freecad: boost-python3 and node
> ==> Installing freecad/freecad/freecad dependency: boost-python3
> ==> Pouring boost-python3-1.73.0.catalina.bottle.tar.gz
> Warning: boost-python3 dependency gcc was built with a different C++ standard
> library (libstdc++ from clang). This may cause problems at runtime.
> 🍺  /usr/local/Cellar/boost-python3/1.73.0: 472 files, 18.1MB
> ==> Installing freecad/freecad/freecad dependency: node
> ==> Pouring node-14.13.0.catalina.bottle.tar.gz
> ==> Caveats
> Bash completion has been installed to:
>   /usr/local/etc/bash_completion.d
> ==> Summary
> 🍺  /usr/local/Cellar/node/14.13.0: 4,698 files, 61.9MB
> ==> Caveats
> ==> node
> Bash completion has been installed to:
>   /usr/local/etc/bash_completion.d
> ➜  FreeCAD brew install --HEAD freecad
> ==> Installing freecad from freecad/freecad
> ==> Cloning https://github.com/vejmarie/FreeCAD.git
> Cloning into '/Users/martinmajewski/Library/Caches/Homebrew/freecad--git'...
> Updating files: 100% (10531/10531), done.
> ==> Checking out branch cloud
> Already on 'cloud'
> Your branch is up to date with 'origin/cloud'.
> ==> pip3 install six
> ==> cmake -DBUILD_QT5=ON -DUSE_PYTHON3=1 -DCMAKE_CXX_FLAGS='-std=c++14' -DBUILD_FEM_NETGEN=1 -DBUILD_FEM=1 -DCMAKE_PREFIX_PATH="/usr/local/opt/qt/lib/cmake;/us
> Last 15 lines from /Users/martinmajewski/Library/Logs/Homebrew/freecad/02.cmake:
>   * libboost_python38-mt.a (3.8, Boost_PYTHON_VERSION=2.7)
> 
>   * libboost_python38.dylib (3.8, Boost_PYTHON_VERSION=2.7)
> 
>   * libboost_python38.a (3.8, Boost_PYTHON_VERSION=2.7)
> 
> Call Stack (most recent call first):
>   /usr/local/lib/cmake/Boost-1.73.0/BoostConfig.cmake:258 (boost_find_component)
>   /usr/local/Cellar/cmake/3.18.3/share/cmake/Modules/FindBoost.cmake:448 (find_package)
>   src/Mod/Path/libarea/CMakeLists.txt:26 (find_package)
> 
> 
> -- Configuring incomplete, errors occurred!
> See also "/tmp/freecad-20201001-33489-sfd6er/Build/CMakeFiles/CMakeOutput.log".
> See also "/tmp/freecad-20201001-33489-sfd6er/Build/CMakeFiles/CMakeError.log".
> 
> If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
>   https://github.com/FreeCAD/homebrew-freecad/issues
> 
> These open issues may also help:
> Auto-bottling for homebrew-freecad tap formulae https://github.com/FreeCAD/homebrew-freecad/issues/63
> Installation of freecad/qtwebkit fails on Mojave. https://github.com/FreeCAD/homebrew-freecad/issues/95
> Please create pull requests instead of asking for help on Homebrew's GitHub,
> Discourse, Twitter or IRC.

peterlama, 2020-10-01:
This seems kinda sus: `libboost_python38.dylib (3.8, Boost_PYTHON_VERSION=2.7)` since it is showing '2.7' but I would need to see the full log. Does `/Users/martinmajewski/Library/Logs/Homebrew/freecad/02.cmake` exist?

MartinMajewski, 2020-10-01:
Sure does! I have attached the file here.

[02.cmake.zip](https://github.com/FreeCAD/homebrew-freecad/files/5313270/02.cmake.zip)

vejmarie, 2020-10-01:
Can you give a list of the various brew packages installed on your system ? I am curious to know if python@3.8 is installed and which boost version is installed.

vejmarie, 2020-10-01:
I think I know where the issue comes from. Let me do a basic PR tonight. The dependency to python3 is not good it shall be python@3.8

MartinMajewski, 2020-10-01:
As I'm using brew and my Mac for various dev projects for over one year now since my last "from scratch re-installation", there are funny things going on for sure. However, I have never dug deeper into how brew works... I'm just a "user"! 😉
So, I will hopefully learn something new out of this.

Here we go...

Python 3.8.5 is available on my system, as well as the macOS default 2.7.16

```
➜  ~ brew list
boost		eigen		gmp		libevent	med-file	openblas	pyqt		sqlite		webp
boost-python	fontconfig	hdf5		libffi		mpfr		opencascade	pyside2		swig		xerces-c
boost-python3	freeimage	hdf5@1.10	libmpc		netcdf		openssl@1.1	pyside2-tools	szip		xz
ccache		freetype	hwloc		libpng		nglib		orocos-kdl	python@3.8	tbb
cmake		gcc		icu4c		libtiff		node		pcre		qt		telnet
coin		gdbm		isl		libyaml		numpy		pcre2		readline	vapor
docker		gettext		jpeg		llvm		oniguruma	pivy		shiboken2	vtk
docker-compose	git		jq		matplotlib	open-mpi	putty		sip		vtk@8.2
➜  ~ brew list --cask
godot                                                                           keepassxc
➜  ~ 
```

Seeing this, I can identify a) python@3.8 should be installed and the error says so, but somehow boost version 2.7 is searched, isn't it? As I can see in the list, boost-python without a version number is there and one with a 3, but none with a 2.7. How are the versions resolved? Filename? Meta-data?

Anyhow, in the freecad.rb it states

`depends_on "boost-python3"`

which is there in the list.

freecad.rb is attached
[freecad.rb.zip](https://github.com/FreeCAD/homebrew-freecad/files/5315098/freecad.rb.zip)

I tried installing FreeCAD with the old Wiki How-To firstly, right before there was the twitter news about the resurrected brew formula.
Then I used the new approach, which you can see in my entry post, you can clearly see the steps:

1. ➜ FreeCAD brew tap FreeCAD/freecad
2. ➜ FreeCAD brew install --only-dependencies freecad --with-packaging-utils
3. ➜ FreeCAD brew install --HEAD freecad

Cheers,
Martin

MartinMajewski, 2020-10-01:
oh, btw.

`➜  ~ which python
/usr/bin/python
➜  ~ which python3
/usr/local/bin/python3
➜  ~ `

vejmarie, 2020-10-01:
ok. Can you try as a workaround.

export PATH=/usr/local/bin:$PATH before launching the command brew install freecad

MartinMajewski, 2020-10-01:
> ok. Can you try as a workaround.
> 
> export PATH=/usr/local/bin:$PATH before launching the command brew install freecad

Same issue. It is not the python3 version that is not called.
Where does this `Boost_PYTHON_VERSION=2.7` come from?

vejmarie, 2020-10-01:
you got boost-python which is installed. Can you remove it ? brew remove boost-python. Looks like that cmake is using it as a default currently with homebrew.

MartinMajewski, 2020-10-01:
Okay, I did. Same issue!

Is there a way to override `Boost_PYTHON_VERSION=2.7`?
I mean, you can not force users to kill other installed versions if their projects depend on them. So maybe you can make your configuration more specific?

vejmarie, 2020-10-01:
a way to switch the library is to use the link option of homebrew, something like brew link libboost-python3 might be helping. I suspect that it is not linked properly due to the previous installation of boot-python

MartinMajewski, 2020-10-01:
Do I have to clear any caches for cmake in order to be reconfigured from scratch?

vejmarie, 2020-10-01:
> Okay, I did. Same issue!
> 
> Is there a way to override `Boost_PYTHON_VERSION=2.7`?
> I mean, you can not force users to kill other installed versions if their projects depend on them. So maybe you can make your configuration more specific?

> Do I have to clear any caches for cmake in order to be reconfigured from scratch?

No. CMake is finding boost_python 2.7 on your system. We just need to remove it, and re-install boost-python3. Are you sure that the brew remove libboost_python has been successful ? If CMake find it back it means it is still there somewhere

vejmarie, 2020-10-01:
Good news. I can reproduce the problem on my machine (I just installed BigSur beta on it). Let me figuring out what is going on.

MartinMajewski, 2020-10-01:
At least brew is not showing boost_python anymore. I also did

```
➜  FreeCAD brew unlink boost-python3 && brew link boost-python3
Unlinking /usr/local/Cellar/boost-python3/1.73.0... 11 symlinks removed
Linking /usr/local/Cellar/boost-python3/1.73.0... 11 symlinks created
```

and on the other hand

```
➜  FreeCAD brew unlink boost_python
Error: No such keg: /usr/local/Cellar/boost_python
```

vejmarie, 2020-10-01:
Yep looks like the issue might be coming from latest XCode. I didn't have it with XCode 11.6

MartinMajewski, 2020-10-01:
Okay, after switching you have to install the cmd tools again.

However

`xcode-select --install`

will not find any software to install. The best way is to do this by opening XCode.app.
If the installation is not taking place immediately, one can open `Xcode -> Preferences... -> Locations` and select the tools there. I am using 12.0.1 now.

MartinMajewski, 2020-10-01:
Please have a look here:

[https://stackoverflow.com/questions/3016448/how-can-i-get-cmake-to-find-my-alternative-boost-installation](https://stackoverflow.com/questions/3016448/how-can-i-get-cmake-to-find-my-alternative-boost-installation)

Can we somehow pin cmake to a specific boost installation?

MartinMajewski, 2020-10-01:
Another interesting observation:
```➜  ~ brew uninstall --ignore-dependencies boost
Uninstalling /usr/local/Cellar/boost/1.73.0... (14,538 files, 509.6MB)
➜  ~ brew uninstall boost-python3
Uninstalling /usr/local/Cellar/boost-python3/1.73.0... (472 files, 18.1MB)
➜  ~ sudo rm -rf /usr/local/include/boost
Password:
➜  ~ sudo rm -rf /usr/local/lib/libboost*
zsh: no matches found: /usr/local/lib/libboost*
➜  ~ brew install --HEAD freecad               
==> Installing freecad from freecad/freecad
==> Downloading https://homebrew.bintray.com/bottles/boost-1.73.0.catalina.bottle.1.tar.gz
Already downloaded: /Users/martinmajewski/Library/Caches/Homebrew/downloads/17ca93cf853c56c6ed876fcb40b33de4b4a16d32508413196accb43296e80855--boost-1.73.0.catalina.bottle.1.tar.gz
==> Downloading https://homebrew.bintray.com/bottles/boost-python3-1.73.0.catalina.bottle.tar.gz
Already downloaded: /Users/martinmajewski/Library/Caches/Homebrew/downloads/42203bfe65080a0fe7869f520b33479f78ca353754c36070de66a47c4d3e4452--boost-python3-1.73.0.catalina.bottle.tar.gz
==> Cloning https://github.com/vejmarie/FreeCAD.git
Updating /Users/martinmajewski/Library/Caches/Homebrew/freecad--git
==> Checking out branch cloud
Already on 'cloud'
Your branch is up to date with 'origin/cloud'.
HEAD is now at f550504 Fix MacOS build with boost 1.73 and updated Homebrew packages
==> Installing dependencies for freecad/freecad/freecad: boost and boost-python3
==> Installing freecad/freecad/freecad dependency: boost
==> Pouring boost-1.73.0.catalina.bottle.1.tar.gz
🍺  /usr/local/Cellar/boost/1.73.0: 14,538 files, 509.6MB
==> Installing freecad/freecad/freecad dependency: boost-python3
==> Pouring boost-python3-1.73.0.catalina.bottle.tar.gz
Warning: boost-python3 dependency gcc was built with a different C++ standard
library (libstdc++ from clang). This may cause problems at runtime.
🍺  /usr/local/Cellar/boost-python3/1.73.0: 472 files, 18.1MB
==> Installing freecad/freecad/freecad --HEAD
patching file /usr/local/Cellar/boost/1.73.0/include/boost/geometry/index/detail/rtree/visitors/insert.hpp
patch: **** Can't rename file /tmp/po3GxHVf to /usr/local/Cellar/boost/1.73.0/include/boost/geometry/index/detail/rtree/visitors/insert.hpp : Operation not permitted
==> pip3 install six
==> cmake -DBUILD_QT5=ON -DUSE_PYTHON3=1 -DCMAKE_CXX_FLAGS='-std=c++14' -DBUILD_FEM_NETGEN=1 -DBUILD_FEM=1 -DCMAKE_PREFIX_PATH="/usr/local/opt/qt/lib/cmake;/us
Last 15 lines from /Users/martinmajewski/Library/Logs/Homebrew/freecad/02.cmake:
  * libboost_python38-mt.a (3.8, Boost_PYTHON_VERSION=2.7)

  * libboost_python38.dylib (3.8, Boost_PYTHON_VERSION=2.7)

  * libboost_python38.a (3.8, Boost_PYTHON_VERSION=2.7)

Call Stack (most recent call first):
  /usr/local/lib/cmake/Boost-1.73.0/BoostConfig.cmake:258 (boost_find_component)
  /usr/local/Cellar/cmake/3.18.3/share/cmake/Modules/FindBoost.cmake:448 (find_package)
  src/Mod/Path/libarea/CMakeLists.txt:26 (find_package)

-- Configuring incomplete, errors occurred!
See also "/tmp/freecad-20201002-76469-dweogx/Build/CMakeFiles/CMakeOutput.log".
See also "/tmp/freecad-20201002-76469-dweogx/Build/CMakeFiles/CMakeError.log".

If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
  https://github.com/FreeCAD/homebrew-freecad/issues

These open issues may also help:
Building FreeCAD on macOS Catalina fails during configuration. https://github.com/FreeCAD/homebrew-freecad/issues/114
Auto-bottling for homebrew-freecad tap formulae https://github.com/FreeCAD/homebrew-freecad/issues/63
Installation of freecad/qtwebkit fails on Mojave. https://github.com/FreeCAD/homebrew-freecad/issues/95
Please create pull requests instead of asking for help on Homebrew's GitHub,
Discourse, Twitter or IRC.
➜  ~ 
```

I uninstalled boost and the formula reinstalls it but then fails with the same error.

MartinMajewski, 2020-10-01:
And I found this interesting post that supports my theory:

> It's a bit more involved. The python component, if more than one Python version variant is present, picks up one at random, unless Boost_PYTHON_VERSION is set. The format of Boost_PYTHON_VERSION is f.ex. 2.7 or 3.8.
> 
> For compatibility with FindBoost, components such as pythonXY are recognized and treated as if the component python is requested with Boost_PYTHON_VERSION set to X.Y.
> 
> The logical behavior for the component python3 (and python2 for symmetry) would be for it to pick up one of the 3.x versions (but not the 2.x ones.) But this doesn't fit well within the current logic.
> 
> I'll try to think of something.

Link: https://github.com/boostorg/boost_install/issues/31

vejmarie, 2020-10-01:
The issue is not coming from boost, but from the fact that CMAKE doesn't identify the right python version. If you look at the early beginning of the output, it does report that it will be using python 2.7 and that it can't find homebrew python binaries. I will fix that in a couple of minutes/hours

MartinMajewski, 2020-10-01:
> The issue is not coming from boost, but from the fact that CMAKE doesn't identify the right python version. If you look at the early beginning of the output, it does report that it will be using python 2.7 and that it can't find homebrew python binaries. I will fix that in a couple of minutes/hours

Where do you see this output?

vejmarie, 2020-10-01:
yep it is in 02.cmake ;)

vejmarie, 2020-10-01:
ok, I found it. The issue comes from CMAKE I think. if you edit the file SetupPython.cmake into cMake/FreeCAD_Helpers the find_program command is trying to initialize a variable which soon exist (PYTHON_EXECUTABLE) and it fails to update it. Let me think about it and update my freecad branch with a fix

vejmarie, 2020-10-01:
It is easy to fix. You can try that. Open /usr/local/Homebrew/Library/Taps/freecad/homebrew-freecad/Formula/freecad.rb . Line 87 add " args << "-DPTYHON_EXECUTABLE=/usr/local/bin/python3"  " and it will work.
You might have to relink a few stuff like VTK, but let's see if it goes farther

MartinMajewski, 2020-10-01:
Yep, it does go further...

currently, it is making a `make -j8 install`

MartinMajewski, 2020-10-01:
And yep, you were right, it has other issues:

```
Last 15 lines from /Users/martinmajewski/Library/Logs/Homebrew/freecad/03.make:
[ 34%] Building CXX object src/Base/CMakeFiles/FreeCADBase.dir/__/zipios++/zipheadio.cpp.o
cd /tmp/freecad-20201002-78731-1jtm7ja/Build/src/Base && /usr/local/bin/ccache /usr/local/Homebrew/Library/Homebrew/shims/mac/super/clang++ -DBOOST_ALL_NO_LIB -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_PP_VARIADICS=1 -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_REGEX_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -DBOOST_THREAD_DYN_LINK -DFreeCADBase_EXPORTS -DHAVE_CONFIG_H -DHAVE_FREEIMAGE -DHAVE_RAPIDJSON -DHAVE_SWIG=1 -DHAVE_TBB -DOCC_CONVERT_SIGNALS -DQT_CORE_LIB -DQT_NO_DEBUG -D_OCC64 -I/tmp/freecad-20201002-78731-1jtm7ja/Build -I/tmp/freecad-20201002-78731-1jtm7ja/Build/src -I/tmp/freecad-20201002-78731-1jtm7ja/src -I/tmp/freecad-20201002-78731-1jtm7ja/Build/src/Base -I/tmp/freecad-20201002-78731-1jtm7ja/src/Base -I/usr/local/opt/python@3.8/Frameworks/Python.framework/Versions/3.8/include/python3.8 -iframework /usr/local/opt/qt/lib -isystem /usr/local/opt/qt/lib/QtCore.framework/Headers -isystem /usr/local/opt/qt/./mkspecs/macx-clang -Wall -Wextra -Wpedantic -Wno-write-strings -std=c++14 -Wno-undefined-var-template -DNDEBUG -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.15.sdk -fPIC -I/usr/local/Cellar/open-mpi/4.0.5/include -fPIC -std=gnu++11 -o CMakeFiles/FreeCADBase.dir/__/zipios++/zipheadio.cpp.o -c /tmp/freecad-20201002-78731-1jtm7ja/src/zipios++/zipheadio.cpp
[ 34%] Building CXX object src/Base/CMakeFiles/FreeCADBase.dir/__/zipios++/zipinputstream.cpp.o
cd /tmp/freecad-20201002-78731-1jtm7ja/Build/src/Base && /usr/local/bin/ccache /usr/local/Homebrew/Library/Homebrew/shims/mac/super/clang++ -DBOOST_ALL_NO_LIB -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_PP_VARIADICS=1 -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_REGEX_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -DBOOST_THREAD_DYN_LINK -DFreeCADBase_EXPORTS -DHAVE_CONFIG_H -DHAVE_FREEIMAGE -DHAVE_RAPIDJSON -DHAVE_SWIG=1 -DHAVE_TBB -DOCC_CONVERT_SIGNALS -DQT_CORE_LIB -DQT_NO_DEBUG -D_OCC64 -I/tmp/freecad-20201002-78731-1jtm7ja/Build -I/tmp/freecad-20201002-78731-1jtm7ja/Build/src -I/tmp/freecad-20201002-78731-1jtm7ja/src -I/tmp/freecad-20201002-78731-1jtm7ja/Build/src/Base -I/tmp/freecad-20201002-78731-1jtm7ja/src/Base -I/usr/local/opt/python@3.8/Frameworks/Python.framework/Versions/3.8/include/python3.8 -iframework /usr/local/opt/qt/lib -isystem /usr/local/opt/qt/lib/QtCore.framework/Headers -isystem /usr/local/opt/qt/./mkspecs/macx-clang -Wall -Wextra -Wpedantic -Wno-write-strings -std=c++14 -Wno-undefined-var-template -DNDEBUG -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.15.sdk -fPIC -I/usr/local/Cellar/open-mpi/4.0.5/include -fPIC -std=gnu++11 -o CMakeFiles/FreeCADBase.dir/__/zipios++/zipinputstream.cpp.o -c /tmp/freecad-20201002-78731-1jtm7ja/src/zipios++/zipinputstream.cpp
[ 34%] Building CXX object src/Base/CMakeFiles/FreeCADBase.dir/__/zipios++/zipinputstreambuf.cpp.o
cd /tmp/freecad-20201002-78731-1jtm7ja/Build/src/Base && /usr/local/bin/ccache /usr/local/Homebrew/Library/Homebrew/shims/mac/super/clang++ -DBOOST_ALL_NO_LIB -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_PP_VARIADICS=1 -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_REGEX_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -DBOOST_THREAD_DYN_LINK -DFreeCADBase_EXPORTS -DHAVE_CONFIG_H -DHAVE_FREEIMAGE -DHAVE_RAPIDJSON -DHAVE_SWIG=1 -DHAVE_TBB -DOCC_CONVERT_SIGNALS -DQT_CORE_LIB -DQT_NO_DEBUG -D_OCC64 -I/tmp/freecad-20201002-78731-1jtm7ja/Build -I/tmp/freecad-20201002-78731-1jtm7ja/Build/src -I/tmp/freecad-20201002-78731-1jtm7ja/src -I/tmp/freecad-20201002-78731-1jtm7ja/Build/src/Base -I/tmp/freecad-20201002-78731-1jtm7ja/src/Base -I/usr/local/opt/python@3.8/Frameworks/Python.framework/Versions/3.8/include/python3.8 -iframework /usr/local/opt/qt/lib -isystem /usr/local/opt/qt/lib/QtCore.framework/Headers -isystem /usr/local/opt/qt/./mkspecs/macx-clang -Wall -Wextra -Wpedantic -Wno-write-strings -std=c++14 -Wno-undefined-var-template -DNDEBUG -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.15.sdk -fPIC -I/usr/local/Cellar/open-mpi/4.0.5/include -fPIC -std=gnu++11 -o CMakeFiles/FreeCADBase.dir/__/zipios++/zipinputstreambuf.cpp.o -c /tmp/freecad-20201002-78731-1jtm7ja/src/zipios++/zipinputstreambuf.cpp
[ 34%] Building CXX object src/Base/CMakeFiles/FreeCADBase.dir/__/zipios++/zipoutputstreambuf.cpp.o
cd /tmp/freecad-20201002-78731-1jtm7ja/Build/src/Base && /usr/local/bin/ccache /usr/local/Homebrew/Library/Homebrew/shims/mac/super/clang++ -DBOOST_ALL_NO_LIB -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_PP_VARIADICS=1 -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_REGEX_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -DBOOST_THREAD_DYN_LINK -DFreeCADBase_EXPORTS -DHAVE_CONFIG_H -DHAVE_FREEIMAGE -DHAVE_RAPIDJSON -DHAVE_SWIG=1 -DHAVE_TBB -DOCC_CONVERT_SIGNALS -DQT_CORE_LIB -DQT_NO_DEBUG -D_OCC64 -I/tmp/freecad-20201002-78731-1jtm7ja/Build -I/tmp/freecad-20201002-78731-1jtm7ja/Build/src -I/tmp/freecad-20201002-78731-1jtm7ja/src -I/tmp/freecad-20201002-78731-1jtm7ja/Build/src/Base -I/tmp/freecad-20201002-78731-1jtm7ja/src/Base -I/usr/local/opt/python@3.8/Frameworks/Python.framework/Versions/3.8/include/python3.8 -iframework /usr/local/opt/qt/lib -isystem /usr/local/opt/qt/lib/QtCore.framework/Headers -isystem /usr/local/opt/qt/./mkspecs/macx-clang -Wall -Wextra -Wpedantic -Wno-write-strings -std=c++14 -Wno [... cut ...] ilesystem-mt.dylib /usr/local/lib/libboost_program_options-mt.dylib /usr/local/lib/libboost_regex-mt.dylib /usr/local/lib/libboost_system-mt.dylib /usr/local/lib/libboost_thread-mt.dylib /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.15.sdk/usr/lib/libz.tbd /usr/local/opt/qt/lib/QtCore.framework/QtCore /usr/local/opt/python@3.8/Frameworks/Python.framework/Versions/3.8/lib/libpython3.8.dylib 
[ 34%] Built target FreeCADBase
make: *** [all] Error 2
```

vejmarie, 2020-10-01:
I'll be needing the whole file to have a look to it. Hav to go offline for 1 hour ;)

vejmarie, 2020-10-02:
I just updated a few formula which shall be fixing some of the issues reported here (#115). By the way I am able to build FreeCAD on Bigsur Beta9 with XCode12 with that PR. Can you try to test the changes ? @peterlama can you upstream this PR ? It does work also on catalina.

MartinMajewski, 2020-10-02:
> I'll be needing the whole file to have a look to it. Hav to go offline for 1 hour ;)

I'm sorry, but it was 2:00am here in Germany and I needed some rest! Will test the current fix later and provide more information and also "whole files"! 😉

MartinMajewski, 2020-10-02:
Okay, @vejmarie, I pulled your changes to freecad.rb and matplotlib into my local git location and rerun `brew install --HEAD freecad` but it failed with the error shown above.

Here is the 03.cmake file:

[03.make.zip](https://github.com/FreeCAD/homebrew-freecad/files/5316285/03.make.zip)

vejmarie, 2020-10-02:
Looks like your command line tool is experiencing issue. Can you check that these files exist
/Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/lib/libz.1.tbd and /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/lib/libz.tbd
According to the output of your build they do not exist as CMAKE is trying to build them
I have them on my Mac.

vejmarie, 2020-10-02:
> > I'll be needing the whole file to have a look to it. Hav to go offline for 1 hour ;)
> 
> I'm sorry, but it was 2:00am here in Germany and I needed some rest! Will test the current fix later and provide more information and also "whole files"! 😉

No problem, I am in California ! so we will have to handle jetlag and latencies ;)

MartinMajewski, 2020-10-02:
> Looks like your command line tool is experiencing issue. Can you check that these files exist
> /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/lib/libz.1.tbd and [...]

This solves it, but it is not a very user-friendly issue! :-)

Details:

The mentioned files were missing. In fact, the entire CommandLineTools folder was not there.
The reason is that XCode installs the CommandLineTools in the XCode.app package under `/Applications/Xcode.app/Contents/Developer`

After the CMD tools are once installed you cannot install it via the terminal and `xcode-select --install` again. A misleading error message will show up telling you that the software is currently no available from the update servers. This is wrong, it should be a "tools are already installed" message. Anyhow, the path `/Library/Developer/CommandLineTools/` was not available but CMake relies on it. Maybe CMake should retrieve the correct path by calling `xcode-select -p`.

To solve this issue, I headed over to [https://developer.apple.com/download](https://developer.apple.com/download) and downloaded the CommandLineTools DMG file and installed them this way. The mentioned path and its content were created.
To switch between the paths you can run the `sudo xcode-select --switch <xcode_folder_path>`)` command. This command only works, if the destination path exists, otherwise an error will occur.

I guess another solution would be to symlink the external path into the XCode package path.

After installing the Tools "externally", brew was able to run the install process beyond the last steps. In fact, it is still compiling. A more verbose output would be nice. Currently, I can only see the activities in the title bar of the Terminal window.

vejmarie, 2020-10-02:
Yes it is weird that they install command line tools in 2 differents places depending from where it is coming :(. Regarding the homebrew UI I can't do that much about it. The compilation time can take some time for sure that is why I am using the -v option from brew which display each compiler command to let me know where it stands !. Just let me know if everything works up to the end.

MartinMajewski, 2020-10-02:
It seems to be working till the end despite some warnings:

```
==> cmake -DBUILD_QT5=ON -DUSE_PYTHON3=1 -DPYTHON_EXECUTABLE=/usr/local/bin/python3 -DCMAKE_CXX_FLAGS='-std=c++14' -DBUILD_FEM_NETGEN=1 -DBUILD_FEM=1 -DCMAKE_P
==> make -j8 install
Warning: freecad/freecad/freecad dependency gcc was built with a different C++ standard
library (libstdc++ from clang). This may cause problems at runtime.
==> pip3 install six
Warning: The post-install step did not complete successfully
You can try again using `brew postinstall freecad/freecad/freecad`
==> Caveats
    After installing FreeCAD you may want to do the following:

    1. Amend your PYTHONPATH environmental variable to point to
       the FreeCAD directory
         export PYTHONPATH=/usr/local/opt/freecad/bin:$PYTHONPATH
==> Summary
🍺  /usr/local/Cellar/freecad/HEAD-f550504: 1,626 files, 342.9MB, built in 41 minutes 7 seconds
==> Caveats
==> ccache
To install symlinks for compilers that will automatically use
ccache, prepend this directory to your PATH:
  /usr/local/opt/ccache/libexec

If this is an upgrade and you have previously added the symlinks to
your PATH, you may need to modify it to the path specified above so
it points to the current version.

NOTE: ccache can prevent some software from compiling.
ALSO NOTE: The brew command, by design, will never use ccache.
==> freecad
    After installing FreeCAD you may want to do the following:

    1. Amend your PYTHONPATH environmental variable to point to
       the FreeCAD directory
         export PYTHONPATH=/usr/local/opt/freecad/bin:$PYTHONPATH
```

vejmarie, 2020-10-02:
can you try to launch FreeCAD ?

MartinMajewski, 2020-10-02:
Post-install is not completing successfully... any drawbacks?

vejmarie, 2020-10-02:
> Post-install is not completing successfully... any drawbacks?

I need to know if FreeCAD starts to check how harmful the post install failure was.

MartinMajewski, 2020-10-02:
I did the mentioned PYTHONPATH export and run `freecad` from Terminal:

[image]

MartinMajewski, 2020-10-02:
Is this the correct version?

14555 seems older than 22522 which can be downloaded from the website.

vejmarie, 2020-10-02:
Good progress. Yep the branch is locked to one of my branch for the moment. It will be updated upstreamed after your successful build no worries to have regarding this ;)

MartinMajewski, 2020-10-02:
> Good progress. Yep the branch is locked to one of my branch for the moment. It will be updated upstreamed after your successful build no worries to have regarding this ;)

Okay, so to summarize:

- We found an issue with multiple Python versions present, especially if it is 2.7. This has been fixed!
- We found an issue with the Command Line Tools, were a solution is to download and install the tools manually. This is something for the wiki, I guess.

vejmarie, 2020-10-02:
yep. I think we got everything. I will update the wiki accordingly

vejmarie, 2020-10-02:
> > Good progress. Yep the branch is locked to one of my branch for the moment. It will be updated upstreamed after your successful build no worries to have regarding this ;)
> 
> Okay, so to summarize:
> 
> * We found an issue with multiple Python versions present, especially if it is 2.7. This has been fixed!
> * We found an issue with the Command Line Tools, where a solution is to download and install the tools manually. This is something for the wiki, I guess.

I think the command line tool issue is addressed by hombrew itself in normal operation (I mean for users which are using it as a default tool without multiple xcode installation). Homebrew is downloading the dmg by itself, and it could explain why we do have this dependancy

MartinMajewski, 2020-10-02:
> Homebrew is downloading the dmg by itself, and it could explain why we do have this dependency

Does it? I don't think that Homebrew is doing anything like that with the CMD Tools (just speculations). If I remember correctly, Homebrew tells the user explicitly to run `xcode-select --install` to install them, because it requires the tools to be present. But if this fails for the reasons mentioned above (which in fact is not even a "multiple XCode" issue, but could be caused by simply updating XCode) there are no further hints given. The error message by Apple is misleading AF!

Even worse: The CMD tools' binaries can be called from the Terminal either way, but the libs are missing at the expected location. So a user could be confused about the root cause. Because the timespan between official FreeCAD builds is so large, users with even less experience than me regarding CMake and Homebrew's internal will be attracted to build FreeCAD themselves. Therefore, the number of users encountering those issues will increase (after the Fusion 360 escape even more so, I expect).

The Wiki should mention this possibility. This would help this project here and also other projects when users searching for a hint.

vejmarie, 2020-10-02:
yep I agree. I rechecked with a fresh installation of homebrew, and confirm that it does install Command Line Tools, but the packaging could evolved as you made it evolved on your systems and led to failures. I will update the wiki during the week-end to be sure that people could workaround these issues.

MartinMajewski, 2020-10-02:
Shall I close this issue?

vejmarie, 2020-10-02:
you are the one who opened it ;). Up to you to decide, but I think we mostly covered your issue.

MartinMajewski, 2020-10-02:
That's true! 😅

Yes, for this issue I'm all set and satisfied with the result.

Thanks for your great support!

Btw: I am re-installing FreeCAD from the official repo right now. I'm curious if this will work. At least brew made it to the make stage already. So, no Python version issue present! 😎

closed 2020-10-02 by MartinMajewski

nachumk, 2021-08-03:
I am getting the same message as the originator got. I tried updating Python3 path in freecad.rb, but that didn't work. The referred to log files don't exist.

I am on MacOS 11.5.1. I have Xcode updated. xcode-select --install says Xcode command line tools are already installed.

[noahk@noahk ~ % brew install --head -s freecad --with-macos-app                                  
Updating Homebrew...
==> Auto-updated Homebrew!
Updated 1 tap (homebrew/core).
==> Updated Formulae
Updated 3 formulae.

==> Installing freecad from freecad/freecad
==> Cloning https://github.com/freecad/FreeCAD.git
Updating /Users/noahk/Library/Caches/Homebrew/freecad--git
==> Checking out branch master
Already on 'master'
Your branch is up to date with 'origin/master'.
HEAD is now at ea9272b1a3 FEM: calculix writer, reaction forces, add some comment
==> cmake -DBUILD_QT5=ON -DUSE_PYTHON3=1 -DCMAKE_CXX_STANDARD=14 -DBUILD_ENABLE_CXX_STD:STRING=C++14 -DBUILD_FEM_NETGEN=1 -DBUILD_FEM=1 -DBUILD_FEM_NETGE
Last 15 lines from /Users/noahk/Library/Logs/Homebrew/freecad/01.cmake:
    boost_python-config.cmake

  Add the installation prefix of "boost_python" to CMAKE_PREFIX_PATH or set
  "boost_python_DIR" to a directory containing one of the above files.  If
  "boost_python" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  /usr/local/lib/cmake/Boost-1.75.0/BoostConfig.cmake:261 (boost_find_component)
  /usr/local/Cellar/cmake/3.21.1/share/cmake/Modules/FindBoost.cmake:594 (find_package)
  src/Mod/Path/libarea/CMakeLists.txt:26 (find_package)

-- Configuring incomplete, errors occurred!
See also "/tmp/freecad-20210803-96061-2mdicn/Build/CMakeFiles/CMakeOutput.log".
See also "/tmp/freecad-20210803-96061-2mdicn/Build/CMakeFiles/CMakeError.log".

If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
  https://github.com/freecad/homebrew-freecad/issues