FreeCAD/homebrew-freecad #118: installation error on macOS 11 (big sur)

turboooooo, 2020-10-07:
Hi,
continuing from [this thread on the forums](https://forum.freecadweb.org/viewtopic.php?f=4&t=50843&start=10) here's the output of my latest attempt to install freecad.
[brewfreecad.txt](https://github.com/FreeCAD/homebrew-freecad/files/5340234/brewfreecad.txt)

vejmarie, 2020-10-08:
Did you installed all dependencies first ? Can you tell me which version of XCode you are using ? I do have a platform which runs BigSur and do not face the same issue than you do.

turboooooo, 2020-10-08:
Yes, I have installed xcode 12.2 beta and done a "brew install --only-dependencies freecad" before the attached attempt. It still fails to build at 5% here.

[freecad_homebrew_bigsur_fail.txt](https://github.com/FreeCAD/homebrew-freecad/files/5347671/freecad_homebrew_bigsur_fail.txt)

turboooooo, 2020-10-09:
That path on my system would be:

/Applications/Xcode-beta.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.0.sdk/System/Library/Frameworks/AppKit.framework

I did an upgrade from Catalina - will do a fresh install when the final is out.

vejmarie, 2020-10-09:
@turboooooo Can you try to get rid of XCode-Beta ? My XCode 12.0.1 (12A7300) is a released and installed into /Applications/Xcode . I suspect (and might be wrong) that the issue is coming from your xcode setup, just liked to get the confirmation

vejmarie, 2020-10-09:
@turboooooo ok I can reproduce the issue on my mac running big sur and latest xcode beta. Let me have a look to it during the week-end. that might be coming from freecad dependencies and requires a lot of recompilation, as homebrew is not ported yet to bigsur.

turboooooo, 2020-10-09:
@vejmarie yes - It looks like you are correct: it is at 66% right now. I am looking forward working with that build. Sketcher had a few annoying bugs in the latest working 0.19 binary release ;)

turboooooo, 2020-10-09:
Installs fine with Xcode 12 (stable) on big sur beta. Thanks!

closed 2020-10-09 by turboooooo

vejmarie, 2020-10-09:
@turboooooo you can even make it work with XCode beta 12.2. To do that you need to re-install opencascade from source. This is this library which is bringing in the dependencies. If you do brew reinstall -s -v opencascade and the brew install -s -v freecad, it will compile like a charm. Which is a good news, because it probably means that the shared binary model might be working with the Apple proprieraty silicon coming up.

vejmarie, 2020-10-09:
OS: macOS 11.0
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.22504 (Git)
Build type: Release
Branch: master
Hash: 3e7e2a9bc4ccbe5913dabcad1297589a17b7117f
Python version: 3.8.5
Qt version: 5.15.1
Coin version: 4.0.0
OCC version: 7.4.0
Locale: English/United States (en_US)