FreeCAD/homebrew-freecad #123: Fail install cask Mac

joaquinpolom, 2020-10-28:
Downloading https://github.com/FreeCAD/FreeCAD/releases/download/0.18.4/FreeCAD_0.18-16146-OSX-x86_64-conda-Qt5-Py3.dmg

curl: (22) The requested URL returned error: 404 Not Found
Error: Download failed on Cask 'freecad' with message: Download failed: https://github.com/FreeCAD/FreeCAD/releases/download/0.18.4/FreeCAD_0.18-16146-OSX-x86_64-conda-Qt5-Py3.dmg

luzpaz, 2020-11-03:
@joaquinpolom  where are you downloading from ?  
Whomever it is needs to update their target to https://github.com/FreeCAD/FreeCAD/releases/download/0.18.4/FreeCAD_0.18-16146-rev1-OSX-x86_64-conda-Qt5-Py3.dmg

ipatch, 2020-11-05:
@luzpaz i'd mark this issue as stale, and then close it.

@joaquinpolom 

the latest pre release of freecad 0.19 can be downloaded within assets group on the releases page of the official [freecad repo](https://github.com/freecad/freecad) which i recommend using over the 0.18.4 release, as the 0.19 pre has become quite stable and offers a significant amount of more complete features than the 0.18.4

---

- 0.18.4 [tag](https://github.com/FreeCAD/FreeCAD/releases/tag/0.18.4) [download](https://github.com/FreeCAD/FreeCAD/releases/download/0.18.4/FreeCAD_0.18-16146-rev1-OSX-x86_64-conda-Qt5-Py3.dmg) if you must, **not recommended**
- 0.19 pre [tag](https://github.com/FreeCAD/FreeCAD/releases/tag/0.19_pre) [download](https://github.com/FreeCAD/FreeCAD/releases/download/0.19_pre/FreeCAD_0.19-22894-macOS-x86_64-conda.dmg)

luzpaz, 2020-11-05:
Thanks @ipatch   
@vejmarie please close ticket.

vejmarie, 2020-11-09:
We can close it now.

closed 2020-11-09 by vejmarie