FreeCAD/homebrew-freecad #125: cannot make macos app

zenz, 2020-11-16:
while using brew install freecad --with-packaging-utils
it brings this:
```
Last 15 lines from /Users/simon/Library/Logs/Homebrew/freecad/01.node:
-g
app_dmg

node:internal/modules/cjs/loader:922
  throw err;
  ^

Error: Cannot find module '/private/tmp/freecad-20201116-39765-74kx2e/install'
    at Function.Module._resolveFilename (node:internal/modules/cjs/loader:919:15)
    at Function.Module._load (node:internal/modules/cjs/loader:763:27)
    at Function.executeUserEntryPoint [as runMain] (node:internal/modules/run_main:76:12)
    at node:internal/main/run_main_module:17:47 {
  code: 'MODULE_NOT_FOUND',
  requireStack: []
}

```

zenz, 2020-11-17:
I found in freecad.rb file, there is a line:
`system "node", "install", "-g", "app_dmg"`
But in npm.js, I cannot find app_dmg, but appdmg, is this the reason makes it failed?

vejmarie, 2020-11-17:
Hi, I am not sure to understand what you are trying to do ? Are you trying to build a dmg image file from homebrew build process ?

zenz, 2020-11-17:
@vejmarie Hi, 
What I want is to create an standalone MacOS app which can be find and click to run in Application Folder or dash board.

zenz, 2020-11-17:
Oh, I found this in the compiling guide, this was missed in the homebrew freecad.rb file.
-DFREECAD_CREATE_MAC_APP=1
I will modify the bottles and give it a try.

vejmarie, 2020-11-17:
No problem for a PR, but I think it might be good to put the MAC APP creation as an input option of the recipe. Homebrew is supported also on linux and I do not really know what might be the behavior if we activate it on linux by default.

zenz, 2020-11-18:
> No problem for a PR, but I think it might be good to put the MAC APP creation as an input option of the recipe. Homebrew is supported also on linux and I do not really know what might be the behavior if we activate it on linux by default.

Oh, I thought that homebrew only works under MacOS.

zenz, 2020-11-18:
I made a pull request, with an compile option --with-macos-app to make macOS app.

vejmarie, 2020-11-18:
Just integrated your PR. Thanks for your contribution !

closed 2020-11-18 by vejmarie