FreeCAD/homebrew-freecad #135: new freecad git source merged some 3rd party code binding to MacOS sdk 11.0 but now is 11.1

zenz, 2020-12-22:
Which makes the compilation failed.
I had to make a symble link for old sdk to pass the compilation.
Should make a patch.

vejmarie, 2020-12-22:
Not sure to fully understand the issue. Can you describe it differently, and tell me which symlink you had to create to workaround it ?

zenz, 2020-12-22:
some 3rd party code use this /Library/Developer/CommandLineTools/SDKs/MacOSX11.0.sdk/libXXX
but xcode seems will remove old sdk after upgrade to new version, so I have to manually to make a symble link from MacOSX11.1.sdk to MacOSX11.0.sdk

vejmarie, 2020-12-22:
Ok, I am aware about that issue. It is coming from the fact that the brew was compiled with XCode 11 and not 12. I am currently updating this for bigsur. Shall be published tonight !

zenz, 2020-12-22:
using MacOS.sdk symbol link to point to latest sdk is more reliable.

ipatch, 2020-12-22:
zenz can you post a link to the line in the code that is referencing this specific sdk version so i can take a look at the upstream code later and possibly submit a pr upstream so this doesnt trickle down to this repo in the future?

presently afk

vejmarie, 2020-12-22:
I just updated the bottles, and that issue shall be fixed as everything on bigsur is compiled using XCode 12.3

zenz, 2020-12-23:
@vejmarie @ipatch 

I updated the bottles, but still cannot pass.
I pick them up from 02.make log:
make[1]: *** [src/3rdParty/salomesmesh/CMakeFiles/Driver.dir/all] Error 2
make[1]: *** [src/3rdParty/salomesmesh/CMakeFiles/MEFISTO2.dir/all] Error 2

vejmarie, 2020-12-23:
Can you attach the full log and the command you type in to get this error ? If you install the bottle normally you do not recompile

zenz, 2020-12-23:
I do compile from source
brew remove --force freecad
brew update
brew tap freecad/freecad
brew install --HEAD freecad --with-macos-app -s -v

Here's the log
[errlog.zip](https://github.com/FreeCAD/homebrew-freecad/files/5732897/errlog.zip)

zenz, 2020-12-23:
I choose the HEAD version since it fixs the unicode problem.

vejmarie, 2020-12-23:
Ok, the error is this, now we need to figure out why MEFISTO needs that. I bet it is a med file dependency.

make[2]: *** No rule to make target `/Library/Developer/CommandLineTools/SDKs/MacOSX11.0.sdk/System/Library/Frameworks/AppKit.framework', needed by `lib/libMEFISTO2.dylib'.  Stop.

vejmarie, 2020-12-23:
Can you give it back a try ? I just updated med-file with xcode 12.3

zenz, 2020-12-23:
@vejmarie 
there was 3 places have this issue:
make[2]: *** No rule to make target `/Library/Developer/CommandLineTools/SDKs/MacOSX11.0.sdk/System/Library/Frameworks/AppKit.framework', needed by `lib/libMEFISTO2.dylib'.  Stop.
make[2]: *** No rule to make target `/Library/Developer/CommandLineTools/SDKs/MacOSX11.0.sdk/System/Library/Frameworks/AppKit.framework', needed by `lib/libDriver.dylib'.  Stop.
make[2]: *** No rule to make target `/Library/Developer/CommandLineTools/SDKs/MacOSX11.0.sdk/usr/lib/libz.tbd', needed by `lib/libSMDS.dylib'.  Stop.

With current update, there's no luck. I delete the freecad tap folder, do a retap and rebuilt, but still failed with same errors.

vejmarie, 2020-12-23:
These 3 places have the same dependencies, I am not surprised that they fail the same way. Let me recheck nglib

vejmarie, 2020-12-23:
Which xcode version do you use ?

zenz, 2020-12-23:
the latest 12.3 (12C33)

vejmarie, 2020-12-23:
I might be wrong but I think you still have a dependencies which is not properly updated with homebrew. I just checked on my system and the MEFISTO library (into link.txt) is dependent to /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.1.sdk and their is no trace from anything else

can you try these commands

brew remove freecad
brew remove matplotlib shiboken2 pyside2 pyside2-tools pivy opencamlib med-file coin nglib

Then brew update and brew upgrade

Can you carefully check if you have any error while performing the upgrade ? Then try to recompile. Do you have multiple XCode version present ? Which MacOS do you run ?

zenz, 2020-12-23:
I do the search 11.0.sdk in FreeCAD git source, there's no place founded have such bindings. but the brew build insisted to link dynlib there.

vejmarie, 2020-12-23:
cmake is putting this for you. This is a quick and dirty output that I got from it on my bigsur system with XCode 12.3. It will be far much faster to find the root cause by hand ;)

You can replicate it by

cloning freecad source into a directory through git clone https://github.com/freecad/freecad
into the same directory create a build directory then go inside and type

cmake -DCMAKE_PREFIX_PATH="/usr/local/Cellar/vtk@8.2/8.2.0_1/lib/cmake;/usr/local/Cellar/qt/5.15.2/lib/cmake;/usr/local/Cellar/nglib/v6.2.2007/Contents/Resources/CMake" -DBUILD_QT5=ON -DUSE_PYTHON3=1 -DPYTHON_EXECUTABLE=/usr/local/bin/python3 -DBUILD_FEM_NETGEN=1 ../freecad

If you have latest brew file installed you shall get something like

into ./src/3rdParty/salomesmesh/CMakeFiles/MEFISTO2.dir/link.txt within your build directory

/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++   -Wall -Wextra -Wpedantic -Wno-write-strings  -Wno-undefined-var-template -Wno-self-assign -Wno-sign-compare -Wno-logical-op-parentheses -Wno-reorder -Wno-switch -Wno-switch-enum -Wno-unknown-pragmas -Wno-unused-variable -Wno-unused-private-field -Wno-unused-function -Wno-sometimes-uninitialized -Wno-overloaded-virtual -Wno-dynamic-class-memaccess -Wno-comment -Wno-unused-parameter -Wno-extra-semi -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.1.sdk -mmacosx-version-min=11.0 -dynamiclib -Wl,-headerpad_max_install_names  -Wl,-undefined,error -o ../../../lib/libMEFISTO2.dylib -install_name @rpath/libMEFISTO2.dylib CMakeFiles/MEFISTO2.dir/src/MEFISTO2/aptrte.cpp.o CMakeFiles/MEFISTO2.dir/src/MEFISTO2/trte.c.o -F/Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/System/Library/Frameworks   -L/usr/local/Cellar/open-mpi/4.0.5/lib  -L/usr/local/opt/libevent/lib  -L/usr/local/lib  -Wl,-rpath,/usr/local/Cellar/open-mpi/4.0.5/lib -Wl,-rpath,/usr/local/opt/libevent/lib -Wl,-rpath,/usr/local/lib -Wl,-rpath,/usr/local/Cellar/nglib/v6.2.2007/Contents/MacOS -lmpi /usr/local/Cellar/nglib/v6.2.2007/Contents/MacOS/libnglib.dylib /usr/local/lib/libTKSTL.7.5.0.dylib /usr/local/lib/libTKBin.7.5.0.dylib /usr/local/lib/libTKOffset.7.5.0.dylib /usr/local/lib/libTKFeat.7.5.0.dylib /usr/local/lib/libTKXDESTEP.7.5.0.dylib /usr/local/lib/libTKXDEIGES.7.5.0.dylib /usr/local/lib/libTKMeshVS.7.5.0.dylib /usr/local/Cellar/nglib/v6.2.2007/Contents/MacOS/libinterface.dylib /usr/local/Cellar/nglib/v6.2.2007/Contents/MacOS/libgeom2d.so /usr/local/Cellar/nglib/v6.2.2007/Contents/MacOS/libcsg.so /usr/local/Cellar/nglib/v6.2.2007/Contents/MacOS/libstl.dylib /usr/local/Cellar/nglib/v6.2.2007/Contents/MacOS/libmesh.so /usr/local/Cellar/nglib/v6.2.2007/Contents/MacOS/libvisual.dylib /usr/local/Cellar/nglib/v6.2.2007/Contents/MacOS/libocc.dylib /usr/local/Cellar/nglib/v6.2.2007/Contents/MacOS/libngcore.dylib /usr/local/lib/libTKBinL.7.5.0.dylib /usr/local/lib/libTKFillet.7.5.0.dylib /usr/local/lib/libTKSTEP.7.5.0.dylib /usr/local/lib/libTKSTEP209.7.5.0.dylib /usr/local/lib/libTKSTEPAttr.7.5.0.dylib /usr/local/lib/libTKSTEPBase.7.5.0.dylib /usr/local/lib/libTKIGES.7.5.0.dylib /usr/local/lib/libTKBool.7.5.0.dylib /usr/local/lib/libTKXSBase.7.5.0.dylib /usr/local/lib/libTKXCAF.7.5.0.dylib /usr/local/lib/libTKVCAF.7.5.0.dylib /usr/local/lib/libTKCAF.7.5.0.dylib /usr/local/lib/libTKBO.7.5.0.dylib /usr/local/lib/libTKPrim.7.5.0.dylib /usr/local/lib/libTKLCAF.7.5.0.dylib /usr/local/lib/libTKCDF.7.5.0.dylib /usr/local/lib/libTKV3d.7.5.0.dylib /usr/local/lib/libTKMesh.7.5.0.dylib /usr/local/lib/libTKShHealing.7.5.0.dylib /usr/local/lib/libTKHLR.7.5.0.dylib /usr/local/lib/libTKTopAlgo.7.5.0.dylib /usr/local/lib/libTKGeomAlgo.7.5.0.dylib /usr/local/lib/libTKBRep.7.5.0.dylib /usr/local/lib/libTKGeomBase.7.5.0.dylib /usr/local/lib/libTKG3d.7.5.0.dylib /usr/local/lib/libTKG2d.7.5.0.dylib /usr/local/lib/libTKService.7.5.0.dylib /usr/local/lib/libTKMath.7.5.0.dylib /usr/local/lib/libTKernel.7.5.0.dylib /usr/local/opt/tbb/lib/libtbb.dylib /usr/local/opt/tbb/lib/libtbbmalloc.dylib -lobjc -framework AppKit -framework IOKit /usr/local/opt/freeimage/lib/libfreeimage.dylib -framework OpenGL /usr/local/opt/freetype/lib/libfreetype.dylib

zenz, 2020-12-23:
> I might be wrong but I think you still have a dependencies which is not properly updated with homebrew. I just checked on my system and the MEFISTO library (into link.txt) is dependent to [...]

Still no luck. same error.

vejmarie, 2020-12-23:
Can you try the stuff I posted just up ? I like to get the output from into ./src/3rdParty/salomesmesh/CMakeFiles/MEFISTO2.dir/link.txt within your build directory

As well as the output of
pkgutil --pkg-info=com.apple.pkg.CLTools_Executables | awk '/version:/ {print $2}'
on your system ;)

zenz, 2020-12-23:
Here's the information after running your requested cmake command

/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++   -Wall -Wextra -Wpedantic -Wno-write-strings  -Wno-undefined-var-template -Wno-self-assign -Wno-sign-compare -Wno-logical-op-parentheses -Wno-reorder -Wno-switch -Wno-switch-enum -Wno-unknown-pragmas -Wno-unused-variable -Wno-unused-private-field -Wno-unused-function -Wno-sometimes-uninitialized -Wno-overloaded-virtual -Wno-dynamic-class-memaccess -Wno-comment -Wno-unused-parameter -Wno-extra-semi -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.1.sdk -dynamiclib -Wl,-headerpad_max_install_names -L/usr/local/opt/sqlite/lib -Wl,-undefined,dynamic_lookup -o ../../../lib/libMEFISTO2.dylib -install_name @rpath/libMEFISTO2.dylib CMakeFiles/MEFISTO2.dir/src/MEFISTO2/aptrte.cpp.o CMakeFiles/MEFISTO2.dir/src/MEFISTO2/trte.c.o   -L/usr/local/Cellar/open-mpi/4.0.5/lib  -L/usr/local/opt/libevent/lib  -L/usr/local/lib  -Wl,-rpath,/usr/local/Cellar/open-mpi/4.0.5/lib -Wl,-rpath,/usr/local/opt/libevent/lib -Wl,-rpath,/usr/local/lib -Wl,-rpath,/usr/local/Cellar/nglib/v6.2.2007/Contents/MacOS -lmpi /usr/local/Cellar/nglib/v6.2.2007/Contents/MacOS/libnglib.dylib /usr/local/lib/libTKSTL.7.5.0.dylib /usr/local/lib/libTKOffset.7.5.0.dylib /usr/local/lib/libTKFeat.7.5.0.dylib /usr/local/lib/libTKBin.7.5.0.dylib /usr/local/lib/libTKBinL.7.5.0.dylib /usr/local/lib/libTKXDESTEP.7.5.0.dylib /usr/local/lib/libTKXDEIGES.7.5.0.dylib /usr/local/lib/libTKMeshVS.7.5.0.dylib /usr/local/lib/libTKRWMesh.7.5.0.dylib /usr/local/Cellar/nglib/v6.2.2007/Contents/MacOS/libinterface.dylib /usr/local/Cellar/nglib/v6.2.2007/Contents/MacOS/libgeom2d.so /usr/local/Cellar/nglib/v6.2.2007/Contents/MacOS/libcsg.so /usr/local/Cellar/nglib/v6.2.2007/Contents/MacOS/libstl.dylib /usr/local/Cellar/nglib/v6.2.2007/Contents/MacOS/libmesh.so /usr/local/Cellar/nglib/v6.2.2007/Contents/MacOS/libvisual.dylib /usr/local/Cellar/nglib/v6.2.2007/Contents/MacOS/libocc.dylib /usr/local/Cellar/nglib/v6.2.2007/Contents/MacOS/libngcore.dylib /usr/local/lib/libTKFillet.7.5.0.dylib /usr/local/lib/libTKSTEP.7.5.0.dylib /usr/local/lib/libTKSTEPAttr.7.5.0.dylib /usr/local/lib/libTKSTEP209.7.5.0.dylib /usr/local/lib/libTKSTEPBase.7.5.0.dylib /usr/local/lib/libTKIGES.7.5.0.dylib /usr/local/lib/libTKBool.7.5.0.dylib /usr/local/lib/libTKXSBase.7.5.0.dylib /usr/local/lib/libTKXCAF.7.5.0.dylib /usr/local/lib/libTKVCAF.7.5.0.dylib /usr/local/lib/libTKCAF.7.5.0.dylib /usr/local/lib/libTKBO.7.5.0.dylib /usr/local/lib/libTKPrim.7.5.0.dylib /usr/local/lib/libTKLCAF.7.5.0.dylib /usr/local/lib/libTKCDF.7.5.0.dylib /usr/local/lib/libTKV3d.7.5.0.dylib /usr/local/lib/libTKMesh.7.5.0.dylib /usr/local/lib/libTKShHealing.7.5.0.dylib /usr/local/lib/libTKHLR.7.5.0.dylib /usr/local/lib/libTKTopAlgo.7.5.0.dylib /usr/local/lib/libTKGeomAlgo.7.5.0.dylib /usr/local/lib/libTKBRep.7.5.0.dylib /usr/local/lib/libTKGeomBase.7.5.0.dylib /usr/local/lib/libTKG3d.7.5.0.dylib /usr/local/lib/libTKG2d.7.5.0.dylib /usr/local/lib/libTKService.7.5.0.dylib /usr/local/lib/libTKMath.7.5.0.dylib /usr/local/lib/libTKernel.7.5.0.dylib -lobjc /Library/Developer/CommandLineTools/SDKs/MacOSX11.0.sdk/System/Library/Frameworks/AppKit.framework /Library/Developer/CommandLineTools/SDKs/MacOSX11.0.sdk/System/Library/Frameworks/IOKit.framework /usr/local/opt/freeimage/lib/libfreeimage.dylib /usr/local/opt/tbb/lib/libtbb.dylib /usr/local/opt/tbb/lib/libtbbmalloc.dylib /Library/Developer/CommandLineTools/SDKs/MacOSX11.0.sdk/System/Library/Frameworks/OpenGL.framework /usr/local/opt/freetype/lib/libfreetype.dylib 

and the result of pkgutil command:
12.3.0.0.1.1607026830

zenz, 2020-12-23:
And this is the brew --env output. every thing seems right, but just link to wrong sdk, so weird.

brew --env
HOMEBREW_CC: clang
HOMEBREW_CXX: clang++
MAKEFLAGS: -j12
CMAKE_PREFIX_PATH: /usr/local
CMAKE_INCLUDE_PATH: /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/OpenGL.framework/Versions/Current/Headers
CMAKE_LIBRARY_PATH: /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/OpenGL.framework/Versions/Current/Libraries
PKG_CONFIG_LIBDIR: /usr/lib/pkgconfig:/usr/local/Homebrew/Library/Homebrew/os/mac/pkgconfig/11.1
HOMEBREW_GIT: git
HOMEBREW_SDKROOT: /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk
ACLOCAL_PATH: /usr/local/share/aclocal
PATH: /usr/local/Homebrew/Library/Homebrew/shims/mac/super:/usr/bin:/bin:/usr/sbin:/sbin

zenz, 2020-12-23:
The same situation with this forum topic?
https://forum.freecadweb.org/viewtopic.php?f=4&t=50843&start=10

zenz, 2020-12-23:
I found that this problem was caused by
/usr/local/Cellar/vtk@8.2/8.2.0_1/lib/cmake/vtk-8.2/VTKTargets.cmake

It has specificed sdk with MacOSX11.0.sdk.

vejmarie, 2020-12-23:
ok, it looks like that your VTK has not been updated accordingly. Did you recompiled it your self ? That is weird. I am thinking at moving back more dependencies into FreeCAD formulas to avoid such issues and being more "independant" from the super fast speed that homebrew maintainer are moving. Your thoughts ?

zenz, 2020-12-23:
> ok, it looks like that your VTK has not been updated accordingly. Did you recompiled it your self ? That is weird. I am thinking at moving back more dependencies into FreeCAD formulas to avoid such [...]

I didn't recompile those dependences, I remove vtk@8.2 and redo 
brew update
brew upgrade
brew install --HEAD freecad --with-macos-app -v, the newly install vtk@8.2 have same bindings to MacOSX11.0.sdk.

But modify the cmake files here didn't fix this issue.

vejmarie, 2020-12-23:
Ok, let me have a look this morning. In fact, I still believe the issue is coming from med-file which is a dependency of vtk

zenz, 2020-12-23:
/usr/local/Cellar/med-file/4.0.0/share/cmake/med-fichier-4.0.0/MEDFileTargets-release.cmake:  IMPORTED_LINK_INTERFACE_LIBRARIES_RELEASE "/usr/local/opt/hdf5@1.10/lib/libhdf5.dylib;/usr/local/lib/libsz.dylib;/Library/Developer/CommandLineTools/SDKs/MacOSX11.1.sdk/usr/lib/libz.tbd;/Library/Developer/CommandLineTools/SDKs/MacOSX11.1.sdk/usr/lib/libdl.tbd;/Library/Developer/CommandLineTools/SDKs/MacOSX11.1.sdk/usr/lib/libm.tbd"

zenz, 2020-12-23:
cd /usr/local/Cellar
find . -type f|grep "\.cmake" | xargs grep "11\.0\.sdk"

./llvm/11.0.0/lib/cmake/llvm/LLVMExports.cmake:  INTERFACE_LINK_LIBRARIES "m;ZLIB::ZLIB;/Library/Developer/CommandLineTools/SDKs/MacOSX11.0.sdk/usr/lib/libcurses.tbd;LLVMDemangle"
./llvm/11.0.0/Toolchains/LLVM11.0.0.xctoolchain/usr/lib/cmake/llvm/LLVMExports.cmake:  INTERFACE_LINK_LIBRARIES "m;ZLIB::ZLIB;/Library/Developer/CommandLineTools/SDKs/MacOSX11.0.sdk/usr/lib/libcurses.tbd;LLVMDemangle"
./xerces-c/3.2.3/lib/cmake/XercesC/XercesCConfigInternal.cmake:  INTERFACE_LINK_LIBRARIES "/Library/Developer/CommandLineTools/SDKs/MacOSX11.0.sdk/usr/lib/libcurl.tbd;/Library/Developer/CommandLineTools/SDKs/MacOSX11.0.sdk/System/Library/Frameworks/CoreServices.framework;Threads::Threads"
./opencascade/7.5.0/lib/cmake/opencascade/OpenCASCADEDrawTargets.cmake:  INTERFACE_LINK_LIBRARIES "TKernel;TKG2d;TKGeomBase;TKG3d;TKMath;TKBRep;TKGeomAlgo;TKTopAlgo;TKShHealing;TKMesh;TKService;TKHLR;/Library/Developer/CommandLineTools/SDKs/MacOSX11.0.sdk/usr/lib/libtcl.tbd;/Library/Developer/CommandLineTools/SDKs/MacOSX11.0.sdk/usr/lib/libtk.tbd;/usr/local/opt/tbb/lib/libtbb.dylib;/usr/local/opt/tbb/lib/libtbbmalloc.dylib;objc;/Library/Developer/CommandLineTools/SDKs/MacOSX11.0.sdk/System/Library/Frameworks/AppKit.framework;/Library/Developer/CommandLineTools/SDKs/MacOSX11.0.sdk/System/Library/Frameworks/IOKit.framework"
./opencascade/7.5.0/lib/cmake/opencascade/OpenCASCADEDrawTargets.cmake:  INTERFACE_LINK_LIBRARIES "TKGeomBase;TKFillet;TKBRep;TKTopAlgo;TKHLR;TKernel;TKMath;TKService;TKShHealing;TKBool;TKPrim;TKGeomAlgo;TKG2d;TKTopTest;TKG3d;TKOffset;TKMesh;TKV3d;TKDraw;TKOpenGl;/Library/Developer/CommandLineTools/SDKs/MacOSX11.0.sdk/usr/lib/libtcl.tbd;/Library/Developer/CommandLineTools/SDKs/MacOSX11.0.sdk/usr/lib/libtk.tbd;/Library/Developer/CommandLineTools/SDKs/MacOSX11.0.sdk/System/Library/Frameworks/OpenGL.framework;objc;/Library/Developer/CommandLineTools/SDKs/MacOSX11.0.sdk/System/Library/Frameworks/AppKit.framework;/Library/Developer/CommandLineTools/SDKs/MacOSX11.0.sdk/System/Library/Frameworks/IOKit.framework;/usr/local/opt/freetype/lib/libfreetype.dylib"
./opencascade/7.5.0/lib/cmake/opencascade/OpenCASCADEVisualizationTargets.cmake:  INTERFACE_LINK_LIBRARIES "TKernel;TKMath;/Library/Developer/CommandLineTools/SDKs/MacOSX11.0.sdk/System/Library/Frameworks/OpenGL.framework;objc;/Library/Developer/CommandLineTools/SDKs/MacOSX11.0.sdk/System/Library/Frameworks/AppKit.framework;/Library/Developer/CommandLineTools/SDKs/MacOSX11.0.sdk/System/Library/Frameworks/IOKit.framework;/usr/local/opt/freeimage/lib/libfreeimage.dylib;/usr/local/opt/freetype/lib/libfreetype.dylib"
./opencascade/7.5.0/lib/cmake/opencascade/OpenCASCADEVisualizationTargets.cmake:  INTERFACE_LINK_LIBRARIES "TKBRep;TKMath;TKernel;TKService;TKShHealing;TKTopAlgo;TKG2d;TKG3d;TKGeomBase;TKMesh;TKGeomAlgo;TKHLR;/Library/Developer/CommandLineTools/SDKs/MacOSX11.0.sdk/System/Library/Frameworks/OpenGL.framework;/usr/local/opt/freetype/lib/libfreetype.dylib;/usr/local/opt/tbb/lib/libtbb.dylib;/usr/local/opt/tbb/lib/libtbbmalloc.dylib"
./opencascade/7.5.0/lib/cmake/opencascade/OpenCASCADEVisualizationTargets.cmake:  INTERFACE_LINK_LIBRARIES "TKernel;TKService;TKMath;/usr/local/opt/tbb/lib/libtbb.dylib;/usr/local/opt/tbb/lib/libtbbmalloc.dylib;/usr/local/opt/freetype/lib/libfreetype.dylib;/Library/Developer/CommandLineTools/SDKs/MacOSX11.0.sdk/System/Library/Frameworks/OpenGL.framework;/Library/Developer/CommandLineTools/SDKs/MacOSX11.0.sdk/System/Library/Frameworks/AppKit.framework;/Library/Developer/CommandLineTools/SDKs/MacOSX11.0.sdk/System/Library/Frameworks/IOKit.framework;objc"
./netcdf/4.7.4_1/lib/cmake/netCDF/netCDFTargets.cmake:  INTERFACE_LINK_LIBRARIES "/usr/local/lib/libhdf5_hl.dylib;/usr/local/lib/libhdf5.dylib;/usr/local/lib/libsz.dylib;/Library/Developer/CommandLineTools/SDKs/MacOSX11.0.sdk/usr/lib/libz.tbd;/Library/Developer/CommandLineTools/SDKs/MacOSX11.0.sdk/usr/lib/libdl.tbd;/Library/Developer/CommandLineTools/SDKs/MacOSX11.0.sdk/usr/lib/libm.tbd;/Library/Developer/CommandLineTools/SDKs/MacOSX11.0.sdk/usr/lib/libcurl.tbd"

Any relative?

vejmarie, 2020-12-23:
mine is giving

./llvm/11.0.0/lib/cmake/llvm/LLVMExports.cmake:  INTERFACE_LINK_LIBRARIES "m;ZLIB::ZLIB;/Library/Developer/CommandLineTools/SDKs/MacOSX11.0.sdk/usr/lib/libcurses.tbd;LLVMDemangle"
./llvm/11.0.0/Toolchains/LLVM11.0.0.xctoolchain/usr/lib/cmake/llvm/LLVMExports.cmake:  INTERFACE_LINK_LIBRARIES "m;ZLIB::ZLIB;/Library/Developer/CommandLineTools/SDKs/MacOSX11.0.sdk/usr/lib/libcurses.tbd;LLVMDemangle"
./xerces-c/3.2.3/lib/cmake/XercesC/XercesCConfigInternal.cmake:  INTERFACE_LINK_LIBRARIES "/Library/Developer/CommandLineTools/SDKs/MacOSX11.0.sdk/usr/lib/libcurl.tbd;/Library/Developer/CommandLineTools/SDKs/MacOSX11.0.sdk/System/Library/Frameworks/CoreServices.framework;Threads::Threads"

Looks like the issue is coming from your opencascade installation in the end. Can you try to remove it and re-install it ?

zenz, 2020-12-23:
no luck! :(
I will use symbol link first to make a usable version to do some job.

vejmarie, 2020-12-23:
what do you mean by no luck ? You still get the wrong toolchain into occt build ? I must admit that I do not remember if I installed mine by recompiling it or just loading it from an existing bottle.

zenz, 2020-12-23:
@vejmarie 

reinstall opencascade didn't solve the issue.

But after I modify all cname files contains specified SDK to MacOSX.sdk, it seems the compilation goes well.
I will confirm it once it's finished.

vejmarie, 2020-12-23:
We need to identify which one is creating the issue to correct it ;). I will have a look on my system today if the opencascade package might be the one broken

zenz, 2020-12-23:
I confirm that opencascade caused the issue.

I will suggest to use /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk as sdk root, since it's always point to latest sdk installed. This way can avoid a lot of problems while after system update.

vejmarie, 2020-12-23:
I agree. The challenge here is that opencascade formula is not maintained by us. Their is a ton of dependencies to take care of, and it could be just tricky to maintain them. I am working at finding the best option with the homebrew people

ipatch, 2020-12-23:
@vejmarie you could create another formula within this repo for opencascade and just set it to a specific version ie. `opencascade@7.5` and have freecad depend on that version of opencascade to slow down the frequent updates that rolling in from upstream. kind of like how there are multiple versions of vtk

bblacey, 2020-12-23:
@vejmarie - or perhaps [brew pin](https://docs.brew.sh/FAQ#how-do-i-stop-certain-formulae-from-being-updated)?  

Managing Homebrew dependencies can be like chasing a stone rolling down a hill and why Conda has merit for FreeCAD.  While the Homebrew team supports personal repos, they have really fallen short in auto-bottling when dependencies change - auto-bottling only happens on Home-brew core, or at least the last time that I looked.  Some [creative souls](https://gist.github.com/maelvls/068af21911c7debc4655cdaa41bbf092#what-to-do-when-some-dependencies-are-updated) have tried to automate bottling on their repos that might help here.

ipatch, 2020-12-23:
@bblacey 

the `brew pin` works on a user level not a upstream system/repo level, so every user would have to manual _pin_ formulas, which i have done in the past, and not a huge fan TBH creating formulas with specific `@` version as i mentioned is way better managing a formula that would be built from source.

vejmarie, 2020-12-23:
@bblacey welcome back !. I looked at brew pin, but the challenge is more how to unpin when you want to upgrade ;). I think the challenge with homebrew is the same than on many linux distro which are packaging independently many libraries, that is also why I created the snap in the end. I like homebrew because it is a good balance for compiling on Mac. Let's keep thinking. But I agree the speed at which they change each package is insane, and when you think that boost has a release every 3 months !!! and we are dependent of it ... this is super challenging. If I had my macpro with me, I would probably got many dependencies out, and recompile everything and create independant bottles which are probably the best option. This can be done with brew extract --version=7.5.0 opencascade freecad/freecad for opencascade as an example. Then we can update the formula whenever we want.

I tried to do it, but qt compilation failed on my Mac for an unknown reason.

vejmarie, 2020-12-23:
ok I found the issue to recompile qt. Roughly ninja doesn't support @ symbol into path name and the forked formula had a format of qt@5.15.2. I replaced it to qt_5.15.2 which means that we probably can recompile most of our dependencies and becomes independant from the high release speed of homebrew while keeping the same infrastructure

vejmarie, 2020-12-24:
I started to look at what dependencies we wanted to be independent from homebrew core. I believe that qt, boost, opencascade, python are key elements to get freecad working into a stable environment (boost for sure, as it is upgraded every 3 months). What are your thoughts ? The challenge if we do that is that other applications which might be dependent from these software might become broken, as symlinks are created into /usr/local to the relevant Cellar directory. What do you think about 'forking' such dependencies into our freecad formulas repo, and being sure that freecad stay functionnal if homebrew is getting upgraded ? I have started to implement such approach to check if it could work or not. I believe it will.

zenz, 2020-12-24:
I don’t know much about how homebrew works. Is it possible to modify some files during compiling and then recover them later? As I checked yesterday. Only some cmake files should be changed can fix the compiling issue. So I will prefer keep them there not fork another copy. or completely modify them.

ipatch, 2020-12-25:
@zenz 

```
man brew
```

or

```
brew help install
```

in short yes, you can keep the temporary files brew uses to build the formula.

zenz, 2020-12-25:
@vejmarie 
I do a re-compiling today, and confirmed that the cmake files should be modified is in vtk@8.2 and opencascade packages.

vejmarie, 2020-12-25:
I know. The issue is not really coming from these packages but more the XCode version which has been used to recompile them. I will be publishing later some update to correct all of these, but my Mac is slow, and some of my compilations failed. Hopefully during next week everything shall be back on track.

vejmarie, 2020-12-31:
Ok, I am a little bit late, but as to fix all of these dependencies issues which are evolving fast with homebrew, I can propose you to maintain these fomula's into freecad tap. It shall shield us against frenetic upgrades (which might be good to other project, but not always to us, as some of our dependencies have unstable API). I have create all of these formulas and they work on my mac on bigsur. I have generated the bottles for bigsur also and not for catalina (I could make it but it is another pass of compilation). If we missed one we still can add it later on. The regular homebrew formula can be installed at the same time, allowing other apps to work also !

Your thoughts before upstreaming ?

boost-python3@1.75.0.rb
pyside2.rb
pyside2-tools.rb
opencamlib.rb
boost@1.75.0.rb
freecad.rb
boost-python@1.75.0.rb
coin@4.0.0.rb
shiboken2.rb
matplotlib.rb
pivy.rb
nglib.rb
sip@4.19.24.rb
python3.9.rb
numpy@1.19.4.rb
tbb@2020_u3.rb
opencascade@7.5.0.rb
cython@0.29.21.rb
pyqt@5.15.2.rb
qt5152.rb
med-file.rb
swig@4.0.2.rb
vtk@8.2.0.rb
icu4c@67.1.rb

zenz, 2020-12-31:
It should be OK.

vejmarie, 2021-01-01:
Ok, I will publish during the week-end.

zenz, 2021-01-01:
@vejmarie thank you for your hard work. Happy New Year!

vejmarie, 2021-01-01:
Just updated it. Hopefully I did not broke everything.

zenz, 2021-01-04:
@vejmarie 

reported big_sur.bottle CRC not correct.

```
% brew install freecad --with-macos-app
Warning: Treating freecad as a formula. For the cask, use homebrew/cask/freecad
==> Installing freecad from freecad/freecad
==> Downloading https://dl.bintray.com/vejmarie/freecad/icu4c%4067.1-67.1.big_sur.bottle.tar.gz
==> Downloading from https://d29vzk4ow07wi7.cloudfront.net/550f2586ae316232721903c627a53932a9a2f2032cade00ed31ec5dab3e27
######################################################################## 100.0%
Error: SHA256 mismatch
Expected: 9a8c506433feac36aa161aaf6a3e579d83293fcf363781e3cda11611c649efb8
  Actual: 550f2586ae316232721903c627a53932a9a2f2032cade00ed31ec5dab3e2727e
    File: /Users/simon/Library/Caches/Homebrew/downloads/e392728adfed46db87163a6d0a4da2896e915bae4d5c028dabba8d6c78cae3be--icu4c@67.1-67.1.big_sur.bottle.tar.gz
To retry an incomplete download, remove the file above.
Warning: Bottle installation failed: building from source.
==> Downloading https://github.com/unicode-org/icu/commit/715d254a02b0b22681cb6f861b0921ae668fa7d6.patch?full_index=1
######################################################################## 100.0%
==> Downloading https://github.com/unicode-org/icu/releases/download/release-67-1/icu4c-67_1-src.tgz
==> Downloading from https://github-production-release-asset-2e65be.s3.amazonaws.com/49244766/38bf9400-849e-11ea-9b02-c3
######################################################################## 100.0%
==> Downloading https://dl.bintray.com/vejmarie/freecad/boost%401.75.0-1.75.0.big_sur.bottle.tar.gz
==> Downloading from https://d29vzk4ow07wi7.cloudfront.net/99aa624e3a19ba01458f9e0be8dcdcfb8df90feabc4728b188682d9174b05
######################################################################## 100.0%
Error: SHA256 mismatch
Expected: 00b9b53fab0a0451e45ed4603af55ca81ab215fd6fedbd01a92701a8cab45c37
  Actual: 99aa624e3a19ba01458f9e0be8dcdcfb8df90feabc4728b188682d9174b05f09
    File: /Users/simon/Library/Caches/Homebrew/downloads/77be0111d5a380ebd2fc1778e4be89a0ef120b1689403926fee643355736afa1--boost@1.75.0-1.75.0.big_sur.bottle.tar.gz
To retry an incomplete download, remove the file above.
Warning: Bottle installation failed: building from source.
==> Downloading https://github.com/boostorg/build/commit/456be0b7ecca065fbccf380c2f51e0985e608ba0.patch?full_index=1
######################################################################## 100.0%
==> Downloading https://dl.bintray.com/boostorg/release/1.75.0/source/boost_1_75_0.tar.bz2
```

vejmarie, 2021-01-04:
You are right. their has been an issue at uploading the new sha256 signature for bigsur. It shall be fixed now

zenz, 2021-01-04:
@vejmarie 
updated, but got this error
```
Last 15 lines from /Users/simon/Library/Logs/Homebrew/freecad/01.cmake:
CMake Error at /usr/local/opt/shiboken2/lib/cmake/Shiboken2-5.15.2/shiboken_helpers.cmake:461 (message):
  The detected Python major version is not compatible with the Python major
  version which was used when Shiboken was built.

  Built with: '3.9' Detected: '.'
Call Stack (most recent call first):
  /usr/local/opt/shiboken2/lib/cmake/Shiboken2-5.15.2/Shiboken2Config.cpython-39-darwin.cmake:45 (shiboken_check_if_built_and_target_python_are_compatible)
  /usr/local/opt/shiboken2/lib/cmake/Shiboken2-5.15.2/Shiboken2Config.cmake:5 (include)
  cMake/FreeCAD_Helpers/SetupShibokenAndPyside.cmake:25 (find_package)
  CMakeLists.txt:70 (SetupShibokenAndPyside)
```

vejmarie, 2021-01-04:
Are you recompiling ? If yes, as you need to post install python3.9 or setup a few environment variables to point to the python directory.

vejmarie, 2021-01-04:
Can you set these before relaunching the brew command. If it works I can update the formula
export PATH="/usr/local/opt/python3.9/bin:$PATH"
export PKG_CONFIG_PATH="/usr/local/opt/python3.9/lib/pkgconfig"

zenz, 2021-01-04:
I set up environment variables as you point out. but got same compile errors.

If I just install your big_sur copy with brew install freecad.
then run freecad from command line. I got this:

`dyld: Library not loaded: /usr/local/opt/gettext/lib/libintl.8.dylib
  Referenced from: /usr/local/opt/python3.9/Frameworks/Python.framework/Versions/3.9/Python
  Reason: image not found
zsh: abort      freecad`

vejmarie, 2021-01-04:
ok I will add the dependencies for python3.9 to gettext. Thanks for the report.

vejmarie, 2021-01-04:
I am working on fixing the shiboken build issue. I am trying to reproduce it on my system first

[6 more replies]

closed 2021-04-06 by vejmarie