FreeCAD/homebrew-freecad #142: building freecad with `--with-macos-app`

ipatch, 2020-12-24:
is the `--with-macos-app` flag / argument suppose to work when building freecad form _head_ ?

i've been running into `make install` issues when using the flag.

```
CMake Error at src/MacAppBundle/cmake_install.cmake:55 (file):
  file INSTALL cannot find
  "/tmp/freecad-20201223-55124-9c422j/src/MacAppBundle/import site": No such
  file or directory.
Call Stack (most recent call first):
  src/cmake_install.cmake:51 (include)
  cmake_install.cmake:68 (include)

make: *** [install] Error 1
/usr/local/Homebrew/Library/Homebrew/shims/scm/git --version
/usr/bin/curl --version
/usr/libexec/java_home --xml --failfast
/usr/libexec/java_home --xml --failfast
git --git-dir /Users/capin/Library/Caches/Homebrew/freecad--git/.git rev-parse --short=7 HEAD
```

last little bit of the error message, let me know if i need to post anything else.

vejmarie, 2020-12-24:
It is supposed to work, even if I never tested it up to now. Did you checked while being detached from head ? I can have a look to it

ipatch, 2020-12-24:
no so much looking around all i did was run above said command but the installation process is failing, i did notice there was a _with-packaging-utils_ a while back that didn't work either. if you can do you mind trying the flag `--with-macos-app` to see if it works on your machine?

vejmarie, 2020-12-24:
I will for sure as soon my qt build is finished ;) (probably not before tomorrow, as I do have only my dual core macbookpro with me :( )

ipatch, 2020-12-24:
no worries, i'll keep investigating to see if i can figure out what's going on. i know qt can take a while to build from source. i believe it requires building chromium which can take like forever.

vejmarie, 2020-12-24:
yep it is currently building chromium ... And my bigsur setup is on a USB drive :( (at least USB 3, but still not the fastest stuff). I kept Catalina on the internal nvme was thinking at dropping it.

ipatch, 2020-12-24:
you run big sur in a vm FWIW, that's what i ended up doing it runs rather slow, but i can start the vm headless on my box and that's how i was testing things out with big sur other day, far from perfect but it's better than nothing. and yeah it'll take ~ 20+ hours to build chromium & qt

zenz, 2020-12-24:
I worked, I just use it to compile my own copy last night. using --with-macos-app option.

vejmarie, 2020-12-24:
> I worked, I just use it to compile my own copy last night. using --with-macos-app option.

At least something which works ;)

zenz, 2020-12-24:
--with-macos-app is just adds this line
`args << "-DFREECAD_CREATE_MAC_APP=1" if build.with? "macos-app"`
which follow the compile document from freecad wifi. It simply works.

And I confirmed --with-packaging-utils doesn't work. since it's using some npm package not exists.
`system "node", "install", "-g", "app_dmg" if build.with? "packaging-utils"`

and also the command to install npm package is not correct, it should be npm install -g npm_package.
I didn't change that part when I submit the pull request.

ithinuel, 2020-12-24:
I have this issue as well and it seems like some dependencies are trying to reach AppKit.Framework from MacOSX11.0.sdk while only MacOSX11.1.sdk is available on my machine.

(I'm new to Mac's universe so I have no idea with this AppKit.framework nor the sdk mean).

vejmarie, 2020-12-24:
It will be fixed before new-year ;). I am hoping to have something during the week-end but it requires a lot of recompilation with XCode 12.3

ipatch, 2020-12-24:
@ithinuel 

a hacky work around would be to create a symlink from your macosx11.1.sdk -> macos11.0.sdk

might you get a little further along in the process, not an elegant solution by any means but could work in a pinch.

ithinuel, 2020-12-24:
@ipatch Thanks. I can still install it without `macos-app` and start it from the terminal.
TBF I'm not even sure what `macos-app`, I assumed it's to have FreeCAD nicely integrated in the App folder and be able to start it from the dock.

ipatch, 2020-12-24:
@ithinuel 

yes i believe that is what it is intended to do. it packages up the install from the homebrew cellar into a `FreeCAD.app` bundle which should be apple run from double clicking the `FreeCAD.app` icon using finder or even as you suggested launching from the dock.

---

out of curiosity what happens when you launch `FreeCAD` from your terminal in an arbitrary directory ie say your `$HOME` dir? do you get issues related to not finding _pyside_ ?

i was getting issues related to that yesterday, but then `cd` into the directory that contained the `FreeCAD` cmd and everything seemed work as it should, ie. opening/launching workbenches and what not.

zenz, 2020-12-24:
> @ipatch Thanks. I can still install it without `macos-app` and start it from the terminal. So that's not a hard blocker for me.
> TBF I'm not even sure what `macos-app`, I assumed it's to have FreeCAD nicely integrated in the App folder and be able to start it from the dock.

with-macos-app only generate a Mac app bundle currently . You need to make a symbol link to /Applications folder to use FreeCAD.app

ithinuel, 2020-12-25:
For what it's worth, I had a similar issue while installing hidapi's python module with `pip install hidapi==0.7.99.post21`.

Creating a simlink from `/Library/Developer/CommandLineTools/SDKs/MacOSX11.0.sdk`to `/Library/Developer/CommandLineTools/SDKs/MacOSX11.1.sdk` solved the issue for both hidapi 0.7.99.post21 and freecad with-macos.

Note that `pip install hidapi==0.10.1` does not have the issue, so it must have been fixed there.

@ipatch when installing FreeCAD with `brew install freecad`, I can run FreeCad from any folder on the shell. When installing FreeCAD with `--with-macos`, FreeCAD is not found in my path. However, symlinking `/usr/local/opt/freecad/FreeCAD.app` to `/Applications/FreeCAD.app` shows FreeCAD in Finder and I'm able to start it from there.

I'm using the MacBookAir (M1) in case that can help narrow down the issue.

ipatch, 2020-12-25:
@ithinuel 

that's really interesting i compile freecad last night `FreeCAD 0.19, Libs: 0.19R23529 (Git)` using brew turned **off** the `--with-macos` build flag and everything installed as should, and am able to launch FreeCAD from an arbitrary directory, and haven't noticed any major breaking changes such as the issues i was coming across with, ie. `PySide` not being found, (more than likely due to path related issues).

that's interesting that the symlink trick/hack resolves the issue. makes me think somewhere there are some hardcoded paths in cmake related files in freecad or a dependency of freecad, or i could totally be off base.  i guess what's even more interesting is i'm using macos mojave 10.14, so not sure if the symlink would resolve my issue. either way i'll give it recompile sometime today and see what i can figure out. i think enabling the `--debug` flag while i'm installing may speed things up a little as it will drop me in an interactive prompt while i can tweak settings and see what is going without have to restart the whole process from scratch. as it takes ~ 50 to 55 minutes to compile freecad on my box from start to finish 😞

zenz, 2020-12-26:
@ithinuel @ipatch 
I think while building Mac Bundle, it needs more dynlib to fit for standard. if you have time, you can use otool -l to inspect what dynlibs are needed with mac bundle app and what's needed for a command line app.

DavidJJ, 2020-12-26:
Just FYI,  

I just built FreeCad from brew on 2013 Macbook pro Big Sur.  
Building without --with-macos-app works and am able to start .19
Building with --with-macos-app crashes but without any readable error message.

zenz, 2021-01-05:
@ipatch @DavidJJ @ithinuel 
Please test with new homebrew fomula,
It seems solve all current issues.

ipatch, 2021-01-05:
@zenz

i'm still on mojave so installing custom deps such as qt take a significant amount of time ~ 20+ hours to build because the tap does not provide a bottle, i'm building pyside2 right now, and will reinstall freecad from a formula file i have with and without the option set. and post my results here.

FWIW i've cloned the freecad src repo and can build freecad without relying a a formula and have installed freecad outside of the homebrew prefix to allow for multiple installs, and enabling the macos app flag without using the formula file also resulted in an error on my system, so i was thinking the error was upstream, ie. something wrong with a cmake file, but could be wrong due to the fact that you say it's working.

ithinuel, 2021-01-06:
@zenz I ran:

```
brew remove freecad
brew bundle dump
brew bundle cleanup
```

That leaves me with:
```
$ brew list
antigen		hidapi		icu4c		libusb		lsusb		ncurses		open-ocd	wabt
confuse		htop		libftdi		libusb-compat	minicom		node		telnet
```

Then `brew install freecad --with-macos-app` results in 
```
Last 15 lines from /Users/.../Library/Logs/Homebrew/freecad/01.cmake:
CMake Error at /usr/local/opt/shiboken2/lib/cmake/Shiboken2-5.15.2/shiboken_helpers.cmake:461 (message):
  The detected Python major version is not compatible with the Python major
  version which was used when Shiboken was built.

  Built with: '3.9' Detected: '.'
Call Stack (most recent call first):
  /usr/local/opt/shiboken2/lib/cmake/Shiboken2-5.15.2/Shiboken2Config.cpython-39-darwin.cmake:45 (shiboken_check_if_built_and_target_python_are_compatible)
  /usr/local/opt/shiboken2/lib/cmake/Shiboken2-5.15.2/Shiboken2Config.cmake:5 (include)
  cMake/FreeCAD_Helpers/SetupShibokenAndPyside.cmake:25 (find_package)
  CMakeLists.txt:70 (SetupShibokenAndPyside)
```

`brew install freecad` does complete but freecad fails to start with `[1]    71145 illegal hardware instruction  freecad` as its only output.

Finally, after removing freecad and bundle cleanup, `brew install freecad --build-from-source` yields the same result as `brew install freecad --with-macos-app`
Because I can no longer run FreeCAD, I guess that would qualify as a regression. :S

zenz, 2021-01-06:
@ithinuel 
Strange, your result just like mine before @vejmarie updated the fomula
Please try manually install gettext first
brew intall gettext
then brew install --HEAD freecad --with-macos-app

ipatch, 2021-01-09:
yeah unfortunately i'm running into issues when enabling the below switch

```
-DFREECAD_CREATE_MAC_APP=ON
```

```
-- Installing: /usr/local/FreeCAD.app/Contents/share/Gui/Stylesheets/images_dark-light/up_arrow_lighter.svg
CMake Error at src/MacAppBundle/cmake_install.cmake:55 (file):
  file INSTALL cannot find
  "/opt/code/github/public/forks/freecad-git/freecad/src/MacAppBundle/import
  site": No such file or directory.
Call Stack (most recent call first):
  src/cmake_install.cmake:51 (include)
  cmake_install.cmake:68 (include)

make: *** [install] Error 1
```

DavidJJ, 2021-01-10:
I'm still getting errors using brew install.  the install runs, but startup
fails.

During initialization the error "UI file does not exist" occurred in
/usr/local/Cellar/freecad/0.19pre/Mod/Draft/InitGui.py

Please look into the log file for further information

During initialization the error "UI file does not exist" occurred in
/usr/local/Cellar/freecad/0.19pre/Mod/Arch/InitGui.py

Please look into the log file for further information

QWidget: Must construct a QApplication before a QWidget

Abort trap: 6

>
> yeah unfortunately i'm running into issues when enabling the below switch
>
> -DFREECAD_CREATE_MAC_APP=ON
>
> -- Installing:
/usr/local/FreeCAD.app/Contents/share/Gui/Stylesheets/images_dark-light/up_arrow_lighter.svg
> CMake Error at src/MacAppBundle/cmake_install.cmake:55 (file):
>   file INSTALL cannot find
"/opt/code/github/public/forks/freecad-git/freecad/src/MacAppBundle/import

vejmarie, 2021-01-10:
> I'm still getting errors using brew install. the install runs, but startup fails. During initialization the error "UI file does not exist" occurred in /usr/local/Cellar/freecad/0.19pre/Mod/Draft/InitGui.py Please look into the log file for further information During initialization the error "UI file does not exist" occurred in /usr/local/Cellar/freecad/0.19pre/Mod/Arch/InitGui.py Please look into the log file for further information QWidget: Must construct a QApplication before a QWidget Abort trap: 6
> /usr/local/FreeCAD.app/Contents/share/Gui/Stylesheets/images_dark-light/up_arrow_lighter.svg
> CMake Error at src/MacAppBundle/cmake_install.cmake:55 (file): file INSTALL cannot find
> "/opt/code/github/public/forks/freecad-git/freecad/src/MacAppBundle/import
> […](#)

Can you tell me with which command you installed FreeCAD ? Which command do you use to launch freecad ? Might be good also to have your environment variables (env)

vejmarie, 2021-01-10:
> yeah unfortunately i'm still getting issues when enabling the below switch
> 
> ```
> -DFREECAD_CREATE_MAC_APP=ON
> ```
> 
> ```
> -- Installing: /usr/local/FreeCAD.app/Contents/share/Gui/Stylesheets/images_dark-light/up_arrow_lighter.svg
> CMake Error at src/MacAppBundle/cmake_install.cmake:55 (file):
>   file INSTALL cannot find
>   "/opt/code/github/public/forks/freecad-git/freecad/src/MacAppBundle/import
>   site": No such file or directory.
> Call Stack (most recent call first):
>   src/cmake_install.cmake:51 (include)
>   cmake_install.cmake:68 (include)
> 
> 
> make: *** [install] Error 1
> ```
> 
> this is the fork / branch i'm trying to build from,
> 
> https://github.com/ipatch/FreeCAD/tree/feature-touchscreen-multi-select
> 
> the above fork/branch builds and installs without any issues, but enabling the above cmake flag still produces errors.

I will have a look during the week ;)

vejmarie, 2021-04-12:
I just fixed that issue (sorry for the delay, got too many things to do at work). Please test latest PR and let me know before I close it !

ipatch, 2021-04-12:
awesome, give me a few. i'm going to have to build most if not all bottles from source unfortunately because i'm still running macos 10.14 mojave.

johnmarinelli, 2022-10-14:
any updates on this?  i am running into the same issue on MacOS Monterey when building from source and homebrew.

ipatch, 2022-10-14:
> any updates on this? i am running into the same issue on MacOS Monterey when building from source and homebrew.

what error are you running into?

johnmarinelli, 2022-10-19:
@ipatch so, when trying to build from source, I got the same issue as https://github.com/FreeCAD/homebrew-freecad/issues/142#issuecomment-757510822:
"I'm still getting errors using brew install.  the install runs, but startup
fails.
During initialization the error "UI file does not exist" occurred in
/usr/local/Cellar/freecad/0.19pre/Mod/Draft/InitGui.py"

now, when I try `brew install freecad` (without `--no-mac-app`), I run into the error described here
https://github.com/FreeCAD/homebrew-freecad/issues/142#issuecomment-757510822

ipatch, 2022-10-21:
@johnmarinelli 

hey john 👋, it seems you've more than likely run into a common runtime issue when building and running freecad.

if you're attempting to install freecad via `brew install freecad -v` and you see error messages in the console when opening a workbench such as the _Draft_ workbench than more than likely you're freecad installation is unable to pick up on your `site-packages` installation of pyside and shiboken.

the homebrew-freecad tap uses tap specific and versioned formula of both shiboken and pyside. homebrew-core provides a pyside@2 version of pyside, but you're gunna want to make sure you have the tap (homebrew-freecad) version of `pyside@5.15.5` installed.

i'm not sure if you're on intel or apple ARM (M1) but it shouldn't make a big difference other than build times. as it takes ~ 60 minutes to build pyside2@5.15.5 on my late 2013 gen 4 i7 macbooks.

there's a couple of ways you can probably resolve your runtime issues,

1. you could setup a `PYTHONPATH` environment variable for the shell if you plan on launching freecad from a terminal of your choosing. the `PYTHONPATH` needs to contain the path to the `site-packages` directory for the `shiboken2@5.15.5` and `pyside2@5.15.5`
2. if you're not familiar with setting up environment variables or locating the paths i'm referring you might try and run

```
brew link shiboken2@5.15.5 -v --force;
brew link pyside2@5.15.5 -v --force;
```

i hope all this helps you with your freecad installation. unfortunately at this time there isn't an easy to just make `brew install freecad/freecad/freecad -v` _just work_™️ out of the box quite yet.

ipatch, 2025-09-29:
the below PR when merged should resolve building an app bundle with homebrew, and the latest stable release of freecad ie. v1.0.2 that is app bundle built with homebrew deps can be found below,

https://github.com/FreeCAD/FreeCAD/pull/24286

https://github.com/FreeCAD/homebrew-freecad/releases/tag/1.0.2

if there is anymore issues concerning this, feel free to open a new issue.