FreeCAD/homebrew-freecad #154: homebrew build-from-source post-install step issue with homebrew-freecad-bundle.pth check

nachumk, 2021-01-11:
When installing FreeCAD from source, the post-install step fails because of this check:
    if !File.exist?("/usr/local/Cellar/freecad/0.19pre/lib/python3.9/site-packages/homebrew-freecad-bundle.pth")
      (lib/"python3.9/site-packages/homebrew-freecad-bundle.pth").write "#{prefix}/MacOS/\n"
    end

Since it is built from HEAD, the folder name should be:
/usr/local/Cellar/freecad/HEAD-6b3936c
and not
/usr/local/Cellar/freecad/0.19pre

The verbose output shows:
/usr/local/Homebrew/Library/Homebrew/brew.rb (Formulary::TapLoader): loading /usr/local/Homebrew/Library/Taps/freecad/homebrew-freecad/Formula/freecad.rb
==> Postinstalling freecad
Warning: The post-install step did not complete successfully
You can try again using `brew postinstall freecad/freecad/freecad`
==> An exception occurred within a child process:
  RuntimeError: Will not overwrite /usr/local/Cellar/freecad/HEAD-6b3936c/lib/python3.9/site-packages/homebrew-freecad-bundle.pth

/usr/local/Homebrew/Library/Homebrew/extend/pathname.rb:176:in `write'
/usr/local/Homebrew/Library/Taps/freecad/homebrew-freecad/Formula/freecad.rb:103:in `post_install'
/usr/local/Homebrew/Library/Homebrew/formula.rb:1062:in `block (2 levels) in run_post_install'
/usr/local/Homebrew/Library/Homebrew/formula.rb:904:in `with_logging'
/usr/local/Homebrew/Library/Homebrew/formula.rb:1061:in `block in run_post_install'
/usr/local/Homebrew/Library/Homebrew/utils.rb:531:in `with_env'
/usr/local/Homebrew/Library/Homebrew/formula.rb:1050:in `run_post_install'
/usr/local/Homebrew/Library/Homebrew/postinstall.rb:22:in `<main>'

vejmarie, 2021-02-11:
thanks for the bug report. If you want make a PR feel free to do it ! Otherwise I will fix the bug next week ;)

vejmarie, 2021-04-12:
I will be fixing that issue shortly as soon as we can close the macapp creation issue.

ipatch, 2024-04-29:
closing this issue as this is no longer relevant with current install of freecad with the current formula file defined within this repo.

closed 2024-04-29 by ipatch