FreeCAD/homebrew-freecad #156: Cannot install after upgrading to Big Sur and Xcode 12.3

andrew-newell, 2021-01-23:
Hello! I'm hoping I might be able to get some guidance on what I've done wrong - I've been trying to compile and install FreeCAD for the last week to no avail.

I started on Mojave with Xcode 11.3 and got the same error that I'm getting now on Big Sur. Unfortunately I can't seem to figure out what the error is, beyond "Error 2". I have python 3.9 installed and I think I've jumped through all the hoops there, but who knows, it's python after all.

Perhaps someone can make sense of this output and guide me in the right direction? Thanks much, I can't wait to use FreeCAD 0.19.

```
andrew.newell in ~/FreeCAD/build(ruby-2.7.2) λ brew install --HEAD freecad --with-macos-app
Updating Homebrew...
==> Auto-updated Homebrew!
Updated 3 taps (homebrew/cask-versions, homebrew/core and homebrew/cask).
==> New Formulae
cloudflare-wrangler                                         osmcoastline
==> Updated Formulae
Updated 85 formulae.
==> Updated Casks
Updated 34 casks.

Warning: Treating freecad as a formula. For the cask, use homebrew/cask/freecad
==> Installing freecad from freecad/freecad
==> Cloning https://github.com/freecad/FreeCAD.git
Updating /Users/andrew.newell/Library/Caches/Homebrew/freecad--git
From https://github.com/freecad/FreeCAD
   24d26a092..125cf099a  master     -> origin/master
==> Checking out branch master
Already on 'master'
Your branch is behind 'origin/master' by 10 commits, and can be fast-forwarded.
  (use "git pull" to update your local branch)
HEAD is now at 125cf099a Sketcher: Invalid Sketch geometry color in preferences
==> cmake -DBUILD_QT5=ON -DUSE_PYTHON3=1 -DPYTHON_EXECUTABLE=/usr/local/opt/python3.9/bin/python3 -std=c++14 -DCMAKE_C
==> make -j16 install
Last 15 lines from /Users/andrew.newell/Library/Logs/Homebrew/freecad/02.make:
[ 29%] Building CXX object src/Base/CMakeFiles/FreeCADBase.dir/__/zipios++/zipheadio.cpp.o
cd /tmp/freecad-20210123-43767-8nyqqk/Build/src/Base && /usr/local/bin/ccache /usr/local/Homebrew/Library/Homebrew/shims/mac/super/clang++ -DBOOST_ALL_NO_LIB -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_PP_VARIADICS=1 -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_REGEX_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -DBOOST_THREAD_DYN_LINK -DFreeCADBase_EXPORTS -DGL_SILENCE_DEPRECATION -DHAVE_CONFIG_H -DHAVE_FREEIMAGE -DHAVE_RAPIDJSON -DHAVE_SWIG=1 -DHAVE_TBB -DOCC_CONVERT_SIGNALS -DQT_CORE_LIB -DQT_NO_DEBUG -D_OCC64 -I/tmp/freecad-20210123-43767-8nyqqk/Build -I/tmp/freecad-20210123-43767-8nyqqk/Build/src -I/tmp/freecad-20210123-43767-8nyqqk/src -I/tmp/freecad-20210123-43767-8nyqqk/Build/src/Base -I/tmp/freecad-20210123-43767-8nyqqk/src/Base -I/usr/local/opt/python3.9/Frameworks/Python.framework/Versions/3.9/include/python3.9 -isystem /usr/local/Cellar/boost@1.75.0/1.75.0/include -iframework /usr/local/opt/qt5152/lib -isystem /usr/local/opt/qt5152/lib/QtCore.framework/Headers -isystem /usr/local/opt/qt5152/./mkspecs/macx-clang -Wall -Wextra -Wpedantic -Wno-write-strings  -Wno-undefined-var-template -DNDEBUG -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk -fPIC -I/usr/local/Cellar/open-mpi/4.0.5/include -fPIC -std=gnu++14 -o CMakeFiles/FreeCADBase.dir/__/zipios++/zipheadio.cpp.o -c /tmp/freecad-20210123-43767-8nyqqk/src/zipios++/zipheadio.cpp
[ 29%] Building CXX object src/Base/CMakeFiles/FreeCADBase.dir/__/zipios++/zipinputstream.cpp.o
cd /tmp/freecad-20210123-43767-8nyqqk/Build/src/Base && /usr/local/bin/ccache /usr/local/Homebrew/Library/Homebrew/shims/mac/super/clang++ -DBOOST_ALL_NO_LIB -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_PP_VARIADICS=1 -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_REGEX_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -DBOOST_THREAD_DYN_LINK -DFreeCADBase_EXPORTS -DGL_SILENCE_DEPRECATION -DHAVE_CONFIG_H -DHAVE_FREEIMAGE -DHAVE_RAPIDJSON -DHAVE_SWIG=1 -DHAVE_TBB -DOCC_CONVERT_SIGNALS -DQT_CORE_LIB -DQT_NO_DEBUG -D_OCC64 -I/tmp/freecad-20210123-43767-8nyqqk/Build -I/tmp/freecad-20210123-43767-8nyqqk/Build/src -I/tmp/freecad-20210123-43767-8nyqqk/src -I/tmp/freecad-20210123-43767-8nyqqk/Build/src/Base -I/tmp/freecad-20210123-43767-8nyqqk/src/Base -I/usr/local/opt/python3.9/Frameworks/Python.framework/Versions/3.9/include/python3.9 -isystem /usr/local/Cellar/boost@1.75.0/1.75.0/include -iframework /usr/local/opt/qt5152/lib -isystem /usr/local/opt/qt5152/lib/QtCore.framework/Headers -isystem /usr/local/opt/qt5152/./mkspecs/macx-clang -Wall -Wextra -Wpedantic -Wno-write-strings  -Wno-undefined-var-template -DNDEBUG -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk -fPIC -I/usr/local/Cellar/open-mpi/4.0.5/include -fPIC -std=gnu++14 -o CMakeFiles/FreeCADBase.dir/__/zipios++/zipinputstream.cpp.o -c /tmp/freecad-20210123-43767-8nyqqk/src/zipios++/zipinputstream.cpp
[ 29%] Building CXX object src/Base/CMakeFiles/FreeCADBase.dir/__/zipios++/zipinputstreambuf.cpp.o
cd /tmp/freecad-20210123-43767-8nyqqk/Build/src/Base && /usr/local/bin/ccache /usr/local/Homebrew/Library/Homebrew/shims/mac/super/clang++ -DBOOST_ALL_NO_LIB -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_PP_VARIADICS=1 -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_REGEX_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -DBOOST_THREAD_DYN_LINK -DFreeCADBase_EXPORTS -DGL_SILENCE_DEPRECATION -DHAVE_CONFIG_H -DHAVE_FREEIMAGE -DHAVE_RAPIDJSON -DHAVE_SWIG=1 -DHAVE_TBB -DOCC_CONVERT_SIGNALS -DQT_CORE_LIB -DQT_NO_DEBUG -D_OCC64 -I/tmp/freecad-20210123-43767-8nyqqk/Build -I/tmp/freecad-20210123-43767-8nyqqk/Build/src -I/tmp/freecad-20210123-43767-8nyqqk/src -I/tmp/freecad-20210123-43767-8nyqqk/Build/src/Base -I/tmp/freecad-20210123-43767-8nyqqk/src/Base -I/usr/local/opt/python3.9/Frameworks/Python.framework/Versions/3.9/include/python3.9 -isystem /usr/local/Cellar/boost@1.75.0/1.75.0/include -iframework /usr/local/opt/qt5152/lib -isystem /usr/local/opt/qt5152/lib/QtCore.framework/Headers -isystem /usr/local/opt/qt5152/./mkspecs/macx-clang -Wall -Wextra -Wpedantic -Wno-write-strings  -Wno-undefined-var-template -DNDEBUG -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk -fPIC -I/usr/local/Cellar/open-mpi/4.0.5/include -fPIC -std=gnu++14 -o CMakeFiles/FreeCADBase.dir/__/zipios++/zipinputstreambuf.cpp.o -c /tmp/freecad-20210123-43767-8nyqqk/src/zipios++/zipinputstreambuf.cpp
[ 29%] Building CXX object src/Base/CMakeFiles/FreeCADBase.dir/__/zipios++/zipoutputstreambuf.cpp.o
cd /tmp/freecad-20210123-43767-8nyqqk/Build/src/Base && /usr/local/bin/ccache /usr/local/Homebrew/Library/Homebrew/shims/mac/super/clang++ -DBOOST_ALL_NO_LIB -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_PP_VARIADICS=1 -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_REGEX_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -DBOOST_THREAD_DYN_LINK -DFreeCADBase_EXPORTS -DGL_SILENCE_DEPRECATION -DHAVE_CONFIG_H -DHAVE_FREEIMAGE -DHAVE_RAPIDJSON -DHAVE_SWIG=1 -DHAVE_TBB -DOCC_CONVERT_SIGNALS -DQT_CORE_LIB -DQT_NO_DEBUG -D_OCC64 -I/tmp/freecad-20210123-43767-8nyqqk/Build -I/tmp/freecad-20210123-43767-8nyqqk/Build/src -I/tmp/freecad-20210123-43767-8nyqqk/src -I/tmp/freecad-20210123-43767-8nyqqk/Build/src/Base -I/tmp/freecad-20210123-43767-8nyqqk/src/Base -I/usr/local/opt/python3.9/Frameworks/Python.framework/Versions/3.9/include/python3.9 -isystem /usr/local/Cellar/boost@1.75.0/1.75.0/include -iframework /usr/local/opt/qt5152/lib -isystem /usr/local/opt/qt5152/lib/QtCore.framework/Headers -isystem /usr/local/opt/qt5152/./mkspecs/macx-clang -Wall -Wextra -Wpedantic -Wno-write-strings  -Wno-undefined-var-template -DNDEBUG -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk -fPIC -I/usr/local/Cellar/open-mpi/4.0.5/include -fPIC -std=gnu++14 -o CMakeFiles/FreeCADBase.dir/__/zipios++/zipoutputstreambuf.cpp.o -c /tmp/freecad-20210123-43767-8nyqqk/src/zipios++/zipoutputstreambuf.cpp
[ 29%] Building CXX object src/Base/CMakeFiles/FreeCADBase.dir/__/zipios++/zipoutputstream.cpp.o
cd /tmp/freecad-20210123-43767-8nyqqk/Build/src/Base && /usr/local/bin/ccache /usr/local/Homebrew/Library/Homebrew/shims/mac/super/clang++ -DBOOST_ALL_NO_LIB -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_PP_VARIADICS=1 -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_REGEX_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -DBOOST_THREAD_DYN_LINK -DFreeCADBase_EXPORTS -DGL_SILENCE_DEPRECATION -DHAVE_CONFIG_H -DHAVE_FREEIMAGE -DHAVE_RAPIDJSON -DHAVE_SWIG=1 -DHAVE_TBB -DOCC_CONVERT_SIGNALS -DQT_CORE_LIB -DQT_NO_DEBUG -D_OCC64 -I/tmp/freecad-20210123-43767-8nyqqk/Build -I/tmp/freecad-20210123-43767-8nyqqk/Build/src -I/tmp/freecad-20210123-43767-8nyqqk/sr [... cut ...]
homebrew build-from-source post-install step issue with homebrew-freecad-bundle.pth check https://github.com/FreeCAD/homebrew-freecad/issues/154
Auto-bottling for homebrew-freecad tap formulae https://github.com/FreeCAD/homebrew-freecad/issues/63
new freecad git source merged some 3rd party code binding to MacOS sdk 11.0 but now is 11.1 https://github.com/FreeCAD/homebrew-freecad/issues/135
Please create pull requests instead of asking for help on Homebrew's GitHub,
Twitter or any other official channels.
```

vejmarie, 2021-01-23:
Hi,

Can you run homebrew like this

brew install -v --HEAD freecad

And let me know the output ? The log doesn't give me where the error stands currently.

vejmarie

andrew-newell, 2021-01-23:
Since that output was huge, here's the console output attached!

[verbose_brew_install.txt](https://github.com/FreeCAD/homebrew-freecad/files/5860625/verbose_brew_install.txt)

vejmarie, 2021-01-23:
let me have a look to it give me a couple of hours

vejmarie, 2021-01-23:
can you give me the package list that you have into your system ?

Can you also run a brew update and brew upgrade, just to be sure that homebrew has been updated. I launched the same compilation than you (without --with-macos-app) and it seems to work currently. I am running big sur and the same xcode than you

brew list output on mine.

boost-python3@1.75.0	gdbm			libffi			numpy@1.19.4		pyside2			swig
boost@1.75.0		gettext			libidn2			open-mpi		pyside2-tools		szip
ccache			gmp			libmpc			openblas		python3.9		tbb@2020_u3
cmake			hdf5			libpng			opencamlib		python@3.9		vtk@8.2.0
coin@4.0.0		hdf5@1.10		libtiff			opencascade@7.5.0	qt			webp
doxygen			hwloc			libunistring		openexr			qt5152			wget
eigen			icu4c			llvm			openssl@1.1		rapidjson		xerces-c
fontconfig		icu4c@67.1		matplotlib		orocos-kdl		readline		xz
freecad			ilmbase			med-file		pcre			shiboken2		zstd
freeimage		isl			mpfr			pivy			sip@4.19.24
freetype		jpeg			netcdf			pkg-config		sphinx-doc
gcc			libevent		nglib			pyqt@5.15.2		sqlite

andrew-newell, 2021-01-23:
Here's the console output (I did `brew list --formula`).

[brew_list.txt](https://github.com/FreeCAD/homebrew-freecad/files/5860676/brew_list.txt)
[brew_update.txt](https://github.com/FreeCAD/homebrew-freecad/files/5860677/brew_update.txt)
[brew_upgrade.txt](https://github.com/FreeCAD/homebrew-freecad/files/5860678/brew_upgrade.txt)

vejmarie, 2021-01-23:
Ok, I think your issue comes from this

make[2]: *** No rule to make target `/Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk/usr/lib/libz.tbd', needed by `lib/libSMDS.dylib'.  Stop.

When you updated your XCode, I think you faced the Command Line Tools update issue. To do this you can download latest command line tools from your Apple developper account. Let me know if you need a link. If you re-install it properly it shall be working.

Mine libz.tbd stand into

/Library/Developer/CommandLineTools/SDKs/MacOSX11.1.sdk/usr/lib/libz.tbd
/Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/lib/libz.tbd

vejmarie

andrew-newell, 2021-01-23:
Ok I see `/Library/Developer/CommandLineTools/SDKs/MacOSX11.1.sdk/usr/lib/` has `libz.tbd` now.  I installed the Command Line Tools shown here: 
[image]

Unfortunately, no dice. Looking into the logs, I see that it is still referencing `MacOSX10.14.sdk` like in this example:
`make[2]: *** No rule to make target `/Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk/usr/lib/libz.tbd', needed by `lib/libSMDS.dylib'.  Stop.`

Should it be using `11.1.sdk` instead?

Here's the full output from this latest attempt after installing the latest CommandLIneTools. 
[brew_install_2_failure.txt](https://github.com/FreeCAD/homebrew-freecad/files/5860894/brew_install_2_failure.txt)

vejmarie, 2021-01-23:
That is weird. Can you give me the output of ls -l /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/lib/libz.*
I do not really understand why you CMake is looking for /Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk/usr/lib/libz.tbd instead of the relevant one which shall be installed up. You can probably link the old directory to the new one, it will work for sure. But it look like that Command Line Tools didn't updated the reference properly. My setup is a fresh big_sur install

vejmarie, 2021-01-23:
you can try this

sudo xcode-select --switch /Applications/Xcode.app
sudo xcode-select --install

Just to be sure that your system consider the latest version of XCode as being the one relevant

andrew-newell, 2021-01-23:
Here's the output of `ls` 
[ls_output.txt](https://github.com/FreeCAD/homebrew-freecad/files/5860913/ls_output.txt)

Here's what those commands output (not much):
```
andrew.newell in ~(ruby-2.7.2) λ sudo xcode-select --switch /Applications/Xcode.app
Password:
andrew.newell in ~(ruby-2.7.2) λ sudo xcode-select --install

xcode-select: error: command line tools are already installed, use "Software Update" to install updates
```
Also, Software Update has no updates for anything Xcode related, which shouldn't be surprising since 12.3 is the latest I guess.

I briefly looked up a way to show which version of CommandLineTools is called so I ran this:
```
andrew.newell in ~(ruby-2.7.2) λ pkgutil --pkg-info=com.apple.pkg.CLTools_Executables
package-id: com.apple.pkg.CLTools_Executables
version: 12.3.0.0.1.1607026830
volume: /
location: /
install-time: 1611428520
groups: com.apple.FindSystemFiles.pkg-group 
```
3rd try, same results. I'm goingt 
Looks like we're still on `10.14` based on the output from attempt 3:

[brew_install_3_failure.txt](https://github.com/FreeCAD/homebrew-freecad/files/5860927/brew_install_3_failure.txt)

I wanted to see what the symlink looked like so I did this:
```
andrew.newell in /(ruby-2.7.2) λ ls -alhF /Library/Developer/CommandLineTools/SDKs/
total 0
drwxr-xr-x  5 root  wheel   160B Jan 23 14:01 ./
drwxr-xr-x  5 root  wheel   160B Jan 23 14:02 ../
lrwxr-xr-x  1 root  wheel    14B Jan 23 14:01 MacOSX.sdk@ -> MacOSX11.1.sdk
drwxr-xr-x  8 root  wheel   256B Jul  9  2020 MacOSX10.15.sdk/
drwxr-xr-x  7 root  wheel   224B Nov 30 07:33 MacOSX11.1.sdk/
```
So where the heck is this `10.14` coming from, I don't even see it installed anymore...

I'm going to do another reboot just in case, then uninstall Xcode completely if possible and reinstall it.

andrew-newell, 2021-01-23:
Is is possible there are leftover config files that are being reused by the brew install process that need to be cleaned up?

andrew-newell, 2021-01-23:
Welp, that changed absolutely nothing. I have no idea why it's so obsessed with `10.14` but reinstalling didn't change anything.

I did a bunch of things like the ones listed here https://stackoverflow.com/questions/31011062/how-to-completely-uninstall-xcode-and-clear-all-settings such as `rm -rf /Library/Developer/CommandLineTools`

as well as these others

```
andrew.newell in ~(ruby-2.7.2) λ sudo rm -rf /Library/Preferences/com.apple.dt.Xcode.plist
Password:
andrew.newell in ~(ruby-2.7.2) λ rm -rf ~/Library/Preferences/com.apple.dt.Xcode.plist
andrew.newell in ~(ruby-2.7.2) λ rm -rf ~/Library/Caches/com.apple.dt.Xcode
andrew.newell in ~(ruby-2.7.2) λ rm -rf ~/Library/Application Support/Xcode
andrew.newell in ~(ruby-2.7.2) λ rm -rf ~/Library/Developer/Xcode
andrew.newell in ~(ruby-2.7.2) λ rm -rf ~/Library/Developer/CoreSimulator
andrew.newell in ~(ruby-2.7.2) λ rm -rf ~/Library/Developer/XCTestDevices 
andrew.newell in ~(ruby-2.7.2) λ xcode-select --print-path                
xcode-select: error: unable to get active developer directory, use `sudo xcode-select --switch path/to/Xcode.app` to set one (or see `man xcode-select`)
andrew.newell in ~(ruby-2.7.2) λ ls /Library/Developer
PrivateFrameworks
```

After uninstalling everything I wanted to make sure the install failed, and it did: 
```
andrew.newell in ~(ruby-2.7.2) λ brew install -v --HEAD freecad
Updating Homebrew...
==> Auto-updated Homebrew!
Updated 2 taps (homebrew/core and homebrew/cask).
==> Updated Formulae
Updated 7 formulae.
==> Updated Casks
Updated 1 cask.

Warning: Treating freecad as a formula. For the cask, use homebrew/cask/freecad
Error: The following flag:
  --HEAD
requires building tools, but none are installed.
Install the Command Line Tools:
  xcode-select --install
```

I wanted to see what would happen after installing Xcode from the App store, but without doing things like `xcode-select --install` and this happened (same as before):

[brew_install_4_failure_after_xcode_reinstall.txt](https://github.com/FreeCAD/homebrew-freecad/files/5861005/brew_install_4_failure_after_xcode_reinstall.txt)

Then I ran `  xcode-select --install` and got this:
[brew_install_5_failure_after_xcodeselect_install.txt](https://github.com/FreeCAD/homebrew-freecad/files/5861006/brew_install_5_failure_after_xcodeselect_install.txt)

So whatever we're doing, the references to these things isn't being affected :(
```
make[2]: *** No rule to make target `/Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/AppKit.framework', needed by `lib/libDriver.dylib'.  Stop.
make[2]: *** No rule to make target `/Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk/System/Library/Frameworks/AppKit.framework', needed by `lib/libMEFISTO2.dylib'.  Stop.
make[2]: *** No rule to make target `/Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk/usr/lib/libz.tbd', needed by `lib/libSMDS.dylib'.  Stop.
```

ipatch, 2021-01-23:
so when you install things with mac homebrew it creates a temporary directory and builds everything within that tmp directory, and generally destroys the directory after installation whether it completes or not to prevent lingering files from causing issues.

so in your first post you said you were using mojave, so mojave is 10.14 so that's a clue as to why your system/brew installation points to a 10.14 sdk. that said a quick hack would be to create a symlink to your existing SDK and name the link with the provided name that the installation is lookign for.  that may move things forward some. no guarantee things will work over an extended period of time or it will fix the immediate issue.

---

_is there any reason your attempting to build freecad from source?_ as you have seen it is a complicated process, and you'll end up spending a considerable amount of time messing around with the build system and building updates to whereas if you install the provided bottle you should be able to launch freecad from a cli and focus less on the tooling and more on the freecad related things that originally got you interested in using freecad in the first place.

that said, i created a cask for freecad a while back you could run `brew cask install freecad` i believe calling cask may be deprecated, but there is a cask available via brew so do a search for that.

and your brew installation may require some updates/fixes, maybe post the output of,

```shell
brew doctor
```

and that may lead to clues as to why your system is looking for a 10.14 SDK, as well there are pre built app bundles provided under the freecad releases page under the freecad git repo that will get up and going with freecad with a simple drap n drop installaltion and a double click to get going.

vejmarie, 2021-01-23:
Most of salome mesh are into
    ${Boost_INCLUDE_DIRS}
    ${OCC_INCLUDE_DIR}
    ${NGLIB_INCLUDE_DIR}
    ${NETGEN_INCLUDE_DIRS}
    ${ZLIB_INCLUDE_DIR}
    ${VTK_INCLUDE_DIRS}
    ${HDF5_INCLUDE_DIR}
    ${MEDFILE_INCLUDE_DIRS}
    ${BOOST_INCLUDE_DIR}
One of them is probably still linked with the previous installation and giving you this challenge. I would remove these dependencies and restart a build to see what happens as homebrew shall reinstall them by itself

vejmarie, 2021-01-23:
i loved to find the root cause just in case the issue comes from one of our fomula

andrew-newell, 2021-01-23:
> _is there any reason your attempting to build freecad from source?_

I've been trying to get to `0.19` as `0.18` addonmanager doesn't work properly (only 3d printing thing is listed and the tweaks to the addonmanager python scripts didn't solve the issue) and there are apparently a lot of improvements to benefit from and since there isn't a `.dmg` release as far as I can find so this was the path I was led to believe I needed to follow. Given I knew I was "behind" on my os it seemed most logical at the time to upgrade to Big Sur after my first couple tries to install didn't work so I could get the latest Xcode and go down the path of debugging why compiling from source wasn't working out. I figure there are other users in a similar boat as this is my main computer for developing software and if I'm running into issues like this it might be worth it to see what we can figure out together to help other potential users.

> so in your first post you said you were using mojave, so mojave is 10.14 so that's a clue as to why your system/brew installation points to a 10.14 sdk. that said a quick hack would be to create a [...]

I'll keep that idea as last-ditch effort as I fear there is some underlying misconfiguration that will come back and haunt me if I don't get to the bottom of this, but I'm not above doing that once we exhaust all other options.

I did take a look at the cask and it seems to be `0.18.4` which is basically the same as the release `.dmg` which puts me right back where I'm at, unfortunately.

`brew doctor` and `brew cleanup` output is here:
```
$ brew doctor
Please note that these warnings are just used to help the Homebrew maintainers
with debugging if you file an issue. If everything you use Homebrew for is
working fine: please don't worry or file an issue; just ignore this. Thanks!

Warning: Some installed kegs have no formulae!
This means they were either deleted or installed with `brew diy`.
You should find replacements for the following formulae:
  node@6
  python@2
  ansible@1.9

Warning: Some installed formulae are deprecated or disabled.
You should find replacements for the following formulae:
  vtk@8.2.0

Warning: Unbrewed dylibs were found in /usr/local/lib.
If you didn't put them there on purpose they could cause problems when
building Homebrew formulae, and may need to be deleted.

Unexpected dylibs:
  /usr/local/lib/libtcl8.6.dylib
  /usr/local/lib/libtk8.6.dylib

Warning: Unbrewed header files were found in /usr/local/include.
If you didn't put them there on purpose they could cause problems when
building Homebrew formulae, and may need to be deleted.

Unexpected header files:
  /usr/local/include/fakemysql.h
  /usr/local/include/fakepq.h
  /usr/local/include/fakesql.h
  /usr/local/include/itcl.h
  /usr/local/include/itcl2TclOO.h
  /usr/local/include/itclDecls.h
  /usr/local/include/itclInt.h
  /usr/local/include/itclIntDecls.h
  /usr/local/include/itclMigrate2TclCore.h
  /usr/local/include/itclTclIntStubsFcn.h
  /usr/local/include/mysqlStubs.h
  /usr/local/include/odbcStubs.h
  /usr/local/include/pqStubs.h
  /usr/local/include/tcl.h
  /usr/local/include/tclDecls.h
  /usr/local/include/tclOO.h
  /usr/local/include/tclOODecls.h
  /usr/local/include/tclPlatDecls.h
  /usr/local/include/tclThread.h
  /usr/local/include/tclTomMath.h
  /usr/local/include/tclTomMathDecls.h
  /usr/local/include/tdbc.h
  /usr/local/include/tdbcDecls.h
  /usr/local/include/tdbcInt.h
  /usr/local/include/tk.h
  /usr/local/include/tkDecls.h
  /usr/local/include/tkPlatDecls.h

Warning: Unbrewed .pc files were found in /usr/local/lib/pkgconfig.
[... 68 lines cut ...]
Removing: /Users/andrew.newell/Library/Caches/Homebrew/openssl@1.1--1.1.1h.mojave.bottle.tar.gz... (5.4MB)
Removing: /usr/local/Cellar/pango/1.44.7... (62 files, 4.3MB)
Removing: /usr/local/Cellar/pango/1.46.2... (64 files, 2.8MB)
Removing: /usr/local/Cellar/pcre2/10.35... (230 files, 6.0MB)
Removing: /usr/local/Cellar/poppler/20.11.0... (476 files, 25.7MB)
Removing: /Users/andrew.newell/Library/Caches/Homebrew/poppler--20.11.0.mojave.bottle.tar.gz... (8.0MB)
Removing: /usr/local/Cellar/python@3.9/3.9.0_1... (4,447 files, 69.5MB)
Removing: /Users/andrew.newell/Library/Caches/Homebrew/python@3.9--3.9.0_1.mojave.bottle.tar.gz... (17.8MB)
Removing: /usr/local/Cellar/qt/5.15.1... (10,691 files, 371.0MB)
Removing: /Users/andrew.newell/Library/Caches/Homebrew/qt--5.15.1.mojave.bottle.tar.gz... (131.6MB)
Removing: /usr/local/Cellar/rav1e/0.3.4... (13 files, 67.0MB)
Removing: /Users/andrew.newell/Library/Caches/Homebrew/rav1e--0.3.4.mojave.bottle.tar.gz... (19.3MB)
Removing: /usr/local/Cellar/readline/8.0.4... (48 files, 1.5MB)
Removing: /usr/local/Cellar/sdl2/2.0.12_1... (89 files, 4.7MB)
Removing: /usr/local/Cellar/shared-mime-info/2.0... (86 files, 4.4MB)
Removing: /Users/andrew.newell/Library/Caches/Homebrew/shared-mime-info--2.0.mojave.bottle.tar.gz... (1MB)
Removing: /usr/local/Cellar/x265/3.4_1... (11 files, 35.7MB)
Removing: /Users/andrew.newell/Library/Caches/Homebrew/portable-ruby-2.6.3_2.yosemite.bottle.tar.gz... (9.1MB)
Pruned 182 symbolic links and 94 directories from /usr/local
```

After running the above things it appears to have failed in the same way.
[brew_install_6_failure_after_brew_cleanup.txt](https://github.com/FreeCAD/homebrew-freecad/files/5861050/brew_install_6_failure_after_brew_cleanup.txt)

@vejmarie is there a file where I can see what dependencies to reinstall?

andrew-newell, 2021-01-23:
I used `brew deps` to help me force uninstall all the dependencies like so:
```
andrew.newell in ~(ruby-2.7.2) λ brew deps freecad | xargs -L brew uninstall --ignore-dependencies --formula
xargs: -L brew: invalid
Warning: Treating freecad as a formula. For the cask, use homebrew/cask/freecad
andrew.newell in ~(ruby-2.7.2) λ brew deps freecad | xargs -n 1 brew uninstall --ignore-dependencies --formula
Warning: Treating freecad as a formula. For the cask, use homebrew/cask/freecad
Uninstalling /usr/local/Cellar/shiboken2/5.15.2... (66 files, 2.6MB)
Uninstalling /usr/local/Cellar/eigen/3.3.9... (493 files, 6.6MB)
Uninstalling /usr/local/Cellar/fontconfig/2.13.1... (531 files, 3.3MB)
Error: No such keg: /usr/local/Cellar/boost-python3@1.75.0
Error: No such keg: /usr/local/Cellar/boost@1.75.0
Error: No such keg: /usr/local/Cellar/coin@4.0.0
Uninstalling /usr/local/Cellar/icu4c@67.1/67.1... (258 files, 71MB)
Uninstalling /usr/local/Cellar/matplotlib/2.1.1... (1,856 files, 22.2MB)
Uninstalling /usr/local/Cellar/med-file/4.0.0... (100 files, 9.4MB)
Uninstalling /usr/local/Cellar/nglib/v6.2.2007... (378 files, 9.6MB)
Uninstalling /usr/local/Cellar/numpy@1.19.4/1.19.4... (859 files, 22.0MB)
Uninstalling /usr/local/Cellar/opencamlib/0.0.1... (52 files, 2.1MB)
Uninstalling /usr/local/Cellar/opencascade@7.5.0/7.5.0... (9,453 files, 165.0MB)
Uninstalling /usr/local/Cellar/pivy/0.6.5... (36 files, 13.6MB)
Uninstalling /usr/local/Cellar/pyqt@5.15.2/5.15.2... (961 files, 27MB)
Uninstalling /usr/local/Cellar/pyside2/5.15.2... (233 files, 40.8MB)
Uninstalling /usr/local/Cellar/pyside2-tools/5.15.2... (14 files, 1.6MB)
Uninstalling /usr/local/Cellar/python3.9/3.9.0_5... (9,229 files, 133.8MB)
Uninstalling /usr/local/Cellar/qt5152/5.15.2... (10,688 files, 365.0MB)
Error: No such keg: /usr/local/Cellar/shiboken2
Uninstalling /usr/local/Cellar/sip@4.19.24/4.19.24_1... (17 files, 1MB)
Uninstalling /usr/local/Cellar/tbb@2020_u3/2020_U3_1... (150 files, 3MB)
Uninstalling /usr/local/Cellar/vtk@8.2.0/8.2.0_1... (4,260 files, 131.1MB)
Uninstalling /usr/local/Cellar/freeimage/3.18.0... (16 files, 30.0MB)
Uninstalling /usr/local/Cellar/freetype/2.10.4... (64 files, 2.3MB)
Uninstalling /usr/local/Cellar/gcc/10.2.0_2... (1,465 files, 338.7MB)
Uninstalling /usr/local/Cellar/gdbm/1.18.1_1... (25 files, 620.8KB)
Uninstalling /usr/local/Cellar/gettext/0.21... (1,959 files, 19MB)
Uninstalling /usr/local/Cellar/gmp/6.2.1... (21 files, 3.2MB)
Uninstalling /usr/local/Cellar/hdf5/1.12.0_1... (268 files, 16.2MB)
Uninstalling /usr/local/Cellar/hdf5@1.10/1.10.7... (263 files, 17.3MB)
Uninstalling /usr/local/Cellar/hwloc/2.4.0... (925 files, 9.3MB)
Uninstalling /usr/local/Cellar/isl/0.23... (72 files, 4.8MB)
Uninstalling /usr/local/Cellar/jpeg/9d... (21 files, 751KB)
Uninstalling /usr/local/Cellar/libevent/2.1.12... (57 files, 1.8MB)
Uninstalling /usr/local/Cellar/libffi/3.3_2... (17 files, 503.9KB)
Uninstalling /usr/local/Cellar/libmpc/1.2.1... (13 files, 381.9KB)
Uninstalling /usr/local/Cellar/libpng/1.6.37... (27 files, 1.2MB)
Uninstalling /usr/local/Cellar/libtiff/4.2.0... (248 files, 3.8MB)
Uninstalling /usr/local/Cellar/llvm/11.0.0_1... (8,922 files, 1.3GB)
Uninstalling /usr/local/Cellar/mpfr/4.1.0... (29 files, 5.1MB)
Uninstalling /usr/local/Cellar/netcdf/4.7.4_2... (90 files, 6.2MB)
Uninstalling /usr/local/Cellar/open-mpi/4.0.5... (752 files, 10.8MB)
Uninstalling /usr/local/Cellar/openblas/0.3.13... (23 files, 120.1MB)
Uninstalling /usr/local/Cellar/openssl@1.1/1.1.1i... (8,067 files, 18.4MB)

Warning: The following openssl@1.1 configuration files have not been removed!
If desired, remove them manually with `rm -rf`:
  /usr/local/etc/openssl@1.1
  /usr/local/etc/openssl@1.1/cert.pem
  /usr/local/etc/openssl@1.1/ct_log_list.cnf
  /usr/local/etc/openssl@1.1/ct_log_list.cnf.default
  /usr/local/etc/openssl@1.1/ct_log_list.cnf.dist
  /usr/local/etc/openssl@1.1/ct_log_list.cnf.dist.default
  /usr/local/etc/openssl@1.1/misc
  /usr/local/etc/openssl@1.1/misc/CA.pl
  /usr/local/etc/openssl@1.1/misc/CA.pl.default
  /usr/local/etc/openssl@1.1/misc/tsget
  /usr/local/etc/openssl@1.1/misc/tsget.default
  /usr/local/etc/openssl@1.1/misc/tsget.pl
  /usr/local/etc/openssl@1.1/misc/tsget.pl.default
  /usr/local/etc/openssl@1.1/openssl.cnf
  /usr/local/etc/openssl@1.1/openssl.cnf.default
  /usr/local/etc/openssl@1.1/openssl.cnf.dist
  /usr/local/etc/openssl@1.1/openssl.cnf.dist.default

Warning: The following may be openssl@1.1 configuration files and have not been removed!
If desired, remove them manually with `rm -rf`:
  /usr/local/etc/openssl
Uninstalling /usr/local/Cellar/orocos-kdl/1.4.0... (97 files, 1002.4KB)
Uninstalling /usr/local/Cellar/pkg-config/0.29.2_3... (11 files, 623.6KB)
Uninstalling /usr/local/Cellar/readline/8.1... (48 files, 1.6MB)
Uninstalling /usr/local/Cellar/sqlite/3.34.0... (11 files, 4MB)
Uninstalling /usr/local/Cellar/szip/2.1.1_1... (11 files, 109.4KB)
Uninstalling /usr/local/Cellar/webp/1.1.0... (39 files, 2.1MB)
Uninstalling /usr/local/Cellar/xerces-c/3.2.3... (1,684 files, 36.6MB)
Uninstalling /usr/local/Cellar/xz/5.2.5... (92 files, 1.1MB)
```

I'm now in the process of running another `brew install -v --HEAD freecad` and will update when I have those results. This was definitely a heavy-handed approach so I'll probably have some cleanup elsewhere to do.

vejmarie, 2021-01-23:
You are more than welcome ! Sorry for the complexity of setting this up and happy to see it working.

closed 2021-01-23 by vejmarie

ipatch, 2021-01-23:
for future reference

```shell
brew search freecad
```

```
==> Casks
freecad                           freecad-ipre ✔                    freecad-pre
```

```shell
brew info freecad-pre

freecad-pre: 0.19_pre,23578
https://freecadweb.org/
Not installed
From: https://github.com/Homebrew/homebrew-cask-versions/blob/HEAD/Casks/freecad-pre.rb
==> Name
FreeCAD
==> Description
3D parametric modeler
==> Artifacts
FreeCAD.app (App)
```

notice above version number.

also that cask / dmg is pulled from the dmg images i mentioned above, which are located here

https://github.com/FreeCAD/FreeCAD/releases

scroll down, click the little expand arrow / triangle

[image]

look for macos conda dmg. that'll get you a double clickable app going.