FreeCAD/homebrew-freecad #159: Freecad from brew not linking on MacOS Big Sur

yves-surrel, 2021-02-11:
brew install --HEAD freecad --with-macos-app --verbose

fails à 69%

[ 69%] Linking CXX shared library ../../lib/libFreeCADGui.dylib
cd /tmp/freecad-20210211-42611-njwxgi/Build/src/Gui && /usr/local/Cellar/cmake/3.19.4/bin/cmake -E cmake_link_script CMakeFiles/FreeCADGui.dir/link.txt --verbose=1
/usr/local/Homebrew/Library/Homebrew/shims/mac/super/clang++ -Wall -Wextra -Wpedantic -Wno-write-strings  -Wno-undefined-var-template -DNDEBUG -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk -dynamiclib -Wl,-headerpad_max_install_names  -Wl,-undefined,dynamic_lookup -o ../../lib/libFreeCADGui.dylib -install_name @rpath/libFreeCADGui.dylib CMakeFiles/FreeCADGui.dir/Application.cpp.o CMakeFiles/FreeCADGui.dir/ApplicationPy.cpp.o CMakeFiles/FreeCADGui.dir/AutoSaver.cpp.o CMakeFiles/FreeCADGui.dir/BitmapFactory.cpp.o CMakeFiles/FreeCADGui.dir/Document.cpp.o CMakeFiles/FreeCADGui.dir/DocumentModel.cpp.o CMakeFiles/FreeCADGui.dir/DocumentPyImp.cpp.o CMakeFiles/FreeCADGui.dir/DocumentObserver.cpp.o CMakeFiles/FreeCADGui.dir/DocumentObserverPython.cpp.o CMakeFiles/FreeCADGui.dir/ExpressionBinding.cpp.o CMakeFiles/FreeCADGui.dir/ExpressionBindingPy.cpp.o CMakeFiles/FreeCADGui.dir/GraphicsViewZoom.cpp.o CMakeFiles/FreeCADGui.dir/ExpressionCompleter.cpp.o CMakeFiles/FreeCADGui.dir/GuiApplication.cpp.o CMakeFiles/FreeCADGui.dir/GuiApplicationNativeEventAware.cpp.o CMakeFiles/FreeCADGui.dir/GuiConsole.cpp.o CMakeFiles/FreeCADGui.dir/Macro.cpp.o CMakeFiles/FreeCADGui.dir/MergeDocuments.cpp.o CMakeFiles/FreeCADGui.dir/resource.cpp.o CMakeFiles/FreeCADGui.dir/Control.cpp.o CMakeFiles/FreeCADGui.dir/SpaceballEvent.cpp.o CMakeFiles/FreeCADGui.dir/Thumbnail.cpp.o CMakeFiles/FreeCADGui.dir/Utilities.cpp.o CMakeFiles/FreeCADGui.dir/WaitCursor.cpp.o CMakeFiles/FreeCADGui.dir/ManualAlignment.cpp.o CMakeFiles/FreeCADGui.dir/TransactionObject.cpp.o CMakeFiles/FreeCADGui.dir/WinNativeGestureRecognizers.cpp.o CMakeFiles/FreeCADGui.dir/iisTaskPanel/src/moc_iisiconlabel.cpp.o CMakeFiles/FreeCADGui.dir/iisTaskPanel/src/moc_iistaskbox.cpp.o CMakeFiles/FreeCADGui.dir/iisTaskPanel/src/moc_iistaskgroup.cpp.o CMakeFiles/FreeCADGui.dir/iisTaskPanel/src/moc_iistaskheader.cpp.o CMakeFiles/FreeCADGui.dir/QSint/actionpanel/moc_actionbox.cpp.o CMakeFiles/FreeCADGui.dir/QSint/actionpanel/moc_actionlabel.cpp.o CMakeFiles/FreeCADGui.dir/QSint/actionpanel/moc_actiongroup.cpp.o CMakeFiles/FreeCADGui.dir/QSint/actionpanel/moc_actionpanel.cpp.o CMakeFiles/FreeCADGui.dir/QSint/actionpanel/moc_taskheader_p.cpp.o CMakeFiles/FreeCADGui.dir/qrc_resource.cpp.o CMakeFiles/FreeCADGui.dir/qrc_translation.cpp.o CMakeFiles/FreeCADGui.dir/qrc_iisTaskPanel.cpp.o CMakeFiles/FreeCADGui.dir/qrc_schemes.cpp.o CMakeFiles/FreeCADGui.dir/Action.cpp.o CMakeFiles/FreeCADGui.dir/ActionFunction.cpp.o CMakeFiles/FreeCADGui.dir/Command.cpp.o CMakeFiles/FreeCADGui.dir/CommandDoc.cpp.o CMakeFiles/FreeCADGui.dir/CommandFeat.cpp.o CMakeFiles/FreeCADGui.dir/CommandMacro.cpp.o CMakeFiles/FreeCADGui.dir/CommandStd.cpp.o CMakeFiles/FreeCADGui.dir/CommandWindow.cpp.o CMakeFiles/FreeCADGui.dir/CommandTest.cpp.o CMakeFiles/FreeCADGui.dir/CommandView.cpp.o CMakeFiles/FreeCADGui.dir/CommandStructure.cpp.o CMakeFiles/FreeCADGui.dir/CommandLink.cpp.o CMakeFiles/FreeCADGui.dir/CommandPyImp.cpp.o CMakeFiles/FreeCADGui.dir/Clipping.cpp.o CMakeFiles/FreeCADGui.dir/DemoMode.cpp.o CMakeFiles/FreeCADGui.dir/DlgActivateWindowImp.cpp.o CMakeFiles/FreeCADGui.dir/DlgUnitsCalculatorImp.cpp.o CMakeFiles/FreeCADGui.dir/DlgDisplayPropertiesImp.cpp.o CMakeFiles/FreeCADGui.dir/DlgInputDialogImp.cpp.o CMakeFiles/FreeCADGui.dir/DlgMacroExecuteImp.cpp.o CMakeFiles/FreeCADGui.dir/DlgRunExternal.cpp.o CMakeFiles/FreeCADGui.dir/DlgEditFileIncludePropertyExternal.cpp.o CMakeFiles/FreeCADGui.dir/DlgMacroRecordImp.cpp.o CMakeFiles/FreeCADGui.dir/DlgMaterialPropertiesImp.cpp.o CMakeFiles/FreeCADGui.dir/DlgParameterImp.cpp.o CMakeFiles/FreeCADGui.dir/DlgParameterFind.cpp.o CMakeFiles/FreeCADGui.dir/DlgProjectInformationImp.cpp.o CMakeFiles/FreeCADGui.dir/DlgProjectUtility.cpp.o CMakeFiles/FreeCADGui.dir/DlgPropertyLink.cpp.o CMakeFiles/FreeCADGui.dir/DlgExpressionInput.cpp.o CMakeFiles/FreeCADGui.dir/TaskDlgRelocation.cpp.o CMakeFiles/FreeCADGui.dir/DlgCheckableMessageBox.cpp.o CMakeFiles/FreeCADGui.dir/TaskCSysDragger.cpp.o CMakeFiles/FreeCADGui.dir/DlgUndoRedo.cpp.o CMakeFiles/FreeCADGui.dir/InputVector.cpp.o CMakeFiles/FreeCADGui.dir/Placement.cpp.o CMakeFiles/FreeCADGui.dir/PropertyPage.cpp.o CMakeFiles/FreeCADGui.dir/SceneInspector.cpp.o CMakeFiles/FreeCADGui.dir/TextureMapping.cpp.o CMakeFiles/FreeCADGui.dir/Transform.cpp.o CMakeFiles/FreeCADGui.dir/DownloadItem.cpp.o CMakeFiles/FreeCADGui.dir/DownloadManager.cpp.o CMakeFiles/FreeCADGui.dir/DocumentRecovery.cpp.o CMakeFiles/FreeCADGui.dir/TaskElementColors.cpp.o CMakeFiles/FreeCADGui.dir/DlgObjectSelection.cpp.o CMakeFiles/FreeCADGui.dir/DlgAddProperty.cpp.o CMakeFiles/FreeCADGui.dir/VectorListEditor.cpp.o CMakeFiles/FreeCADGui.dir/DlgActionsImp.cpp.o CMakeFiles/FreeCADGui.dir/DlgCommandsImp.cpp.o CMakeFiles/FreeCADGui.dir/DlgCustomizeImp.cpp.o CMakeFiles/FreeCADGui.dir/DlgCustomizeSpaceball.cpp.o CMakeFiles/FreeCADGui.dir/DlgCustomizeSpNavSettings.cpp.o CMakeFiles/FreeCADGui.dir/DlgKeyboardImp.cpp.o CMakeFiles/FreeCADGui.dir/DlgToolbarsImp.cpp.o CMakeFiles/FreeCADGui.dir/DlgWorkbenchesImp.cpp.o CMakeFiles/FreeCADGui.dir/QListWidgetCustom.cpp.o CMakeFiles/FreeCADGui.dir/DlgEditorImp.cpp.o CMakeFiles/FreeCADGui.dir/DlgGeneralImp.cpp.o CMakeFiles/FreeCADGui.dir/DlgOnlineHelpImp.cpp.o CMakeFiles/FreeCADGui.dir/DlgPreferencesImp.cpp.o CMakeFiles/FreeCADGui.dir/DlgReportViewImp.cpp.o CMakeFiles/FreeCADGui.dir/DlgSettings3DViewImp.cpp.o CMakeFiles/FreeCADGui.dir/DlgSettingsNavigation.cpp.o CMakeFiles/FreeCADGui.dir/DlgSettingsSelection.cpp.o CMakeFiles/FreeCADGui.dir/DlgSettingsUnitsImp.cpp.o CMakeFiles/FreeCADGui.dir/DlgSettingsViewColor.cpp.o CMakeFiles/FreeCADGui.dir/DlgSettingsColorGradientImp.cpp.o CMakeFiles/FreeCADGui.dir/DlgSettingsDocumentImp.cpp.o CMakeFiles/FreeCADGui.dir/DlgSettingsImageImp.cpp.o CMakeFiles/FreeCADGui.dir/DlgSettingsLazyLoadedImp.cpp.o CMakeFiles/FreeCADGui.dir/DlgSettingsMacroImp.cpp.o CMakeFiles/FreeCADGui.dir/ComboView.cpp.o CMakeFiles/FreeCADGui.dir/DockWindow.cpp.o CMakeFiles/FreeCADGui.dir/PropertyView.cpp.o CMakeFiles/FreeCADGui.dir/ReportView.cpp.o CMakeFiles/FreeCADGui.dir/SelectionView.cpp.o CMakeFiles/FreeCADGui.dir/ToolBox.cpp.o CMakeFiles/FreeCADGui.dir/Tree.cpp.o CMakeFiles/FreeCADGui.dir/TreeView.cpp.o CMakeFiles/FreeCADGui.dir/ProjectView.cpp.o CMakeFiles/FreeCADGui.dir/DAGView/DAGView.cpp.o CMakeFiles/FreeCADGui.dir/DAGView/DAGModel.cpp.o CMakeFiles/FreeCADGui.dir/DAGView/DAGRectItem.cpp.o CMakeFiles/FreeCADGui.dir/DAGView/DAGModelGraph.cpp.o CMakeFiles/FreeCADGui.dir/DAGView/DAGFilter.cpp.o CMakeFiles/FreeCADGui.dir/CallTips.cpp.o CMakeFiles/FreeCADGui.dir/EditorView.cpp.o CMakeFiles/FreeCADGui.dir/TextDocumentEditorView.cpp.o CMakeFiles/FreeCADGui.dir/PythonConsole.cpp.o CMakeFiles/FreeCADGui.dir/PythonConsolePy.cpp.o CMakeFiles/FreeCADGui.dir/PythonDebugger.cpp.o CMakeFiles/FreeCADGui.dir/PythonEditor.cpp.o CMakeFiles/FreeCADGui.dir/SyntaxHighlighter.cpp.o CMakeFiles/FreeCADGui.dir/TextEdit.cpp.o CMakeFiles/FreeCADGui.dir/Assistant.cpp.o CMakeFiles/FreeCADGui.dir/NetworkRetriever.cpp.o CMakeFiles/FreeCADGui.dir/OnlineDocumentation.cpp.o CMakeFiles/FreeCADGui.dir/WhatsThis.cpp.o CMakeFiles/FreeCADGui.dir/Inventor/SoDrawingGrid.cpp.o CMakeFiles/FreeCADGui.dir/Inventor/SoAutoZoomTranslation.cpp.o CMakeFiles/FreeCADGui.dir/Inventor/MarkerBitmaps.cpp.o CMakeFiles/FreeCADGui.dir/Inventor/SmSwitchboard.cpp.o CMakeFiles/FreeCADGui.dir/SoFCBackgroundGradient.cpp.o CMakeFiles/FreeCADGui.dir/SoFCBoundingBox.cpp.o CMakeFiles/FreeCADGui.dir/SoFCColorBar.cpp.o CMakeFiles/FreeCADGui.dir/SoFCColorGradient.cpp.o CMakeFiles/FreeCADGui.dir/SoFCColorLegend.cpp.o CMakeFiles/FreeCADGui.dir/SoFCDB.cpp.o CMakeFiles/FreeCADGui.dir/SoFCInteractiveElement.cpp.o CMakeFiles/FreeCADGui.dir/SoFCOffscreenRenderer.cpp.o CMakeFiles/FreeCADGui.dir/SoFCSelection.cpp.o CMakeFiles/FreeCADGui.dir/SoFCUnifiedSelection.cpp.o CMakeFiles/FreeCADGui.dir/SoFCSelectionContext.cpp.o CMakeFiles/FreeCADGui.dir/SoFCSelectionAction.cpp.o CMakeFiles/FreeCADGui.dir/SoFCVectorizeSVGAction.cpp.o CMakeFiles/FreeCADGui.dir/SoFCVectorizeU3DAction.cpp.o CMakeFiles/FreeCADGui.dir/SoNavigationDragger.cpp.o CMakeFiles/FreeCADGui.dir/SoAxisCrossKit.cpp.o CMakeFiles/FreeCADGui.dir/SoTextLabel.cpp.o CMakeFiles/Fre [... cut ...] bin:/usr/local/opt/pyqt@5.15.2/bin:/usr/local/opt/vtk@8.2.0/bin:/usr/local/opt/webp/bin:/usr/local/opt/xerces-c/bin:/usr/bin:/bin:/usr/sbin:/sbin

Error: freecad/freecad/freecad HEAD-32200b6 did not build
Logs:
     /Users/yves/Library/Logs/Homebrew/freecad/00.options.out
     /Users/yves/Library/Logs/Homebrew/freecad/01.cmake.cc
     /Users/yves/Library/Logs/Homebrew/freecad/01.cmake
     /Users/yves/Library/Logs/Homebrew/freecad/02.make.cc
     /Users/yves/Library/Logs/Homebrew/freecad/02.make

vejmarie, 2021-02-11:
Hi, thanks for the bug report. I will have a look to it in about a week, as I am off grid up to next monday.

pilhuhn, 2021-02-14:
This may not be BigSur specific, as I am getting the same error on Catalina 10.15.7

vejmarie, 2021-03-10:
Can one of you post the whole output of the build ? I can't reproduce the issue.

yves-surrel, 2021-03-12:
Breaks now at 71%, still linking ../../lib/libFreeCADGui.dylib

Full log in attached file
[freecad.txt.zip](https://github.com/FreeCAD/homebrew-freecad/files/6130687/freecad.txt.zip)

yves-surrel, 2021-03-12:
Another trial, with --HEAD and without --with-macos-app

Fails with message

CMake Error in src/Gui/CMakeLists.txt:
  Imported target "Shiboken2::libshiboken" includes non-existent path

Full build text:
---------------------------------------------------------------------------
(base) MacBook-Pro-de-Yves:~ yves$ brew install --HEAD freecad --verbose
Updating Homebrew...
==> Auto-updated Homebrew!
Updated 1 tap (homebrew/core).
==> Updated Formulae
Updated 2 formulae.

Warning: Treating freecad as a formula. For the cask, use homebrew/cask/freecad
==> Installing freecad from freecad/freecad
==> Cloning https://github.com/freecad/FreeCAD.git
git --git-dir /Users/yves/Library/Caches/Homebrew/freecad--git/.git status -s
Updating /Users/yves/Library/Caches/Homebrew/freecad--git
git config remote.origin.url https://github.com/freecad/FreeCAD.git
git config remote.origin.fetch \+refs/heads/master:refs/remotes/origin/master
git config remote.origin.tagOpt --no-tags
git fetch origin
==> Checking out branch master
git checkout -f master --
Previous HEAD position was f35d30bc58 Fix travis build for MacOS
Switched to branch 'master'
Your branch is up to date with 'origin/master'.
git reset --hard origin/master --
HEAD is now at bfce108083 [OpenSCAD] Add scale parameter to linear_extrude
git --git-dir /Users/yves/Library/Caches/Homebrew/freecad--git/.git rev-parse --short=7 HEAD
Error: freecad 0.19pre is already installed
To install HEAD-bfce108, first run:
  brew unlink freecad
Warning: Skipping (old) /usr/local/Cellar/freecad/0.19pre due to it being linked
(base) MacBook-Pro-de-Yves:~ yves$ brew unlink freecad
Unlinking /usr/local/Cellar/freecad/0.19pre... 79 symlinks removed.
(base) MacBook-Pro-de-Yves:~ yves$ brew install --HEAD freecad --verbose
Warning: Treating freecad as a formula. For the cask, use homebrew/cask/freecad
==> Installing freecad from freecad/freecad
==> Cloning https://github.com/freecad/FreeCAD.git
git --git-dir /Users/yves/Library/Caches/Homebrew/freecad--git/.git status -s
Updating /Users/yves/Library/Caches/Homebrew/freecad--git
git config remote.origin.url https://github.com/freecad/FreeCAD.git
git config remote.origin.fetch \+refs/heads/master:refs/remotes/origin/master
git config remote.origin.tagOpt --no-tags
git fetch origin
==> Checking out branch master
git checkout -f master --
Already on 'master'
Your branch is up to date with 'origin/master'.
git reset --hard origin/master --
HEAD is now at bfce108083 [OpenSCAD] Add scale parameter to linear_extrude
git --git-dir /Users/yves/Library/Caches/Homebrew/freecad--git/.git rev-parse --short=7 HEAD
/usr/bin/sandbox-exec -f /private/tmp/homebrew20210312-2959-1anqq10.sb nice ruby -W1 -- /usr/local/Homebrew/Library/Homebrew/build.rb /usr/local/Homebrew/Library/Taps/freecad/homebrew-freecad/Formula/freecad.rb --verbose --HEAD
git --git-dir /Users/yves/Library/Caches/Homebrew/freecad--git/.git rev-parse --short=7 HEAD
hdiutil imageinfo -format /Users/yves/Library/Caches/Homebrew/freecad--git
cp -pR /Users/yves/Library/Caches/Homebrew/freecad--git/CODE_OF_CONDUCT.md /private/tmp/d20210312-2960-ngkcvx/CODE_OF_CONDUCT.md
cp -pR /Users/yves/Library/Caches/Homebrew/freecad--git/.arcconfig /private/tmp/d20210312-2960-ngkcvx/.arcconfig
cp -pR /Users/yves/Library/Caches/Homebrew/freecad--git/subuser/. /private/tmp/d20210312-2960-ngkcvx/subuser
cp -pR /Users/yves/Library/Caches/Homebrew/freecad--git/environment-linux.yml /private/tmp/d20210312-2960-ngkcvx/environment-linux.yml
cp -pR /Users/yves/Library/Caches/Homebrew/freecad--git/CMakeLists.txt /private/tmp/d20210312-2960-ngkcvx/CMakeLists.txt
cp -pR /Users/yves/Library/Caches/Homebrew/freecad--git/.subuser.json /private/tmp/d20210312-2960-ngkcvx/.subuser.json
cp -pR /Users/yves/Library/Caches/Homebrew/freecad--git/LICENSE /private/tmp/d20210312-2960-ngkcvx/LICENSE
cp -pR /Users/yves/Library/Caches/Homebrew/freecad--git/requirements.txt /private/tmp/d20210312-2960-ngkcvx/requirements.txt
cp -pR /Users/yves/Library/Caches/Homebrew/freecad--git/cMake/. /private/tmp/d20210312-2960-ngkcvx/cMake
cp -pR /Users/yves/Library/Caches/Homebrew/freecad--git/conda/. /private/tmp/d20210312-2960-ngkcvx/conda
cp -pR /Users/yves/Library/Caches/Homebrew/freecad--git/vagrant/. /private/tmp/d20210312-2960-ngkcvx/vagrant
[... 520 lines cut ...]
PKG_CONFIG_LIBDIR: /usr/lib/pkgconfig:/usr/local/Homebrew/Library/Homebrew/os/mac/pkgconfig/11
HOMEBREW_GIT: git
HOMEBREW_SDKROOT: /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk
ACLOCAL_PATH: /usr/local/share/aclocal
PATH: /usr/local/Homebrew/Library/Homebrew/shims/mac/super:/usr/local/opt/ccache/bin:/usr/local/opt/cmake/bin:/usr/local/opt/swig/bin:/usr/local/opt/icu4c@67.1/bin:/usr/local/opt/gdbm/bin:/usr/local/opt/openssl@1.1/bin:/usr/local/opt/sqlite/bin:/usr/local/opt/xz/bin:/usr/local/opt/gettext/bin:/usr/local/opt/python3.9/bin:/usr/local/opt/coin@4.0.0/bin:/usr/local/opt/libpng/bin:/usr/local/opt/freetype/bin:/usr/local/opt/gcc/bin:/usr/local/opt/numpy@1.19.4/bin:/usr/local/opt/hdf5@1.10/bin:/usr/local/opt/med-file/bin:/usr/local/opt/opencascade@7.5.0/bin:/usr/local/opt/qt5152/bin:/usr/local/opt/llvm/bin:/usr/local/opt/shiboken2/bin:/usr/local/opt/pyside2-tools/bin:/usr/local/opt/hwloc/bin:/usr/local/opt/libevent/bin:/usr/local/opt/open-mpi/bin:/usr/local/opt/pkg-config/bin:/usr/local/opt/fontconfig/bin:/usr/local/opt/hdf5/bin:/usr/local/opt/jpeg/bin:/usr/local/opt/libtiff/bin:/usr/local/opt/netcdf/bin:/usr/local/opt/sip@4.19.24/bin:/usr/local/opt/pyqt@5.15.2/b [... cut ...]
Auto-bottling for homebrew-freecad tap formulae https://github.com/FreeCAD/homebrew-freecad/issues/63
homebrew build-from-source post-install step issue with homebrew-freecad-bundle.pth check https://github.com/FreeCAD/homebrew-freecad/issues/154
new freecad git source merged some 3rd party code binding to MacOS sdk 11.0 but now is 11.1 https://github.com/FreeCAD/homebrew-freecad/issues/135
Please create pull requests instead of asking for help on Homebrew's GitHub,
Twitter or any other official channels.

vejmarie, 2021-03-12:
can you try to reinstall shiboken2 ? brew reinstall freecad/freecad/shiboken2

yves-surrel, 2021-03-12:
Got:

Warning: freecad/freecad/shiboken2 dependency gcc was built with a different C++ standard

vejmarie, 2021-03-12:
This warning is ok. Do you have any other shiboken library installed ? ( brew search shiboken2)

yves-surrel, 2021-03-12:
No.
Apparently it has fixed the shiboken problem, it is now building (takes ages !)

yves-surrel, 2021-03-12:
Sorry, it did not fix !

CMake Error in src/Gui/CMakeLists.txt:
  Imported target "Shiboken2::libshiboken" includes non-existent path

    "/usr/local/Cellar/python3.9/3.9.0_5/Frameworks/Python.framework/Versions/3.9/include/python3.9

yves-surrel, 2021-03-12:
Only one shiboken2 after search

vejmarie, 2021-03-12:
it is more a python related issue.
What about brew search python

can you also do brew reinstall freecad/freecad/python3.9

yves-surrel, 2021-03-12:
(base) MacBook-Pro-de-Yves:~ yves$ brew search python
==> Formulae
app-engine-python                              gst-python                                     python-yq
boost-python                                   ipython                                        python@3.7
boost-python3                                  micropython                                    python@3.8
bpython                                        ptpython                                       python@3.9 ✔
freecad/freecad/boost-python3@1.75.0 ✔         python-markdown                                reorder-python-imports
freecad/freecad/python3.9 ✔                    python-tabulate                                wxpython
==> Casks
awips-python                                   kk7ds-python-runtime                           mysql-connector-python

If you meant "python" specifically:
It was migrated from homebrew/cask to homebrew/core.

vejmarie, 2021-03-12:
can you check ls /usr/local/Cellar/python3.9/3.9.0_5/Frameworks/Python.framework/Versions/3.9/include/python3.9
I got Python include file into the directory

yves-surrel, 2021-03-12:
Still
Imported target "Shiboken2::libshiboken" includes non-existent path

after having done 
brew reinstall freecad/freecad/python3.9

yves-surrel, 2021-03-12:
ls /usr/local/Cellar/python3.9/3.9.0_5/Frameworks/Python.framework/Versions/3.9/include/python3.9
Python-ast.h		dictobject.h		longintrepr.h		pyconfig.h		pythread.h
Python.h		dynamic_annotations.h	longobject.h		pyctype.h		pytime.h
abstract.h		enumobject.h		marshal.h		pydebug.h		rangeobject.h
asdl.h			errcode.h		memoryobject.h		pydtrace.h		setobject.h
ast.h			eval.h			methodobject.h		pydtrace_probes.h	sliceobject.h
bitset.h		exports.h		modsupport.h		pyerrors.h		structmember.h
bltinmodule.h		fileobject.h		moduleobject.h		pyexpat.h		structseq.h
boolobject.h		fileutils.h		namespaceobject.h	pyfpe.h			symtable.h
bytearrayobject.h	floatobject.h		node.h			pyframe.h		sysmodule.h
bytesobject.h		frameobject.h		object.h		pyhash.h		token.h
cellobject.h		funcobject.h		objimpl.h		pylifecycle.h		traceback.h
ceval.h			genericaliasobject.h	odictobject.h		pymacconfig.h		tracemalloc.h
classobject.h		genobject.h		opcode.h		pymacro.h		tupleobject.h
code.h			graminit.h		osdefs.h		pymath.h		typeslots.h
codecs.h		grammar.h		osmodule.h		pymem.h			ucnhash.h
compile.h		import.h		parsetok.h		pyport.h		unicodeobject.h
complexobject.h		internal		patchlevel.h		pystate.h		warnings.h
context.h		interpreteridobject.h	picklebufobject.h	pystrcmp.h		weakrefobject.h
cpython			intrcheck.h		py_curses.h		pystrhex.h
datetime.h		iterobject.h		pyarena.h		pystrtod.h
descrobject.h		listobject.h		pycapsule.h		pythonrun.h

yves-surrel, 2021-03-12:
I run

conda deactivate

and once again brew install --HEAD freecad --verbose without any change.

vejmarie, 2021-03-12:
I never tried the setup with conda and homebrew installed at the same time. But let's pursue the debugging. What about the output of brew search freecad

vejmarie, 2021-03-12:
I think you are missing pyside2 (shiboken is a sibling of it ;))

yves-surrel, 2021-03-12:
MacBook-Pro-de-Yves:~ yves$ brew search freecad
==> Formulae
freecad/freecad/boost-python3@1.75.0 ✔         freecad/freecad/nglib ✔                        freecad/freecad/python3.9 ✔
freecad/freecad/boost@1.75.0 ✔                 freecad/freecad/numpy@1.19.4 ✔                 freecad/freecad/qt5152 ✔
freecad/freecad/coin@4.0.0 ✔                   freecad/freecad/opencamlib ✔                   freecad/freecad/shiboken2 ✔
freecad/freecad/cython@0.29.21                 freecad/freecad/opencascade@7.5.0 ✔            freecad/freecad/sip@4.19.24 ✔
freecad/freecad/freecad ✔                      freecad/freecad/pivy ✔                         freecad/freecad/swig@4.0.2
freecad/freecad/icu4c@67.1 ✔                   freecad/freecad/pyqt@5.15.2 ✔                  freecad/freecad/tbb@2020_u3 ✔
freecad/freecad/matplotlib ✔                   freecad/freecad/pyside2 ✔                      freecad/freecad/vtk@8.2.0 ✔
freecad/freecad/med-file ✔                     freecad/freecad/pyside2-tools ✔
==> Casks
freecad                                                                homebrew/cask-versions/freecad-pre

vejmarie, 2021-03-12:
can you reinstall pyside2 from your CMake output pyside2 is not their

yves-surrel, 2021-03-12:
MacBook-Pro-de-Yves:~ yves$ brew search pyside2
==> Formulae
freecad/freecad/pyside2 ✔                      freecad/freecad/pyside2-tools ✔                pyside@2

vejmarie, 2021-03-12:
remove the pyside@2 and brew reinstall freecad/freecad/pyside2

yves-surrel, 2021-03-12:
brew uninstall pyside@2 does not work. How should I do ?

vejmarie, 2021-03-12:
brew remove pyside@2

yves-surrel, 2021-03-12:
MacBook-Pro-de-Yves:~ yves$ brew remove pyside@2
Error: No available formula or cask with the name "pyside@2".

pilhuhn, 2021-03-12:
brew remove pyside2
Error: Refusing to uninstall /usr/local/Cellar/pyside2/5.15.2
because it is required by pyside2-tools, which is currently installed.
You can override this and force removal with:
  brew uninstall --ignore-dependencies pyside2

vejmarie, 2021-03-12:
ok, just reinstall it then

brew reinstall freecad/freecad/pyside2

yves-surrel, 2021-03-12:
MacBook-Pro-de-Yves:~ yves$ brew uninstall --ignore-dependencies pyside2
Uninstalling /usr/local/Cellar/pyside2/5.15.2... (233 files, 40.8MB)
MacBook-Pro-de-Yves:~ yves$ brew search pyside
==> Formulae
freecad/freecad/pyside2            freecad/freecad/pyside2-tools ✔    pyside ✔                           pyside@2

vejmarie, 2021-03-12:
remove also pyside

yves-surrel, 2021-03-12:
No success

vejmarie, 2021-03-12:
ok. Let me know what was the latest error, and keep posting logs. You do have a conflict somewhere for sure

pilhuhn, 2021-03-12:
I have now also tried several permutations of the above wrt python, shiboken and so and always end up in breakage with a message of 

```
CMake Error in src/Gui/CMakeLists.txt:
  Imported target "Shiboken2::libshiboken" includes non-existent path

    "/usr/local/Cellar/python3.9/3.9.0_5/Frameworks/Python.framework/Versions/3.9/include/python3.9

  "

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

```

Of course the listed directory is filled with files like for @yves-surrel . What puzzles me though is how the erraneous path is listed. It has `"` directly before `/usr/local..` but not directly after `pyhton3.9` which makes me wonder if there are several LF and even space chars at the end, which cause the breakage

vejmarie, 2021-03-12:
I can't reproduce the issue on my machine yet. Working on it, but as long as I can't see it is tough to fix. Do you have conda installed also as @yves-surrel ? Or just homebrew ? Which XCode version do you run ?

vejmarie, 2021-03-12:
@pilhuhn can you post the whole output of your cmake command. I am trying to understand where does this error happens.

yves-surrel, 2021-03-13:
The comment of @pilhuhn about possible extraneous LF's in the path construction makes sense to me...

yves-surrel, 2021-03-13:
If you look at the "summary report" part of the build log, you see the entry

-- Shiboken2:           5.15.2 [/usr/local/opt/shiboken2/lib/cmake/Shiboken2-5.15.2] [/usr/local/opt/shiboken2/include/shiboken2;/usr/local/Cellar/python3.9/3.9.0_5/Frameworks/Python.framework/Versions/3.9/include/python3.9
]

with a LF before the closing bracket. Other entries do not show this.

==============
Summary report
==============

-- Build type:          Release
-- Compiler:            /usr/local/Homebrew/Library/Homebrew/shims/mac/super/clang++ (12.0.0.12000032)
-- Flags:               -Wall -Wextra -Wpedantic -Wno-write-strings  -Wno-undefined-var-template
-- Standard:            Requires C++14
-- Python:              3.9 [/usr/local/opt/python3.9/bin/python3.9] [.cpython-39-darwin]
-- PCL:                 not enabled
-- pybind11:            not enabled
-- Boost:               1.75.0 (1.75.0)
-- XercesC:             3.2.3 [/usr/local/lib/libxerces-c.dylib] [/usr/local/include]
-- ZLIB:                1.2.11
-- PyCXX:               6.2.8 [/tmp/freecad-20210313-27599-amrta6/src]
-- OCC:                 7.5.0 [TKFillet;TKMesh;TKernel;TKG2d;TKG3d;TKMath;TKIGES;TKSTL;TKShHealing;TKXSBase;TKBool;TKBO;TKBRep;TKTopAlgo;TKGeomAlgo;TKGeomBase;TKOffset;TKPrim;TKSTEPBase;TKSTEPAttr;TKSTEP209;TKSTEP;TKHLR;TKFeat] [/usr/local/opt/opencascade@7.5.0/lib] [/usr/local/opt/opencascade@7.5.0/include/opencascade]
-- SMESH:               build internal
--  MEDFile:            4.0.0 [/usr/local/lib/libmedC.dylib;/usr/local/lib/libmed.dylib] [/usr/local/include]
--  HDF5:               1.10.7
--  VTK:                8.2.0
-- NETGEN:              6.2.2007 (6.2.2007) [-DNO_PARALLEL_THREADS;-DOCCGEOMETRY;-DNETGEN_VERSION=395735] [] [/usr/local/opt/nglib/Contents/Resources/include/include;/usr/local/opt/nglib/Contents/Resources/include] [nglib] [/usr/local/opt/nglib/Contents/Resources/include/include;/usr/local/opt/nglib/Contents/Resources/include]
-- SWIG:                4.0.2
-- Eigen3               3.3.9
-- Qt5Core:             5.15.2
-- Qt5Network:          5.15.2
-- Qt5Xml:              5.15.2
-- Qt5XmlPatterns:      5.15.2
-- Qt5Widgets:          5.15.2
-- Qt5PrintSupport:     5.15.2
-- Qt5OpenGL:           5.15.2
-- Qt5Svg:              5.15.2
-- Qt5UiTools:          5.15.2
-- Qt5Concurrent:       5.15.2
-- Qt5WebEngineWidgets: 5.15.2
-- Shiboken2:           5.15.2 [/usr/local/opt/shiboken2/lib/cmake/Shiboken2-5.15.2] [/usr/local/opt/shiboken2/include/shiboken2;/usr/local/Cellar/python3.9/3.9.0_5/Frameworks/Python.framework/Versions/3.9/include/python3.9
]
-- PySide2:             5.15.2 [/usr/local/opt/pyside2/include/PySide2]
-- PySide2Tools:        [/usr/local/opt/qt5152/bin/uic] [/usr/local/opt/qt5152/bin/rcc]
-- Freetype:            2.10.4
-- OpenGL:              /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/OpenGL.framework
-- OpenGLU:             [/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/OpenGL.framework][/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/OpenGL.framework]
-- Coin3D:              4.0.0 [/usr/local/opt/coin@4.0.0/lib/libCoin.dylib] [/usr/local/opt/coin@4.0.0/include]
-- Pivy:                0.6.5
-- SPNAV:               [SPNAV_LIBRARY-NOTFOUND] [SPNAV_INCLUDE_DIR-NOTFOUND]
-- Matplotlib:          2.1.1
-- Rift:                not enabled (BUILD_VR)
-- Doxygen:             not found

yves-surrel, 2021-03-13:
Also in 

-- SHIBOKEN_PYTHON_INCLUDE_DIRS computed to value: '/usr/local/Cellar/python3.9/3.9.0_5/Frameworks/Python.framework/Versions/3.9/include/python3.9
'

yves-surrel, 2021-03-13:
Modified /usr/local/Cellar/shiboken2/5.15.2/lib/cmake/Shiboken2-5.15.2/shiboken_helpers.cmake

line 353

    if (SHIBOKEN_COMPUTE_INCLUDES_IS_CALLED_FROM_EXPORT)
        #TODO target_include_directories works on imported targets only starting with v3.11.0.
	message(STATUS
		"*** DEBUG *** '${PYTHON_INCLUDE_DIRS}'")
        set_property(TARGET Shiboken2::libshiboken
                     APPEND PROPERTY INTERFACE_INCLUDE_DIRECTORIES ${PYTHON_INCLUDE_DIRS})

and I get in the build text:

-- Found PythonLibs: /usr/local/opt/python3.9/Frameworks/Python.framework/Versions/3.9/lib/libpython3.9.dylib (Required is at least version "3") 
-- *** DEBUG *** '/usr/local/Cellar/python3.9/3.9.0_5/Frameworks/Python.framework/Versions/3.9/include/python3.9
'
-- SHIBOKEN_PYTHON_INCLUDE_DIRS computed to value: '/usr/local/Cellar/python3.9/3.9.0_5/Frameworks/Python.framework/Versions/3.9/include/python3.9
'
-- SHIBOKEN_PYTHON_LIBRARIES computed to value: '-undefined dynamic_lookup'

yves-surrel, 2021-03-13:
I succeeded in getting rid of the shxiboken error, by editing /usr/local/Cellar/shiboken2/5.15.2/lib/cmake/Shiboken2-5.15.2/shiboken_helpers.cmake

Line 355, I included

	set(PYTHON_INCLUDE_DIRS ${Python3_INCLUDE_DIRS})

as PYTHON_INCLUDE_DIRS includes a trailing LF

However, we are back to the initial problem, i.e. linking fails at 73 %

[ 73%] Built target FreeCADGui
make: *** [all] Error 2
git --git-dir /Users/yves/Library/Caches/Homebrew/freecad--git/.git rev-parse --short=7 HEAD

==> Formula
Tap: freecad/freecad
Path: /usr/local/Homebrew/Library/Taps/freecad/homebrew-freecad/Formula/freecad.rb
==> Configuration
HOMEBREW_VERSION: 3.0.5
ORIGIN: https://github.com/Homebrew/brew
HEAD: 8b677c0ddd5f96b5521c1b353a9ca07dfe57b80e
Last commit: 4 days ago
Core tap ORIGIN: https://github.com/Homebrew/homebrew-core
Core tap HEAD: 0ab28b6b836c125813f6b8c088bef8c36b7249b0
Core tap last commit: 2 hours ago
Core tap branch: master
HOMEBREW_PREFIX: /usr/local
HOMEBREW_CASK_OPTS: []
HOMEBREW_MAKE_JOBS: 8
Homebrew Ruby: 2.6.3 => /System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/bin/ruby
CPU: octa-core 64-bit kabylake
Clang: 12.0 build 1200
Git: 2.24.3 => /Library/Developer/CommandLineTools/usr/bin/git
Curl: 7.64.1 => /usr/bin/curl
macOS: 11.2.1-x86_64
CLT: 12.4.0.0.1.1610135815
Xcode: N/A
==> ENV
HOMEBREW_CC: clang
HOMEBREW_CXX: clang++
MAKEFLAGS: -j8
CMAKE_PREFIX_PATH: /usr/local/opt/icu4c@67.1:/usr/local/opt/openssl@1.1:/usr/local/opt/readline:/usr/local/opt/sqlite:/usr/local/opt/python3.9:/usr/local/opt/boost-python3@1.75.0:/usr/local/opt/coin@4.0.0:/usr/local/opt/openblas:/usr/local/opt/hdf5@1.10:/usr/local/opt/qt5152:/usr/local/opt/libffi:/usr/local/opt/llvm:/usr/local/opt/sip@4.19.24:/usr/local/opt/pyqt@5.15.2:/usr/local/opt/vtk@8.2.0:/usr/local
CMAKE_INCLUDE_PATH: /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/OpenGL.framework/Versions/Current/Headers
CMAKE_LIBRARY_PATH: /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/OpenGL.framework/Versions/Current/Libraries
CMAKE_FRAMEWORK_PATH: /usr/local/opt/python3.9/Frameworks:/usr/local/opt/qt5152/Frameworks
PKG_CONFIG_PATH: /usr/local/opt/icu4c@67.1/lib/pkgconfig:/usr/local/opt/openssl@1.1/lib/pkgconfig:/usr/local/opt/readline/lib/pkgconfig:/usr/local/opt/sqlite/lib/pkgconfig:/usr/local/opt/xz/lib/pkgconfig:/usr/local/opt/python3.9/lib/pkgconfig:/usr/local/opt/coin@4.0.0/lib/pkgconfig:/usr/local/opt/libpng/lib/pkgconfig:/usr/local/opt/freetype/lib/pkgconfig:/usr/local/opt/gmp/lib/pkgconfig:/usr/local/opt/isl/lib/pkgconfig:/usr/local/opt/mpfr/lib/pkgconfig:/usr/local/opt/openblas/lib/pkgconfig:/usr/local/opt/qt5152/lib/pkgconfig:/usr/local/opt/libffi/lib/pkgconfig:/usr/local/opt/shiboken2/lib/pkgconfig:/usr/local/opt/pyside2/lib/pkgconfig:/usr/local/opt/hwloc/lib/pkgconfig:/usr/local/opt/libevent/lib/pkgconfig:/usr/local/opt/open-mpi/lib/pkgconfig:/usr/local/opt/orocos-kdl/lib/pkgconfig:/usr/local/opt/fontconfig/lib/pkgconfig:/usr/local/opt/jpeg/lib/pkgconfig:/usr/local/opt/libtiff/lib/pkgconfig:/usr/local/opt/netcdf/lib/pkgconfig:/usr/local/opt/webp/lib/pkgconfig:/usr/local/opt/xerces-c/lib/pkgconfig:/usr/local/opt/eigen/share/pkgconfig
PKG_CONFIG_LIBDIR: /usr/lib/pkgconfig:/usr/local/Homebrew/Library/Homebrew/os/mac/pkgconfig/11
HOMEBREW_GIT: git
HOMEBREW_SDKROOT: /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk
ACLOCAL_PATH: /usr/local/share/aclocal
PATH: /usr/local/Homebrew/Library/Homebrew/shims/mac/super:/usr/local/opt/ccache/bin:/usr/local/opt/cmake/bin:/usr/local/opt/swig/bin:/usr/local/opt/icu4c@67.1/bin:/usr/local/opt/gdbm/bin:/usr/local/opt/openssl@1.1/bin:/usr/local/opt/sqlite/bin:/usr/local/opt/xz/bin:/usr/local/opt/gettext/bin:/usr/local/opt/python3.9/bin:/usr/local/opt/coin@4.0.0/bin:/usr/local/opt/libpng/bin:/usr/local/opt/freetype/bin:/usr/local/opt/gcc/bin:/usr/local/opt/numpy@1.19.4/bin:/usr/local/opt/hdf5@1.10/bin:/usr/local/opt/med-file/bin:/usr/local/opt/opencascade@7.5.0/bin:/usr/local/opt/qt5152/bin:/usr/local/opt/llvm/bin:/usr/local/opt/shiboken2/bin:/usr/local/opt/pyside2-tools/bin:/usr/local/opt/hwloc/bin:/usr/local/opt/libevent/bin:/usr/local/opt/open-mpi/bin:/usr/local/opt/pkg-config/bin:/usr/local/opt/fontconfig/bin:/usr/local/opt/hdf5/bin:/usr/local/opt/jpeg/bin:/usr/local/opt/libtiff/bin:/usr/local/opt/netcdf/bin:/usr/local/opt/sip@4.19.24/bin:/usr/local/opt/pyqt@5.15.2/bin:/usr/local/opt/vtk@8.2.0/bin:/usr/local/opt/webp/bin:/usr/local/opt/xerces-c/bin:/usr/bin:/bin:/usr/sbin:/sbin

Error: freecad/freecad/freecad HEAD-89b8e81 did not build
Logs:
     /Users/yves/Library/Logs/Homebrew/freecad/00.options.out
     /Users/yves/Library/Logs/Homebrew/freecad/01.cmake.cc
     /Users/yves/Library/Logs/Homebrew/freecad/01.cmake
     /Users/yves/Library/Logs/Homebrew/freecad/02.make.cc
     /Users/yves/Library/Logs/Homebrew/freecad/02.make
If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
  https://github.com/freecad/homebrew-freecad/issues

/usr/bin/curl --disable --globoff --show-error --user-agent Homebrew/3.0.5\ \(Macintosh\;\ Intel\ Mac\ OS\ X\ 11.2.1\)\ curl/7.64.1 --header Accept-Language:\ en --retry 3 --location https://api.github.com/search/issues\?q=freecad\+repo\%3Afreecad\%2Fhomebrew-freecad\+state\%3Aopen\+in\%3Atitle\&per_page=100 --header Accept:\ application/vnd.github.v3\+json --write-out '
[... cut ...]
Auto-bottling for homebrew-freecad tap formulae https://github.com/FreeCAD/homebrew-freecad/issues/63
homebrew build-from-source post-install step issue with homebrew-freecad-bundle.pth check https://github.com/FreeCAD/homebrew-freecad/issues/154
new freecad git source merged some 3rd party code binding to MacOS sdk 11.0 but now is 11.1 https://github.com/FreeCAD/homebrew-freecad/issues/135
Please create pull requests instead of asking for help on Homebrew's GitHub,
Twitter or any other official channels.

pilhuhn, 2021-03-13:
@yves-surrel Your shiboken fix also works for me. 

I think what would be very helpful is if the build would not throw away temp artefacts. My 12CPU cores spin for solid 10 minutes at 100% on each compile run. @vejmarie Do you have an idea if it is possible to keep the intermediate artefacts. Should also help with debugging.

pilhuhn, 2021-03-13:
```
[ 70%] Linking CXX shared library ../../lib/libFreeCADGui.dylib
cd /tmp/freecad-20210313-42191-3hhbtk/Build/src/Gui && /usr/local/Cellar/cmake/3.19.6/bin/cmake -E cmake_link_script CMakeFiles/FreeCADGui.dir/link.txt --verbose=1
/usr/local/Homebrew/Library/Homebrew/shims/mac/super/clang++ -Wall -Wextra -Wpedantic -Wno-write-strings  -Wno-undefined-var-template -DNDEBUG -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk -dynamiclib -Wl,-headerpad_max_install_names  -Wl,-undefined,dynamic_lookup -o ../../lib/libFreeCADGui.dylib -install_name @rpath/libFreeCADGui.dylib CMakeFiles/FreeCADGui.dir/Application.cpp.o CMakeFiles/FreeCADGui.dir/ApplicationPy.cpp.o CMakeFiles/FreeCADGui.dir/AutoSaver.cpp.o CMakeFiles/FreeCADGui.dir/BitmapFactory.cpp.o CMakeFiles/FreeCADGui.dir/Document.cpp.o CMakeFiles/FreeCADGui.dir/DocumentModel.cpp.o CMakeFiles/FreeCADGui.dir/DocumentPyImp.cpp.o CMakeFiles/FreeCADGui.dir/DocumentObserver.cpp.o CMakeFiles/FreeCADGui.dir/DocumentObserverPython.cpp.o CMakeFiles/FreeCADGui.dir/ExpressionBinding.cpp.o CMakeFiles/FreeCADGui.dir/ExpressionBindingPy.cpp.o CMakeFiles/FreeCADGui.dir/GraphicsViewZoom.cpp.o CMakeFiles/FreeCADGui.dir/ExpressionCompleter.cpp.o CMakeFiles/FreeCADGui.dir/GuiApplication.cpp.o CMakeFiles/FreeCADGui.dir/GuiApplicationNativeEventAware.cpp.o CMakeFiles/FreeCADGui.dir/GuiConsole.cpp.o CMakeFiles/FreeCADGui.dir/Macro.cpp.o CMakeFiles/FreeCADGui.dir/MergeDocuments.cpp.o CMakeFiles/FreeCADGui.dir/resource.cpp.o CMakeFiles/FreeCADGui.dir/Control.cpp.o CMakeFiles/FreeCADGui.dir/SpaceballEvent.cpp.o CMakeFiles/FreeCADGui.dir/Thumbnail.cpp.o CMakeFiles/FreeCADGui.dir/Utilities.cpp.o CMakeFiles/FreeCADGui.dir/WaitCursor.cpp.o CMakeFiles/FreeCADGui.dir/ManualAlignment.cpp.o CMakeFiles/FreeCADGui.dir/TransactionObject.cpp.o CMakeFiles/FreeCADGui.dir/WinNativeGestureRecognizers.cpp.o CMakeFiles/FreeCADGui.dir/iisTaskPanel/src/moc_iisiconlabel.cpp.o CMakeFiles/FreeCADGui.dir/iisTaskPanel/src/moc_iistaskbox.cpp.o CMakeFiles/FreeCADGui.dir/iisTaskPanel/src/moc_iistaskgroup.cpp.o CMakeFiles/FreeCADGui.dir/iisTaskPanel/src/moc_iistaskheader.cpp.o CMakeFiles/FreeCADGui.dir/QSint/actionpanel/moc_actionbox.cpp.o CMakeFiles/FreeCADGui.dir/QSint/actionpanel/moc_actionlabel.cpp.o CMakeFiles/FreeCADGui.dir/QSint/actionpanel/moc_actiongroup.cpp.o CMakeFiles/FreeCADGui.dir/QSint/actionpanel/moc_actionpanel.cpp.o CMakeFiles/FreeCADGui.dir/QSint/actionpanel/moc_taskheader_p.cpp.o CMakeFiles/FreeCADGui.dir/qrc_resource.cpp.o CMakeFiles/FreeCADGui.dir/qrc_translation.cpp.o CMakeFiles/FreeCADGui.dir/qrc_iisTaskPanel.cpp.o CMakeFiles/FreeCADGui.dir/qrc_schemes.cpp.o CMakeFiles/FreeCADGui.dir/Action.cpp.o CMakeFiles/FreeCADGui.dir/ActionFunction.cpp.o CMakeFiles/FreeCADGui.dir/Command.cpp.o CMakeFiles/FreeCADGui.dir/CommandDoc.cpp.o CMakeFiles/FreeCADGui.dir/CommandFeat.cpp.o CMakeFiles/FreeCADGui.dir/CommandMacro.cpp.o CMakeFiles/FreeCADGui.dir/CommandStd.cpp.o CMakeFiles/FreeCADGui.dir/CommandWindow.cpp.o CMakeFiles/FreeCADGui.dir/CommandTest.cpp.o CMakeFiles/FreeCADGui.dir/CommandView.cpp.o CMakeFiles/FreeCADGui.dir/CommandStructure.cpp.o CMakeFiles/FreeCADGui.dir/CommandLink.cpp.o CMakeFiles/FreeCADGui.dir/CommandPyImp.cpp.o CMakeFiles/FreeCADGui.dir/Clipping.cpp.o CMakeFiles/FreeCADGui.dir/DemoMode.cpp.o CMakeFiles/FreeCADGui.dir/DlgActivateWindowImp.cpp.o CMakeFiles/FreeCADGui.dir/DlgUnitsCalculatorImp.cpp.o CMakeFiles/FreeCADGui.dir/DlgDisplayPropertiesImp.cpp.o CMakeFiles/FreeCADGui.dir/DlgInputDialogImp.cpp.o CMakeFiles/FreeCADGui.dir/DlgMacroExecuteImp.cpp.o CMakeFiles/FreeCADGui.dir/DlgRunExternal.cpp.o CMakeFiles/FreeCADGui.dir/DlgEditFileIncludePropertyExternal.cpp.o CMakeFiles/FreeCADGui.dir/DlgMacroRecordImp.cpp.o CMakeFiles/FreeCADGui.dir/DlgMaterialPropertiesImp.cpp.o CMakeFiles/FreeCADGui.dir/DlgParameterImp.cpp.o CMakeFiles/FreeCADGui.dir/DlgParameterFind.cpp.o CMakeFiles/FreeCADGui.dir/DlgProjectInformationImp.cpp.o CMakeFiles/FreeCADGui.dir/DlgProjectUtility.cpp.o CMakeFiles/FreeCADGui.dir/DlgPropertyLink.cpp.o CMakeFiles/FreeCADGui.dir/DlgExpressionInput.cpp.o CMakeFiles/FreeCADGui.dir/TaskDlgRelocation.cpp.o CMakeFiles/FreeCADGui.dir/DlgCheckableMessageBox.cpp.o CMakeFiles/FreeCADGui.dir/TaskCSysDragger.cpp.o CMakeFiles/FreeCADGui.dir/DlgUndoRedo.cpp.o CMakeFiles/FreeCADGui.dir/InputVector.cpp.o CMakeFiles/FreeCADGui.dir/Placement.cpp.o CMakeFiles/FreeCADGui.dir/PropertyPage.cpp.o CMakeFiles/FreeCADGui.dir/SceneInspector.cpp.o CMakeFiles/FreeCADGui.dir/TextureMapping.cpp.o CMakeFiles/FreeCADGui.dir/Transform.cpp.o CMakeFiles/FreeCADGui.dir/DownloadItem.cpp.o CMakeFiles/FreeCADGui.dir/DownloadManager.cpp.o CMakeFiles/FreeCADGui.dir/DocumentRecovery.cpp.o CMakeFiles/FreeCADGui.dir/TaskElementColors.cpp.o CMakeFiles/FreeCADGui.dir/DlgObjectSelection.cpp.o CMakeFiles/FreeCADGui.dir/DlgAddProperty.cpp.o CMakeFiles/FreeCADGui.dir/VectorListEditor.cpp.o CMakeFiles/FreeCADGui.dir/DlgActionsImp.cpp.o CMakeFiles/FreeCADGui.dir/DlgCommandsImp.cpp.o CMakeFiles/FreeCADGui.dir/DlgCustomizeImp.cpp.o CMakeFiles/FreeCADGui.dir/DlgCustomizeSpaceball.cpp.o CMakeFiles/FreeCADGui.dir/DlgCustomizeSpNavSettings.cpp.o CMakeFiles/FreeCADGui.dir/DlgKeyboardImp.cpp.o CMakeFiles/FreeCADGui.dir/DlgToolbarsImp.cpp.o CMakeFiles/FreeCADGui.dir/DlgWorkbenchesImp.cpp.o CMakeFiles/FreeCADGui.dir/QListWidgetCustom.cpp.o CMakeFiles/FreeCADGui.dir/DlgEditorImp.cpp.o CMakeFiles/FreeCADGui.dir/DlgGeneralImp.cpp.o CMakeFiles/FreeCADGui.dir/DlgOnlineHelpImp.cpp.o CMakeFiles/FreeCADGui.dir/Dl [... cut ...] local/opt/pyqt@5.15.2/bin:/usr/local/opt/vtk@8.2.0/bin:/usr/local/opt/webp/bin:/usr/local/opt/xerces-c/bin:/usr/bin:/bin:/usr/sbin:/sbin

Error: freecad/freecad/freecad HEAD-89b8e81 did not build
Logs:
     /Users/hrupp/Library/Logs/Homebrew/freecad/00.options.out
     /Users/hrupp/Library/Logs/Homebrew/freecad/01.cmake.cc
     /Users/hrupp/Library/Logs/Homebrew/freecad/01.cmake
     /Users/hrupp/Library/Logs/Homebrew/freecad/02.make.cc
     /Users/hrupp/Library/Logs/Homebrew/freecad/02.make
```

pilhuhn, 2021-03-13:
The logs from `/Users/hrupp/Library/Logs/Homebrew/freecad` - from previous comment
[fc-build.zip](https://github.com/FreeCAD/homebrew-freecad/files/6134590/fc-build.zip)

yves-surrel, 2021-03-13:
About the shiboken bug:

The problem comes from the macro 'shiboken_find_required_python' in /usr/local/Cellar/shiboken2/5.15.2/lib/cmake/Shiboken2-5.15.2/shiboken_helpers.cmake

As ${ARGC} is GREATER 0  (the macro is called with parameter '3'), the lines

        find_package(PythonInterp ${ARGV0} REQUIRED)
        find_package(PythonLibs ${ARGV0} REQUIRED)

are executed. The variable PYTHON_INCLUDE_DIRS comes out with an extraneous LF after the second one (PythonLibs), God knows why.

I replaced these lines by

      find_package (Python3 COMPONENTS Interpreter Development)

following [cmake documentation](https://cmake.org/cmake/help/latest/module/FindPython3.html#module:FindPython3) 

For @vejmarie information...

pilhuhn, 2021-03-13:
The link line `clang++ -Wall ....` has 18318 chars in my case. Could it be that some path length limit is hit and perhaps not all needed dependencies are on the argument list?

ipatch, 2021-03-13:
@yves-surrel

can please format your terminal output using code blocks, ie. use triple backticks <kbd>`</kbd> ?

ie. like

<pre>
```
terminal output
```
</pre>

makes reading your output much easier thus aiding in resolving the issue. it's that little key above your <kbd>tab</kbd> key

vejmarie, 2021-03-13:
> The logs from `/Users/hrupp/Library/Logs/Homebrew/freecad` - from previous comment
> [fc-build.zip](https://github.com/FreeCAD/homebrew-freecad/files/6134590/fc-build.zip)

Hi,

Your issue comes from tbb installation. Can you try:

brew search tbb (post the output)
brew remove tbb
brew install freecad/freecad/tbb@2020_u3

It shall be fixing your issue

ls -lta /usr/local/Cellar/tbb@2020_u3/2020_U3_1/include/tbb/parallel_for.h

vejmarie

vejmarie, 2021-03-13:
> /usr/local/Cellar/shiboken2/5.15.2/lib/cmake/Shiboken2-5.15.2/shiboken_helpers.cmake

@yves-surrel is it working with your proposed fix ? What is your MacOS default language, I am wondering if that could come from a LANG package issue. Mine is in english even if I am french ;)

pilhuhn, 2021-03-14:
```
snert:~ hrupp$ brew search tbb
==> Formulae
freecad/freecad/tbb@2020_u3 ✔                          libbtbb                                                tbb ✔
snert:~ hrupp$ brew remove tbb
Error: Refusing to uninstall /usr/local/Cellar/tbb/2020_U3_1
because it is required by opencascade and tbb@2020_u3, which are currently installed.
You can override this and force removal with:
  brew uninstall --ignore-dependencies tbb
snert:~ hrupp$ brew uninstall --ignore-dependencies tbb
Uninstalling /usr/local/Cellar/tbb/2020_U3_1... (144 files, 3.0MB)
snert:~ hrupp$ ls -lta /usr/local/Cellar/tbb@2020_u3/2020_U3_1/include/tbb/parallel_for.h
-rw-r--r--  1 hrupp  staff  21162 10 Jul  2020 /usr/local/Cellar/tbb@2020_u3/2020_U3_1/include/tbb/parallel_for.h
snert:~ hrupp$ brew reinstall freecad/freecad/tbb@2020_u3
==> Downloading https://dl.bintray.com/vejmarie/freecad/tbb%402020_u3-2020_U3_1.catalina.bottle.tar.gz
Already downloaded: /Users/hrupp/Library/Caches/Homebrew/downloads/ee306379c83938f4d979e4fd320e273690cbce6a1be27a3cba53f106ef39bd2e--tbb@2020_u3-2020_U3_1.catalina.bottle.tar.gz
==> Reinstalling freecad/freecad/tbb@2020_u3
==> Pouring tbb@2020_u3-2020_U3_1.catalina.bottle.tar.gz
🍺  /usr/local/Cellar/tbb@2020_u3/2020_U3_1: 144 files, 3MB
snert:~ hrupp$ ls -lta /usr/local/Cellar/tbb@2020_u3/2020_U3_1/include/tbb/parallel_for.h
-rw-r--r--  1 hrupp  staff  21162 10 Jul  2020 /usr/local/Cellar/tbb@2020_u3/2020_U3_1/include/tbb/parallel_for.h
snert:~ hrupp$
```

And my LANG is not English: 

```
$ echo $LANG
de_DE.UTF-8
```

vejmarie, 2021-03-14:
@pilhuhn Did it better found tbb during the build process now ?

pilhuhn, 2021-03-14:
@vejmarie It is still running at ~50% now. Wanted to paste the output nevertheless.

yves-surrel, 2021-03-14:
```
(base) MacBook-Pro-de-Yves:~ yves$ brew search tbb
==> Formulae
freecad/freecad/tbb@2020_u3 ✔                                 libbtbb                                                       tbb
(base) MacBook-Pro-de-Yves:~ yves$ brew remove tbb
Error: No available formula or cask with the name "tbb".
(base) MacBook-Pro-de-Yves:~ yves$ brew uninstall tbb
Error: No available formula or cask with the name "tbb".
```

yves-surrel, 2021-03-14:
```
(base) MacBook-Pro-de-Yves:~ yves$ brew doctor
Please note that these warnings are just used to help the Homebrew maintainers
with debugging if you file an issue. If everything you use Homebrew for is
working fine: please don't worry or file an issue; just ignore this. Thanks!

Warning: Some installed formulae are deprecated or disabled.
You should find replacements for the following formulae:
  vtk@8.2.0

Warning: "config" scripts exist outside your system or Homebrew directories.
`./configure` scripts often look for *-config scripts to determine if
software packages are installed, and which additional flags to use when
compiling and linking.

Having additional scripts in your path can confuse software installed via
Homebrew if the config script overrides a system or Homebrew-provided
script of the same name. We found the following "config" scripts:
  /Users/yves/miniconda3/bin/icu-config
  /Users/yves/miniconda3/bin/freetype-config
  /Users/yves/miniconda3/bin/smpeg2-config
  /Users/yves/miniconda3/bin/libpng16-config
  /Users/yves/miniconda3/bin/libpng-config
  /Users/yves/miniconda3/bin/xml2-config
  /Users/yves/miniconda3/bin/python3-config
  /Users/yves/miniconda3/bin/ncursesw6-config
  /Users/yves/miniconda3/bin/pcre-config
  /Users/yves/miniconda3/bin/python3.8-config
  /Users/yves/miniconda3/bin/sdl2-config

Warning: Unbrewed dylibs were found in /usr/local/lib.
If you didn't put them there on purpose they could cause problems when
building Homebrew formulae, and may need to be deleted.

Unexpected dylibs:
  /usr/local/lib/libStatTimer.2.14.0.dylib
  /usr/local/lib/libclFFT.2.14.0.dylib

Warning: Unbrewed header files were found in /usr/local/include.
If you didn't put them there on purpose they could cause problems when
building Homebrew formulae, and may need to be deleted.

Unexpected header files:
  /usr/local/include/clAmdFft.h
  /usr/local/include/clAmdFft.version.h
  /usr/local/include/clFFT.h
  /usr/local/include/clFFT.version.h

Warning: You have unlinked kegs in your Cellar.
Leaving kegs unlinked can lead to build-trouble and cause formulae that depend on
those kegs to fail to run properly once built. Run `brew link` on these:
  opencascade@7.5.0
  qt
  pyside2-tools
  freecad
  tbb@2020_u3
  netcdf
  boost@1.75.0
  python@3.9
```

pilhuhn, 2021-03-14:
This progressed now until the following. 
The logs from /Users/hrupp/Library/Logs/Homebrew/freecad: [fc-build2.zip](https://github.com/FreeCAD/homebrew-freecad/files/6136210/fc-build2.zip)

```
[100%] Built target PartDesignGui
/usr/local/Cellar/cmake/3.19.6/bin/cmake -E cmake_progress_start /tmp/freecad-20210314-73001-1l47hzj/Build/CMakeFiles 0
/Applications/Xcode.app/Contents/Developer/usr/bin/make  -f CMakeFiles/Makefile2 preinstall
make[1]: Nothing to be done for `preinstall'.
Install the project...
/usr/local/Cellar/cmake/3.19.6/bin/cmake -P cmake_install.cmake
-- Install configuration: "Release"
-- Installing: /usr/local/Cellar/freecad/HEAD-8ec3766/FreeCAD.app
-- Installing: /usr/local/Cellar/freecad/HEAD-8ec3766/FreeCAD.app/Contents
[...]

-- Installing: /usr/local/Cellar/freecad/HEAD-8ec3766/FreeCAD.app/Contents/libexec/matplotlib/lib/python3.9/site-packages/dateutil/tz/tz.py
-- Installing: /usr/local/Cellar/freecad/HEAD-8ec3766/FreeCAD.app/Contents/libexec/matplotlib/lib/python3.9/site-packages/dateutil/tz/_common.py
-- Installing: /usr/local/Cellar/freecad/HEAD-8ec3766/FreeCAD.app/Contents/libexec/matplotlib/lib/python3.9/site-packages/dateutil/tz/__init__.py
-- Installing: /usr/local/Cellar/freecad/HEAD-8ec3766/FreeCAD.app/Contents/libexec/matplotlib/lib/python3.9/site-packages/dateutil/tz/win.py
CMake Error at src/MacAppBundle/cmake_install.cmake:74 (file):
  file INSTALL cannot find
  "/tmp/freecad-20210314-73001-1l47hzj/src/MacAppBundle/import site": No such
  file or directory.
Call Stack (most recent call first):
  src/cmake_install.cmake:51 (include)
  cmake_install.cmake:68 (include)

make: *** [install] Error 1
git --git-dir /Users/hrupp/Library/Caches/Homebrew/freecad--git/.git rev-parse --short=7 HEAD

==> Formula
Tap: freecad/freecad
Path: /usr/local/Homebrew/Library/Taps/freecad/homebrew-freecad/Formula/freecad.rb
==> Configuration
HOMEBREW_VERSION: 3.0.5
ORIGIN: https://github.com/Homebrew/brew
HEAD: 8b677c0ddd5f96b5521c1b353a9ca07dfe57b80e
Last commit: 5 days ago
Core tap ORIGIN: https://github.com/Homebrew/homebrew-core
Core tap HEAD: 2b8a8a558e0b1e7f7cc0d962fdb35c5c3aa4c36e
Core tap last commit: 3 hours ago
Core tap branch: master
HOMEBREW_PREFIX: /usr/local
HOMEBREW_CASK_OPTS: []
HOMEBREW_DISPLAY: /private/tmp/com.apple.launchd.nEt0oRn2QL/org.macosforge.xquartz:0
HOMEBREW_GITHUB_API_TOKEN: set
HOMEBREW_MAKE_JOBS: 12
Homebrew Ruby: 2.6.3 => /usr/local/Homebrew/Library/Homebrew/vendor/portable-ruby/2.6.3_2/bin/ruby
CPU: dodeca-core 64-bit kabylake
Clang: 12.0 build 1200
Git: 2.30.2 => /usr/local/bin/git
Curl: 7.64.1 => /usr/bin/curl
macOS: 10.15.7-x86_64
CLT: 12.0.32.29
Xcode: 12.4
XQuartz: 2.7.11 => /opt/X11
==> ENV
HOMEBREW_CC: clang
HOMEBREW_CXX: clang++
MAKEFLAGS: -j12
CMAKE_PREFIX_PATH: /usr/local/opt/icu4c@67.1:/usr/local/opt/openssl@1.1:/usr/local/opt/readline:/usr/local/opt/sqlite:/usr/local/opt/python3.9:/usr/local/opt/boost-python3@1.75.0:/usr/local/opt/coin@4.0.0:/usr/local/opt/openblas:/usr/local/opt/hdf5@1.10:/usr/local/opt/qt5152:/usr/local/opt/libffi:/usr/local/opt/llvm:/usr/local/opt/sip@4.19.24:/usr/local/opt/pyqt@5.15.2:/usr/local/opt/vtk@8.2.0:/usr/local
CMAKE_INCLUDE_PATH: /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/System/Library/Frameworks/OpenGL.framework/Versions/Current/Headers
CMAKE_LIBRARY_PATH: /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/System/Library/Frameworks/OpenGL.framework/Versions/Current/Libraries
CMAKE_FRAMEWORK_PATH: /usr/local/opt/python3.9/Frameworks:/usr/local/opt/qt5152/Frameworks
PKG_CONFIG_PATH: /usr/local/opt/icu4c@67.1/lib/pkgconfig:/usr/local/opt/openssl@1.1/lib/pkgconfig:/usr/local/opt/readline/lib/pkgconfig:/usr/local/opt/sqlite/lib/pkgconfig:/usr/local/opt/xz/lib/pkgconfig:/usr/local/opt/python3.9/lib/pkgconfig:/usr/local/opt/coin@4.0.0/lib/pkgconfig:/usr/local/opt/libpng/lib/pkgconfig:/usr/local/opt/freetype/lib/pkgconfig:/usr/local/opt/gmp/lib/pkgconfig:/usr/local/opt/isl/lib/pkgconfig:/usr/local/opt/mpfr/lib/pkgconfig:/usr/local/opt/openblas/lib/pkgconfig:/usr/local/opt/qt5152/lib/pkgconfig:/usr/local/opt/libffi/lib/pkgconfig:/usr/local/opt/shiboken2/lib/pkgconfig:/usr/local/opt/pyside2/lib/pkgconfig:/usr/local/opt/hwloc/lib/pkgconfig:/usr/local/opt/libevent/lib/pkgconfig:/usr/local/opt/open-mpi/lib/pkgconfig:/usr/local/opt/orocos-kdl/lib/pkgconfig:/usr/local/opt/fontconfig/lib/pkgconfig:/usr/local/opt/jpeg/lib/pkgconfig:/usr/local/opt/libtiff/lib/pkgconfig:/usr/local/opt/netcdf/lib/pkgconfig:/usr/local/opt/webp/lib/pkgconfig:/usr/local/opt/xerces-c/lib/pkgconfig:/usr/local/opt/eigen/share/pkgconfig
PKG_CONFIG_LIBDIR: /usr/lib/pkgconfig:/usr/local/Homebrew/Library/Homebrew/os/mac/pkgconfig/10.15
HOMEBREW_GIT: git
HOMEBREW_SDKROOT: /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk
ACLOCAL_PATH: /usr/local/share/aclocal
PATH: /usr/local/Homebrew/Library/Homebrew/shims/mac/super:/usr/local/opt/ccache/bin:/usr/local/opt/cmake/bin:/usr/local/opt/swig/bin:/usr/local/opt/icu4c@67.1/bin:/usr/local/opt/gdbm/bin:/usr/local/opt/openssl@1.1/bin:/usr/local/opt/sqlite/bin:/usr/local/opt/xz/bin:/usr/local/opt/gettext/bin:/usr/local/opt/python3.9/bin:/usr/local/opt/coin@4.0.0/bin:/usr/local/opt/libpng/bin:/usr/local/opt/freetype/bin:/usr/local/opt/gcc/bin:/usr/local/opt/numpy@1.19.4/bin:/usr/local/opt/hdf5@1.10/bin:/usr/local/opt/med-file/bin:/usr/local/opt/opencascade@7.5.0/bin:/usr/local/opt/qt5152/bin:/usr/local/opt/llvm/bin:/usr/local/opt/shiboken2/bin:/usr/local/opt/pyside2-tools/bin:/usr/local/opt/hwloc/bin:/usr/local/opt/libevent/bin:/usr/local/opt/open-mpi/bin:/usr/local/opt/pkg-config/bin:/usr/local/opt/fontconfig/bin:/usr/local/opt/hdf5/bin:/usr/local/opt/jpeg/bin:/usr/local/opt/libtiff/bin:/usr/local/opt/netcdf/bin:/usr/local/opt/sip@4.19.24/bin:/usr/local/opt/pyqt@5.15.2/bin:/usr/local/opt/vtk@8.2.0/bin:/usr/local/opt/webp/bin:/usr/local/opt/xerces-c/bin:/usr/bin:/bin:/usr/sbin:/sbin

```

yves-surrel, 2021-03-14:
Hurray

After 
```
brew install tbb
```
```
brew remove tbb
```
```
brew link tbb@2020_u3
```
```
brew uninstall --ignore-dependencies freecad/freecad/tbb@2020_u3
```
```
brew install --HEAD freecad --verbose
```
I ended up with a splendid Freecad 0.20 (surprise!)

```
OS: macOS 11.2
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.20.24434 (Git)
Build type: Release
Branch: master
Hash: 8ec3766b00d672c15ce050d6a29a5d76f5602153
Python version: 3.9.0
Qt version: 5.15.2
Coin version: 4.0.0
OCC version: 7.5.0
Locale: French/France (fr_FR)
```
Will try now with the problematic --with-macos-app flag

yves-surrel, 2021-03-14:
Succeeded in having the MacOS app.

Thanks a lot.

@vejmarie what did you put on track of the tbb problem ?

pilhuhn, 2021-03-14:
@yves-surrel Did you change anything since yesterday except for what you describe in https://github.com/FreeCAD/homebrew-freecad/issues/159#issuecomment-798868680 ?

pilhuhn, 2021-03-14:
Yieha, the build now also succeeded after 33mins when not using the mac-app option.

Is it expected to get this warning?

```
Warning: The post-install step did not complete successfully
You can try again using:
  brew postinstall freecad/freecad/freecad

```

Running `/usr/local/bin/freecad` works as expected though (didn't do any throughout testing)

pilhuhn, 2021-03-14:
@vejmarie Thank you for all your help.
@vejmarie Sorry if that is a stupid question: can I prevent that git is overwriting local changes? I want to try a patch to the FC source before opening a PR. And/or how could I use a local development version from a different source dir instead of the one that `brew install` checks out?

yves-surrel, 2021-03-14:
@pilhuhn I had the same post-install message; seems not to be crucial.

[10 more replies]

closed 2021-04-12 by vejmarie