FreeCAD/homebrew-freecad #160: SHA256 mismatch with `brew install freecad --HEAD`

ornamentist, 2021-03-08:
On an M1 Mac mini running Big Sur I tried a:

brew install freecad --HEAD

With the M1 version of Homebrew.  I'm seeing this error:

Error: SHA256 mismatch
Expected: 4e59d1b8b2c672ae571966f3f7ce8d0c66dd3844d6eb3727012dd98c9e897a25
  Actual: 49574bb914e2564b21ab0fb23cadcccd7dd323ae7f0f26f64fd6346c3db14cd7
    File: /Users/.../Library/Caches/Homebrew/downloads/67c4afda4db78785c4cfbfe6d3707a39619f81d65489b81ff500e1ac97827a4a--ca3b5a50d945b6e65f0e764b3138cad17bd7eb8d.patch

Clearing that file and re-brewing doesn't seem to help.

vejmarie, 2021-03-08:
I am not sure which file we are speaking about. I am still waiting for my M1, but shall get it soon

ornamentist, 2021-03-08:
Just updated issue to include context download lines which I forgot to copy.

ipatch, 2021-03-08:
@vejmarie 

he is referring to the [patches](https://github.com/FreeCAD/homebrew-freecad/blob/master/Formula/vtk@8.2.0.rb#L26) in the vtk formula. for what it is work i believe homebrew core recently updated vtk

brainstorm, 2021-03-09:
I second this, more context after running `brew install --only-dependencies freecad --with-packaging-utilsim` as stated in the [official OSX compile docs](https://wiki.freecadweb.org/Compile_on_MacOS):

[image]

brainstorm, 2021-03-10:
After PR #160, there's another hash that is failing:

```
==> Downloading https://gitlab.kitware.com/vtk/vtk/commit/257b9d7b18d5f3db3fe099dc18f230e23f7dfbab.patch
-#O#- #   #
Error: SHA256 mismatch
Expected: d5eef4a022d7d18087c9267c632c79bd6ef312fc6a287aeacfc44e9d47a5ec91
  Actual: 41914057adc511527167b812f9604f1ff0aed74e12dc20c98bbc5c52ccab9cda
    File: /Users/rvalls/Library/Caches/Homebrew/downloads/d2791e69f47a5ed24bcd90999382ba7c6d18ef5088e55b95a53dd7a51053b6aa--257b9d7b18d5f3db3fe099dc18f230e23f7dfbab.patch
To retry an incomplete download, remove the file above.
```

@vejmarie Current mood: agreeable, please take the following comment as constructive criticism as it's what it is ;)

Now I see the real reason behind the comment on https://github.com/FreeCAD/homebrew-freecad/pull/158#issuecomment-794703718, this repo does not seem to have CI (Continuous Integration) configured?

I'm therefore not surprised that "maintainers just got exhausted due to dependencies which where breaking too often", this repo is in urgent need of a basic CI setup to shield devs from frequent breakage!

vejmarie, 2021-03-10:
@brainstorm No offenses at all. We need to setup a CI. Keep in mind that all FreeCAD developers work on their spare time. Now to get back on the issue with homebrew and why we had to "fork" formulas. Homebrew keeps updating their based formulas into core. We do have dependencies coming from many or a ton of libraries. Many of them are evolving fast with non compatible API. A good example is boost, which is running release cycle every 2 to 3 months and each time we have a new release we have compilation error as well as runtime one's. OpenCascade did accelerated also, and python is also evolving fast as well as Qt. Roughly an end user which is running homebrew has a very high probability to see its freeCAD installation breaking due to library updates from the core. We spent a lot of time and energy trying to work around this and discussing with homebrew people. The way we implemented it by getting Formulas from the dependencies where we have API that evolve and break our binary has been the best tradeoff found between us and homebrew core developpers. We agreed that this was the most stable way to get people happy with the software delivery even if that could create some frustration because you might not be using state of the art release. We are doing the same principles when building the snap and it works. It is easier for us to debug the code, and understand what broke or what did evolve and broke our code.

brainstorm, 2021-03-10:
I see, glad to see you have it in mind... is this XerCes error also a known issue with the current build/release system?:

```
(freecad_dev) rvalls@Romans-MBP build % cmake \
  -DCMAKE_BUILD_TYPE="Release"   \
  -DBUILD_QT5=1                  \
  -DWITH_PYTHON3=1               \
  -DCMAKE_PREFIX_PATH="$PREFIX_PATH" \
  -DPYTHON_EXECUTABLE="/Users/rvalls/.mambaforge/bin/python" \
  -DFREECAD_USE_EXTERNAL_KDL=1   \
  -DCMAKE_CXX_FLAGS='-std=c++14' \
  -DBUILD_FEM_NETGEN=1           \
  -DFREECAD_CREATE_MAC_APP=1     \
  -DCMAKE_INSTALL_PREFIX="./.."  \
  ..
CMake Deprecation Warning at CMakeLists.txt:8 (cmake_policy):
  The OLD behavior for policy CMP0072 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

-- Compiler: AppleClang, version: 12.0.0.12000032
-- Force BOOST_PP_VARIADICS=1 for clang
-- prefix: /Users/rvalls/dev/personal/FreeCAD
-- bindir: bin
-- datadir: share
-- docdir: share/doc/FreeCAD
-- includedir: include
-- libdir: lib
-- cmake: 3.19.6
-- Detected Homebrew install at /opt/homebrew
CMake Error at /opt/homebrew/Cellar/cmake/3.19.6/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:218 (message):
  Failed to find XercesC (missing: XercesC_LIBRARY XercesC_INCLUDE_DIR
  XercesC_VERSION)
Call Stack (most recent call first):
  /opt/homebrew/Cellar/cmake/3.19.6/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:582 (_FPHSA_FAILURE_MESSAGE)
  /opt/homebrew/Cellar/cmake/3.19.6/share/cmake/Modules/FindXercesC.cmake:105 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  cMake/FreeCAD_Helpers/SetupXercesC.cmake:4 (find_package)
  CMakeLists.txt:49 (SetupXercesC)

-- Configuring incomplete, errors occurred!
See also "/Users/rvalls/dev/personal/FreeCAD/build/CMakeFiles/CMakeOutput.log".
See also "/Users/rvalls/dev/personal/FreeCAD/build/CMakeFiles/CMakeError.log".
```

It seems that it got installed succesfully from the tap:

```==> Downloading https://homebrew.bintray.com/bottles/xerces-c-3.2.3.arm64_big_sur.bottle.1.tar.gz
Already downloaded: /Users/rvalls/Library/Caches/Homebrew/downloads/4f6c685b443d61ead3e35e9bdaa1a5bd4516c27e08d03051e9da60b50aed7afc--xerces-c-3.2.3.arm64_big_sur.bottle.1.tar.gz
==> Installing dependencies for freecad/freecad/freecad: gdbm, /python3.9, /cython@0.29.21, /numpy@1.19.4, /boost-python3@1.75.0, /coin@4.0.0, /matplotlib, /swig@4.0.2, hdf5@1.10, /med-file, rapidjson, freeimage, /tbb@2020_u3, /opencascade@7.5.0, /nglib, /opencamlib, /pivy, mpdecimal, python@3.9, sphinx-doc, /qt5152, libffi, llvm, /shiboken2, /pyside2, /pyside2-tools, hwloc, open-mpi, orocos-kdl, /sip@4.19.24, /pyqt@5.15.2, /vtk@8.2.0 and xerces-c
(...)
```

brainstorm, 2021-03-10:
> I take my CI comments back, it seems that there's [TravisCI mentioned](https://github.com/FreeCAD/homebrew-freecad#continuous-integration-support) in the README.

I take back my taking back...:

1) It seems like OSX is NOT even tested routinely on CI. It's disabled... Checking the build logs here: https://travis-ci.org/github/FreeCAD/FreeCAD/branches , only linux and windows seem to be built :-!

2) There are currently **three (!) different places** that document how to building for OSX, all of them with slight variations:

https://github.com/FreeCAD/FreeCAD/blob/master/BUILD_OSX.md
https://github.com/FreeCAD/FreeCAD/blob/master/.travis.yml#L209
https://wiki.freecadweb.org/Compile_on_MacOS

This should be consolidated to avoid confusion and perhaps just move to GitHub Actions, TBH.

vejmarie, 2021-03-10:
> > I take my CI comments back, it seems that there's [TravisCI mentioned](https://github.com/FreeCAD/homebrew-freecad#continuous-integration-support) in the README.
> 
> I take back my taking back...:
> 
> 1. It seems like OSX is NOT even tested routinely on CI. It's disabled... Checking the build logs here: https://travis-ci.org/github/FreeCAD/FreeCAD/branches , only linux and windows seem to be built :-!
> 2. There are currently **three (!) different places** that document how to building for OSX, all of them with slight variations:
> 
> https://github.com/FreeCAD/FreeCAD/blob/master/BUILD_OSX.md
> https://github.com/FreeCAD/FreeCAD/blob/master/.travis.yml#L209
> https://wiki.freecadweb.org/Compile_on_MacOS
> 
> This should be consolidated to avoid confusion and perhaps just move to GitHub Actions, TBH.
> 
> Given this evidence, it's unsurprising that so many issues crop up when trying to consolidate the release engineering as attempted in PR #158 for Apple M1's.

The MacOS build is disabled, because it is taking more than 2 hours on Travis CI machines even when we activate CCACHE. We need to negotiate more time or build up an infrastructure that allows us to run this on faster machine to build on MacOS with the CI. 

The Wiki shall be the source for latest infos (it doesn't mean it is free of bugs). Now as I said FreeCAD is community driven project, we have no financial backup and only volunteers which are doing their best to keep moving. If you want run a donation with a Mac machine to enhance Mac support we will be glad to take it (but we will be needing to find a datacenter where to host it). Let's work together to enhance all of these issues. I am building FreeCAD regularly on my Mac machine, but it is still an x86 build and not yet an M1. 

I have no issue to integrate patches and ideas, but take some time to discover the past of this 10+ years project.

brainstorm, 2021-03-10:
> The MacOS build is disabled, because it is taking more than 2 hours on Travis CI machines even when we activate CCACHE. We need to negotiate more time or build up an infrastructure that allows us to run this on faster machine to build on MacOS with the CI.

Instead of building Boost and other big dependencies from scratch, use bottles, as @andysan proposed in his PR. Don't throw more iron into the problem, fix the software :)

> The Wiki shall be the source for latest infos (it doesn't mean it is free of bugs). Now as I said FreeCAD is community driven project, we have no financial backup and only volunteers which are doing [...]

I'm fully aware this is volunteer driven, not arguing about that. No donations needed, I have a very similar experience with [MUSL-make build times (dependency for AWS Rust Lambdas)](https://github.com/FiloSottile/homebrew-musl-cross/issues/13#issuecomment-602173990). I forked that formula, bottled the dependencies up and it went from unavoidable CI timeouts to less than 2-5min installation.

TL;DR: 

Bottle dependencies up, use the upstream ones when possible and fix FreeCAD accordingly when dependencies change (via a coordinated effort and communication between devs, not stepping into each other).

With good CI, your developers should not suffer too much breakage across platforms, or at least, not as much as now.

vejmarie, 2021-03-10:
Ok, let's do it differently. I will approve the pull request and you will take care about the code breaking. I won't revert the code and you will take care about the next complains of the end users on the forum ?

By the way, even with bottles, wish you all the best to build freeCAD on travisCI in less than 2 hours. Right now their is bottles available for most of the formulas which have been "forked". They are x86 only I agree, but they do exist, which make their deployment fast, but you still need at least 15 minutes to deploy them as TravisCI (which is a VM) is really slow on the I/O side and homebrew is doing a crazy amount of I/O's

I understand your wish to improve all of this, but please take into consideration the work which has been made by previous developer and there is no magic, we really tested a lot of things. We know the app, the constraints and the complexity.

vejmarie, 2021-03-10:
Not sure you followed what I said, it is taking 2 hours to build FreeCAD only on TravisCI VM's. We do have 2 hours slot, and most of the build are ending prematurely. We use bottles to build FreeCAD on the CI and do not rebuild everything from source. Did you looked at the current formulas ? They all have a bottle. 

> > The MacOS build is disabled, because it is taking more than 2 hours on Travis CI machines even when we activate CCACHE. We need to negotiate more time or build up an infrastructure that allows us to run this on faster machine to build on MacOS with the CI.
> 
> Instead of building Boost and other big dependencies from scratch, use bottles, as @andysan proposed in his PR. Don't throw more iron into the problem, just fix the software :)

What we do is we forked the boost formula, we run a build of it, and publish our own bottles. Then when you install FreeCAD from the formula without the build from source switch you get it from a binary if you run on bigsur. If you want to build it you can install the dependencies without building them. The challenge with Travis is the time required to setup the build environment and building FreeCAD. Did you ever tried to do it  ?

ipatch, 2021-08-08:
closing this issue, as the topics discussed deviated to several issues not related to the original issue opened.

as for _sha256sum mismatch_ errors, all of those issues should be resolved with the [latest bottle release i published][1] a few days ago. if there are any future _sha256sum mismatch_ issues please open a new issue.

as for updating the _repo/tap_ with CI, that is definitely in the pipeline (no pun intended there) it's something i messing around with my current fork of this repo, but haven't fully got everything working/finished. i plan on setting up this repo to use github actions so that pr's can be sent to the repo, and repo maintainers can accept PR request, and new bottles should be built and generated if there are no issues with the PR, thus allowing for a more streamlined approach to keep the formulae in the repo updated. 🤞

[1]: <https://github.com/FreeCAD/homebrew-freecad/releases/tag/07.28.2021>

closed 2021-08-08 by ipatch