FreeCAD/homebrew-freecad #163: Freecad build fails on Big Sur during Install

dmingram66, 2021-03-12:
Everything complies but I get the following error during installation phase

brew reinstall freecad    
Warning: Treating freecad as a formula. For the cask, use homebrew/cask/freecad
==> Cloning https://github.com/freecad/freecad.git
Updating /Users/dingram/Library/Caches/Homebrew/freecad--git
==> Checking out revision f35d30bc58cc2000754d4f30cf29d063416cfb9e
HEAD is now at f35d30bc5 Fix travis build for MacOS
HEAD is now at f35d30bc5 Fix travis build for MacOS
==> Reinstalling freecad/freecad/freecad --with-macs-app
==> cmake -DBUILD_QT5=ON -DUSE_PYTHON3=1 -DPYTHON_EXECUTABLE=/usr/local/opt/pyth
==> make -j8 install

-- Installing: /usr/local/Cellar/freecad/0.19pre/FreeCAD.app/Contents/share/Gui/Stylesheets/images_dark-light/up_arrow_lighter.svg
CMake Error at src/MacAppBundle/cmake_install.cmake:55 (file):
  file INSTALL cannot find
  "/tmp/freecad-20210312-96801-9rbsq9/src/MacAppBundle/import site": No such
  file or directory.
Call Stack (most recent call first):
  src/cmake_install.cmake:51 (include)
  cmake_install.cmake:68 (include)

vejmarie, 2021-03-12:
Can you execute the resulting binaries ? It seems to fail when it build the App package. You shall still be able to use FreeCAD (even if this needs to be fixed)

vejmarie, 2021-03-12:
Sorry can you try to use the tap instead of the cask ?

zisoft, 2021-04-11:
Are there any news on this? I have the same problem on my Mac with MacOS Big Sur 11.2.3:

`CMake Error at src/MacAppBundle/cmake_install.cmake:74 (file):
  file INSTALL cannot find
  "/Users/mario/FreeCAD/source/src/MacAppBundle/import site": No such file or
  directory.
Call Stack (most recent call first):
  src/cmake_install.cmake:51 (include)
  cmake_install.cmake:68 (include)

make: *** [install] Error 1
`

vejmarie, 2021-04-12:
Can you test the upcoming PR ? It is fixing the issue (normally)

zisoft, 2021-04-12:
Thanks for your quick reply.
Unfortunately, this doesn't seem to have any effect.

- cloned your repo, branch macbundle
- emptied the build folder
- run cmake
- make -j6 install

Compilation went fine, but when it comes to the final install step it still says:

`
CMake Error at src/MacAppBundle/cmake_install.cmake:74 (file):
  file INSTALL cannot find
  "/Users/mario/FreeCAD/source/src/MacAppBundle/import site": No such file or
  directory.
Call Stack (most recent call first):
  src/cmake_install.cmake:51 (include)
  cmake_install.cmake:68 (include)

make: *** [install] Error 1
`

vejmarie, 2021-04-12:
What is your cmake command ? Because it look like that you are not building with hombrew (which is fine). The import site issue can be fixed by using an export PKG_CONFIG_PATH = "/usr/local/opt/icu4c@67.1/lib/pkgconfig//lib/pkgconfig" and remove default icu4c from your system (brew remove icu4c)

If that still doesn't work, I need your full cmake command. I have met your error yesterday while debugging, it was coming from a conflict. Can you also give me your src/MacAppBundle/cmake_install.cmake from your build directory. I can remember easily which package was causing the error.

zisoft, 2021-04-12:
Ok, I followed the steps.

brew remove --ignore-dependencies icu4c
export PKG_CONFIG_PATH...

This is my shell script to compile FreeCAD:

export PREFIX_PATH="/usr/local/Cellar/qt5152/5.15.2/lib/cmake;/usr/local/Cellar/nglib/v6.2.2007/Contents/Resources;/usr/local/Cellar/vtk@8.2.0/8.2.0_1/lib/cmake/vtk-8.2;/usr/local/opt/boost-python3@1.75.0/lib/cmake"
export PKG_CONFIG_PATH = "/usr/local/opt/icu4c@67.1/lib/pkgconfig//lib/pkgconfig"

cd ./build

cmake \
  -DCMAKE_BUILD_TYPE="Release"   \
  -DBUILD_QT5=1                  \
  -DWITH_PYTHON3=1               \
  -DCMAKE_PREFIX_PATH="$PREFIX_PATH" \
  -DPYTHON_EXECUTABLE="/usr/local/bin/python3" \
  -DPYTHON_INCLUDE_DIR=$(python -c "from distutils.sysconfig import get_python_inc; print(get_python_inc())") \
  -DFREECAD_USE_EXTERNAL_KDL=1   \
  -DCMAKE_CXX_FLAGS='-std=c++14' \
  -DBUILD_FEM_NETGEN=1           \
  -DFREECAD_CREATE_MAC_APP=1     \
  -DCMAKE_INSTALL_PREFIX="./.."  \
  ../source/

I need to set the PYTHON_ICNLUDE_DIR here because I am also a victim of this newline character at the end of the path.

But still no change in the result:

CMake Error at src/MacAppBundle/cmake_install.cmake:74 (file):
  file INSTALL cannot find
  "/Users/mario/FreeCAD/source/src/MacAppBundle/import site": No such file or
  directory.
Call Stack (most recent call first):
  src/cmake_install.cmake:51 (include)
  cmake_install.cmake:68 (include)

make: *** [install] Error 1

vejmarie, 2021-04-12:
ok I think you do have sip installed on your machine (the default one). Just do
brew remove sip
brew  install freecad/freecad/sip@4.19.24
brew link --force sip@4.19.24

This step shall fix your initial error into cmake (remove it before rebuilding otherwise it keeps it into the cache)

Be sure that all the package from freecad are properly linked on your syste,
brew link icu4c@67.1
brew link --force freecad/freecad/coin@4.0.0
and that you are using embedded python3.9 not @3.9
brew link --force python3.9
It shall work

zisoft, 2021-04-12:
Ok, I have followed all the steps. The PKG_CONFIG_PATH should be "/usr/local/opt/icu4c@67.1/lib/pkgconfig" right? (I assume you have a type in your comment above).

Did all the link steps you mentioned.

...still the exact same result...

Sorry, I'm long time developer veteran but I'm completely lost here.

vejmarie, 2021-04-12:
Yes got a typo sorry.  Can you post src/MacAppBundle/cmake_install.cmake from your build directory. I can have a look to the error.  The file is not properly generated during the cmake phase by one of imported module which is probably not linked as it shall on your machine. On mine the error was coming from the SIP package.

No worries to be lost, debugging freecad installation is not a complex task, but it involve so many dependencies that it could be frustrating. I am on discord or slack, it could be easier to have a look to if you want. Also on gitter.

vejmarie, 2021-04-12:
Look like your issue is coming from protobuf

/usr/local/Cellar/protobuf/3.15.8

Can you remove it from your machine ? I do not have it installed

zisoft, 2021-04-12:
That was as success. Installation completes without error now.
But the resulting FreeCAD.app can not be opened.

"FreeCAD.app cannot be opened due to an error" (I get this message in german).

vejmarie, 2021-04-12:
hey we are progressing. Weird message. Can you send me through email, or put the output of your compilation as an attachement into this issue. I can have a look if everything is ok. That might be an issue with the fact that your Mac is prohibiting the execution of non-signed app. You can open the app by rick clicking on the App and then open. It will force to remove the protection during the first run and your O/S will learn about it

You can remove your ccache from your home directory. Normally it is here ~/.ccache or $HOME/.ccache

zisoft, 2021-04-12:
Will try, this will take much longer of course. Please see my EDIT above

vejmarie, 2021-04-12:
Ok, do not recompile. I know from where this error comes
You need to use my freecad branch you are not using it I think.

git clone https://github.com/vejmarie/freecad

then checkout macbundle. It will fix this issue. Other option cherry pick the modifcations from this branch, that is a very small one.

If you pick up the branch, the error might comes from the fact that you need to fully clean your build environment and that the cmake has not been regenerated properly.

zisoft, 2021-04-12:
Hmmm, no, that's where I started (cloned your repo, branch macbundle).
Tried again, deleted the source directory and cloned again.
Deleted the contents of the build folder.

There is no .ccache folder in my HOME.

Run cmake again, then build.

Here are the last lines of the build process:

-- Making bundle relocatable...
-- INFO: Analyzing bundle dependencies...
-- ERROR: Unable to find LC_DYLD_LOAD entry: @rpath/libc++abi.1.dylib
-- ERROR: Failed to resolve dependency in /usr/local/opt/llvm/lib/libc++.1.dylib
Traceback (most recent call last):
  File "/Users/mario/FreeCAD/source/src/Tools/MakeMacBundleRelocatable.py", line 388, in <module>
    main()
  File "/Users/mario/FreeCAD/source/src/Tools/MakeMacBundleRelocatable.py", line 371, in main
    build_deps_graph(graph, bundle_path, dir_filter, search_paths)
  File "/Users/mario/FreeCAD/source/src/Tools/MakeMacBundleRelocatable.py", line 238, in build_deps_graph
    deps = create_dep_nodes(list_install_names(k2), s_paths)
  File "/Users/mario/FreeCAD/source/src/Tools/MakeMacBundleRelocatable.py", line 163, in create_dep_nodes
    raise LibraryNotFound(lib_name + " not found in given search paths")
__main__.LibraryNotFound: libc++abi.1.dylib not found in given search paths
-- Installing: /Users/mario/FreeCAD/FreeCAD.app/Contents/share/License.txt
-- Installing: /Users/mario/FreeCAD/FreeCAD.app/Contents/share/examples/Schenkel.stp
-- Installing: /Users/mario/FreeCAD/FreeCAD.app/Contents/share/examples/draft_test_objects.FCStd
-- Installing: /Users/mario/FreeCAD/FreeCAD.app/Contents/share/examples/EngineBlock.FCStd
-- Installing: /Users/mario/FreeCAD/FreeCAD.app/Contents/share/examples/PartDesignExample.FCStd
-- Installing: /Users/mario/FreeCAD/FreeCAD.app/Contents/share/examples/RobotExample.FCStd
-- Installing: /Users/mario/FreeCAD/FreeCAD.app/Contents/share/examples/ArchDetail.FCStd
-- Installing: /Users/mario/FreeCAD/FreeCAD.app/Contents/share/examples/FemCalculixCantilever2D.FCStd
-- Installing: /Users/mario/FreeCAD/FreeCAD.app/Contents/share/examples/FemCalculixCantilever3D.FCStd
-- Installing: /Users/mario/FreeCAD/FreeCAD.app/Contents/share/examples/FemCalculixCantilever3D_newSolver.FCStd

Giving up for today, will continue tomorrow, if you have any more hints.
Thank you for being so helpful with this!

vejmarie, 2021-04-12:
I think your cmake is lacking a lot of entries. Let me give them to you
cmake -DBUILD_QT5=ON -DUSE_PYTHON3=1 -DCMAKE_CXX_FLAGS:STRING="-I/usr/local/include -I/usr/local/opt/icu4c@67.1/include" -DPYTHON_EXECUTABLE=/usr/local/bin/python3 -DPYTHON_INCLUDE_DIR=/usr/local/opt/python3.9/Frameworks/Python.framework/Headers -DCMAKE_CXX_STANDARD=14 -DBUILD_ENABLE_CXX_STD:STRING=C++14 -DBUILD_FEM_NETGEN=1 -DBUILD_FEM=1 -DBUILD_FEM_NETGEN:BOOL=ON -DFREECAD_USE_EXTERNAL_KDL=ON -DCMAKE_PREFIX_PATH="/usr/local/opt/llvm/lib/cmake;/usr/local/opt/qt5152/lib/cmake;/usr/local/opt/nglib/Contents/Resources;/usr/local/opt/vtk@8.2.0/lib/cmake;/usr/local/opt/opencascade@7.5.0/lib/cmake/;/usr/local/opt/med-file/lib/cmake;/usr/local/opt/shiboken2/lib/cmake;/usr/local/opt/pyside2/lib/cmake;/usr/local/opt/coin@4.0.0/lib/cmake;/usr/local/opt/boost@1.75.0/lib/cmake;/usr/local/opt/boost-python3@1.75.0/lib/cmake;" -DFREECAD_CREATE_MAC_APP=1  ..

You need the llvm cmake path and the other remaining one's. It shall work with that now.

zisoft, 2021-04-12:
This gives me a bunch of compilation errors.
Will investigate further tomorrow.

Thank you again!

zisoft, 2021-04-12:
Last attempt for today :-)

After adding /usr/local/opt/llvm/lib/cmake to the PREFIX_PATH I now get

-- Making bundle relocatable...
-- INFO: Analyzing bundle dependencies...
error: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/otool-classic: can't open file: /usr/local/opt/icu4c/lib/libicuuc.68.dylib (No such file or directory)
-- ERROR: Failed to resolve dependency in /usr/local/opt/icu4c/lib/libicuuc.68.dylib
Traceback (most recent call last):
  File "/Users/mario/FreeCAD/source/src/Tools/MakeMacBundleRelocatable.py", line 388, in <module>
    main()
  File "/Users/mario/FreeCAD/source/src/Tools/MakeMacBundleRelocatable.py", line 371, in main
    build_deps_graph(graph, bundle_path, dir_filter, search_paths)
  File "/Users/mario/FreeCAD/source/src/Tools/MakeMacBundleRelocatable.py", line 238, in build_deps_graph
    deps = create_dep_nodes(list_install_names(k2), s_paths)
  File "/Users/mario/FreeCAD/source/src/Tools/MakeMacBundleRelocatable.py", line 111, in list_install_names
    output = check_output(["otool", "-L", path_macho])
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/subprocess.py", line 223, in check_output
    raise CalledProcessError(retcode, cmd, output=output)
subprocess.CalledProcessError: Command '['otool', '-L', '/usr/local/opt/icu4c/lib/libicuuc.68.dylib']' returned non-zero exit status 1

vejmarie, 2021-04-12:
Did you recompiled any other dependencies ? Look like that /usr/local/opt/icu4c/lib/libicuuc.68.dylib is one of your dependency. So what you can try is to re-install the regular icu4c through brew and see where the next error will be. (brew install icu4c)

zisoft, 2021-04-12:
brew install icu4

this (re)installed a bunch of other packages as dependencies, i.e. freecad/freecad/vtk@8.2.0

I just ran `make install` again in the build directory.
Couldn't see any errors anymore, so I launched FreeCAD.app and IT WORKS :-)

YOU ARE MY HERO!! :-)

Thank you so much for your efforts!
So does that mean once your PR is accepted I should switch to the official repo again?

This stuff drives me nuts. How can the documentation be updated to make this work for other developers out of the box?

Thank you again!!

vejmarie, 2021-04-12:
I think that when both of my PR will be merged, that will work transparently from homebrew (brew install --HEAD) and you will be able to rebuild nightly freecad on Mac.

Good to see it working. Enjoy FreeCAD !

ipatch, 2021-04-16:
@zisoft

i've been bottling up the freecad dependencies for macos mojave, and have been building and installing freecad using the formula file and also [manually building](https://github.com/ipatch/homebrew-us-05#-development--freecad--non-formula-build--install) from the git source using cmake from a cli.

I have installed all freecad dependencies from this tap, and when i tried to build freecad manually (not using the formula file) i ran into linking errors related to `-licudata -licui18n -licuuc` which is indirectly related to boost [from what i understand](https://github.com/Homebrew/homebrew-core/issues/67427#issuecomment-754187345).

either way i am able to resolve linking issues when defining a `LIBRARY_PATH` env var with the below setting

```shell
export LIBRARY_PATH="/usr/local/opt/icu4c/lib"
```

i know you resolved your issue but if something pops up in the future this could be something to experiment with down the road. and glad you got your issue resolved.

zisoft, 2021-04-16:
@ipatch 

Thank you for your hints. I now a have a configuration which compiles and links fine:

export PREFIX_PATH="/usr/local/opt/llvm/lib/cmake;/usr/local/Cellar/qt5152/5.15.2/lib/cmake;/usr/local/Cellar/nglib/v6.2.2007/Contents/Resources;/usr/local/Cellar/vtk@8.2.0/8.2.0_1/lib/cmake/vtk-8.2;/usr/local/opt/boost-python3@1.75.0/lib/cmake"
export PKG_CONFIG_PATH="/usr/local/opt/icu4c@67.1/lib/pkgconfig"

cd ./build

cmake \
  -DCMAKE_BUILD_TYPE="Release"   \
  -DBUILD_QT5=1                  \
  -DWITH_PYTHON3=1               \
  -DCMAKE_PREFIX_PATH="$PREFIX_PATH" \
  -DPYTHON_EXECUTABLE="/usr/local/bin/python3" \
  -DPYTHON_INCLUDE_DIR=$(python -c "from distutils.sysconfig import get_python_inc; print(get_python_inc())") \
  -DFREECAD_USE_EXTERNAL_KDL=1   \
  -DCMAKE_CXX_FLAGS='-std=c++14' \
  -DBUILD_FEM_NETGEN=1           \
  -DFREECAD_CREATE_MAC_APP=1     \
  -DFREECAD_COPY_PLUGINS_BIN_TO_BUILD=1 \
  -DFREECAD_USE_OCC_VARIANT="Official Version"
  -DCMAKE_INSTALL_PREFIX="./.."  \
  ../source/

Now, when I run FreeCAD, create a new document and open the Draft workbench, FreeCAD crashes.

Terminal output:

Persistent toolbars: Snapper toolbar not managed.
'NoneType' object has no attribute 'pnts'
Traceback (most recent call last):
  File "<string>", line 132, in Activated
  File "/Users/mario/FreeCAD/build/Mod/Draft/draftguitools/gui_snapper.py", line 1643, in show
    self.setTrackers()
  File "/Users/mario/FreeCAD/build/Mod/Draft/draftguitools/gui_snapper.py", line 1701, in setTrackers
    self.dim1 = trackers.archDimTracker(mode=2)
  File "/Users/mario/FreeCAD/build/Mod/Draft/draftguitools/gui_trackers.py", line 1317, in __init__
    self.dimnode.pnts.setValues([p1node, p2node])

[1]    91404 segmentation fault  /Users/mario/FreeCAD/build/bin/FreeCAD
Saving session...completed.

This is all very frustating.

ipatch, 2021-04-16:
presently i don't have a big sur box setup, but i plan to have one going sooner than latter to hopefully test these things. that said are you building freecad on an m1 mac?

also something i've been meaning to do is update some of my documentation related to building freecad, and something i learned from a few posts on the freecad forum was how to setup the build and source directories. and looking at your response, it looks like your doing an in source build which is something that has presented issues to me in the past.

i'd recommend setting up your folder/directory structure as follows,

```
/path/to/freecad-git
/path/to/freecad-git/freecad-source/
/path/to/freecad-git/freecad-build-dir/
```

with following directory structure you keep your build directory as a peer folder of the source directory and not within the source directory itself. so you'd end with something like,

```
echo "cd into build directory";
cd /path/to/freecad-git/freecad-build-dir;
cmake \
[INSERT CMAKE ARGS] \
../freecad-source
```

zisoft, 2021-04-17:
Yes, that is exactly what I'm doing. This is my directory structure:

    ~/FreeCAD/source
    ~/FreeCAD/build

I'm running a MacBook Pro 16" (2019), Intel Core I9

vejmarie, 2021-04-17:
can you tell me which git commit you are compiling. I will try to reproduce your issue on my machine. Can you also display into freecad the python console and tell me which python interpreter version is displayed.

zisoft, 2021-04-17:
I have forced a 

    brew link --overwrite python@3.9

and this seemed to solve the problem. No crashes anymore so far.

Python console in freecad says:

    Python 3.9.4 (default, Apr  5 2021, 01:50:46) 
    [Clang 12.0.0 (clang-1200.0.32.29)] on darwin

My git commit is master from https://github.com/FreeCAD/FreeCAD

The dependencies with brew are a nightmare. I really don't know what steps I've tried, just followed what you told me to do. But if I have to setup a new environment I probably cannot reproduce it.

vejmarie, 2021-04-21:
> I have forced a
>
> ```
> brew link --overwrite python@3.9
> ```
>
> and this seemed to solve the problem. No crashes anymore so far.
>
> Python console in freecad says:
>
> ```
> Python 3.9.4 (default, Apr  5 2021, [...]

Ok, it probably means that you needed to link python before the build ;). Good to see it working now

zisoft, 2021-04-27:
The FreeCAD.app is only runnable in the folder in which it was built. As soon as I move it to another folder (i.e. /Applications) it quits with an error message. The report says:

    Termination Reason:    DYLD, [0x1] Library missing

    Application Specific Information:
    dyld: launch, loading dependent libraries

    Dyld Error Message:
      dyld: Using shared cache: 0781246A-4850-3632-8FD5-CCB70AA8F9B8
    Library not loaded: @rpath/libFreeCADGui.dylib
      Referenced from: /Users/USER/FreeCAD.app/Contents/MacOS/FreeCAD
      Reason: image not found

Seems that the libs are linked dynamically. Is there any setting to link them statically into the app?

ipatch, 2021-04-27:
https://github.com/vejmarie/FreeCAD/commit/a5a448c85766d59bb3e71168906c704857a0db16#commitcomment-50072628

ipatch, 2024-06-10:
this issue should be resolved now, a simple `brew install freecad/freecad/freecad -v -s` should _just work_.

closed 2024-06-10 by ipatch