FreeCAD/homebrew-freecad #164: Cannot create relocatable bundle when building MacOS app with brew

yves-surrel, 2021-03-16:
At the end of the macOS X build 

```
brew install --HEAD --with-macos-app freecad --verbose
```
I get the error:

```
-- Making bundle relocatable...
-- INFO: Analyzing bundle dependencies...
-- ERROR: Unable to find LC_DYLD_LOAD entry: @rpath/libc++abi.1.dylib
-- ERROR: Failed to resolve dependency in /usr/local/opt/llvm/lib/libc++.1.dylib
Traceback (most recent call last):
  File "/tmp/freecad-20210316-74220-1lg8g5b/src/Tools/MakeMacBundleRelocatable.py", line 388, in <module>
    main()
  File "/tmp/freecad-20210316-74220-1lg8g5b/src/Tools/MakeMacBundleRelocatable.py", line 371, in main
    build_deps_graph(graph, bundle_path, dir_filter, search_paths)
  File "/tmp/freecad-20210316-74220-1lg8g5b/src/Tools/MakeMacBundleRelocatable.py", line 238, in build_deps_graph
    deps = create_dep_nodes(list_install_names(k2), s_paths)
  File "/tmp/freecad-20210316-74220-1lg8g5b/src/Tools/MakeMacBundleRelocatable.py", line 163, in create_dep_nodes
    raise LibraryNotFound(lib_name + " not found in given search paths")
__main__.LibraryNotFound: libc++abi.1.dylib not found in given search paths
```

This does not prevent to successfully obtain the Mac app, however I cannot just copy the Mac app into /Applications and then brew remove freecad (housekeeping), probably because of this error.

ipatch, 2021-03-16:
#142

vejmarie, 2021-04-12:
Please can you test with the upcoming PR (warning you need to use my own branch of freecad for testing. Edit freecad.rb and point github entry to it)

ipatch, 2021-04-13:
i haven't forgotten about your PR. presently i've built a majority of the formula from source and have created bottles for all the formula i have built so far. they can be found [here](https://github.com/ipatch/homebrew-freecad/releases/tag/april.12.2021)

started building the taps version of qt5152 which will probably take ~ 7 to 12 hours. so i'll post back when that finishes.

vejmarie, 2021-04-21:
@ipatch any news on your build ?

ipatch, 2021-04-21:
my bad, give me a few (by end of business day CST) and i'll let you know. i was working an updated PR for the freecad formula BTW

and thanks for the bump 🙏 totally slipped my mind.

vejmarie, 2021-04-21:
No worries ;). Just want to try to integrate latest PR to keep moving forward and being sure that we are improving instead of regressing

ipatch, 2021-04-21:
just started the build process of the formula using

```shell
brew install Formula/freecad.rb --verbose --with-macos-app
```

ipatch, 2021-04-24:
for future me 👴🏼

https://github.com/vejmarie/FreeCAD/commit/a5a448c85766d59bb3e71168906c704857a0db16

ipatch, 2021-12-02:
still working on trying to get `brew install freecad` to _just work™️_ as a _batteries included 🔋🔋_ type of install. pretty much have most if not all the build time issues resolved. i'm working through the run time issues presently, and have been running into issues getting the install process to properly recognize a pyside2 installation. until then i'm holding on removing any of the excess formula in this repo as this is the main issue i'm working on.

a little heads up of what it is i'm currently troubleshooting. freecad has a runtime dependency of pyside even when built into a self contained macos app bundle ie. a `FreeCAD.app`. so if the [python 2.7 command](https://github.com/FreeCAD/FreeCAD/blob/501c3dc65795baedff5daa4660e5addce5e51a56/src/MacAppBundle/CMakeLists.txt#L131) (yes that needs to be upgraded to work with python 3 eventually (that's an upstream issue onto its own)). presently freecad `0.19.2` has a dependency on qt5 and a runtime dependency of pyside2. using pyside2 a dependency install via macos homebrew begins to complicate things as it's a `keg-only` dependency, as certain files aren't symlinked into certain directories which complicates matters when cmake and python command begin to make this relocatable as "for reasons i don't understand yet" the dependencies aren't copied into the bundle, thus the when app bundle `FreeCAD.app` is launched by double clicking freecad, the `$USER` will be presented with the below runtime error message

```
18:31:43  During initialization the error "No module named 'PySide2'" occurred in /Users/brewmaster/homebrew/Cellar/freecad/0.19/FreeCAD.app/Contents/Mod/Draft/InitGui.py
18:31:43  Please look into the log file for further information
18:31:45  During initialization the error "No module named 'PySide2'" occurred in /Users/brewmaster/homebrew/Cellar/freecad/0.19/FreeCAD.app/Contents/Mod/Arch/InitGui.py
18:31:45  Please look into the log file for further information
18:31:45  During initialization the error "No module named 'PySide2'" occurred in /Users/brewmaster/homebrew/Cellar/freecad/0.19/FreeCAD.app/Contents/Mod/AddonManager/InitGui.py
18:31:45  Please look into the log file for further information
18:31:46  During initialization the error "No module named 'PySide2'" occurred in /Users/brewmaster/homebrew/Cellar/freecad/0.19/FreeCAD.app/Contents/Mod/Tux/InitGui.py
18:31:46  Please look into the log file for further information
18:31:48  Unknown command 'Std_AddonMgr'
18:31:48  Traceback (most recent call last):
  File "<string>", line 2, in <module>
  File "/Users/brewmaster/homebrew/Cellar/freecad/0.19/FreeCAD.app/Contents/Mod/Start/StartPage/StartPage.py", line 29, in <module>
    from . import TranslationTexts
  File "/Users/brewmaster/homebrew/Cellar/freecad/0.19/FreeCAD.app/Contents/Mod/Start/StartPage/TranslationTexts.py", line 24, in <module>
    from PySide import QtGui
  File "/Users/brewmaster/homebrew/Cellar/freecad/0.19/FreeCAD.app/Contents/MacOS/PySide/__init__.py", line 2, in <module>
    from PySide2 import __version__
<class 'ModuleNotFoundError'>: No module named 'PySide2'
18:31:48  No module named 'PySide2'
```

which will make the interface / GUI look like something below,

[image]

ipatch, 2021-12-10:
i'm closing this issue as `brew install freecad` now creates a relocatable `FreeCAD.app` will open a new issue related to the bundling of the pyside2@5.15.2 dependencies. and will add a reference to this issue.

closed 2021-12-10 by ipatch