FreeCAD/homebrew-freecad #165: Building FreeCAD --head CMake error in SHIBOKEN_PYTHON_INCLUDE_DIRS

nachumk, 2021-04-02:
I am building FreeCAD with HomeBrew:
brew install --HEAD freecad --with-macos-app --verbose

I get this error:
=================================================
Now run 'cmake --build /tmp/freecad-20210402-12157-8z9y7j/Build' to build FreeCAD
=================================================

-- Configuring done
CMake Error in src/Gui/CMakeLists.txt:
  Imported target "Shiboken2::libshiboken" includes non-existent path

    "/usr/local/Cellar/python3.9/3.9.0_5/Frameworks/Python.framework/Versions/3.9/include/python3.9

  "

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.

I suspect it might have to do with the extra newline that is shown earlier:
-- SHIBOKEN_PYTHON_INCLUDE_DIRS computed to value: '/usr/local/Cellar/python3.9/3.9.0_5/Frameworks/Python.framework/Versions/3.9/include/python3.9
'

I've attempted updating different dependencies to --HEAD, but so far I've not managed to resolve the issue.

hyarion, 2021-04-02:
I had the same issue and solved it like this:
In `/usr/local/opt/pyside2/lib/cmake/Shiboken2-5.15.2/shiboken_helpers.cmake` just above the line:
`if (SHIBOKEN_COMPUTE_INCLUDES_IS_CALLED_FROM_EXPORT)` (it should be around line 353), add the following line:
`string(STRIP "${PYTHON_INCLUDE_DIRS}" PYTHON_INCLUDE_DIRS)`

vejmarie, 2021-04-02:
I am going to fix these issues during the week-end. We need to take care about the shutdown of bintray services and relocate the bottles to another place. I will while doing this rebuild the bottles.

nachumk, 2021-04-02:
> I had the same issue and solved it like this:
> In `/usr/local/opt/pyside2/lib/cmake/Shiboken2-5.15.2/shiboken_helpers.cmake` just above the line:
> `if [...]

Thanks - I found the file in a different place, but doing what you suggested resolved the problem.

nachumk, 2021-04-02:
FYI - another change I had to make earlier to get this working was removing this line:
      -std=c++14
from:
/usr/local/Homebrew/Library/Taps/freecad/homebrew-freecad/Formula/freecad.rb

It was causing an error with cmake - unknown argument

vejmarie, 2021-04-06:
This in fact easier than that. Just add the second -D include into the cmake within freecad.rb and it shall work without changing a thing into shiboken2. I confirm that removing the std c++14 is required with latest compiler. 

-DPYTHON_EXECUTABLE=Formula["#@tap/python3.9"].opt_prefix + "/bin/python3"
-DPYTHON_INCLUDE_DIR=Formula["#@tap/python3.9"].opt_prefix + "/Frameworks/Python.framework/Headers"

closed 2021-04-06 by vejmarie