FreeCAD/homebrew-freecad #167: All bottled installs fail with "404 page not found"

slaarti, 2021-04-08:
It looks like the justyour dot parts server doesn't actually have _any_ of the bottles that these formulae attempt to install by default. It's just 404s all the way down. Using the `-s` flag to install from source appears to work fine, though.

(Also, I'm glad I was able to find at least one pull request about this "just your parts" server being a real choice you were making, because otherwise, the domain name looks super suspicious. I get that bintray is going away and you needed somewhere else, but still.)

vejmarie, 2021-04-08:
which MacOS version are you using ? Currently only big_sur bottles are available. Can you ping the server ? I suspect more a dns resolution issue as others are able to download the bottle. And you are right that is a temporary solution to find a better option. It has been turned on to avoid a rush when bintray will go down early may.

slaarti, 2021-04-08:
Ah, that might be it; I'm still on Catalina. (Ping is fine; it's very specifically 404, not that I can't get to the server.)

vejmarie, 2021-04-08:
ok, let me check if I still have archive of catalina bottles. They are not on the server for sure and this explain why you are unable to load them.

vejmarie, 2021-04-08:
> Ah, that might be it; I'm still on Catalina. (Ping is fine; it's very specifically 404, not that I can't get to the server.)

Can you retry ? I just loaded the Catalina bottle archive that I had. Unfortunatly I can't recompile freecad bottles as I do not have Catalina anymore.

slaarti, 2021-04-08:
Looks like it _mostly_ worked, except for two items:

```
Warning: freecad/freecad/vtk@8.2.0 has been deprecated because it is a versioned formula!
```

(I assume that this has been remedied for the Big Sur bottles, but figured I'd mention it.)

```
==> Downloading https://justyour.parts:8080/freecad/freecad-0.19.catalina.bottle.tar.gz
/usr/bin/curl --disable --globoff --show-error --user-agent Homebrew/3.0.11-68-g53e35d2\ \(Macintosh\;\ Intel\ Mac\ OS\ X\ 10.15.7\)\ curl/7.64.1 --header Accept-Language:\ en --fail --retry 3 --location --remote-time --continue-at 0 --output /Users/slarti/Library/Caches/Homebrew/downloads/573bc0e4d28e38e05c1167fe6a5e33e9659f0de9a92e1d0fea2ebfd7d0ddf3b8--freecad-0.19.catalina.bottle.tar.gz.incomplete https://justyour.parts:8080/freecad/freecad-0.19.catalina.bottle.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  9210  100  9210    0     0  25583      0 --:--:-- --:--:-- --:--:-- 25512
==> Verifying checksum for '573bc0e4d28e38e05c1167fe6a5e33e9659f0de9a92e1d0fea2ebfd7d0ddf3b8--freecad-0.19.catalina.bottle.tar.gz'
Error: SHA256 mismatch
Expected: 8ef75eb7cea8ca34dc4037207fb213332b9ed27976106fd83c31de1433c2dd29
  Actual: a73297b9733462589ca44991a1bb7a96e6fc9c3fe31e1634c83b93dc382dbf7c
    File: /Users/slarti/Library/Caches/Homebrew/downloads/573bc0e4d28e38e05c1167fe6a5e33e9659f0de9a92e1d0fea2ebfd7d0ddf3b8--freecad-0.19.catalina.bottle.tar.gz
To retry an incomplete download, remove the file above.
```

This one, however, I assume is going to be a lost cause and I'll have to resort to building from source instead, which is fine. I'm just glad I don't _actually_ have to wade through doing that for all of the dependencies. (I got through maybe the first half-dozen, one at a time, because `-s --only-dependencies` doesn't appear to source-build all dependencies like I might've expected it would.)

vejmarie, 2021-04-08:
Just retry it shall work now ;)

slaarti, 2021-04-08:
Well, the download did work, but...

```
slarti@panopticon:~$ ls /usr/local/Cellar/freecad/
ls: /usr/local/Cellar/freecad/: No such file or directory
slarti@panopticon:~$ brew -v install freecad/freecad/freecad
==> Installing freecad from freecad/freecad
==> Downloading https://justyour.parts:8080/freecad/freecad-0.19.catalina.bottle.tar.gz
/usr/bin/curl --disable --globoff --show-error --user-agent Homebrew/3.0.11-68-g53e35d2\ \(Macintosh\;\ Intel\ Mac\ OS\ X\ 10.15.7\)\ curl/7.64.1 --header Accept-Language:\ en --fail --retry 3 --location --remote-time --continue-at 0 --output /Users/slarti/Library/Caches/Homebrew/downloads/573bc0e4d28e38e05c1167fe6a5e33e9659f0de9a92e1d0fea2ebfd7d0ddf3b8--freecad-0.19.catalina.bottle.tar.gz.incomplete https://justyour.parts:8080/freecad/freecad-0.19.catalina.bottle.tar.gz
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  222M  100  222M    0     0  6913k      0  0:00:32  0:00:32 --:--:-- 9175k
==> Verifying checksum for '573bc0e4d28e38e05c1167fe6a5e33e9659f0de9a92e1d0fea2ebfd7d0ddf3b8--freecad-0.19.catalina.bottle.tar.gz'
==> Pouring freecad-0.19.catalina.bottle.tar.gz
tar xof /Users/slarti/Library/Caches/Homebrew/downloads/573bc0e4d28e38e05c1167fe6a5e33e9659f0de9a92e1d0fea2ebfd7d0ddf3b8--freecad-0.19.catalina.bottle.tar.gz -C /private/tmp/d20210407-56041-x9a7f6
cp -pR /private/tmp/d20210407-56041-x9a7f6/freecad/. /usr/local/Cellar/freecad
chmod -Rf +w /private/tmp/d20210407-56041-x9a7f6
Error: /usr/local/Cellar/freecad/0.19 is not a directory
slarti@panopticon:~$ ls /usr/local/Cellar/fre
freecad/   freeimage/ freetype/  frei0r/
slarti@panopticon:~$ ls /usr/local/Cellar/freecad/
0.19pre/
```

vejmarie, 2021-04-08:
I just updated it to make a last try. brew remove freecad and then brew install freecad
Can you edit /usr/local/Homebrew/Library/Taps/freecad/homebrew-freecad/Formula/freecad.rb and change the sha256sum for catalina on line 51 to 85798a42290e257568301b811b6c187fef53f4c3aa20a150677a54c4f805971c and let me know if that is better.

slaarti, 2021-04-08:
It installed this time, although it balked at the postinstall step of `pip3 install six` (possibly because it's already installed?). More importantly:

```
slarti@panopticon:~$ FreeCAD
dyld: Library not loaded: /usr/local/opt/libspnav/lib/libspnav.dylib
  Referenced from: /usr/local/bin/FreeCAD
  Reason: image not found
Abort trap: 6
```

I see there is a `libspnav` formula in `homebrew-core`; do I just install that?

vejmarie, 2021-04-08:
> It installed this time, although it balked at the postinstall step of `pip3 install six` (possibly because it's already installed?). More importantly:
>
> ```
> slarti@panopticon:~$ FreeCAD
> dyld: Library [...]

Yes give it a try brew install libspnav

slaarti, 2021-04-08:
Well, that was interesting... `libspnav` installed fine, and this time FreeCAD did start up... but threw a bunch of errors and tracebacks about not being able to find the PySide module. After chasing down some ideas about my having both `python3.9` from this tap and `python@3.9` from upstream installed, it turned out that the problem was that `homebrew-freecad-bundle.pth` referenced `/usr/local/Cellar/freecad/0.19pre/MacOS/`. Once I edited it to be just `0.19`,  it started up just fine.

Thanks so much for your help! 😃

vejmarie, 2021-04-08:
I have update the formula to reflect the bug you reported and issue a new PR. Thanks a lot !

closed 2021-04-08 by vejmarie

ipatch, 2021-04-08:
pretty sure i've mentioned this before in case i haven't. it would behoove us to use github actions to do CI/CD for distributing the bottles. i've been setting up a macos VM (mojave) on an extra macbook i have, and plan to use it as a self hosted runner, but definitely using github actions seems like the path to least resistance.

https://brew.sh/2020/11/18/homebrew-tap-with-bottles-uploaded-to-github-releases/

vejmarie, 2021-04-09:
I like this idea, but do see a couple of challenge that we have to overcome to make it works. We definitely need a CI/CD for our homebrew package. I can probably add a running VM on my macpro at home also, was more thinking at putting a system within a DC. Right now justyour.parts server is hosted into a datacenter (that is a baremetal machine), I can generate some API keys to facilitate upload automation.