FreeCAD/homebrew-freecad #176: Unable to build using brew in OSX Big Sur

jrhubott, 2021-04-30:
Get the following error when building using the brew instructions:
```
brew install --HEAD freecad --verbose

Make Error at cMake/FreeCAD_Helpers/SetupShibokenAndPyside.cmake:5 (find_package):
  Could not find a package configuration file provided by "Shiboken" with any
  of the following names:

    ShibokenConfig.cmake
    shiboken-config.cmake

  Add the installation prefix of "Shiboken" to CMAKE_PREFIX_PATH or set
  "Shiboken_DIR" to a directory containing one of the above files.  If
  "Shiboken" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  CMakeLists.txt:75 (SetupShibokenAndPyside)

Configuration
HOMEBREW_VERSION: 3.1.5
ORIGIN: https://github.com/Homebrew/brew
HEAD: 227445ece9414a82aa8929b91ca2e0cca28a561b
Last commit: 23 hours ago
Core tap ORIGIN: https://github.com/Homebrew/homebrew-core
Core tap HEAD: c8b7fdfca0c71da4a9e535621e159371317845c9
Core tap last commit: 28 minutes ago
Core tap branch: master
HOMEBREW_PREFIX: /usr/local
HOMEBREW_CASK_OPTS: []
HOMEBREW_MAKE_JOBS: 16
Homebrew Ruby: 2.6.3 => /System/Library/Frameworks/Ruby.framework/Versions/2.6/usr/bin/ruby
CPU: 16-core 64-bit kabylake
Clang: 12.0 build 1205
Git: 2.31.1 => /usr/local/bin/git
Curl: 7.64.1 => /usr/bin/curl
macOS: 11.2.3-x86_64
CLT: 12.5.0.0.1.1617976050
Xcode: 12.5
==> ENV
HOMEBREW_CC: clang
HOMEBREW_CXX: clang++
MAKEFLAGS: -j16
CMAKE_PREFIX_PATH: /usr/local/opt/icu4c@67.1:/usr/local/opt/openssl@1.1:/usr/local/opt/readline:/usr/local/opt/sqlite:/usr/local/opt/python3.9:/usr/local/opt/boost-python3@1.75.0:/usr/local/opt/coin@4.0.0:/usr/local/opt/openblas:/usr/local/opt/hdf5@1.10:/usr/local/opt/qt5152:/usr/local/opt/llvm:/usr/local/opt/sip@4.19.24:/usr/local/opt/pyqt@5.15.2:/usr/local/opt/vtk@8.2.0:/usr/local
CMAKE_INCLUDE_PATH: /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/OpenGL.framework/Versions/Current/Headers
CMAKE_LIBRARY_PATH: /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/System/Library/Frameworks/OpenGL.framework/Versions/Current/Libraries
CMAKE_FRAMEWORK_PATH: /usr/local/opt/python3.9/Frameworks:/usr/local/opt/qt5152/Frameworks:/usr/local/opt/python@3.9/Frameworks
PKG_CONFIG_PATH: /usr/local/opt/icu4c@67.1/lib/pkgconfig:/usr/local/opt/openssl@1.1/lib/pkgconfig:/usr/local/opt/readline/lib/pkgconfig:/usr/local/opt/sqlite/lib/pkgconfig:/usr/local/opt/xz/lib/pkgconfig:/usr/local/opt/python3.9/lib/pkgconfig:/usr/local/opt/coin@4.0.0/lib/pkgconfig:/usr/local/opt/gmp/lib/pkgconfig:/usr/local/opt/isl/lib/pkgconfig:/usr/local/opt/mpfr/lib/pkgconfig:/usr/local/opt/openblas/lib/pkgconfig:/usr/local/opt/libpng/lib/pkgconfig:/usr/local/opt/freetype/lib/pkgconfig:/usr/local/opt/qt5152/lib/pkgconfig:/usr/local/opt/python@3.9/lib/pkgconfig:/usr/local/opt/shiboken2/lib/pkgconfig:/usr/local/opt/pyside2/lib/pkgconfig:/usr/local/opt/hwloc/lib/pkgconfig:/usr/local/opt/libevent/lib/pkgconfig:/usr/local/opt/open-mpi/lib/pkgconfig:/usr/local/opt/orocos-kdl/lib/pkgconfig:/usr/local/opt/fontconfig/lib/pkgconfig:/usr/local/opt/jpeg/lib/pkgconfig:/usr/local/opt/libtiff/lib/pkgconfig:/usr/local/opt/netcdf/lib/pkgconfig:/usr/local/opt/webp/lib/pkgconfig:/usr/local/opt/xerces-c/lib/pkgconfig:/usr/local/opt/eigen/share/pkgconfig
PKG_CONFIG_LIBDIR: /usr/lib/pkgconfig:/usr/local/Homebrew/Library/Homebrew/os/mac/pkgconfig/11
HOMEBREW_GIT: git
HOMEBREW_SDKROOT: /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk
ACLOCAL_PATH: /usr/local/share/aclocal
PATH: /usr/local/Homebrew/Library/Homebrew/shims/mac/super:/usr/local/opt/ccache/bin:/usr/local/opt/cmake/bin:/usr/local/opt/swig@4.0.2/bin:/usr/local/opt/icu4c@67.1/bin:/usr/local/opt/gdbm/bin:/usr/local/opt/gettext/bin:/usr/local/opt/openssl@1.1/bin:/usr/local/opt/sqlite/bin:/usr/local/opt/xz/bin:/usr/local/opt/python3.9/bin:/usr/local/opt/coin@4.0.0/bin:/usr/local/opt/gcc/bin:/usr/local/opt/numpy@1.19.4/bin:/usr/local/opt/libpng/bin:/usr/local/opt/freetype/bin:/usr/local/opt/hdf5@1.10/bin:/usr/local/opt/med-file/bin:/usr/local/opt/opencascade@7.5.0/bin:/usr/local/opt/qt5152/bin:/usr/local/opt/python@3.9/bin:/usr/local/opt/llvm/bin:/usr/local/opt/shiboken2/bin:/usr/local/opt/pyside2-tools/bin:/usr/local/opt/hwloc/bin:/usr/local/opt/libevent/bin:/usr/local/opt/open-mpi/bin:/usr/local/opt/pkg-config/bin:/usr/local/opt/fontconfig/bin:/usr/local/opt/hdf5/bin:/usr/local/opt/jpeg/bin:/usr/local/opt/libtiff/bin:/usr/local/opt/netcdf/bin:/usr/local/opt/sip@4.19.24/bin:/usr/local/opt/pyqt@5.15.2/bin:/usr/local/opt/vtk@8.2.0/bin:/usr/local/opt/webp/bin:/usr/local/opt/xerces-c/bin:/usr/bin:/bin:/usr/sbin:/sbin
```

I have tried everything and even uninstalled brew and started fresh with just the FreeCAD tap.  Not sure what is going on.  Any advise would be great.

Thanks

ipatch, 2021-04-30:
not the answer you're looking but this is an upstream issue with shiboken that has been addressed with this tap/formula, for you viewing pleasure [see this](https://github.com/FreeCAD/homebrew-freecad/issues/165)

personally i stopped seeing this issue when the formula file was updated with [the following line](https://github.com/FreeCAD/homebrew-freecad/blob/master/Formula/freecad.rb#L73)

did you try installing freecad from this tap/formula **without** the `--HEAD` flag?

jrhubott, 2021-04-30:
I just figured it out.  It is from having MacPorts installed.

From [SetupShibokenAndPyside.cmake](https://raw.githubusercontent.com/FreeCAD/FreeCAD/master/cMake/FreeCAD_Helpers/SetupShibokenAndPyside.cmake)

```
macro(SetupShibokenAndPyside)
# -------------------------------- Shiboken/PySide ------------------------

    if(DEFINED MACPORTS_PREFIX)
        find_package(Shiboken REQUIRED HINTS "${PYTHON_LIBRARY_DIR}/cmake")
        find_package(PySide REQUIRED HINTS "${PYTHON_LIBRARY_DIR}/cmake")
    endif(DEFINED MACPORTS_PREFIX)

```

closed 2021-04-30 by jrhubott