FreeCAD/homebrew-freecad #178: heads up compiling latest version of freecad from git source has a xerces-c dep now

ipatch, 2021-07-07:
```
git rev-list --count HEAD                                                                                                                
25197
```

aka

github.com/freecad/freecad/commit/01c901066

seems to now have a dependency on **xerces-c**

ageeye, 2021-07-29:
The commit do not add an include, require or other dependence. Maybe it is require from other formula?

ipatch, 2021-07-29:
here's another person on the freecad forum documenting the issue similar to what i ran into.

https://forum.freecadweb.org/viewtopic.php?f=4&t=60663

ageeye, 2021-07-30:
The Base/Reader.cpp  has long been included exerces-c. From this point of view must search therefore one completely other starting point. E.g. changes from cmake version.

ipatch, 2021-08-08:
does not seem to be issue anymore.

closing now. if i run into this issue down the road i'll open another issue.

closed 2021-08-08 by ipatch